machinekit/machinekit #873: icomp compiled components using posix path

darenschwenke, 2016-02-15:
Pardon the brevity as this is from memory.  When I can rebuild, I will grab more.

Using bone-debian-7.9-machinekit-armhf-2016-02-07-4gb.img.xz
Ran apt-get update
Ran apt-get install machinekit-dev
Loaded a configuration with icomps present.
The icomps compiled, but then attempted to place themselves in: /usr/lib/linuxcnc/posix
That directory did not exist.  I created it.  They compiled again and ended up there.
Startup then fails though as subsequent runs look for them in: /usr/lib/linuxcnc/xenomai
I moved the compiled components from /usr/lib/linuxcnc/posix to /usr/lib/linuxcnc/xenomai and all was well.

ArcEye, 2016-02-17:
instcomp has never been separated from the Machinekit package, so you should have had that anyway without installing the -dev package.
However you needed that for the headers.

You probably need to ask Robert Nelson, looks like his image.
https://rcn-ee.com/rootfs/bb.org/testing/2016-02-15/machinekit/bone-debian-7.9-machinekit-armhf-2016-02-15-4gb.img.xz

AFAIK that sort of thing would normally only occur, if you configured without a rt kernel being present and just got posix paths set from there.

mhaberler, 2016-03-13:
is this still pending?

darenschwenke, 2016-03-14:
I forwarded it to the email I had for robertnelson since it seems to be a
build issue over there.  I haven't tried a new image to see yet.

closed 2016-09-29 by darenschwenke