machinekit/machinekit #875: TODO: pending stuff for cmake conversion

kinsamanka, 2016-02-21:
Here's what's still missing

- [ ] `comp/icomp` out-of-tree builds
- [ ] packaging
- [ ] race condition on python targets (parallel make builds)
- [ ] finalise codebase split
- [ ] runtests and nosetests
- [ ] build configuration (`configure ...` equivalent)
- [ ] travis-ci integration
- [ ] documentation on how to merge updates from the legacy repo
- [ ] arm builds

ArcEye, 2016-02-21:
> Here's what's still missing
    comp/icomp out-of-tree builds

This is quite easy to achieve, but the bigger issue is related to #874 and your own

> build configuration (configure ... equivalent)

item.

Nearly all my problems with a Makefile.modinc file, came from trying to have a generic build file, which relied upon FLAVOR variables that were not carried over.
I can just fudge it for now and build quite happily because only posix and rt-preempt are currently catered for, but a more complete solution is probably required.

I am increasingly favouring a more radical and simple approach to building, which I will try to expound on #874

ArcEye, 2018-08-01:
Has been done another way now :)

closed 2018-08-01 by ArcEye