machinekit/machinekit #877: PktUART driver for MESA

sirop, 2016-02-21:
_I do not get how to make a pull request for machinekit as I am always redirected to linuxcnc repo._

https://github.com/sirop/machinekit/commit/3446d4a448eb961421c3ee7211f94f1a1a2462d5

A simple test component working with Microstrain 3DM-GX3-15 gyro is in the commit. 

Yet missing:
- man files
- test with over 115200 bits/s 

Some discussion with PWC : beginning from https://forum.linuxcnc.org/forum/24-hal-components/30262-reading-uart-rx-count-register-on-mesa-5i25-6i25?limitstart=0#69504

luminize, 2016-02-21:
> On 21 Feb 2016, at 22:25, Boris Skegin <notifications@github.com> wrote:
> 
> I do not get how to make a pull request for machinekit as I am always redirected to linuxcnc repo.

Google is your friend:
https://help.github.com/articles/using-pull-requests/

Look at the section about changing destination repository

sirop, 2016-02-21:
Ok, I get it now.

So let me first do a test with over 115200 bits/s the following week and then I'll do this pull request.
Yes, and man files...

luminize, 2016-02-21:
> On 21 Feb 2016, at 22:57, Boris Skegin <notifications@github.com> wrote:
> 
> Yes, and man files...

Yes please!

sirop, 2016-02-29:
Added man files: https://github.com/sirop/machinekit/commit/bc16eccae13a28d292f7137d7be4a1bfa024e41e

Have not yet got the converter for 1Mps. Have to wait some days more.

mhaberler, 2016-03-13:
is this still active?

sirop, 2016-03-13:
Yes, there is only some delay, as I get the 1Mps converter only on next Friday.

amitgoradia, 2016-03-19:
I have tried the pktUart with 1.5 Mbps and have seen many errors in the bitstream. Could never figure out the fault. However, same hardware connections with normal mesa UART at the same 1.5 Mbps baud rate works fine.

Below 1Mbps the pktUart also works fine for days. But above 1.5 Mbps, it will start giving packet overrun or missing stop bit errors in less than 10 to 15 minutes.

PCW from mesa said that he is using pktUART for the SSerial drivers and has it working at much higher baud rates. So, I am sure I am doing something wrong in the driver.

I will try to post the driver source somewhere over the weekend.

-automata

sirop, 2016-03-19:
@amitgoradia 
I have corrected some typos on my repo.
So did you use the latest commit?

I have seen also some errors like  RxRCFIFOError even at 115200 baud rate,
but that was definitely due to some soldering faults of our self made converter.
These errors occurred at sunset or dawn time, so there was some dependency on
environmental temperature.

I have got now a better converter which will work with at least 1Mbps.
I will test it either tomorrow or on Monday.

Which converter do you use: one for RS232 or one for RS485?

>I will try to post the driver source somewhere over the weekend

You have written your own driver?
Or do you actually mean a component?
Could you let me have a look at your component?

sirop, 2016-03-19:
@amitgoradia 
There is actually a way to test the PktUART driver without any converter.

Define two PktUART instances in your bit file,
and then cross wire the RX/TX of the corresponding instances on the daughter card.

amitgoradia, 2016-03-19:
@sirop: I have tried the local cross wire for RX and TX. That works flawlessly for even 6 MBaud. I have tried running it for days on end (I think I ran it for 7 days and nights straight).
However when I use 2 pktuarts on 2 different FPGA's on 2 different PC's, the pktUART starts giving bit errors.

I have eliminated the level translator hardware as a source of error as I have used the same hardware from the same pins on the FPGA to communicate at 1.5 MBaud using regular mesa UARTS. So there is something in the pktUART that does not like higher baud rates.

Will try to find that driver code and put it up somewhere.
-amit

sirop, 2016-03-19:
@amitgoradia 
So as I understand, you used your own driver  and not the one I want to commit.

UART and PktUART have similar error registers,
but you can - actually you have to - set  Bits 15..8 (_InterFrame delay in bit times_)
for PktUART.
Did you experiment with this option?

>Will try to find that driver code and put it up somewhere.  

Yes, please.

amitgoradia, 2016-03-20:
@sirop: I used my own driver for the pktuart. Actually I started by writing a bare bones driver without any link from HAL or even realtime kernel (pure linux driver) just to learn how to interface with the PCI card.

For the pktUART, thanks for pointing out the interframe delay setting. I had totally missed that part. The driver is at my office burried in someold computer, will have to dig it out on Monday. This driver is written using HAL and integrated into the linuxcnc codebase. I vaguely remember adding some functions to hostmot2 driver to get access to the pktUART. 

Thanks again for pointing out the interframe delay bits... 
-amit

amitgoradia, 2016-03-20:
@sirop: PCW had pointed me to a resource on the mesa website on how to interface with the pktuart:
http://freeby.mesanet.com/lbpcom16.pas
hope that is of some help.

sirop, 2016-03-26:
PktUART driver is now merged into the master branch.

closed 2016-03-26 by sirop

sirop, 2016-03-26:
Just for the record:
CS-428/9AT-mini2
http://sysbas.com/modules/catalogue/detail.php?inoGood_no=189&ino_code=1810&list_change=&field=&keyword=&page=1 
 seems to be the a good and an inexpensive option if one needs a longer distance than RS232 permits.

machinekoder, 2017-05-16:
@sirop If I understand correctly the difference between the UART and the PktUART driver is that the PktUART driver creates gaps between "packages." Is there any additional difference?

sirop, 2017-05-16:
@machinekoder 

The main difference is that the default buffer of PktUART is 1024 bytes ( and can be increased up to 2048 bytes  ). Compare this to only 16 byte deep Rx FIFO of simple UART.

Your wording _" PktUART driver creates gaps between "packages." "_  is a bit misleading as one can speak about gap creation only in case of Tx direction, in case of Rx it is rather detection of interframe delay/gap.

machinekoder, 2017-05-16:
@sirop Thanks for the explanation.

machinekoder, 2017-05-17:
@sirop Can you maybe share your pin config? I have some problems getting RX to work properly, even with a local cross-wire.

sirop, 2017-05-17:
@machinekoder 
https://gist.github.com/sirop/224ab739b00dbe31689dd3b9e74b678a  has one Tx and one Rx instance.

But you do understand the hm2* functions of PktUART?

machinekoder, 2017-05-17:
I think so. Without crossing out the wires I get no errors on RX and TX seems to work.
However, when I cross out the wires I get something along the lines:
```
hal_lib:13113:rt hm2_7i92.0.pktuart.1 sent: bytes 24, frames 6
hal_lib:13113:rt hm2/hm2_7i92.0: hm2_7i92.0.pktuart.1: Rx Logic active
hal_lib:13113:rt hm2: hm2_7i92.0.pktuart.1: Overrun error, no stop bit
hal_lib:13113:rt PktUART receive: got -111 bytes, 0 frames
```

The error does not go away until I reset the MESA card.

sirop, 2017-05-17:
Does at least the first line

> hal_lib:13113:rt hm2_7i92.0.pktuart.1 sent: bytes 24, frames 6

make sense?

machinekoder, 2017-05-17:
Yes, TX works as expected.

sirop, 2017-05-17:
I do not know your cross wiring, so it is hard to guess if you add any noise which is then detected 
as an incomplete UART byte and therefore 

> hal_lib:13113:rt hm2: hm2_7i92.0.pktuart.1: Overrun error, no stop bit

Which voltage is on the wires when you cross  them out? Do you crosswire before or after the converter?
Is your baud rate the maximal baud rate your converter can operate with?

You can also try to increase the interframe delay hoping that any noise during the interframe delay is then omitted.

As I was testing this driver, I did not do any crosswiring but used a scope and a logic analyzer.

machinekoder, 2017-05-17:
I did cross-wire straight from the DB25 pin of the MESA card so before the converter. The Baudrate is just 115200 so that shouldn't be a problem.

I have just ordered a scope for my office. I will report back when I have measured the signals.

machinekoder, 2017-05-19:
Okay, I have the scope and my setup up and running now. My serial port USB adapter outputs a nice 5V signal, so does the MESA card. However, I still can't get the read function of the driver to work. @sirop Did you need to modify anything with the PIN configuration for the FPGA other then the `PIN_*` file?

sirop, 2017-05-19:
@machinekoder 
I remember there was some talk about InstanceStride.

To be sure let's compare the following lines of TopPCIHostMot2.vhd :
```
entity TopPCIHostMot2 is -- for 5I24,5I25, 5I26, 6I25 PCI target mode
	generic 
	(
		ThePinDesc: PinDescType := PinDesc;
		TheModuleID: ModuleIDType := ModuleID;
		PWMRefWidth: integer := 13;			-- PWM resolution is PWMRefWidth-1 bits 
		IDROMType: integer := 3;		
		UseIRQLogic: boolean := true;			--- note this will pull in PWM ref
		UseWatchDog: boolean := true;
		OffsetToModules: integer := 64;
		OffsetToPinDesc: integer := 448;
		BusWidth: integer := 32;
		AddrWidth: integer := 16;
		InstStride0: integer := 4;			-- instance stride 0 = 4 bytes = 1 x 32 bit
--		InstStride1: integer := 64;		-- instance stride 1 = 64 bytes = 16 x 32 bit registers !! UARTS need 0x10
		InstStride1: integer := 16;		
		RegStride0: integer := 256;		-- register stride 0 = 256 bytes = 64 x 32 bit registers
		RegStride1: integer := 256;      -- register stride 1 = 256 bytes - 64 x 32 bit
		FallBack: boolean := false			-- is this a fallback config?
	);    
```
Do InstStride0 and InstStride1 have the same values in your TopPCIHostMot2.vhd ?

And just to be sure: how wide is your UART word? 10 or 11 bits?

You could also share the code of your component...

machinekoder, 2017-05-20:
Indeed my `InstanceStride1` was set to 64 and not 16. However, changing the value in the configuration did not solve the problem.

Here is my `TopEthernet16HostMot2.vhd`:
```vhdl
entity TopEthernetHostMot2 is -- for 7I80DB and 7I80HD, 7I77E and 7I76E
	 generic 
	 (
		ThePinDesc: PinDescType := PinDesc;
		TheModuleID: ModuleIDType := ModuleID;
		PWMRefWidth: integer := 13;	-- PWM resolution is PWMRefWidth-1 bits 
		IDROMType: integer := 3;		
		UseStepGenPrescaler : boolean := true;
		UseIRQLogic: boolean := true;
		UseWatchDog: boolean := true;
		OffsetToModules: integer := 64;
		OffsetToPinDesc: integer := 448;
		BusWidth: integer := 32;
		AddrWidth: integer := 16;
		InstStride0: integer := 4;			-- instance stride 0 = 4 bytes = 1 x 32 bit
--		InstStride1: integer := 64;		-- instance stride 1 = 64 bytes = 16 x 32 bit registers sserial
		InstStride1: integer := 16;		-- 4..7 16 for BSPI/UART Ick double Ick
		RegStride0: integer := 256;		-- register stride 0 = 256 bytes = 64 x 32 bit registers
		RegStride1: integer := 256;      -- register stride 1 = 256 bytes - 64 x 32 bit
		FallBack: boolean := false			-- is this a fallback config?

		);
```

For testing purposes I create a simple comp based on the `mesa_pktgyro_comp` which also sends data:
https://gist.github.com/machinekoder/37de8d4bbe73d60335b7d2cb3fc6c118

As counterpart I have small Qt application running on my desktop computer, answering the single `aaaa` frame with another `aaaa` frame. On the oscilloscope, both received message and answer look fine.

My UART word is 10 bits: start bit + 8 data bits + 1 stop bit

Is there anything else that needs changing in the VHDL files?

machinekoder, 2017-05-20:
@sirop I found the problem. At least the PktUART of the Ethernet Hostmost has now the `FilterReg` enabled. Therefore, one needs to set the bits 29..22 of the RX register to receive something.

sirop, 2017-05-20:
@machinekoder 
Then one has to change 
```
if (rx_mode >= 0) {
        buff = ((u32)rx_mode) & 0xffff;
        r += hm2->llio->write(hm2->llio, inst->rx_mode_addr, &buff, sizeof(u32));
    }
``` 
to 
```
if (rx_mode >= 0) {
        buff = ((u32)rx_mode) & 0x3fc0ffff; // =0011 1111 1100 0000 1111 1111 1111 1111
        r += hm2->llio->write(hm2->llio, inst->rx_mode_addr, &buff, sizeof(u32));
    }
``` 
in https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/pktuart.c#L227-L230 .

I could do it tomorrow before the noon german time if you like.

sirop, 2017-05-20:
@machinekoder 
Or I'd rather ask PWC on LinuxCNC foum first.

If you are sure about this change, then you could apply this little patch ( see my comment above) locally.

Otherwise you could use https://github.com/LinuxCNC/hostmot2-firmware which has not got any `FilterReg`  yet.

machinekoder, 2017-05-21:
@sirop I read out the `rx_mode` register before setting the `rx_mode` to ensure the default value is not touched.

machinekoder, 2017-05-21:
```c
/* http://freeby.mesanet.com/regmap
      The PktUARTrMode register is used for setting and checking the PktUARTr's 
      operation mode, timing, and status
      Bit  31..30      Unused
      Bit  29..22      Filter Register
      Bit  21          FrameBuffer has data 
      Bits 20..16      Frames received
      Bits 15..8       InterFrame delay in bit times
      Bit  7           Receive Logic active, not an error
      Bit  6           RXMask
      Bit  5           Unused
      Bit  4           RCFIFO Error
      Bit  3           RXEnable (must be set to receive packets) 
      Bit  2           RXMask Enable (enables input data masking when transmitting)
      Bit  1           Overrun error (no stop bit when expected) (sticky)
      Bit  0           False Start bit error (sticky)
    */
    if (rx_mode >= 0) {
        // read filter settings from rx register before setting new values
        r += hm2->llio->read(hm2->llio, inst->rx_mode_addr, &buff, sizeof(u32));
        buff = (((u32)rx_mode) & 0xffff) | (buff & (0xFF << 22));
        r += hm2->llio->write(hm2->llio, inst->rx_mode_addr, &buff, sizeof(u32));
    }
```

sirop, 2017-05-21:
@machinekoder 
See.

So this patch of yours lets you use the newest VHDL sources of MESA? Does Rx of packet UART work now?

I have got no answer yet to https://forum.linuxcnc.org/27-driver-boards/32811-mesa-digital-filters .

Both https://github.com/LinuxCNC/hostmot2-firmware and https://github.com/machinekit/mksocfpga still have the old VHDL code. Is your patch backwards compatible?

machinekoder, 2017-05-21:
Yes, it works now.

It should be compatible, because `pktuart.vhd` ignores the rest of the settings bits and they should be 0 anyway: https://github.com/LinuxCNC/hostmot2-firmware/blob/master/src/pktuartr.vhd#L367

sirop, 2017-05-21:
Ok.
I'll add your patch to my question to PCW on LinuxCNC forum if you do not mind. Just to be sure ...

machinekoder, 2017-05-21:
Yes, please go ahead.

sirop, 2017-05-21:
Ok, as I understand PCw's answer with  FilterReg = 0.5*bittime*ClockLow -1 and 
at 128,000 bit/s with bittime=7.81 µs  FilterReg = 0.5*7.81*100-1 = 389.5 and should be set down to 255=0xff . For ClockLow see IDROMConst.vhd .  So you might have had luck with the "default" value.

So I'd patch as:`
```
if (rx_mode >= 0) {
        buff = ((u32)rx_mode) & 0x3fc0ffff; // =0011 1111 1100 0000 1111 1111 1111 1111
        r += hm2->llio->write(hm2->llio, inst->rx_mode_addr, &buff, sizeof(u32));
}
```
The user should know which version of VHDL code he uses and how to calculate  FilterReg .

machinekoder, 2017-05-22:
@sirop Sounds good. I suggest updating the register description comment and maybe updating the example component to reflect the changes. Another option would be to add an additional function to pktuart to calculate the filter setting for you.

sirop, 2017-05-22:
@machinekoder 
I hope that you can work with your local version for a while as I'll have time to make patch only next weekend. 

@cdsteinkuehler 
What is the update policy for MESA firmware within machinekit? As I'll have to refer to "older" and "newer" versions for the patch discussed above.

dkhughes, 2017-05-22:
@sirop 
There is a firmware ID version tag embedded that you can use for this purpose provided the newer VHDL did bump it up.

sirop, 2017-05-23:
@dkhughes 
I guess you mean https://github.com/machinekit/mksocfpga . As https://github.com/machinekit/mesanet-firmware is empty, neither the "newer" nor the "older" VHDL has thus any version hash.

dkhughes, 2017-05-23:
@sirop I mean inside of the VHDL itself, there is a runtime readable register called IDROM Type. Any compiled firmware will have that embedded internally. Here is where it gets populated in hostmot2:

[https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L527](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L527)

If there is a breaking change in the VHDL, that number should have been bumped up, and you can check it and decide which version of the serial driver to load. We did something similar to eliminate the hardcoded board pin count checks based on legacy connectors without breaking backwards compatibility.

sirop, 2017-05-23:
@dkhughes 
Well, we could expect the user to know if the VHDL code he uses needs this 8 bit digital filter or not.
The newest MESA VHDL has not been yet merged anyway. 
And the patch for the digital filter will be backwards compatible.