machinekit/machinekit #894: Install machinekit from LinuxCNC

JonnyTech, 2016-03-05:
I installed LinuxCNC from its ISO: http://www.linuxcnc.org/linuxcnc-2.7-wheezy.iso

Then I discovered machinekit so I decided to install it on together with the above since the realtime kernels are already in a working state there.

Using instructions foe Debian at https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/getting-started/installing-packages.asciidoc

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv 43DDF224
sudo sh -c "echo 'deb http://deb.machinekit.io/debian wheezy main' > /etc/apt/sources.list.d/machinekit.list"
sudo apt-get update
sudo apt-get install machinekit-rt-preempt

There was a warning that linuxcnc and linuxcnc-dev would be removed. OK then.

It completed correctly but machinekit does not start. What am I doing wrong? Can I install both together? I would love to try machinekit without a full new OS install.

luminize, 2016-03-05:
> On 05 Mar 2016, at 14:44, JonnyTech <notifications@github.com> wrote:
> 
> It completed correctly but machinekit does not start. What am I doing wrong? Can I install both together? I would love to try machinekit without a full new OS install.

Hi. We can't see inside your system. What do the log files say for example?

Have a look here: https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/getting-help.asciidoc

Bas

JonnyTech, 2016-03-05:
> Hi. We can't see inside your system. What do the log files say for example?

Of course, I wanted to ask the question before throwing too much information at you:

```
cnc@linuxcnc:~$ uname -a
Linux linuxcnc 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux
cnc@linuxcnc:~$ uname -v
#1 SMP PREEMPT Debian 3.4.55-4linuxcnc
cnc@linuxcnc:~$ sudo apt-get install machinekit-rt-preempt
[sudo] password for cnc: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  tclx8.4 yapps2-runtime
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  machinekit
The following packages will be REMOVED:
  linuxcnc linuxcnc-dev
The following NEW packages will be installed:
  machinekit machinekit-rt-preempt
0 upgraded, 2 newly installed, 2 to remove and 0 not upgraded.
Need to get 0 B/7,267 kB of archives.
After this operation, 17.5 MB disk space will be freed.
Do you want to continue [Y/n]? 
(Reading database ... 155081 files and directories currently installed.)
Removing linuxcnc-dev ...
Removing linuxcnc ...
Processing triggers for man-db ...
Processing triggers for python-support ...
Processing triggers for desktop-file-utils ...
Processing triggers for shared-mime-info ...
Selecting previously unselected package machinekit-rt-preempt.
(Reading database ... 152556 files and directories currently installed.)
Unpacking machinekit-rt-preempt (from .../machinekit-rt-preempt_0.1.1457049044-1mk.travis.master.git463f053f~1wheezy_i386.deb) ...
Selecting previously unselected package machinekit.
Unpacking machinekit (from .../machinekit_0.1.1457049044-1mk.travis.master.git463f053f~1wheezy_i386.deb) ...
Processing triggers for desktop-file-utils ...
Processing triggers for shared-mime-info ...
Processing triggers for man-db ...
Setting up machinekit-rt-preempt (0.1.1457049044-1mk.travis.master.git463f053f~1wheezy) ...
Setting up machinekit (0.1.1457049044-1mk.travis.master.git463f053f~1wheezy) ...
[ ok ] Stopping the hotplug events dispatcher: udevd.
[ ok ] Starting the hotplug events dispatcher: udevd.
[ ok ] Stopping enhanced syslogd: rsyslogd.
[ ok ] Starting enhanced syslogd: rsyslogd.
cnc@linuxcnc:~$ machinekit
MACHINEKIT - 0.1
Machine configuration directory is '/home/cnc/linuxcnc/configs/franco-cnc'
Machine configuration file is 'franco-cnc.ini'
Starting Machinekit...
Warning - /usr/libexec/linuxcnc/rtapi_app_posix not setuid
'sudo make setuid' missing?
/usr/bin/realtime: line 192: /usr/libexec/linuxcnc/rtapi_app_posix: No such file or directory
rtapi_app startup failed - aborting
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
io started
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
halcmd loadusr io started
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
Shutting down and cleaning up Machinekit...
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout
[... 85 lines cut ...]
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/usr/bin
LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/usr/share/linuxcnc/tcl/msgs
INIVAR=/usr/libexec/linuxcnc/inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.5
INIFILE=/home/cnc/linuxcnc/configs/franco-cnc/franco-cnc.ini
PARAMETER_FILE=linuxcnc.var
TASK=milltask
HALUI=
DISPLAY=axis
Starting Machinekit server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting Machinekit IO program: io
Killing task linuxcncsvr, PID=7580
Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments
cnc@linuxcnc:~$ dmesg >dmesg.txt
```
And here is the dmesg file...
[dmesg.txt](https://github.com/machinekit/machinekit/files/160075/dmesg.txt)

mhaberler, 2016-03-05:
machinekit does not support RTAI any more.

ArcEye, 2016-03-06:
> Setting up machinekit-rt-preempt (0.1.1457049044-1mk.travis.master.git463f053f~1wheezy) ...
Setting up machinekit (0.1.1457049044-1mk.travis.master.git463f053f~1wheezy) ...

You have actually pulled the package for rt-preempt.

The repo will still build on rtai currently, but there are no packages.

mhaberler, 2016-03-06:
did you by any chance _both_ install packages and build from source?

JonnyTech, 2016-03-06:
Thanks, I assumed it to be a kernel issue but was not sure.
Is there an advantage of one kernel over another?
Would installing a rt-preempt kernel break my existing linuxcnc setup?
Why can linuxcnc and machinekit not exist together on the system?
Is there a prebuilt .ISO for machinekit like the linuxcnc one mentioned?

ArcEye, 2016-03-06:
> Is there an advantage of one kernel over another?

Yes, but not if you are using a parallel port and software stepping.  In that case rtai is likely to be the only one that works for you.  Just stick with linuxcnc.

>Why can linuxcnc and machinekit not exist together on the system?

Because machinekit is forked from linuxcnc, installing it system wide will overwrite most of the linuxcnc install.

You can however have machinekit and linuxcnc both installed a RIP builds quite happily, because they are 'contained' and don't interfere with each other.

> Is there a prebuilt .ISO for machinekit like the linuxcnc one mentioned?

No, that was an option of necessity to reach a wider audience for linuxcnc, but one that is a maintenance nightmare and a path we have no intention of following.

closed 2016-03-06 by ArcEye