machinekit/machinekit #903: Xilinx Zynq port not available

mhaberler, 2016-04-06:
This is intended as an anchor issue for making machinekit work on Zynq boards, including the mesanet firmware in the FPGA. Let's collect Zynq-related discussion here.

mhaberler, 2016-04-06:
@claudiolorini has significant experience with Zynq systems and has contributed some drivers for the Zedboard. He kindly shared some suggestions in private mail which I include here for reference:

>@work we started with the zedboard that is a good platform for development but lately we switched all new 
and old projects to the MYIR Z-Turn which has a better price/performance tag, and more manageable connectors.
For sure i'd exclude red-pitaya for lacking of io-pins, 

So I've ordered some MYIR Z-Turn boards as well (I mentioned I own a Redpitaya).

Re kernel Claudio recommends:

>i suggest you to start from the git of a colleague of mine that is maintaining our kernel 
in line with the official kernel development, mine is not always as up to date...
 https://github.com/andreamerello/linux-zynq-stable  
start with the branch: 
 linux-4.1.13-zynq
and apply the appropriate RT_PREEMPT patch from kernel.org (should be very straightforward).
Soon (hopefully) we will jump to 4.4.something that should be the latest RT patchable..

>first time you compile you have to do:
 make ARCH=arm  zynq_edl_defconfig

>the .dts for the MYIR is 
 .../linux-zynq/arch/arm/boot/dts/zynq-zturn-myir.dts
this should give you all peripherals serials, USB, Eth, accelerometers, temperature, and CAN!
A free IP for HDMI is not (yet) available but is in our TODO list (with a medium priority) so maybe soon... 
to compile the dtb:
 make ARCH=arm zynq-zturn-myir.dtb

>AFAIR pitaya has a ramdisk (non persistent crap), for the myir (and zed) with Debian you should 
have 3 partiions a small FAT32 for boot that should contaiin:
- uImage: the kernel you've just compiled
- devicetree.dtb: device tree you've just cmpiled
- BOOT.bin - a blob containing the FPGA bitfile the firststage bootloader and the u-boot... and here comes the pain:
the BOOT.bin can only be created by a rather messy process using the Xilinx Vivado suite. 
The easiest way is that you use mine BOOT.bin in attachment, otherwise you have to install all the 
Vivado SW (that is huge). We are working on making a simplified non graphical procedure (as has 
done cdsteinkuehler) to do so bit we are still far from that.

>The second partition EXT4 contains the FS, i think that you can use the same the other guys are using 
for the Altera. I've started from scratch with a Debian netinstall but is for sure a longer way to walk...

>if you want you can have a third partition for swap but i don't think is necessary (never seen used 
a single bit of swap)

Re u-boot:

>no, uboot can be compiled but the u-boot.elf must be packed together with the FPGA bitfile and the FSBL.elf
using a Xilinx utility from the Vivado suite. Maybe i's simply a cat of the 3 files, i've never analyzed the problem. 
I can try to ask the colleague of mine(Francesco) that manages the FPGA...
BTW the u-boot with modifications for the initialization of the Eth is here:
 https://github.com/francescodiotalevi/u-boot-xlnx

>...well, for u-boot no problem, the FSBL.elf can be problematic, i *think* is propietary stuff 
(i'm not sure...) but the real problem is below.

>AFAIK at the moment the FPGA is not (yet) re-configurable at run-time, it's loaded in the boot stage, 
not sure if in the FSBL or prior to that. 
We have seen in the latest kernels some movement in the direction of run-time reload of the FPGA 
but we have not tried it yet.  

>>I assume the processor boots with an empty FPGA?

>THIS is the big issue, the SOC with Linux *does*not*work* without a ton of things in the FPGA side!!
At least the FPGA must route all external memory lines (4SPI&RAM) clocks, reset... so without a proper bitfile 
nothing works. The bad news is that this can only be done (well, I'm able only able to do it) with Vivado suite 
graphical tools...  

>The Vivado suite is a 'Graphical' tool, under the hood it's all scriptable and configurable but Xilinx want you
to click with the mouse ;-) and all the scripting is well hidden...

nilrods, 2016-04-06:
Attaching my notes on the github issue.

Here are a few notes I will add to what Claudio wrote. I used the Linux kernel he mentioned in the email to you. I haven't tried his updated uboot yet.

I believe most of the Vivado work can be done via TCL scripting if you have the skillset http:www.xilinx.com/support/documentation/sw_manuals/xilinx2015_4/ug894-vivado-tcl-scripting.pdf. I have used this to export a base project and recreate in a new project. Unfortunately, you still need most of the *large multi-gig* Vivado install installed on the machine, but should be able to be mostly automated.

Attached is a document with a screenshot of my SD card boot partition. I have the .bit file (FPGA configuration) separated from the BOOT.bin and devicetree and all. This is nice when you just need to correct an FPGA component and nothing else changes. I just generate a new .bit file and overwrite the existing one on the sd card and reboot. I am using an older version of uboot though than Claudio is though, but I would think it is still supported.(edit:sounds like may not be used)

I have read that you can reprogram the FPGA from Linux using the xdevcfg device, but I have not tried it. To do so though the .bit file(7z020.bit in my example) needs to bit reversed using the Xilinx promgen utility and a .bin file produced. I would think this would only work though if the FPGA side memory addresses are the same, unless there is some sort of dynamic reconfiguration of the device-tree available in Linux(you may know more about this area with your work on the BBB) to map to the new addresses. But should work fine with only updating a hm2 device configuration with another hm2 device configuration in the FPGA, since they would both present the same memory addresses to Linux.

I used the filesystem partition from Michael's mksocfpga, so that should all be 100% reusable, all I did was updated the kernel modules on the filesystem to the ones used by the Xilinx kernel and it worked fine.(edit:sounds like not the planned direction going the omap-image-builder)

the-snowwhite, 2016-04-06:
I decided to follow the link @nilrods gave to the Snickerdoodle board. 
Upon very much liking the infrastructure they have built up I wrote them a mail mentioning I would very much like to add their Snickerdoodle board as the next port.
They have now after I ordered one of each promoted me to the Alpha hardware list.
Under all circumstances I expect the documentation they already provide here:

https://github.com/krtkl/snickerdoodle-manual

Will serve as providing me a crash course Into the Vivado Xilinx universe.

This documentation can possibly also be utilized for getting the Z-Turn board up and running (no online docs from there ?), However I am right now very strongly biased towards supporting the open-sourced (and funded) Snickerdoodle project and their very friendly developer minded infrastructure (+ all the I/O pins on the board).

The 4.4 kernel's fpga manager is supposed to support configuration of both Altera And Xilinx socs.

The DTS --> Linux Driver problems are proposed to be solved by the New fpga-manager Framework and device support built into the kernel (4.4  +).

THis is whats behind the (altera 3.10-rt-ltsi) kernel decision, to be able to get that release "quickly out of the door".

What my intuition is telling me is more time is needed.

for the new 4.x kernel based linux device tree handling, new fpga manager driver framework +  the intelligent  driver loading, with updated device tree overlays.  

To mature.
Which will put a stable unified Xilinx / Altera soc 4.x -rt-ltsi MK release into a late summer august timeframe.

nilrods, 2016-04-06:
Just an FYI. 

Both the Z-turn and the snickerdoodle use the same zynq part for the boards so a few pins mappings and maybe memory part config(depending on what chip the snickerdoodle uses) would probably be the main differences between the two boards when it comes to the FPGA side and bit file.

So really the same work can support both boards with only trivial differences in the vivado projects or probably even just one project, if it is scripted. 

I have the Z-turn manuals if anyone is interested in them. They are actually pretty good compared to some of the other boards I have bought over the years.

nilrods, 2016-04-08:
I will take a look at trying to build a vivado docker when I get some extra time. I haven't used it before but am very familiar with VMware so hopefully the process is somewhat similar.

Any particular debian release or configuration required?

dkhughes, 2016-04-08:
A couple of quick notes from my experiences with the Zynq on a microzed in reference to the emails:

Like @nilrods  mentioned, the FSBL for the zynq doesn't require a bitfile to be able to boot. When you build the boot image (BOOT.bin) you can omit any bit file at all and it will simply skip programming the FPGA. The Xilinx bootloader generator tool, however, does require a base design that defines pin locations for the ddr, ethernet, etc., which is used in their code generator to create the ps7-xxx.c/.h files. IIRC, u-boot in the xilinx git tree now includes a SBL, which basically integrates a sanely configured default FSBL that the arm can use to get to u-boot. I haven't tried the SBL out yet, on my boards I simply compile u-boot and FSBL into a BOOT.bin, but I don't know how to automate this through scripting (all gui so far).

I found some other discussion about run time configuration of the FPGA on the web, and, if your in a kernel that supports /dev/xdevcfg, it is functioning correctly with the caveat that the bit file must expose minimally the interface required by any loaded kernel drivers or it will hang after config. The workaround is to build any drivers that depend on the fpga being configured as modules, that are loaded after configuration succeeds. To update without a reboot, unload those modules, reconfig FPGA, reload modules. I will test this process out tonight if I get a chance but it seems promising.

@nilrods I agree the trouble remaining is how to handle the device tree. I think it is fair to assume the interface of the FPGA configs will be relatively constant though, e.g. containing a unified h2mot entry point, which should allow reconfig without a hang. If you do make significant enough changes to the bit file to require a *dts change, you can perform a reboot with a new .dtb in the sd-card. I use a uEnv.txt to be able to quickly swap those out.

I think I will try the 4.4.6 kernel git to get on the same page as you guys.

cdsteinkuehler, 2016-04-08:
On 4/8/2016 2:11 PM, nilrods wrote:
> 
> Any particular debian release or configuration required?

We tend to standardize on Debian, but you can use something else if
required.  Neither the Altera or Xilinx tools are officially supported
on Debian, but they can generally be made to work.

You may find my VM and Docker work for the Altera Quartus tools helpful:

VM Install of Quartus:
https://github.com/cdsteinkuehler/mksocfpga/blob/master/HW/README.BuildSystem.txt

Docker files for Quartus build system:
https://github.com/cdsteinkuehler/QuartusBuildVMs

-- 
Charles Steinkuehler
charles@steinkuehler.net

nilrods, 2016-04-08:
Charles,
Oh wow nice notes. Those will help for sure.

Thanks

dkhughes, 2016-04-11:
So, I took a look at getting Vivado in docker yesterday. I installed a new linux machine to use at my house and I figured it would be a good time to see if I could isolate my dev work from the main fs. Keep in mind, unlike Altera, Xilinx has made their web version still very locked down probably due to export restrictions. There is no perma-link that I can find to access the downloads, its all served per user after accepting their terms, and I didn't think I should circumvent their download terms.

**Try #1 **
I downloaded the net installer manually. It can be run in batch mode now, and I was able to have docker build extract the web install script, patch it to run on the Jessie image, and then launch from a config file. That's where the wheels fell off. To finish the download and install from a web file, you must provide a xilinx username and password, but there seems to be no mechanism to pass those in through the command line.

**Try #2 **
If you download the full installer locally, it will run in batch mode correctly since it doesn't need the web server credentials for the download. Once the program is installed, Xilinx still requires a license on their free edition, and unfortunately it's now activation based, not file based. I need to mess with this some more and see if there is a way to get it activated that is per image instead of per container. Besides the licensing, which affects all of the install methods, this try had the downside of a HUGE build context.

**Try #3 **
I didn't have as much time with this method. Instead of embedding the full installer in the docker build script, I mounted it on a share on my host. The advantage is a much smaller build context, and you have the choice to install it to the image or back into the share which may help with licensing. Need to check this out more.

The biggest advantage to running vivado with docker on a dev machine is that it will automatically patch the xsetup script to launch on a 64 bit Debian install, and keep all the vivado required libraries nicely isolated - plus the Dockerfile now remembers all of the required packages for me.

If anyone is interested in messing around with this I can put together a readme and upload it to my github.

cdsteinkuehler, 2016-04-11:
On 4/11/2016 7:25 AM, dkhughes wrote:
> So, I took a look at getting Vivado in docker yesterday.

Peter at Mesanet is still using ISE (version 13 IIRC).  I don't know
if that would side-step any of the issues you have with Vivado, or if
the ISE installs are locked-down as well.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-04-11:
is this helpful? https://github.com/mithro/docker-xilinx

dkhughes, 2016-04-11:
@mhaberler That link would have been helpful! They did it about the same way.

I just tried downloading a new license file from Xilinx instead of using my old one, and good news! They now use a wildcard for the host id. That fixed the licensing issue. You still have to generate your own license through their web page - it links to your user id - and download the full installer, but now I can generate a working vivado-gui docker image properly licensed.

You can try it at if[](https://github.com/dkhughes/vivado-docker). Drop your downloaded installer and license file in the release image folder and run the build script. Then run the generated run script with your image name and you should see a licensed vivado gui.

I hope its helpful. Now back to the rtl!

the-snowwhite, 2016-04-12:
@dkhughes 
I would very much like to work with Vivado instead of ISE for the mksocfpga's so I hope to pursade you to try this:

Make sure you have a lan interface named eth0 in the docker:
https://github.com/the-snowwhite/mksocfpga/blob/master/docs/Quartus-setup-notes/Altera-15.1-depedencies-and-udev-rules.md

Then create and download into the docker setup:

a new free node locked licensefile with * as mac address, and see if you can point the install script to that license file.

---

The eth0 trick is still a Quartus requirement(15.1) for node locked licenses,
So as I noticed upon installing Vivado on a different Debian Stretch PC that the mac address in the license setup read all 0's 

until I applied the eth0 rename udev rule fix.

licensing now workes fine, after pointing to the downloaded license file.

dkhughes, 2016-04-12:
@the-snowwhite I managed to get the licensing all worked out since Xilinx offers a license with the * in the mac address position. Xilinx does, however, still include your Xilinx username in the license file which is why I didn't publish my own with the git tree at [https://github.com/dkhughes/vivado-docker](url). 

The docker image at that url kicks out a working Vivado gui image, and a little helper script to launch the gui without having to remember all of the command line switches. By default, it also mounts a user specified share that you can point to your project files so they will persist across containers. Because of the export regulation stuff, and the personal account link stuff, I haven't published a prebuilt image to docker hub. Excluding the download time of the full installer, which granted is large, the docker build only takes around 10 minutes on my machines, and I've been successful with it on three.

If Xilinx adds a way to give user/pass to the netinstaller then that can easily be substituted instead of the full image.

the-snowwhite, 2016-04-12:
Well its great to know it's possible to get it to run.

About the export restriction stuff:

Both Xilinx and Altera are listed in wikipedia as US Companies.

AFAIK the in the US a corporation entity can wave the same human civil and rights as a normal (flesh and blod) human.

So a legal workaround possibility could be to create a Machinekit Xilinx account for the free web versions mksocfpga docker licensing ?

mhaberler, 2016-04-18:
noticed @RobertCNelson has started collecting telemetry on how to build Zynq images :https://www.eewiki.net/display/linuxonarm/Zynq-7000

dkhughes, 2016-04-18:
Little update. I have written an AXI-Lite shim for the hostmot2 that compiles in Vivado. I still have to finish the test bench and verify that it operates as expected. Next step is to get the device tree sorted out, and then compile machinekit to try and debug.

Do we have a documented memory map of where we would like the hm2 to exist - or do we just leave that up to the individual board and pull it out of the device tree? Should we try to standardize an address just for simplicity sake when reading the source of the different dts files? I have no preference so I figured I would get your input.

the-snowwhite, 2016-04-18:
The last implemention plan suggesion (for the first round ) we have been following is here:

https://github.com/machinekit/mksocfpga/issues/5

And my best suggestion for adding the first Zynq port is here:

https://github.com/machinekit/mksocfpga/tree/master/HW/VivadoProjects
---
That way the memory maps would be identical.
----
I expect / hope to be able to change the whole configuration scheme into something different than the original mesa source static config style.

By utilising the newer incremental partition / partial reconfiguration styles that were not possible earlier.

So the idea is old style for fast turnout now or soon.

New more flexible and dynamic style over the next number of months...

nilrods, 2016-04-18:
Nice! 
One thing to add, unless you plan on coding the dts manually most of the Xilinx tools generate a different string for the dts than the Altera ones. If I recall they all make the vendor Xilinx, no matter what you set the vendor field to(but I may be remembering wrong)

Just my thoughts.

the-snowwhite, 2016-04-18:
With the Altera chips it seems like it would be preferable to switch to using the kernel generated .dtb file from kernel version 4.4+ , I have no idea yet as to how far the Xilinx soc device-tree integration is.

I used these Xilinx doc sources underway as info about the device tree and drivers:

http://blog.idv-tech.com/wp-content/uploads/2014/09/drivers-session3-uio-4public.pdf

http://xillybus.com/tutorials/device-tree-zynq-5

http://fpga.org/2013/05/28/how-to-design-and-access-a-memory-mapped-device-part-one/

http://fpga.org/2013/05/28/how-to-design-and-access-a-memory-mapped-device-part-two/

----

But I was looking at it from an Altera perspective and not looking for how to include dts info via a Xilinx hm2 ip interface core.

Hand edited is ok for getting it to work first prototyping, but later it has to be automated / scripted somehow for image releases.

TomKeddie, 2016-04-18:
FYI I built an i/o expander for parallella that could be evolved into a cramps style board or act as a host for a grblshield.

https://parallella.org/forums/viewtopic.php?f=11&t=3081

dkhughes, 2016-04-19:
Okay, the shim is simulating acceptably now. 

@nilrods It's been a while since I've used Quartus for my SocKit so I'll have to keep that in mind. Right now I let Vivado kick out the dts and then I hand edit a little. This will definitely have to be automated.

I'm ready to get back to bundling the IP. Vivado is not happy about the pin files/IDROM being external when I try to package the hm2 IP, but this seems to be my inexperience with the tools. I will figure out a way to have hm2 customizable IP so the automatic system wrappers continue to work correctly, and we don't have to maintain a manual hdl top level file per configuration.

I can see two downsides to the plan at the moment. The first is physical IP project size. When you bundle IP for Vivado the package is pretty large. The other potential problem is vendor lock. Altera and Xilinx are not the only vendors making SoC's, and the other guys tend to make much more affordable chips. The mesa config style remains vendor independent, whereas the QSYS/Vivado IP pack methods require maintenance of multiple items going forward. 

That being said, I am trying to stick to @the-snowwhite 's README plan. Vivado handles some things fundamentally differently, though. For example, you don't need to add a .qip (xci) style file to all of the source folders; Vivado has an add_directory command in the tcl script that can also add all of the subdirectories. I'm able to add the entire hm2 directory source with one line of tcl in the project creation script, and I thought that was easy enough.

@TomKeddie  That's cool. I think there are a LOT of configurations that the Zynq/CVSoC chips could manifest in because the boards they ride on tend to be SoM. Right now I have my project stuff organized in a SoM/Carrier folder structure as discussed previously to try and keep track.

dkhughes, 2016-04-19:
I have managed to package the interface shim, and the hostmot2 as generic, configurable IP for Vivado. This required a couple minor things:

1) hostmot2 in the mksocfpga tree has been edited to look for a library package called pins, but I couldn't find any vhdl file defining the library. I just commented out the `library pins` and changed `use pins.PINXXX...` to `use work.PINXXX...` like the original source ball from mesanet had it. I can't think of any reason that would fail in an altera build, but I haven't tried one.

2) I had to add a wrapper hiding 4 of the generics exposed by hostmot2: the pin description, the moduleid, the boardnamehigh, and boardname low. Vivado's IP packager cannot handle user defined types exposed at the top level, and it doesn't like std_logic_vector either on my machine. The wrapper removes these from the generic statement, and simply maps them to the generic map of the hostmot2. That being said, all of the other generic configurables survived, and are exposed in the Vivado IP customization.

I learned that after packaging the Vivado IP, the content is actually very small. The large file sizes I was worried about before are only due to the temporary project created to edit the IP package, and they are deleted on completion. The IP packages for these components are happily in my git tree now.

Here is a little screenshot of the block design in vivado:
[image]

After constraining the pinouts of the design - viola! - a successful bitfile is produced. I would like to test it out on the software side before enabling interrupts. More to come...

nilrods, 2016-04-19:
@dkhughes
Wow, looks good! The bus shim was just beyond my skillset in vhdl. Glad to see someone be able to take that on. If you need any assistance on the uio driver or hm2 driver let me know. I am somewhat familiar with both from working on them with Michael for the nano port and my own machinekit FPGA components.
 
Looking forward to giving it a try once you get to a point you want testers.

the-snowwhite, 2016-04-20:
@dkhughes 
Great work...

I have just made a make all with

    changed use pins.PINXXX... to use work.PINXXX...

here:
https://github.com/dkhughes/mksocfpga/blob/master/HW/hm2/hostmot2.vhd#L70

        Info: Peak virtual memory: 1088 megabytes
        Info: Processing ended: Wed Apr 20 10:53:18 2016
        Info: Elapsed time: 00:07:34
        Info: Total CPU time (on all processors): 00:13:05
    quartus_cpf -c -o bitstream_compression=on output_files/soc_system.sof output_files/soc_system.rbf

I can confirm it compiles fine with that improvement.

The pins library is a leftover due to newer having worked with vhdl language before and my attempts to isolate the config files.
Just delete the faulty pins.PINXXX. line no need to leave that comment.

the-snowwhite, 2016-04-20:
Ip sharing between Altera and Xilinx would only be possible by having a (free..open) 3.rd party ip tool that could create ip cores for both flavors imho (it will eventually happen, but how soon is uncertain).

The (quartus) philosophy of using partial compilation to add configs is:

1)  add 1 revision (persona), including the basic 5i25 + the pin config now used.

then add new revisions (personas), using all the other PIN_xxx files we want to distribuate and create partial config files for each config (replacable hm2 core), and load them from linux.

This should reduce the compile time as only the hm2 part needs to be recompiled for each config added.

2) What I think of as the modular approach:

Create a static top interface structure inside the hostmot2.vhd file that can interface to all hm2 cores.
(hardware hardened)
create similar sized  blocks / chunks of grouped cores representing smallest unit of all practically needed combinations for running machines in the real world.

Each block is then used as a loadable / unload able persona as in example 1 , but more modular and fine grained than just daughter board configs.

If the rest of the design is staticially locked down into physical hardware correctly,  the dynamic module chunks / blocks should be able to compile very quickly... I think.

dkhughes, 2016-04-20:
Thanks for the input guys.

@nilrods Driver time is here, and I'm open for any suggestions on how to proceed.
@the-snowwhite I was thinking of managing the hm2 ip independently, not necessarily the vendor specific core implementation. The more I look at this, wrapping hm2 into vendor specific IP as we have done is easier to manage than keeping a generic hm2 and manually tying together via a top level HDL. Vivado's top level management abilities are too good to give up, and IIRC Qsys is much easier as well. 

Partial configs sound really cool, but I think the devil is in constraining the design so we can make closure. I know nothing about partial config yet, and how well it's integrated into the relevant tools.

Okay, here is the question I have: The hm2_soc driver makes Altera specific calls and checks as it programs the device before becoming device agnostic when it actually reads/writes to the hm2 in the fpga. I don't know the state of the Altera kernel, but in the Xilinx tree they have implemented Altera's fpga manager interface for the zynq. Should I take a look at being able to refactor the hm2_soc driver to use this higher level interface, or is the Altera side just not ready yet (i.e., do the generic fpga manager headers exist)? If the Altera kernel is too old for that, I would try to break the Zynq stuff into a new driver that is generically interfaced. Later, we could work back in the  CV's when the Altera tree catches up to have a monolithic driver.

I feel the kernel version we are running on the different boards could be an issue if Altera is still back in the 3.xxx's.

nilrods, 2016-04-20:
@dkhughes
Something must have changed since I last looked at it then if there are Altera specific calls now.
The way I understood it was working was as follows
1. When kernel boots it loads the uio kernel driver which maps the hm2 driver to userspace memory.
2. When machinekit is started it loads the hm2_soc hal driver and attaches to the uio kernel driver memory space and creates the hal pins.
3. Machinekit uses this and runs the write and read functions to write and read from hm2 fpga component.

None of this was Altera specific, since I used the same process on my AXI-lite basic stepper hal component I wrote and run on the Z-Turn board.

Just my thoughts

mhaberler, 2016-04-20:
@dkhughes yes, hm2_soc.c is currently assuming an altera environment (sysfs, device entries etc)

please work off machinekit/machinekit - the code in machinekit/mksocfpga is dead  and will be removed

once you have something working we can fold it in, we need to do automatic platform detection (usually involves checks in /sysfs /dev /proc/cpuinfo and whatnot entries and can be a bit heuristic; the raspberry gpio driver does a similar thing to figure rpi1 vs rpi2)

for the kernel drivers, see machinekit/machinekit-dkms - either adapt a driver there, or post a PR with a new one.

nilrods, 2016-04-20:
Just an observation. Xilinx puts all the fpga side items under amba_pl in the devicetree. Altera just lumps them in with the other devices. So a quick check would be to check that. If it is not there then you are running on Altera or your running on a Zynq with no FPGA components (so there would be nothing for the hm2 driver to do).

Just a thought.

dkhughes, 2016-04-20:
Looks like this is a really active topic over at Altera. Alan Tull has a device tree overlay version [here](https://github.com/altera-opensource/linux-socfpga/blob/socfpga-4.4/Documentation/devicetree/bindings/fpga/fpga-region.txt).

Effectively, all of the usermod direct managing of the FPGA has been removed and hidden. Vendor specific stuff, e.g., bridge enabling/disabling for reconfiguration, is also transparent to the end user. The fpga interface also handles probing/rmmoding any modules that are in the device tree in a proper order to avoid a crash. Sounds like a win-win-win. It requires a device tree overlay file that you would write to the configfs/sysfs interface, which kicks off programming the fpga, updating the active tree, and inserting any modules required by new devices. Of course, the opaque vendor specific bitfile is also required to be present.

I would think a query of the fpga manager sysf to get the vendor, and then a lookup based on a carrier eeprom id would be enough info for hostmot2 to pick the correct overlay and fire it into the pipe. Each persona would then just have a device-tree overlay and a bitfile to deploy. Or, maybe just pass in a config string as a parameter from upstream configuration files.

I think I will try to get his patches running on a 4.4 kernel to play with a little...

the-snowwhite, 2016-04-20:
@dkhughes 

I have already parked the altera socfpga 4.1-rt-ltsi release aside here:

https://github.com/machinekit/mksocfpga/issues/12

As a separate study item relevant for moving forward towards the Next release.

The rev 1 @mhaberler has in the release cannon crrently, is locked down to the 2013 GHRD Altera generated device trees the old 3.10 rt-ltsi kernel and 1 u-boot config load (that had to be changed to comply with the omap builder system).

I intentionally drew that line in the sand so we could finish and get that release out for people to play with ASAP.

4.1 mainline kernel is interresting, because it is supposed to bring in full BBB overlay support.

Whats interesting about the 4.4-rt mainline kernel is that it is supposed to bring into life the new fpga manager framework with device tree overlay support for both Xilinx And Altera chips and some partial config support, all (supposed to be) non vendor specific.

The basic idea is Altera / Xilinx ip core --> kernel compile(+ u-boot)  --> device tree.

No more need to have Quartus or Vivado generate the dtb binary.

Even thou some of this work may have been (vendor broken as usual ?)back ported into the Altera 4.1xxxxx repo this is IMHO only interesting for study purposes not mk production use.

I will still point towards mainline 4.4-rt-ltsi as the better alternative for rev 2. when it arrives.

I dont know where that leaves the possibility of making a quick rev 1. zync release (1 static config). That must be up to demand.

-----

The hm2 ip core idea seems like a better idea / concept for me to handle also instead of the top level parameter feed method.

I have no actual experience with partial reconfig either have only so far been studying it for some time.

What I have noticed is the Both Xilinx and Altera have provisions where you can have physical "black boxes" in a fixed region with flexible interface signals at the boundaries. 

This would help to minimize the timing closure problem ?

dkhughes, 2016-04-22:
Sooooo.....
```
root@mctrl:/home/administrator# cd /lib/firmware/
root@mctrl:/lib/firmware# mkdir /sys/kernel/config/device-tree/overlays/1
root@mctrl:/lib/firmware# echo microzed_jd2cb_overlay.dtbo > /sys/kernel/config/device-tree/overlays/1/path 
fpga_manager fpga0: writing soc_system_wrapper.bit.bin to Xilinx Zynq FPGA Manager
root@mctrl:/lib/firmware# hm2cvsoc 43c00000.hm2-socfpga: setting: uioinfo->irq = UIO_IRQ_NONE 

root@mctrl:/lib/firmware# /home/administrator/hm2-reg-mem-test -i -g 0
GPIO access through /dev/mem.

Address: 1040 Value: 4153454d
Address: 1044 Value: 35324935
root@mctrl:/lib/firmware# 
```
And to help understand from IDROMConst.vhd:
```
...
constant BoardNameMesa : std_logic_vector(31 downto 0) := x"4153454D";		-- MESA
...
constant BoardName5I25 : std_logic_vector(31 downto 0) := x"35324935";		-- 5I25
...
```
The address might be shifted by one position but i think that's in my c program reading the addresses. I think it would be rather coincidental to see those values right next to each other in memory (how they're organized in hostmot2) without indicating a successful read from the device. The device is automatically programmed via a dts overlay, and the framework Alan Tull, et. al. wrote also loads/unloads the hm2reg driver automatically via configfs. I had to cherry pick commits to get this into the Xilinx kernel, but it patched cleanly along side of RTPREEMPT in kernel 4.4.0. What's really exciting about this is that the Altera tree already has this working (that's where they are developing most of the fpga config stuff).  I can write more details a little later, but now I need to take a break.

But, yeah, hostmot2 says "Hello" from Zynq...

mhaberler, 2016-04-22:
@dkhughes - congratulations!

mhaberler, 2016-04-24:
@dkhughes - I think we could work ahead as follows:

if you provide me with a Zynq FPGA bitfile and the shell commands to load/unload it, I can look into hm2_soc.c to adapt it for the Zynq specifics - it does not even have to be running machinekit or an RT kernel to do that

the more  think about it, probably hm2_soc.c could support both platforms quite easily

what I would do is extend environment autodetection in [rtapi_kdetect.c](https://github.com/machinekit/machinekit/blob/90ba9f1b712bce4ba27a73e0d5c0704c32acb44e/src/rtapi/rtapi_kdetect.c) to detect socfpga and zynq specifics, add evidence bits to the syndrome bitmask, and make that available as a global variable in rtapi

this way autodetection happens once at startup, and any component could consult the syndrome bit vector to decide how to handle the task at hand like loading the FPGA, mapping memory, loading drivers and whatnot

the upside - no need to document anything, the standard hostmot2 driver command line arguments apply ;)

dkhughes, 2016-04-25:
Okay, I'm back. Right after I posted the other morning my wife came in and said it was time to go to the hospital. So, now with my newest little one helping, I'm ready to get the rest of this into mk. 

@mhaberler - I think that is a good idea and it should be really simple to detect an altera or zynq based build. The device tree for an fpga framework device will have a specific driver loaded depending on the hardware its riding on - socfpga-fpga-mgr for altera, and zynq-fpga-mgr for xilinx. We could look for these modules or investigate the device-tree compatible strings directly in /proc. The tricky part is going to be how to handle the different carrier boards available. Each one will require a specific bitfile that has been constrained for the available physical pinouts. Given a bitfile, it all becomes completely generic. Programming the device is done by writing to configfs - talking to hm2 is done by 32 bit r/w through uio.

> the more think about it, probably hm2_soc.c could support both platforms quite easily

Yep, I think hm2_soc.c can be completely generic and totally unaware of the fpga its on, once the altera side is updated to kernel 4.4 with the manager. I was going to work on this today by copying the hm2_soc to a new one, call it hm2_soc_ol (for overlay based) and work on it there. That way, any fpga manager designs could reference hm2_soc_ol, and all of the kernel 3 work everyone did for the altera side remains stable. Speaking of which, once the zynq stuff is working happily, I can pull out my SOCKit and work to help with v2 mentioned on the other thread.

OK, as for device files so other can start helping that's no problem. I can post a compiled kernel/uboot/bitfile right now. The rootfs I built using a hacked up omap-image-builder script is a little buggy so I don't think it's all that helpful. If there is another common Zynq board everyone else is working with I can also generate a u-boot and bitfile specific for that build. The kernel should be identical. Just let me know a board and where to drop the files.

Next post coming with details on the project status and todos.

dkhughes, 2016-04-25:
For the Zynq port I have it broken down into bitfile generation, u-boot, kernel, and then mk/driver work.

Bitfile generation is done through vivado projects. The project for the microzed on my custom carrier is the one you can find on my [zynq_port ](https://github.com/dkhughes/mksocfpga/tree/zynq_port) branch of mksocfpga. With the exception of removing the pin library reference discussed before, I did not need to change any of the hm2 files. The vivado project is autogenerated from a tcl script that is run in the Vivado gui. It will automatically create the block design required to drop the AXI components, hm2, and the Zynq device into the project. Todos:

- Automatic compilation of project without a gui - shouldn't be hard, just run vivado tcl in batch mode...

- AXI interface component - Functions in simple read case. Need to test write and debug. Once the last HAL layer details are figured out this will go much quicker.

- hostmot2 wrapper - This is a super thin component that just helps with compilation into IP for Vivado. Only thing that would be tweaked here is the address/data size and some of the other generic data to make it match hostmot2 from Altera side. It *should* already match.

- Other Zynq based board configs! - I only have a microzed but I'm willing to buy another make to help out once the microzed is functional.

U-Boot - U-boot itself is actually very generic for the zynq. The only reason we have to compile this from source is if you use the u-boot-spl, you can skip Xilinx SDK tools to get a FSBL. You *must* compile u-boot-spl for the specific board you are running, because it carries the DDR address layout/timings/etc. Todos:

- omap-image-builder customizations to auto build the correct u-boot-spl automatically?

Kernel - The kernel for zynq is 4.4.0 taken from Xilinx kernel git tree, and then patched with RTPREEMPT, and then patched with the latest fpga manager framework from altera's 4.4 kernel branch. This is the only moving target of the project. But(!), the only portion in flux is the configfs overlay handling. Everything else is pretty mature and stable for months or years. fpga-mgrs themselves are already mainlined, the bridges for altera devices are also stable and about to go mainline.  The ease of configfs is totally worth maintaining a patch set for the zynq kernel, and the altera kernel gets it for free. You can find the patched up kernel sources [here](https://github.com/dkhughes/linux-xlnx/tree/zynq_4_4_rt) on the zynq_4_4_rt branch. TODOS:

- Autobuild of kernel or download of built kernel in omap-image-builder?
- Figure out occasional hang on boot - this could be because of my rootfs. Hang only shows up when there is no ethernet connected and the device is dhcp. Maybe timeout too long in rfs init?

HAL and machine kit - TODOS:

- I think the hm2 uio driver in dkms will work without update. We need a hm2_soc that is capable of working with zynq. I don't think the original hm2_soc is very far off. I'm compiling mk now to begin work on this.

- Auto recognition of carrier to determine proper bitfile (the physical pinouts are embedded, the hm2 "pins" are still configured in the same manner as any hostmot2 design).
- ???
- Profit!

the-snowwhite, 2016-04-26:
@dkhughes Thank you very much for pointing in the right direction:

Got it working on my own buildscript repo in 4.1-ltsi-rt branch

@mhaberler Sorry: looking at the overlay stuff made me go into my silent introvert mode, until now:

the-snowwhite, 2016-04-26:
Loading hm2 uio driver and configuring fpga in socfpga 4.1-ltsi-rt  kernel

Via loading a device tree overlay fragment

      Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
      permitted by applicable law.
      Last login: Tue Apr 26 07:02:06 2016 from debian9-ws.holotronic.lan
      machinekit@mksocfpga:~$ 
      machinekit@mksocfpga:~$ 
      machinekit@mksocfpga:~$ lsmod
      Module                  Size  Used by
      autofs4                23922  1 
      machinekit@mksocfpga:~$ ls /dev/uio*
      ls: cannot access /dev/uio*: No such file or directory

      machinekit@mksocfpga:~$ sudo mkdir /config/device-tree/overlays/uio0
      machinekit@mksocfpga:~$ sudo sh -c 'cat hm2reg_uio.dtbo > /config/device-tree/overlays/uio0/dtbo'

      machinekit@mksocfpga:~$ lsmod
      Module                  Size  Used by
      hm2reg_uio              3113  0 
      autofs4                23922  1 
      machinekit@mksocfpga:~$ ls /dev/uio*
      /dev/uio0
      machinekit@mksocfpga:~$ 

Running machinekit:

      machinekit@mksocfpga:~$ machinekit
      Creating machinekit directory
      MACHINEKIT - 0.1
      Machine configuration directory is '/home/machinekit/machinekit/configs/hm2-soc-stepper'
      Machine configuration file is '5i25-soc.ini'
      Starting Machinekit...
      io started
      halcmd loadusr io started
      hm2-soc-stepper-5i25.hal:39: insmod failed, returned -2:
      rtapi_app_main(hm2_soc): -2 No such file or directory

      See /var/log/linuxcnc.log for more information.
      Shutting down and cleaning up Machinekit...
      Cleanup done
      Machinekit terminated with an error.  You can find more information in the log:
         /home/machinekit/linuxcnc_debug.txt
      and
         /home/machinekit/linuxcnc_print.txt
      as well as in the output of the shell command 'dmesg' and in the terminal
      machinekit@mksocfpga:~$

the-snowwhite, 2016-04-26:
Note Im using (pantoniou dt-overlays6  ) dtc-compile-tools v1.4.1:
https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/dtc-overlay.sh

overlay compile load and notes:
https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/inst-rbf-dtb.sh

Kernel patches:
https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/SD-Image-Gen/socfpga-4.1-ltsi-rt-changeset.patch

the-snowwhite, 2016-04-26:
This Method should work for Xilinx also with the kernel config:

CONFIG_FPGA_MGR_ZYNQ_FPGA=y 

instead of the Altera:

CONFIG_FPGA_MGR_SOCFPGA=y

Ref:  http://qiita.com/ikwzm/items/9eb4d49545636162057e

dkhughes, 2016-04-26:
@the-snowwhite Can you try two things? 1st - grab the kernel from altera's 4.4 branch. That's where the most stable config overlay stuff comes from, and that is where I pulled the fpga config patches for the Xilinx kernel. Then we'll be on equal footing for both device families. Also, you can use the in tree dtc to compile the device tree binaries, as long as your kernel version supports the -@ (compile symbol) flag - no need for an additional download. You compile it with make scripts in the kernel directory.

Second, Can you post dmesg from the fail?

mhaberler, 2016-04-26:
@dkhughes: ok, happy to try your bitfile/uboot/kernel but all I have is a Myir z-turn and a red pitaya

I want the current de0 image to continue using the 3.10 kernel for now as the newer altera kernel arent there yet AFAICT and that is one moving part too much - but then I will take care of modifiying hm2_soc.c as needed

re u-boot: shall be built outside omap-image-builder as I do not want any huge unmaintainable 'does it all' scripts but separate jobs and packages which can individually be upgraded; we currently have for the de0:

- the kernel deb
- u-boot - a deb too
- machinekit-dkms - any kernel driver needed; also debs
- the Quartus bitfile - also wrapped in a deb

finally (currently) omap-image-builder which just pulls the parts together
it's sufficient to point me to repo/branch/instructions  of partsand I will take care of the build automation

what we still DO need is a docker image with Vivado for machinekit/mksocfpga; please also the the issues in machinekit/mksocfpga; for instance naming bitfiles and configs are discussed there

mhaberler, 2016-04-26:
@the-snowwhite:  is that the image which I posted on sunday?  re your error - as the message says:

  See /var/log/linuxcnc.log for more information.

so pastebin that please - impossible to guess

dkhughes, 2016-04-26:
@mhaberler I posted a docker image script a while back on this issue. You can find it [here](https://github.com/dkhughes/vivado-docker). The licensing restrictions don't allow me to post a built image to docker hub.

Right, I wanted to work outside hm2_soc so you could keep 3.1 kernel for the working parts. I was talking about an upgrade for v2 as mentioned on the other issue.

Separate debs are no problem. I can package a set of u-boot and kernel patched up for your z-turn today probably if you want it.

mhaberler, 2016-04-26:
@dkhughes ah sorry, I overlooked that; ok - I can build that image

yep, moving parts for the zturn would be great. Is that the same stuff Claudio recommended?

dkhughes, 2016-04-26:
@mhaberler No worries, this thread has really grown. 
It isn't exactly the same. Claudio is pulling a kernel that is newer I think maintained by one of his colleagues. I was a little more conservative and stuck with the xilinx maintained version from their github before applying patches. U-boot I imagine is identical, the current xilinx u-boot has SPL support which eliminates the need to use xilinx SDK. 

I just realized...@the-snowwhite Is that an unmodified hm2_soc? I need to hurry up and build my mk debs. Zynq might be closer than I thought!

the-snowwhite, 2016-04-26:
@dkhughes   @mhaberler 
I am currently tied up in trying to rebuild the original first working version of the hm2_soc driver 
(Without all the fpga confugured ? checks and fpga config loading added later).

Reason:  This is what's making me not able to run machinekit, when using the device overlay to load the driver and configure the fpga. (in kernels 4.x +).

the-snowwhite, 2016-04-26:
> @the-snowwhite: is that the image which I posted on sunday? re your error - as the message says:
> 
> See /var/log/linuxcnc.log for more information.
> 
> so pastebin that please - impossible to guess

No this is a new image using device tree overlay method to configure the fpga and load the uio driver via the 2 commands:

    sudo mkdir /config/device-tree/overlays/uio0
    sudo sh -c 'cat hm2reg_uio.dtbo > /config/device-tree/overlays/uio0/dtbo'

Thats all ...!

Now if I could just get rid of all the "old style" hm2_soc driver checks related to the 3.10 kernel, preventing Machinekit from starting up in 4+ kernel overlay load style ?

That is what that error is about  + and no dmesg messages show up.

dkhughes, 2016-04-26:
Yeah I was hoping dmsg would say the file missing. I was working on getting the overlay automatically applied from hm2_soc on load because I quickly realized manually making the directory and writing the overlay each boot got tiresome. 

Can you call hm2_soc without any arguments? Doesn't that bypass loading the fpga firmware and basically use the already loaded one? That's what I meant when I said Zynq could be closer than I thought.

the-snowwhite, 2016-04-26:
@dkhughes 
oh  ...!
I had'nt noticed that the hostmot2_soc example had been changed to add firmware ....
moment  ..
need to rebuild image.

the-snowwhite, 2016-04-26:
!!¤%&

Nope ... this is annoying... I'm currently unable to compile a custom machinekit without the checks for fpga0 /dev status   that is totally missing in the new fpga manager framework ...!

I got this with and without trying to load a firmware file in hal:

    # load low-level driver
    loadrt [HOSTMOT2](DRIVER)
    #config=[HOSTMOT2](CONFIG)

    machinekit@mksocfpga:~$ machinekit
    MACHINEKIT - 0.1
    Machine configuration directory is '/home/machinekit/machinekit/configs/hm2-soc-stepper2'
    Machine configuration file is '5i25-soc.ini'
    Starting Machinekit...
    io started
    halcmd loadusr io started
    hm2-soc-stepper-5i25.hal:39: insmod failed, returned -2:
    rtapi_app_main(hm2_soc): -2 No such file or directory

    See /var/log/linuxcnc.log for more information.
    Shutting down and cleaning up Machinekit...
    Cleanup done
    Machinekit terminated with an error.  You can find more information in the log:
        /home/machinekit/linuxcnc_debug.txt
    and
        /home/machinekit/linuxcnc_print.txt
    as well as in the output of the shell command 'dmesg' and in the terminal

    machinekit@mksocfpga:~$ cat /home/machinekit/linuxcnc_debug.txt
    2174
    PID TTY      STAT   TIME COMMAND
    Stopping realtime threads
    Unloading hal components
    machinekit@mksocfpga:~$ cat /home/machinekit/linuxcnc_print.txt
    RUN_IN_PLACE=no
    LINUXCNC_DIR=
    LINUXCNC_BIN_DIR=/usr/bin
    LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
    LINUXCNC_SCRIPT_DIR=
    LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc
    LINUXCNC_CONFIG_DIR=
    LINUXCNC_LANG_DIR=/usr/share/linuxcnc/tcl/msgs
    INIVAR=/usr/libexec/linuxcnc/inivar
    HALCMD=halcmd
    LINUXCNC_EMCSH=/usr/bin/wish8.6
    INIFILE=/home/machinekit/machinekit/configs/hm2-soc-stepper2/5i25-soc.ini
    PARAMETER_FILE=hm2-soc-stepper.var
    TASK=milltask
    HALUI=
    DISPLAY=axis
    Starting Machinekit server program: linuxcncsvr
    Loading Real Time OS, RTAPI, and HAL_LIB modules
    Starting Machinekit IO program: io
    Killing task linuxcncsvr, PID=2174
    Removing HAL_LIB, RTAPI, and Real Time OS modules
    Removing NML shared memory segments

    machinekit@mksocfpga:~$ dmesg
    .....
    [   24.175471] random: nonblocking pool is initialized
    [   98.223752] fpga_manager fpga0: writing socfpga.rbf to Altera SOCFPGA FPGA Manager
    [   98.438761] hmsoc ff240000.hm2-socfpg0: setting: uioinfo->irq = UIO_IRQ_NONE 

Default example with:

    # load low-level driver
    loadrt [HOSTMOT2](DRIVER) config=[HOSTMOT2](CONFIG)

    machinekit@mksocfpga:~$ machinekit
    MACHINEKIT - 0.1
    Machine configuration directory is '/home/machinekit/machinekit/configs/hm2-soc-stepper'
    Machine configuration file is '5i25-soc.ini'
    Starting Machinekit...
    io started
    halcmd loadusr io started
    hm2-soc-stepper-5i25.hal:39: insmod failed, returned -2:
    rtapi_app_main(hm2_soc): -2 No such file or directory

    See /var/log/linuxcnc.log for more information.
    Shutting down and cleaning up Machinekit...
    Cleanup done
    Machinekit terminated with an error.  You can find more information in the log:
        /home/machinekit/linuxcnc_debug.txt
    and
        /home/machinekit/linuxcnc_print.txt
    as well as in the output of the shell command 'dmesg' and in the terminal

    machinekit@mksocfpga:~$ cat /home/machinekit/linuxcnc_debug.txt
    3330
    PID TTY      STAT   TIME COMMAND
    Stopping realtime threads
    Unloading hal components

    machinekit@mksocfpga:~$ cat /home/machinekit/linuxcnc_print.txt
    RUN_IN_PLACE=no
    LINUXCNC_DIR=
    LINUXCNC_BIN_DIR=/usr/bin
    LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
    LINUXCNC_SCRIPT_DIR=
    LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc
    LINUXCNC_CONFIG_DIR=
    LINUXCNC_LANG_DIR=/usr/share/linuxcnc/tcl/msgs
    INIVAR=/usr/libexec/linuxcnc/inivar
    HALCMD=halcmd
    LINUXCNC_EMCSH=/usr/bin/wish8.6
    INIFILE=/home/machinekit/machinekit/configs/hm2-soc-stepper/5i25-soc.ini
    PARAMETER_FILE=hm2-soc-stepper.var
    TASK=milltask
    HALUI=
    DISPLAY=axis
    Starting Machinekit server program: linuxcncsvr
    Loading Real Time OS, RTAPI, and HAL_LIB modules
    Starting Machinekit IO program: io
    Killing task linuxcncsvr, PID=3330
    Removing HAL_LIB, RTAPI, and Real Time OS modules
    Removing NML shared memory segments

dmesg unchanged:

    [   12.715989] stmmaceth ff702000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
    [   24.175471] random: nonblocking pool is initialized
    [   98.223752] fpga_manager fpga0: writing socfpga.rbf to Altera SOCFPGA FPGA Manager
    [   98.438761] hmsoc ff240000.hm2-socfpg0: setting: uioinfo->irq = UIO_IRQ_NONE

the-snowwhite, 2016-04-26:
OK In an effort to stop what I see as madness due to not having u-boot configure the kernel in rev 1 release, resulting in that every later effort to put something into the hm2_soc.c driver related to fpga configuration.

Had been contraindicated towards the 4+ kernels device-tree overlay support, and fpga manager. this has totally broken any easy upgrade path.

Here I try to spell out in more detail ...!

-----

The (delayed rev1 ) 3.10 fpga configure (not in u-boot as preposed) 

Growing kernel 4+ upstream compatibility problem

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.h#L52

This has been removed in the newer 4.1 & 4.4 + kernels

    const char *fpga_dev;       // "fpga0"
    

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L245-L247

Does not check if bridges are available.

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L297

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L300

if (!board->firmware_given)

Hmmm trouble ...

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L211
firmware_given = 1:

Whaaat ...!   -->  uncontional fpga config, by method only available in 3.10 kernel ?

dkhughes, 2016-04-27:
Nearly completed the overlay based driver for userspace. It is generic for Xilinx or Altera devices. I did separate it into hm2_soc_ol. That way, the kernel 3 version of the driver does just work as well. Once we get past this functioning I can clean it up / merge the files however everyone would like. Here is a summary of what had to change:

- Removed all references of bridge handling (that's done automatically now by the kernel drivers)

- The uio driver check in the beginning of main is now moved. This driver is automatically inserted and probed when the overlay is applied, right now that's done in the program function callback.

- FPGA state is now found by just checking the overlay directory status. If the overlay was applied successfully, you are guaranteed that the FPGA bit file was programmed correctly and all dependent modules were inserted.

- Overlay is removed on exit by deleting the created directory

Update soon...

mhaberler, 2016-04-28:
just noting I did a 'developer image' for the Altera socfpga - has the [machinekit src repo installed and compiled ready to go](https://github.com/machinekit/machinekit/issues/915#issuecomment-215223440) so any on-target HAL or driver mods are easy

the only thing specific to the socfpga is kernel, uboot, FPGA blob+dtb's
so this can be mutated  quickly into a Zynq developer image by removing those parts, and adding any Zynq specifics:

`````
#remove socfpga specifics:
apt remove -y	linux-headers-socfpga-rt	\
	linux-image-socfpga-rt	\
	socfpga-rbf	\
	u-boot-socfpga
# add any Zynq specifics
....
`````
Any drivers can be modified and tested on-target in the dkms directory (eg `/var/lib/dkms/hm2reg_uio/0.0.2`)
any update or new driver added to the [driver repo](https://github.com/machinekit/machinekit-dkms) (PR's as usual) will the corresponding package in the package stream to be added/updated automatically, so an apt update / apt install foo-dkms will pull those in and built for the current kernel

for the moving parts - kernel, uboot, bitfile generation - just point me to repo/branch/desired package name and I'll add those to CI build/apt repo insertion

dkhughes, 2016-04-28:
Okay, the hm2 driver I have now configures the fpga correctly, and I think I'm at the last hurdle. The driver is failing to load at the cookie read - time to debug the hardware more...

@mhaberler As soon as I get this bus debugged I will look into providing the info you listed in your last comment. A dev image would be nice for working on this HAL layer stuff.

dkhughes, 2016-04-29:
Machinekit is now running on Zynq with hostmot2. Still need to iron some kinks out, I'm getting a joint following error in axis about 15s into running a program. I must have a mistake in my config. As soon as I get access to my LSA I'll post some data from physical pins. Processor usage is around 40% while running axis over ssh. 

I would like to clean up the permissions scheme some, then begin submitting some PRs for code review. As for the bitfiles/kernel/etc. I will work on getting a base line package together for a z-turn since that seems to be popular.

mhaberler, 2016-04-29:
@dkhughes congratulations - superb news!

dkhughes, 2016-04-30:
I was going to grab some data with an LSA but I thought this would be more fun:

[https://youtu.be/_AcWxu475aY](https://youtu.be/_AcWxu475aY)

The Zynq is running the machine at 800ipm feeds with 1400ipm rapids. Gantry has two motors running in tandem. Seems to be working well. I'm cleaning up and organizing the work this weekend to start PRs.

cdsteinkuehler, 2016-04-30:
On 4/30/2016 8:38 AM, dkhughes wrote:
> I was going to grab some data with an LSA but I thought this would be more fun:
> 
> https://youtu.be/_AcWxu475aY

Very nice!

-- 
Charles Steinkuehler
charles@steinkuehler.net

[139 more replies]

closed 2018-08-01 by ArcEye