machinekit/machinekit #907: PR tests can pass on broken code

cdsteinkuehler, 2016-04-11:
The Travis-CI automated tests run for each pull request can pass, even if the code in the pull request is broken.  For an example, see the recent PR #904 which modified the encoder logic in the hal_pru_generic driver.  

The PR test build succeeded:
https://travis-ci.org/machinekit/machinekit/builds/122116394

But after a merge, the build from master failed:
https://travis-ci.org/machinekit/machinekit/builds/122133668

It appears the PR builds are not performing enough checks.  The build jobs that fail after apx. 15 minutes in master all run successfully in less than a minute in the PR test.

mhaberler, 2016-04-11:
I think what happened is:

- the PR checks only run the amd64 build - a shortcut @kinsamanka and me decided back then 
- the merge commit triggers the works, including armhf

this might mean any issues in ARM-only code is discovered during the merge only

cdsteinkuehler, 2016-04-11:
On 4/11/2016 2:22 PM, Michael Haberler wrote:
> I think what happened is:
> 
>   * the PR checks only run the amd64 build - a shortcut @kinsamanka
>     <https://github.com/kinsamanka> and me decided back then
>   * the merge commit triggers the works, including armhf
> 
> this might mean any issues in ARM-only code is discovered during the merge only

ARM-only code seems like a large chunk to not check.  It seems like
there should be at least one build from each choice (Wheezy/Jessie,
x86/amd64/armhf, posix/xenomai/rt_preempt), but I can understand not
running all permutations.

...and maybe build the posix target for all arches, since that seems
to be what triggers the build of all the HAL modules?

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-04-17:
@kinsamanka  - see https://travis-ci.org/machinekit/machinekit/jobs/123687503
did we exceed the travis free time limit?

kinsamanka, 2016-04-17:
I don’t think so.

Maybe some issue with Google’s compute engine.

mhaberler, 2016-04-27:
fixed by Gemi. Arm builds are checked now too.

closed 2016-04-27 by mhaberler