machinekit/machinekit #908: automatic generation of manpages for documentation

luminize, 2016-04-12:
(please correct title if something more appropriate is needed)

@mhaberler I think we can remove the machinekit-man repo from the machinekit organisation. Agree?

I have merged @mhaberler branch groff-convert-make-support into @ArcEye branch [groff-convert](https://github.com/ArcEye/machinekit/tree/groff-convert) into my own branch [groff-convert](https://github.com/ArcEye/machinekit/tree/groff-convert).

a first run of `make whole-sausage` yields this [pastebin](http://pastebin.com/y1PM1GRc) which raises some warnings and errors.
I'll take these info and work on the source files to get no warnings and no errors. When I have updated some files I'll make a PR to @ArcEye (and merge PR's to my repo+branch when there's something new)

luminize, 2016-04-12:
or would it make more sense to make PR's of commits of updated sources against machinekit:master and rebasing the local branches?

ArcEye, 2016-04-12:
> @mhaberler I think we can remove the machinekit-man repo from the machinekit organisation. Agree?

On the contrary, if we create a branch `groff-convert` under that, we can all PR to the same MK branch rather than having copies of private repos dotted around.

* Whatever conversion tool(s) is / are used they must be available as packages to Wheezy and Jessie.
in order to meet requirements of docker or whatever containers, plus building on supported platforms.

pandoc does not do direct groff to html/xml conversion, but does lots of others.
Knowing the output format required will determine the tool probably.

I will look at the components that are throwing errors, a lot of the old .comps had troff formatting embedded in them, especially in the comment headers.

mhaberler, 2016-04-12:
I'm trying now two formats:

docbook to html - just to see what the result looks like, but not a good base format thereafter:

	`xsltproc -o $@.html /usr/share/xml/docbook/stylesheet/docbook-xsl/xhtml-1_1/docbook.xsl $@`

and see what that yields:
also https://blogs.gnome.org/pmkovar/2015/10/27/converting-docbook-into-asciidoc/, to see if the resulting asciidoc can be carried further as a base format

ArcEye, 2016-04-12:
All the .icomp man pages are already formatted and spaced to produce good .asciidoc pages

They are converted by outputting to a .txt file and then converting with pandoc, with will also do .md and .html
conversions

https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/maintainence/man2asciidoc.sh

ArcEye, 2016-04-12:
https://github.com/ArcEye/machinekit/tree/groff-convert/man/examples

has a txt file generated directly from an icomp man page and the subsequent pandoc conversions 
of that txt file into asciidoc, md and html

luminize, 2016-04-12:
@mhaberler @ArcEye I will continue with making sausages and correcting the component sources until I have no errors with doclifter. 

Since we aim to generate readable/renderable information, let's not get blinded by sticking with asciidoc at least for these generated manpages. Seems pointless to me since one shouldn't change the generated asciidoc anyway. So a chain with as little conversions/steps seems easiest to me.

mhaberler, 2016-04-12:
yes, agree  - step1 is to correctly convert anything to DocBook - that is the better representation than a .txt generated by `man` 

A cursory initial conversion docbook to html can be had by adding to the rule like so: https://github.com/mhaberler/machinekit/commit/14db7d18f9cacfe70b5529af36d6c52306438819

Step 2 is the conversion to whatever we decide to carry forward as manpage source format ; I think asciidoc is better than github markdown even if it requires a formatting step; also the rest of the docs is asciidoc anyway

ArcEye, 2016-04-12:
OK, but just be mindful that errors like this
```
"man9/bitslice.9icomp", line 27: error - ill-formed NAME section ' \fBbitslice\fR' in man9/bitslice.9icomp, giving up.
"man9/bitslice.9icomp", line 27: error - uninterpreted 'SH' command
"man9/bitslice.9icomp", line 78: there were 2 errors during translation
```
that occur when running doclifter on the man file from icomps, are failing on something which has already
been generated by another program, so the source cannot be 'corrected'

Need to find out why exactly the same formatting which works with nearly every other icomp, throws errors
with bitslice, lutn and wcompn

The comp man pages that fail are also generated, but I have not altered that formatting at all, so the error could be embedded in the .comp file itself

luminize, 2016-04-12:
@ArcEye My feeling it's mostly with omitting some trivial info in the source files, like the string after the name in the groff source like here https://github.com/luminize/machinekit/commit/cfc46aaa3bf2610c80e0708ff1e26982d38abade

I just dumbly ran Michaels addition for `.html` and `.asciidoc` rules and there's a big difference between the output of files man/xml-docs/man9/weighted_sum.9.xml.asciidoc is one of the best yet, while man/xml-docs/man9/siggen.9.xml.asciidoc doesn't come near.
The .html generated files look OK so far.

I'll look into the differences later today. I think there is a big cleanup needed in the different source files. No shortcuts which were "ok" up to now :)

ArcEye, 2016-04-12:
It is largely the source files - I agree.
Found 2 instances just now where `/f` is not followed by `B` or `I` and thus it swallows the first letter of the word which should be highlighted or italicised and throws a warning.

Check the ones I have already committed before going back to it later.

luminize, 2016-04-12:
Just a moment of reflection...

Are we doing the right thing?

Shouldn't we stop using troff in the sources, and use asciidoc/markdown/reST or whatnot, and use that to generate other formats (docbook, HTML etc plus generate manpages) from that.

I mean, troff as source isn't very inviting for people to make contributions.

mhaberler, 2016-04-12:
static manpages: yes, once converted to asciidoc, let's forget about the nroff source

now, of course if somebody comes around to generate asciidoc from comp, then... 

but if the conversion nroff/docbook/asciidoc work without manual intervention, we're good for that part too

ArcEye, 2016-04-12:
The problem with abandoning legacy man pages, is on the minimalist, single board, non internet connected setup.

I personally don't use them and can always look up what I need on the web, but nearly all the other users seem to think it is mandatory to use just this type of low power ARM board for everything with MK.

>  now, of course if somebody comes around to generate asciidoc from comp, then... 

A challenge I shall resist for now :tongue:

luminize, 2016-04-12:
Well, I think that we could still have manpages for those who want them, but have them generated from asciidoc in the source.

Instead of: source with troff -> manpage -> docbook/whatnot -> html 

Source with asciidoc -> manpage or docbook/whatnot or html 

I understand that this needs comp to generate a proper asciidoc. I'm wondering aloud if reviewing troff in the sources and manpages as part of the building documentation chain is a future proof setup.

The main reason: lowering the barrier. It's not really a 2016 way for documenting in troff. Big hurdle teeth grinding experience. Especially first time. Think about manpage for socfpga, or pru documentation... I would certainly be held back by troff. I did the delayline manpage and i didnt really got excited by that.

We can discuss this in a separate issue if that's more appropriate.

ArcEye, 2016-04-13:
@luminize 
After a nights rest, I think the only sane way to go is to have one base format for all docs as you say.

This will entail a fair bit of work, but it will only have to be done once.

There is no hurry to get the end docs produced, that was simply prompted by @mhaberler's discovery of the ease of use of Jenkins for CI.

The underlying mess we have discovered converting from an unsatisfactory base format (troff / groff / nroff) with additional formatting embedded in actual sources of binaries, probably requires a root and branch approach.

1.  Convert all static man1 and man3 pages to asciidoc.  Probably just requires scripting a mass conversion and manually checking results and tweaking as required.

2.  Change comp / instcomp to generate asciidoc pages for components

3.  Change .comp and .icomp component code to alter groff `/fB... /fR` type bold highlighting embedded in the source, to asciidoc `*...*` etc.

4.  Change .c sources to generate asciidoc.  They are run through comp, so the sources again will need any groff `/fB... /fR` type bold highlighting embedded in the source, replaced with asciidoc `*...*` etc.  AFAIK there is not much or any, most of the text used is lifted from the MODULE_XXXX  module information, so those need to exist and be as meaningful as possible.  The other formatting is applied by comp and will be dealt with.

If this seems like a plan, I suppose I will have to do No. 2 :frowning: 

Once we have all the docs present or being built in asciidoc format, the makefile conversions, web hooks etc will be the easy part.

Then we probably back to the aborted document conversion discussion of last year and may yet get a CI based `machinekit.io/docs` !!

mhaberler, 2016-04-13:
[image]

from:  http://elcubanocafe.blogspot.co.at/2011/07/hook-line-and-sinker.html

luminize, 2016-04-13:
> On 13 Apr 2016, at 11:03, ArcEye <notifications@github.com> wrote:
> 
> @luminize <https://github.com/luminize> 
> After a nights rest, I think the only sane way to go is to have one base format for all docs as you say.
> 
> This will entail a fair bit of work, but it will only have to be done once.
> 
I agree, let’s not make rash decisions. It will cost us a lot of sweat :)
> There is no hurry to get the end docs produced, that was simply prompted by @mhaberler <https://github.com/mhaberler>'s discovery of the ease of use of Jenkins for CI.
> 
we can still decide to first get some output to publish (getting results as quickly and basic as possible). I think the html generated pages are pretty good so far. That will at least get us some continuous up-to-date basic documentation out into the open.
> The underlying mess we have discovered converting from an unsatisfactory base format (troff / groff / nroff) with additional formatting embedded in actual sources of binaries, probably requires a root and branch approach.
> 
I think that roo. Also, there’s one step before No 1. Test if the .asciidoc is good enough to use as a source. If we have trouble later on with converting asciidoc to manpage or the other formats we might have chosen the wrong format.
A bit of searching yielded me this: http://www.methods.co.nz/asciidoc/chunked/ch24.html but also asciidoctor should be able to do this http://asciidoctor.org/docs/render-documents/#selecting-an-output-format (see ‘manpage’)
> Convert all static man1 and man3 pages to asciidoc. Probably just requires scripting a mass conversion and manually checking results and tweaking as required
> 
I will look if I can find a good converted .asciidoc manpage in the sausage build and convert it back to manpage. That will give us an idea what it takes (we’ll need that for No. 2 )
> Change comp / instcomp to generate asciidoc pages for components
> 
> Change .comp and .icomp component code to alter groff /fB... /fR type bold highlighting embedded in the source, to asciidoc *...* etc.
> 
> Change .c sources to generate asciidoc. They are run through comp, so the sources again will need any groff /fB... /fR type bold highlighting embedded in the source, replaced with asciidoc *...* etc. AFAIK there is not much or any, most of the text used is lifted from the MODULE_XXXX module information, so those need to exist and be as meaningful as possible. The other formatting is applied by comp and will be dealt with.
> 
> If this seems like a plan, I suppose I will have to do No. 2 
> 
happy to help out when we decide to do this 
> Once we have all the docs present or being built in asciidoc format, the makefile conversions, web hooks etc will be the easy part.
> 
> Then we probably back to the aborted document conversion discussion of last year and may yet get a CI based machinekit.io/docs !!
>

ArcEye, 2016-04-13:
Ignoring @mhaberler's joy at suckering us into hard work...... :facepunch: 

If you can do *1.* for now,  I am already part way through doing *2.*

When I have instcomp producing full .asciidoc man pages, you will be able to apply reverse conversion to troff and conversions to html, then we will see how well it converts and what tweaks may be required.

luminize, 2016-04-13:
> On 13 Apr 2016, at 12:53, ArcEye <notifications@github.com> wrote:
> 
> Ignoring @mhaberler's joy at suckering us into hard work...... 
> 
> If you can do 1. for now, I am already part way through doing 2.
> 
I will do 1. :)

RunningLight, 2016-04-13:
Just noticed this flurry of activity. Have you tried the Ruby gem `asciidoctor`? From its documentation

> Asciidoctor is a fast text processor and publishing toolchain for converting AsciiDoc content 
to HTML5, EPUB3, PDF, DocBook 5 (or 4.5) slidedecks and other formats.

Well, it looks like they lied about PDF---at least for the version I'm using, and asciidoctor-pdf is in its infancy---but I was able easily to create an asciidoc file for `comp` by hand and then generate comp.1 and comp.html files from it. 

To generate comp.html
```
$ asciidoctor -d manpage comp.adoc
```

To generate comp.1
```
$ asciidoctor -d manpage -b manpage comp.adoc
```

Have a look at https://drive.google.com/folderview?id=0BxgxhFDH9ROSYWxSX0dkN2o4SWM&usp=sharing

Note: I chose to install asciidoctor using the Ruby infrastructure to make sure I got a recent version
```
sudo gem install asciidoctor
```

This loaded asciidoctor 1.5.4 (into Ruby 1.9.3p194) on my Debian Wheezy 7.9 VM.

RunningLight, 2016-04-13:
PS - There is an anomaly in my manpage output. The asciidoc markup for italic text was converted to underscored text (compare _sudo_ in the HTML vs the manpage).

ArcEye, 2016-04-13:
> Have you tried the Ruby gem asciidoctor? 

Yes I have used it previously.

Presently we are getting all the man page documentation _written_ in .asciidoc

Then we will try all the different conversion programs to see which gives the best results for conversion to man page and to html, it may not be the same one!

ArcEye, 2016-04-13:
@luminize 

I am doing *3.* now for icomps

Out of necessity, because until they are cleaned up, I cannot test the output from instcomp properly and be sure what is causing any glitches.

Thus far looking promising, instcomp is generating .asciidoc files direct into man/man9 for now

luminize, 2016-04-13:
> On 13 Apr 2016, at 15:51, ArcEye <notifications@github.com> wrote:
> 
> @luminize
> 
> I am doing 3. now for icomps
> 
> Out of necessity, because until they are cleaned up, I cannot test the output from instcomp properly and be sure what is causing any glitches.
> 
> Thus far looking promising, instcomp is generating .asciidoc files direct into man/man9 for now
> 
@ArcEye Oh my, sounds like your up to speed. Do you have something to push to github? Then i'll pull and work on no 1 with your generated asciidocs

ArcEye, 2016-04-13:
> Do you have something to push to github?

Just done it
https://github.com/ArcEye/machinekit/commits/groff-convert
Last 4 commits with todays date

*Current state of play*

instcomp is modified to produce .asciidoc files in man/doc/man9
Most .icomps have been edited to remove groff formatting and to improve their appearence in asciidoc

Some icomps remain to be done
bldc_hall3 for example has huge wadges of groff formatting in tables, which need converting

That's me for today :smile:

mhaberler, 2016-04-13:
whoa! great to see the end of the seventies.. ;)

mhaberler, 2016-04-13:
hm2_soc status:

- usability now improved - firmware can be loaded HAL-time
- boot-time fpga loading not needed
- firmware will live in the standard `/lib/firmware/...` location, using the hostmot2-built in firmware code

any number of 
`````
loadrt hostmot2
loadrt hm2_soc config="firmware=socfpga.rbf"
unloadrt hm2_soc
loadrt hm2_soc config="firmware=socfpga.rbf"
unloadrt hm2_soc
````
commands succeeds - no more crashes.

The key contribution was - thanks to @the-snowwhite's stamina: the `gpio_altera` module wreaks havoc on fpga loading.

the proper incantation (from shell) is [this](https://github.com/mhaberler/machinekit/blob/431ee9c459661744bbc242e545503b0089243e37/scripts/hm2soc_program_fpga). The hm2_soc.c driver [sticks exactly to this sequence](https://github.com/mhaberler/machinekit/blob/431ee9c459661744bbc242e545503b0089243e37/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L180-L258) - all peachy.

luminize, 2016-04-13:
I did some tinkering and the asciidoc files that got generated are a beauty to read.

I first tried generating the manpage with `asciidoctor` (the ruby gem) but I ran into troubles with the backend. I wanted to use something working out of the box, so I `sudo apt-get install asciidoc` which unfortunately pulls in tex atcetera. Not sure if we want that dependency on the beaglebones etcetera, but that's for later. I can imagine this being a problem if you build from source PLUS you'd want to generate manpages next to nicer) asciidoc files.

So after installing `asciidoc` i only had to do some minor changes to get a resulting manpage. It got generated with the following command `a2x --doctype manpage --format manpage test2_and2.asciidoc` (test2_and2 was my copy of the generated and2.asciidoc)

The source which was used (revision 1 was the generated asciidoc, revision 2 were the changes that had to be made to get the manpage to be generated by a2x):
https://gist.github.com/luminize/6b63537b68c2cd506a9f28a536309767/revisions

the resulting manpage
https://gist.github.com/luminize/c660fb695cf10656fa2d43d6320e1f24

some notes:
* following conventions must be used regarding the description header (first line), first NAME section and second SYNOPSIS section http://www.methods.co.nz/asciidoc/chunked/ch24.html
* I had to unindent some headers from `=== Header` to just 2 like `== Header` but I saw that adding more indentation works too (so `=== another header` gives more indentation).
* the manpage that gets generated gets it's file name from the NAME section, so if I were to change this line https://gist.github.com/luminize/6b63537b68c2cd506a9f28a536309767#file-asciidoc-to-roff-L8 in `foobar - Two-input AND gate` then the manpage would be `foobar.9`
* there's apparently no way of generating a component with other extensions than digits, like `.9icomp`
* the date when the the manpage was made seems to be in the middle of the last line https://gist.github.com/luminize/c660fb695cf10656fa2d43d6320e1f24/revisions#diff-751dc2761e629b132579370b7b9217ddR48 and in this case the `:man version:` tag too
* these 3 lines https://gist.github.com/luminize/6b63537b68c2cd506a9f28a536309767#file-asciidoc-to-roff-L2-L4 seem to bring some extra info like the manual (HAL Components), source (and2.icomp) and version (date? commit?)
* these lines https://gist.github.com/luminize/6b63537b68c2cd506a9f28a536309767/revisions#diff-9eab729d04b39dd001b319f419b37043L9 couldn't be used since a2x first converts to `.xml` and that step throws an error

mhaberler, 2016-04-13:
@luminize whatever formatting is needed could possibly be shifted to build time - the asciidoc toolchain on-target certainly doesnt make sense

luminize, 2016-04-13:
@mhaberler ok, was just more of an observation. I find it a bit hard to think that the youth of today is in very urgent need of manpages anyways, especially if one is developping.

luminize, 2016-04-13:
@ArcEye I did some reshuffeling and small fixes and now instcomp should generate convertable asciidoc files. https://github.com/luminize/machinekit/commit/d2b0f1b6432de9515a4dc396f2e34c754f0d6420 the thing I didn't do yet is to get the info on the source file name+extension (like component.icomp) into the attributes like here https://github.com/luminize/machinekit/blob/d2b0f1b6432de9515a4dc396f2e34c754f0d6420/src/hal/utils/instcomp.g#L977
That would be nice to have the manpage show the source file

ArcEye, 2016-04-14:
> the resulting manpage
https://gist.github.com/luminize/c660fb695cf10656fa2d43d6320e1f24

The man page looks OK but what does the asciidoc look like?
All the things taken out were there to improve the format and appearance of the asciidoc.

> these 3 lines https://gist.github.com/luminize/6b63537b68c2cd506a9f28a536309767#file-asciidoc-to-roff-L2-L4 seem to bring some extra info like the manual (HAL Components), source (and2.icomp) and version (date? commit?)

They don't add anything, that info (except the source, which is obvious but could easily be added) was in the top of the page anyway.
That is how the existing instcomp man pages were, because that is how instcomp has written them for some while, so writing that way directly to man page works fine, it is just converting back from asciidoc that is causing problems.

**The answer to this is so glaringly obvious that we have all failed to see it in our concentration upon conversion to and from different formats**.

The current code produces perfectly good manpages which work in the legacy manner.

If I duplicate the writing process to **also** produce an .asciidoc for the same component, for minimal time processing, we get a custom formatted version which is best able to render good quality output in the chosen medium.

I don't have a lot of time today, but since both outputs are already written, it will just be a case of combining and creating switches to trigger the output required and building those into the Submakefile.

What do you think?

Should make things much easier - any tweaks to improve output will not have knock-on effects into the other format.

luminize, 2016-04-14:
> On 14 Apr 2016, at 10:37, ArcEye <notifications@github.com> wrote:
> 
> the resulting manpage
> https://gist.github.com/luminize/c660fb695cf10656fa2d43d6320e1f24 <https://gist.github.com/luminize/c660fb695cf10656fa2d43d6320e1f24>
> The man page looks OK but what does the asciidoc look like?
> All the things taken out were there to improve the format and appearance of the asciidoc.
> 
mostly the things I did were to get a2x to convert asciidco ->  xml -> manpage without errors.
the resulting asciidoc:  https://gist.github.com/luminize/6b63537b68c2cd506a9f28a536309767
> these 3 lines https://gist.github.com/luminize/6b63537b68c2cd506a9f28a536309767#file-asciidoc-to-roff-L2-L4 [...]
I thought that it would more clear to see if you’re dealing with an icomp or comp
> That is how the existing instcomp man pages were, because that is how instcomp has written them for some while, so writing that way directly to man page works fine, it is just converting back from asciidoc that is causing problems.
> 
correct
> The answer to this is so glaringly obvious that we have all failed to see it in our concentration upon conversion to and from different formats.
> 
> 
> The current code produces perfectly good manpages which work in the legacy manner.
> 
yes, but at least taking manpages out of the building chain for documentation has advantages.
> If I duplicate the writing process to also produce an .asciidoc for the same component, for minimal time processing, we get a custom formatted version which is best able to render good quality [...]
smart thinking. Provided that we keep asciidoc as the way to document, and forget all about roff in the component sources. That’s really a big hurdle and downplays the ease of comp. I really like that this way asciidoc and manpage get generated without additional dependencies. Asciidoc can be used for building/converting to other formats, and the manpages should not be part of building other stuff. I think there’s not a lot maintenance on instcomp for that. And we pave the way for a way to get a modern way of source documentation with little effort.
> Should make things much easier - any tweaks to improve output will not have knock-on effects into the other format.
> 
yes. added bonus!

ArcEye, 2016-04-14:
Good.

I am building in switches to instcomp for man page docs and asciidocs.

These can be used by the Submakefile, but also work internally if someone uses instcomp to generate docs themselves.
By selecting or not, man page builds can be completely removed from the build, we could even make it a configure switch (or cmake define) which determines the doc build and does not do so unless specifically selected.  (one for later)

> They don't add anything, that info (except the source, which is obvious but could easily be added) was in the top of the page anyway.
I thought that it would more clear to see if you’re dealing with an icomp or comp

When I get to carry the same changes over to comp, I will put something in to instcomp and comp docs to show which it is, albeit the section about how instcomps are invoked, should be a clue for those :wink:

ArcEye, 2016-04-14:
OK, instcomp now creates both man pages and asciidocs
https://github.com/ArcEye/machinekit/commit/c1c3243a892db3ce6102237b3d7ff6da6f650bea

These are selectable using --document (-d) and --ascii-document (-a) switches
The install option will work for both types, I have created the code but not updated the switches yet.
The view option is just restricted to man page, as this is the only one that makes sense in a terminal environment.

The Submakefile currently creates both types by default - we need that to test for now.
`make clean` for asciidoc not working, need to look at it later.

I have changed some component source formatting to something that renders a reasonable page in both groff and asciidoc (and2, or2, xor2 all had asciidoc lists which did not work for groff)
The rest of the .icomps output need checking between the icomp man pages and aciidoc, but that is all I have time for until later today

Feel free to play:)

luminize, 2016-04-14:
> On 14 Apr 2016, at 12:33, ArcEye wrote:
> 
> Feel free to play:)

Will do!

ArcEye, 2016-04-14:
https://github.com/ArcEye/machinekit/commit/ee26eda60fda48e03052e1b0f93640ddf2bd2b6f
Fixes the make clean and inserts file origin for the docs as requested.

The --install-???? switches should now work too, not had time to test yet.

Should have more time tomorrow, so will try to sort out the source files with heavy groff formatting in tables etc. eg. bldc_hall3 and then check through all the icomp generated files in both formats.

luminize, 2016-04-14:
> On 14 Apr 2016, at 18:50, ArcEye <notifications@github.com> wrote:
> 
> ArcEye@ee26eda
> Fixes the make clean and inserts file origin for the docs as requested.
> 
> The --install-???? switches should now work too, not had time to test yet.
> 
> Should have more time tomorrow, so will try to sort out the source files with heavy groff formatting in tables etc. eg. bldc_hall3 and then check through all the icomp generated files in both formats.
> 
Been a bit busy myself. Hope to have some more time end of tomorrow afternoon.

ArcEye, 2016-04-15:
Starting to look at the components that have major groff formatting for tables etc.

What is actually present is a very 'clever' groff if /else block which uses one set of formatting for a html table and another for man page.

This results in man pages being rendered well and asciidoc being unrecognisable.

The tables (bldc, lut etc) are embedded in the DESCRIPTION field which is lifted as a single block by comp / instcomp from the option `description ".................";` in the component, so no easy possibility of intervention.

We either need to completely rewrite components that display perfectly as man pages, or perhaps introduce another option heading for specialised formatting, such as tables, after `description`.

Something like options of `special-format-man ".............";`  and  `special-format-doc ".............";`
Then comp / instcomp can simply read in the option field that applies and render that.

We will just need to format an asciidoc table to replace the html one then, should be able to cut and paste into a blank doc and convert with pandoc etc.

It is a bit of work for a few components, but with those as examples, it will pretty much 'future-proof' the system for any similar uses.

Unless I hear stunning alternative thoughts, that is where I am heading.

ArcEye, 2016-04-15:
> Something like options of special-format-man "............."; and special-format-doc ".............";
Then comp / instcomp can simply read in the option field that applies and render that.

That works perfectly :+1: 

man pages exactly as before, just need to convert the html tables to asciidoc 

Commissioning new laptop interfering with progress right now!

luminize, 2016-04-15:
> On 15 Apr 2016, at 13:57, ArcEye <notifications@github.com> wrote:
> 
> Something like options of special-format-man "............."; and special-format-doc ".............";
> Then comp / instcomp can simply read in the option field that applies and render that.
> 
> That works perfectly 
> 
> man pages exactly as before, just need to convert the html tables to asciidoc
> 
need help?
> Commissioning new laptop interfering with progress right now!
> 
that's a perfectly good reason.

ArcEye, 2016-04-15:
> need help?

If you could update your repo and do a build, that should give you a set of man pages in /man/man9 and a set of asciidoc in /man/doc/man9

Ignore any with tables etc in them, I will be dealing with those

Need to compare the man page, asciidoc page and the source component and make any alterations required to the source to get the best output.

(eg. Noticed that if there is a blank line after or before `"""` multi-line text, a '*' charactor gets output in groff  -- that kind of thing)_

Don't alter the formatting in instcomp for now or we are not comparing like with like.
Now that we are not converting groff > asciidoc, that seems to work well.

ArcEye, 2016-04-16:
Sitrep

Tables converted to asciidoc instead of HTML and instcomp options work perfectly in deciding which version to render.

src/hal/i_components edited from a - hbridge and committed
https://github.com/ArcEye/machinekit/commit/542b540e33aa876481465135ce2cf52f31a989b9

I'll edit in ascending alphabetical order until we meet somewhere, so keep me updated.

Enough for now, my eyes need a rest ;)

mhaberler, 2016-04-16:
@ArcEye - I feel with you ;)

luminize, 2016-04-16:
> On 16 Apr 2016, at 13:26, ArcEye <notifications@github.com> wrote:
> 
> Sitrep
> 
> Tables converted to asciidoc instead of HTML and instcomp options work perfectly in deciding which version to render.
> 
> src/hal/i_components edited from a - hbridge and committed
> ArcEye@542b540 <https://github.com/ArcEye/machinekit/commit/542b540e33aa876481465135ce2cf52f31a989b9>
> I'll edit in ascending alphabetical order until we meet somewhere, so keep me updated.
> 
will do, bit of a chaotic weekend so my time will be a bit low. Mostly in the evening when the noise levels in the house have died down.
> Enough for now, my eyes need a rest ;)
> 
well earned!

ArcEye, 2016-04-17:
Next batch done 
https://github.com/ArcEye/machinekit/commit/41ff20c1a0eb410f72df1040ef511ddc03c18a4b

@luminize What did you find was the best conversion from the built in man pages to asciidoc?

Once the icomps are done, need to convert all the pre-existing xxxx.9 man pages.

luminize, 2016-04-18:
@ArcEye 
I did some small tests with converting the manual written manpages (man9/*.9) and I'll convert all of those manpages to .asciidoc this evening with a small adapted script. They only need small manual cleanup afterwards to be able to generate the asciidoc back to manpages for those who can't live without manpages.

After that we need to check the formatting of instcomp that gets the same layout (in asciidoc) for both the manually written (which need to have the ability to be converted back to manpage) and the ones generated by instcomp. (headers and other trivial things)

mhaberler, 2016-04-18:
just a heads up on the next step - build integration - and what is happening there:

@kinsamanka did the docker images for building machinekit on all of amd64,i386,armhf and for wheezy, jessie and raspbian - those are used by the travis builds and also the jenkins server I've set up

now we're about to extend hose images by a few packages needed for other tasks like kernel builds and uboot (https://github.com/machinekit/mk-builder/issues/2) 

rebuilding the docker images is a very slow process, so not phun to do too often - we have not started this yet as there is no urgency (I did an interim docker image which addresses my needs)

but it would be good to think through which packages you need for manpage generation eventually so we can slide these packages into the next image build

then the mk-builder scripts (jessie-64 - need to do this on one target only) can take care of any formatting and upload the results (to wherever - TBD)

I guess `asciidoc pandoc doclifter` and then what?

no urgency on this - good enough to state your shopping list when you are done

luminize, 2016-04-18:
>but it would be good to think through which packages you need for manpage generation eventually so we can slide these packages into the next image build

>then the mk-builder scripts (jessie-64 - need to do this on one target only) can take care of any formatting and upload the results (to wherever - TBD)

>I guess asciidoc pandoc doclifter and then what?

even better, I converted the asciidoc to manpage with `asciidoctor -d manpage -b manpage classicladder.9.asciidoc` afaik `asciidoctor` is a very minimal package (other than asciidoc which needs tex and whatnot)

mhaberler, 2016-04-18:
yeah the stock asciidoc is a tad top-heavy ;)

ArcEye, 2016-04-18:
@mhaberler 
> I guess asciidoc pandoc doclifter and then what?
no urgency on this - good enough to state your shopping list when you are done

We should not need doclifter at all, as we are going to generate man pages from asciidoc.

I suspect just pandoc and asciidoctor, those are the only 2 that I normally use, we should know soon.

@luminize 
> After that we need to check the formatting of instcomp that gets the same layout (in asciidoc) for both the manually written (which need to have the ability to be converted back to manpage) and the ones generated by instcomp.

Rather than change instcomp to fit the formatting of handwritten man pages, which have no standardisation and no common generator, we would be better changing the far smaller number of handwritten pages to match instcomp, which we know does produce consistent rendered man pages and asciidocs.

Then we have one standard - our own - not something we inherited from what whoever wrote a page decided to do.

luminize, 2016-04-18:
> Rather than change instcomp to fit the formatting of handwritten man pages, which have no standardisation and no common generator, we would be better changing the far smaller number of handwritten [...]
I'll check the conversion from one of the asciidoc generated from instcomp to manpage and see how much slack we have in our own sectionning. 

I think in the end the limitations (if any) from generating manual written asciidoc to manpages with asciidoctor will determine the level of required standardization.

I'll update in a few hours :)

ArcEye, 2016-04-18:
You beat me to the post @luminize 

I have been experimenting with converting the existing legacy static man pages into asciidoc.
Have tried all sorts of combinations, doclifter into xml, pandoc xml to asciidoc etc etc.
I am not impressed by the results.  

They all require manual reformatting to look anything like as good as the asciidocs and man pages from instcomp.
Certainly not any sort of automatic process.

This is a one-off effort with the legacy man pages and the required corresponding asciidocs.  
Once they are converted, they will only ever require editing for changes.
We don't need to build these man pages or asciidocs each build
  
Everything else regards components, is generated automatically from a build.

I know the answer, but have been resisting it, hoping that an 'off-the-shelf' conversion tool would make it all easy for us.

I think I need to write a documenting GUI.
It should have all the possible fields from a standard man page and a free text one for AOB.
The basic engine already exists in instcomp.
(subject to a couple of tweaks, one @luminize  has already highlighted, using a progression of header sizes = / == / === for grumble free conversion to html )

In converting the existing man pages, all we then would need to do is cut and paste the actual text from each heading into it and it would render man page and asciidoc according to the formatting rules already established.

The same GUI would be able to be used for any new documentation of a similar nature, with the benefit
that it would all be standardised.

NB.  Conversion from asciidoc into html is another matter.
I know that works very well, it is the native conversion for asciidoctor and it comes out little changed from the asciidoc but in html format.
So conversion of asciidocs for web base documentation is not the problem

Check out the examples
https://github.com/ArcEye/machinekit/commit/10b202e156bc19f248e2d7828731d3f60433a731

ArcEye, 2016-04-18:
Something like this

[image]

luminize, 2016-04-18:
> On 18 Apr 2016, at 17:51, ArcEye <notifications@github.com> wrote:
> 
> In converting the existing man pages, all we then would need to do is cut and paste the actual text from each heading into it and it would render man page and asciidoc according to the formatting rules already established.
> 
that's not needed in this case. I’ll have them done this evening.

luminize, 2016-04-18:
great! Could serve as a template

do you plan to read an asciidoc back into those text area’s?

One thing we could do is make some bare template (in asciidoc format) which people can copy and extend upon.
(but I guess that the amount of new manpage writings will be a bit low)

luminize, 2016-04-18:
@ArcEye well... not really finished this evening, but tomorrow will be better since I nog have the workflow ready. Strange that there are always small glitches that need to be worked out :)

The converted manpage -> asciidoc : https://github.com/luminize/machinekit/commit/620c16f20640ffa790033841866f4ff18e2badbe
and the fixed up page for generating for the to be automated asciidoc -> manpage : https://github.com/luminize/machinekit/commit/c9709cec1a52cacd4288cc04714ab1267071fe73

I continue tomorrow again ;)

ArcEye, 2016-04-19:
Good work, I will plough on with the.icomp changes and output checking.

Once that is done, need to carry over the instcomp changes into comp, so that it too automatically produces both types of docs and then we are not far from the light.

ArcEye, 2016-04-19:
instcomps finished https://github.com/ArcEye/machinekit/commit/088b764a14bf1d61880c1e9d46eae7f361d30fd2

Now amending `comp` to produce same output as `instcomp`

(That and 6 other things at once - as per usual)

RunningLight, 2016-04-19:
You folks are doing great work.

A minor quibble about rebranding in man pages. I think it's a bit of a stretch to say, for example, "This man page written by Alex Joni, as part of the Machinekit project". I know I'm always too wordy for other peoples' taste, but it would be more accurate and less confrontational to say something like "This man page originally written by Alex Joni, as part of the LinuxCNC project, and subsequently edited for use in the Machinekit project".

[41 more replies]

closed 2016-05-30 by ArcEye