machinekit/machinekit #915: Terasic de0 nano SD image  build incomplete

mhaberler, 2016-04-18:
just an anchor issue for the rest of the process of the mksocfpga project

mhaberler, 2016-04-18:
where we are:

the hm2_soc driver is now bulletproof (ha!), part of the stock packages in deb.machinekit.io

the following jobs are fully automated by push against their respective repos on github:

- the FPGA firmware build, rbf, dtb, dts generation and packaging - repo https://github.com/machinekit/mksocfpga, deb: socfgpa-rbf - thanks, @cdsteinkuehler !
- uboot for the de0 nano: build and packaging, repo https://github.com/mhaberler/u-boot/tree/v2016.03-terasic-de0-nano-machinekit-squashed,  deb: u-boot-socfpga
- kernel modules: dkms-packaged, repo https://github.com/machinekit/machinekit-dkms, debs: hm2reg-uio-dkms adcreg-dkms 

all of those are apt-upgrade-able, the dkms kmods rebuild fine on target; the uboot dd-into-partition command is not automated yet

lacking automation: linux-image-3.10.37-ltsi-rt37-05841-g81c6be3 linux-headers-3.10.37-ltsi-rt37-05841-g81c6be3, still a manual upload but good enough for a start with the SD image

so now turning to final pass over omap-image-builder for the de0 nano - I guess I'll step back a bit and will start from Robert's jessie minfs console image, the X stuff doesnt make sense due to lack of framebuffer/hdmi

the builds are all done through jenkins (many thanks to @RobertCNelson !) 

the dkms vehicle actually works fine for all architectures, so any drivers for mk in the future will go into the machinekit-dkms repo and things happen automatically

the package naming will have to be cleaned up once we figure out a taxonomy for FPGA configuration naming

ps: I put the work-on-progress debs into 'deb [arch=armhf] https://deb.mah.priv.at/ jessie socfpga' for now - for  less confusion in deb.machinekit.io - will switch eventually

mhaberler, 2016-04-20:
Here is the fairly final image for the de0-nano: 

    http://deb.mah.priv.at/uploads/de0-nano/debian-8.4-machinekit-de0-armhf-2016-04-20/

Please download, test and post feedback here.

This is a fully automatic build (except for the kernel which follows soonish). 

Hence, to fix any issues, post PR's as follows:

- hm2_soc.c hostmot2 HAL driver - directly against https://github.com/machinekit/machinekit
- FPGA firmware: https://github.com/machinekit/mksocfpga
- hm2reg_uio etc kernel drivers: https://github.com/machinekit/machinekit-dkms
- u-boot: https://github.com/mhaberler/u-boot/tree/v2016.03-terasic-de0-nano-machinekit-squashed
- root file system content:  https://github.com/mhaberler/omap-image-builder/blob/socfpga-try2/configs/machinekit-debian-jessie-de0.conf
- post-rootfs-install massage: https://github.com/mhaberler/omap-image-builder/blob/socfpga-try2/target/chroot/machinekit-jessie-de0.sh
- target description for SD image build: https://github.com/mhaberler/omap-image-builder/blob/socfpga-try2/tools/hwpack/socfpga-de0-nano.conf

@the-snowwhite - can I ask you to produce a pinout description comprehensible by mortals, and in proper asciidoc documentation format? This must be added to https://github.com/machinekit/machinekit-docs which is the _only_ place for final documentation.

ArcEye, 2016-04-20:
Can't wait to get out from under the mound of documents someone tipped on me and try it;)

My interface pcb card courtesy of @cdsteinkuehler's brilliant work is already on order.

luminize, 2016-04-20:
> On 20 Apr 2016, at 20:39, ArcEye <notifications@github.com> wrote:
> 
> My interface pcb card courtesy of @cdsteinkuehler's brilliant work is already on order.

As are mine :)

cdsteinkuehler, 2016-04-20:
On 4/20/2016 3:57 PM, Bas de Bruijn wrote:
>  > On 20 Apr 2016, at 20:39, ArcEye <notifications@github.com> wrote:
>  >
>  > My interface pcb card courtesy of @cdsteinkuehler's brilliant work is already 
> on order.
> 
> As are mine :)

Are you ordering boards from OSH Park?

If so, please be aware that the design is untested so far, and at a
minimum will require fiddling with the FPGA pinout to get the I/O pins
to line up with how I routed them on the adapter board.  At worst,
there could be something fundamental I did wrong.

Caveat emptor!  ;-)

If everything _does_ work as planned, I have parts to build up three
boards.  I'm keeping one, but the other two will be available for
someone who wants to do testing but doesn't like SMT soldering.

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-04-20:
> On 20 Apr 2016, at 23:06, cdsteinkuehler <notifications@github.com> wrote:
> 
> On 4/20/2016 3:57 PM, Bas de Bruijn wrote:
> > > On 20 Apr 2016, at 20:39, ArcEye <notifications@github.com> wrote:
> > >
> > > My interface pcb card courtesy of @cdsteinkuehler's brilliant work is already 
> > on order.
> > 
> > As are mine :)
> 
> Are you ordering boards from OSH Park?

yes

> If so, please be aware that the design is untested so far, and at a
> minimum will require fiddling with the FPGA pinout to get the I/O pins
> to line up with how I routed them on the adapter board. At worst,
> there could be something fundamental I did wrong.
> 
> Caveat emptor! ;-)

no pain, no gain

> If everything _does_ work as planned, I have parts to build up three
> boards. I'm keeping one, but the other two will be available for
> someone who wants to do testing but doesn't like SMT soldering.

I was planning to do the soldering after I hear success stories :)
So that I at least know if/what to correct.

the-snowwhite, 2016-04-20:
zzzzz .. wow zzzz :-)

ArcEye, 2016-04-21:
> Caveat emptor! ;-)

I am getting mine through someone who is far more competant than I am in surface mount soldering etc.

> If so, please be aware that the design is untested so far, and at a minimum will require fiddling with the FPGA pinout to get the I/O pins to line up with how I routed them on the adapter board. 

That is the thing I need to get my head around, it is far from clear to me what the actual physical pinout currently is or how to change it.

mhaberler, 2016-04-21:
ok, now onto the final frontier: _usability_!

We need a blinkenleds demo - I tried here, does not work: 

https://gist.github.com/mhaberler/fbadbb07f54cebde045fd9fff9d5f43a

@the-snowwhite - can you get it going and explain the pin relation for mortals?

the-snowwhite, 2016-04-21:
PInout file:
https://drive.google.com/open?id=0BwyLvgyVIdi8b0NUM0pwWkUxUXc

mhaberler, 2016-04-21:
ok.. @the-snowwhite just revealed.. CR0 and CR1 -  remember to wire your own 2 leds to gpio 0 pins 39 & 40
it's not the on-board leds.. searching for leds..

mhaberler, 2016-04-21:
ok, bug1 in image: ssh -X display forwarding doesnt work - xauth is missing

sudo apt update
sudo apt install xauth

logout/login again

will be in next version

mhaberler, 2016-04-21:
ok, after an embarrassing blunder on my side the blinkenleds.hal demo works

the layout of the GPIO 0 header in [p26 Figure 3.18](http://www.terasic.com.tw/attachment/archive/941/DE0-Nano-SoC_User_manual.pdf), lower left, is confusing:  it shows pin 1 left top whereas the photograph shows pin 1 right bottom.. so the leds are the toplevel 2 pins on the left connector as per photograph!

the-snowwhite, 2016-04-21:
Running the current image and installing the auth package gives a working system that boots very fast after a re-login.

However setting the hw mac address in u-boot invites disaster:

U-Boot SPL 2016.03-00014-g9f7ba25 (Apr 20 2016 - 16:46:59)
drivers/ddr/altera/sequencer.c: Preparing to start memory calibration
drivers/ddr/altera/sequencer.c: CALIBRATION PASSED
drivers/ddr/altera/sequencer.c: Calibration complete
Trying to boot from MMC

U-Boot 2016.03-00014-g9f7ba25 (Apr 20 2016 - 16:46:59 +0200), Build: jenkins-u-boot-socfpga-on-arm-13

CPU:   Altera SoCFPGA Platform
FPGA:  Altera Cyclone V, SE/A4 or SX/C4, version 0x0
BOOT:  SD/MMC Internal Transceiver (3.0V)
       Watchdog enabled
I2C:   ready
DRAM:  1 GiB
MMC:   dwmmc0@ff704000: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Terasic DE0-Nano(Atlas)
Net:
Error: ethernet@ff702000 address not set.
No ethernet found.
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Checking for: /boot/uEnv.txt ...
trying interface=mmc bootpart0:2 ...
279 bytes read in 34 ms (7.8 KiB/s)
Loaded environment from /boot/uEnv.txt
fpgaimage= not set, skipping FPGA load...
Checking if uname_r is set in /boot/uEnv.txt...
Running uname_boot ...
loading /boot/vmlinuz-3.10.37-ltsi-rt37-05841-g81c6be3 ...
3552688 bytes read in 220 ms (15.4 MiB/s)
loading /lib/firmware/socfpga//soc_system.dtb ...
21547 bytes read in 74 ms (284.2 KiB/s)
loading /boot/initrd.img-3.10.37-ltsi-rt37-05841-g81c6be3 ...
2199444 bytes read in 146 ms (14.4 MiB/s)
debug: [console=ttyS0,115200 root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait coherent_pool=1M quiet] ...
debug: [bootz 0x01000000 0x07080000:218f94 0x07000000] ...
Kernel image @ 0x1000000 [ 0x000000 - 0x3635b0 ]
## Flattened Device Tree blob at 07000000
   Booting using the fdt blob at 0x7000000
   Loading Ramdisk to 03de7000, end 03ffff94 ... OK
   Loading Device Tree to 03dde000, end 03de642a ... OK

Starting kernel ...

dwc2 ffb40000.usb: unable to find phy
dwc2 ffb40000.usb: 2560 invalid for host_rx_fifo_size. Check HW configuration.
systemd-fsck[72]: rootfs: clean, 71765/217728 files, 357088/869888 blocks
hm2cvsoc ff240000.hm2-socfpga: setting: uioinfo->irq = UIO_IRQ_NONE

Debian GNU/Linux 8 mksocfpga ttyS0

Machinekit Debian Image 2016-04-20

Support/FAQ: http://www.machinekit.io/

default username:password is [machinekit:machinekit]

mksocfpga login: machinekit
Password:
Last login: Thu Apr 21 10:47:23 UTC 2016 from debian9-ws.holotronic.lan on pts/0
[... 185 lines cut ...]
    the exact distribution terms for each program are described in the
    individual files in /usr/share/doc/*/copyright.

    Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
    permitted by applicable law.
    Last login: Thu Apr 21 12:27:46 2016
    machinekit@mksocfpga:~$ machinekit
    MACHINEKIT - 0.1
    Machine configuration directory is '/home/machinekit/machinekit/configs/hm2-soc-stepper2'
    Machine configuration file is '5i25-soc.ini'
    Starting Machinekit...
    io started
    halcmd loadusr io started
    hm2-soc-stepper-5i25.hal:39: insmod failed, returned -5:
    rtapi_app_main(hm2_soc): -5 Input/output error

    See /var/log/linuxcnc.log for more information.
    Shutting down and cleaning up Machinekit...
    Cleanup done
    Machinekit terminated with an error.  You can find more information in the log:
        /home/machinekit/linuxcnc_debug.txt
    and
        /home/machinekit/linuxcnc_print.txt
    as well as in the output of the shell command 'dmesg' and in the terminal
    machinekit@mksocfpga:~$

mhaberler, 2016-04-21:
@the-snowwhite - any suspicion? memory ranges used by u-boot? could well be I bungled it up

mhaberler, 2016-04-21:
I stepped back to v2016.01 and the hang is gone
@the-snowwhite - to verify, could you download this test binary and see if this more meets your palate: http://static.mah.priv.at/public/socfpga/u-boot-with-spl-dtb.sfp 

this is the branch - just loading the kernel and dtb, no fpgaimage no initrd: https://github.com/mhaberler/u-boot/tree/v2016.01-machinekit-testfix

I'm only interested if the environment save/reset/reboot stuff works, not if the environment is clever or not - this is just for testing

if that works, I'll carry over the environment and we go with v2016.01 as you always suggested

the-snowwhite, 2016-04-21:
I can confirm I worke fine with u-boot 2016.01

    sudo bmaptool copy debian-8.4-machinekit-de0-armhf-2016-04-20-4gb.img.xz /dev/sdj
    sudo dd if=u-boot-with-spl-dtb.sfp of=/dev/sdj1

set hw mac address (without hang)

   
ssh -X machinekit@mksocfpga.local

    sudo apt update
    sudp apt install -y xauth
    
    machinekit

mhaberler, 2016-04-21:
@the-snowwhite  - thanks a lot!

the-snowwhite, 2016-04-21:
Same 2u !! :-)

mhaberler, 2016-04-21:
there is one issue - I wonder if you see that as well with the v2016.01 u-boot as just posted:

`apt install -y strace`

immediately after reboot, login and run `top` 

I see `systemd-udevd` running 100% on end - take it's pid and run `strace -fvvp <pid>`

you should see tons of:
`````
symlink("../uio0", "/dev/uio/")         = -1 ENOENT (No such file or directory)
stat64("/dev/uio", 0x7e874430)          = -1 ENOENT (No such file or directory)
mkdir("/dev", 0755)                     = -1 EEXIST (File exists)
symlink("../uio0", "/dev/uio/")         = -1 ENOENT (No such file or directory)
stat64("/dev/uio", 0x7e874430)          = -1 ENOENT (No such file or directory)
mkdir("/dev", 0755)                     = -1 EEXIST (File exists)
symlink("../uio0", "/dev/uio/")         = -1 ENOENT (No such file or directory)
stat64("/dev/uio", 0x7e874430)          = -1 ENOENT (No such file or directory)
mkdir("/dev", 0755)                     = -1 EEXIST (File exists)
symlink("../uio0", "/dev/uio/")         = -1 ENOENT (No such file or directory)
stat64("/dev/uio", 0x7e874430)          = -1 ENOENT (No such file or directory)
mkdir("/dev", 0755)                     = -1 EEXIST (File exists)
`````

this seems a more general systemd issue, @cdsteinkuehler has observed it on a BB with jessie as well. 

See also http://unix.stackexchange.com/questions/233247/why-is-systemd-udev-pegging-my-cpu

a workaround which does the trick for me is:
`````

the-snowwhite, 2016-04-21:
THe uio driver without fpga config is  the culprit:
this seemt so solve the problem:

   root@mksocfpga:~# cat /etc/modprobe.d/fbdev-blacklist.conf
   # This file blacklists most old-style PCI framebuffer drivers.

   blacklist hm2reg_uio
   blacklist arkfb
   blacklist aty128fb
   blacklist atyfb
   blacklist radeonfb
   blacklist cirrusfb
   blacklist cyber2000fb
   blacklist kyrofb
   blacklist matroxfb_base
   blacklist mb862xxfb
   blacklist neofb
   blacklist pm2fb
   blacklist pm3fb
   blacklist s3fb
   blacklist savagefb
   blacklist sisfb
   blacklist tdfxfb
   blacklist tridentfb
   blacklist vt8623fb

mhaberler, 2016-04-21:
oh ;)

mhaberler, 2016-04-21:
well that gets rid of the looping udevadm, unfortunately it also gets rid of hm2_soc ;)

````
Apr 21 18:22:41 mksocfpga msgd:0: hal_lib:626:rt hm2: loading Mesa HostMot2 driver version 0.15
Apr 21 18:22:41 mksocfpga msgd:0: hal_lib:626:rt hm2_soc: loading Mesa AnyIO HostMot2 socfpga driver version 0.8
Apr 21 18:22:41 mksocfpga msgd:0: hal_lib:626:rt hm2_soc: hm2reg_uio not loaded!
.. copout...
`````
so I dont see how blacklisting hm2reg_uio can be a fix :-/

the options I see:

- always load an rbf file - hm2_soc will overwrite it anyway but could become a support issue
- rename or remove `/etc/udev/rules.d/uio.rules` which makes udev ignore uio events but I'm unsure about side effects

tick box please

the-snowwhite, 2016-04-21:
Well yes the simple fast out of the door 3.10 release is grounded in u-boot --> fpga config ... hm2_uio autoload. (period).... as stated earlier :-)

the 4.4 release is founded in dynamic devicetree updates (via overlays), while loading the driver. (fpga mgr framework).

The line in the sand concept between release 1 (GHRD 2013 --> u-boot 2016.01) and release 2 (of which software mostly still to be in rapid development mode 4.4.x kernel), 
Points out that keeping these 2 conceps (static  vs dunamic) configuration methods separate leads to a lot less pain and frustration by having to deal with weird flaky problems.

A bit like the man moving stone uphill story ...

the-snowwhite, 2016-04-21:
So uio driver load has to be put into hm2_soc.c driver to modprobe or insmod the hm2_uio kernel driver load,,, after it has configured the fpga ... ?

mhaberler, 2016-04-21:
that would be an option, yes - but not now

I go for the 'always load rbf in uboot' and be done with it

the-snowwhite, 2016-04-21:
Agree totally.

the-snowwhite, 2016-04-21:
Otherwise next struggle would be how to put the correct info into the device tree....

mhaberler, 2016-04-21:
ok, Michael and me did more investigation offlist:

The udev looping problem is **only** conditional on.

- the hm2reg_uio driver loaded (automatic boot-time or manually later makes no difference, as well as blacklisting or not).
- udevd is triggered into action on a uio device by a non-empty  `/etc/udev/rules.d/uio.rules` file.

It has no relation to the fpga being loaded or not, wether by u-boot or userland.

so it seems the only way forward to avoid the hang on reset AND to avoid the looping issue (which is btw temporary - 30secs) is to remove the `/etc/udev/rules.d/uio.rules` file. 

Given the fact that lots of people have that problem (which seems related to the udevd -> systemd transition), I assume a fix will appear in due course

It is a wart, we file it as 'Known issue' for now and go ahead.

mhaberler, 2016-04-22:
uio/udev: just noting there is a [related thread in progress ](https://groups.google.com/forum/#!searchin/beagleboard/High$20CPU$20usage$20after$20loading$20a$20cape$20on$20Jessie/beagleboard/VmkFhClgGaY/EvDR-92kBgAJ) at the Beagleboard google group

cdsteinkuehler, 2016-04-22:
On 4/22/2016 5:43 AM, Michael Haberler wrote:
> uio/udev: just noting there is a related thread in progress 
> <https://groups.google.com/forum/#!searchin/beagleboard/High%2420CPU%2420usage%2420after%2420loading%2420a%2420cape%2420on%2420Jessie/beagleboard/VmkFhClgGaY/EvDR-92kBgAJ> 
> at the Beagleboard google group

For the BBB Jessie image, commenting out the first line of uio.rules
(with the SYMLINK+ entry) fixes the CPU usage issue.

Perhaps this will help with the DE0 image as well.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-04-22:
that line reads
`````
SUBSYSTEM=="uio", SYMLINK+="uio/%s{device/of_node/uio-alias}"
`````
the failing systemcall is: `symlink("../uio0", "/dev/uio/") `
this would suggest that there is something wrong with the `{device/of_node/uio-alias}` replacement

could be a devictree issue, copying how it's supposed to work from http://forum.43oh.com/topic/9072-wip-uio/  :
`````
# put in /etc/modprobe.d/uio.conf
#
# make driver match on  compatible = "uio";
options uio_pdrv_genirq of_id=uio
 
 
# put in /etc/udev/rules.d/uio.rules
#
# create named symlink to locate the device easily
# (assumes your kernel is new enough to have of_node symlinks in sysfs)
SUBSYSTEM=="uio", SYMLINK+="uio/%s{device/of_node/uio-alias}"
#
# give some group access rights (adjust as needed)
SUBSYSTEM=="uio", GROUP="users", MODE="0660"
 
 
# example device tree node to give userspace direct access to ADC
&tscadc {
        status = "okay"; 
        compatible = "uio";    # <------- I guess we need 
        uio-alias = "adc";       # those lines - adapted for hm2reg_uio and adcreg
};
 
 
# device will appear as /dev/uio/adc     # <---- that would be great - no guesswork in hm2_soc.c!!
`````` 

I'll give it a spin

cdsteinkuehler, 2016-04-22:
On 4/22/2016 7:42 AM, Michael Haberler wrote:
> |# device will appear as /dev/uio/adc # <---- that would be great - no
> guesswork in hm2_soc.c!! |
> 
> I'll give it a spin

The other way to fix the problem is to create a /dev/uio directory.

On the BBB, when I did this the systemd-udevd tasks no longer hung,
but there were no symlinks created in /dev/uio/ either.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-04-22:
so this is happenening at least for the socfpga 3.10 kernel:

- the `of_node`  symlink feature [5590f319](https://sourceforge.net/p/android-x86/kernel/ci/5590f3196b293574a12be58d06d5e1120d8856ec/) was added in 2015
- the current 3.10 kernel [doesnt have that feature ](https://github.com/mhaberler/linux-socfpga/commits/gemify/drivers/base/core.c)(source at 2013)

no wonder the string replacement fails 

however, to get a driver-specific uio name for our use case this does work:

`````
#SUBSYSTEM=="uio", SYMLINK+="uio/%s{device/of_node/uio-alias}"
SUBSYSTEM=="uio", ATTR{name}=="hm2-socfpga", SYMLINK+="uio/hm2-socfpga"
SUBSYSTEM=="uio", GROUP="users", MODE="0660"
...modprobe hm2reg_uio...
root@mksocfpga:~# ls -l /dev/uio/hm2-socfpga 
lrwxrwxrwx 1 root root 7 Apr 22 15:03 /dev/uio/hm2-socfpga -> ../uio0
`````
I'll use that later, not a priority for this version

figured via `udevadm info -a -p /devices/sopc.0/c0000000.bridge/ff240000.hm2-socfpga/uio/uio0`

summary: `SYMLINK+="uio/%s{device/of_node/uio-alias}"` works only for very recent kernels, and in RT land that is unlikely to be _now_

mhaberler, 2016-04-24:
I've prepared a new test image debian-8.4-machinekit-de0-armhf-2016-04-24

see[ instructons in this gist](https://gist.github.com/mhaberler/89a813dc70688e35d8848e8e467a1337)

changes from [previous image](https://github.com/machinekit/machinekit/issues/915#issuecomment-212545008):
- bootloader fixed [(reverted to v2016.01)](https://github.com/mhaberler/u-boot/tree/v2016.01-machinekit)
- xauth issue fixed
- the kernel is now cross-built on amd64, fully automated and apt-upgradeable from [this branch](https://github.com/mhaberler/linux-socfpga/commits/socfpga-3.10-ltsi-rt-machinekit) - thanks, @kinsamanka for the build magic!
- SD image built from[ this branch, config machinekit-debian-jessie-de0.conf](https://github.com/mhaberler/omap-image-builder/commits/machinekit-jessie) (brings in @RobertCNelson's uio/udev issue fix)

I am pretty confident this is next to final for the first round - please test and post feedback here

this took a while, but overall I'm quite excited - all moving parts are now built on amd64 machines using docker - I would especially thank Gemi for all the help; he has [upgraded the machinekit docker images so they can be used for kernel builds now](https://github.com/machinekit/mk-builder/commits/base)!

only omap-image-builder still runs on an ARM host and that will be history soon - downstream we will be able to run the complete process from source to final SD image on any docker-enabled amd64 box!

mhaberler, 2016-04-24:
machinekit seems to perform pretty well on the de0 (modulo lacking graphics on board), lots of cycles and memory left:

`````
top - 08:08:24 up 51 min,  2 users,  load average: 0.49, 0.96, 0.55
Tasks:  85 total,   1 running,  84 sleeping,   0 stopped,   0 zombie
%Cpu(s):  6.4 us,  5.9 sy,  0.0 ni, 86.5 id,  0.0 wa,  0.0 hi,  1.2 si,  0.0 st
KiB Mem:   1029436 total,   355180 used,   674256 free,    23272 buffers
KiB Swap:        0 total,        0 used,        0 free.   221196 cached Mem

 2940 root      20   0   61472  58552  10672 S   3.6  5.7   0:00.52 rtapi:0
 2974 machine+  20   0  102788  32120  16024 S  10.9  3.1   0:09.56 axis
  524 machine+  20   0   10300   1712    844 S   6.6  0.2   0:05.85 sshd
 2940 root      20   0   61472  58552  10672 S   5.0  5.7   0:04.49 rtapi:0
  288 root     -51   0       0      0      0 S   2.0  0.0   0:02.08 irq/152-eth0
 2973 machine+  20   0   21788  10520   8192 S   1.3  1.0   0:01.55 milltask
````

luminize, 2016-04-24:
Hi Guys,

I'm totally blank regarding this device (so is the device, fresh out of the box), so although I can search around for clues, i thought i'd stick with this thread (as to prevent future questions)

I followed the gist to the letter (with Jessie VM). Two minor things:
* bmaptools should be bmap-tools
* remember to unmount the SD card before copying the image

After copying the image to a 16GB card I attached the USB, ran `screen /dev/ttyUSB0 115200`

```
U-Boot SPL 2016.01-00001-ge56465e (Apr 22 2016 - 01:40:32)
drivers/ddr/altera/sequencer.c: Preparing to start memory calibration
drivers/ddr/altera/sequencer.c: CALIBRATION FAILED
drivers/ddr/altera/sequencer.c: Calibration complete
SDRAM calibration failed.
### ERROR ### Please RESET the board ###
```

now unplug and plug power back in, same situation.
Are there any things I needed to have done with jumpers? This board is unmodified straight out of it's wrapping.

Plugging in the 4GB SD card that came with the box, I get a lot of startup info back until the login prompt.

cdsteinkuehler, 2016-04-24:
@luminize That sounds to me like a device-tree or perhaps a u-Boot problem.  I'll try to test the new image soon and see what happens here.

mhaberler, 2016-04-24:
@luminize try an SD card from another manufacturer - I found the de0 nano to be a lot more fussy than the BB wrt SD card manufacturer

- known-to-work: Samsung HC, Kingston SDC4 HC
- problematic:  Sandisk Ultra HC1, Transcend Ultimate 633x

mhaberler, 2016-04-24:
let me verify I can actually boot from what I uploaded..

cdsteinkuehler, 2016-04-24:
I have verified the image boots here, but don't have time to fully test.  The image comes up with a bad environment checksum and loads the default environment, and there's no Ethernet address set.  Is this to be expected or shouldn't the Ethernet MAC come form the on-board EEPROM?

uSD card is a Transcend Premium 400x 16G Class-10 (TS16GUSDU1PE):
http://www.amazon.com/Transcend-MicroSDHC-Memory-Adapter-TS16GUSDU1PE/dp/B015J44QS8

mhaberler, 2016-04-24:
@cdsteinkuehler AFAICT this is to be expected as no environment is set, I always saw that message on first boot

but I'll stand corrected in case it's an u-boot issue

I think it is harmless as it means u-boot went with compiled-in defaults (which includes the Ethernet snag)

the Terasic de0-nano SoC has no MAC EEPROM capability AFAIK, also nothing printed on board and [the manual is silent on the issue](http://www.terasic.com.tw/attachment/archive/941/DE0-Nano-SoC_User_manual.pdf)

btw the ONLY consequence of not setting the Ethernet address is: linux comes up with a different MAC address on each boot, so the assigned IP's keep changing

cdsteinkuehler, 2016-04-24:
Yep...no EEPROM on the Nano.  I was getting my development boards
confused.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-04-24:
@luminize - eject the SD card and wiggle the card a bit before gently pushing in again

the SD card connector is a tad flakey and easy to break - @the-snowwhite has a broken one

mhaberler, 2016-04-24:
the 'CALIBRATION FAILED' error [seems to come up quite a bit in google](http://www.alteraforum.com/forum/showthread.php?t=51782&p=213833#post213833)

@luminize - since we'll meet tomorrow anyway: bring along your board, SD card and power supply, I bring mine, and we'll have a test swapping boards and cards

luminize, 2016-04-24:
> On 24 Apr 2016, at 18:25, Michael Haberler <notifications@github.com> wrote:
> 
> @luminize <https://github.com/luminize> try an SD card from another manufacturer - I found the de0 nano to be a lot more fussy than the BB wrt SD card manufacturer
> 
> known-to-work: Samsung HC, Kingston SDC4 HC
> problematic: Sandisk Ultra HC1, Transcend Ultimate 633x

hmm, my SD card is a SanDisk Ultra 16GB HC1 (Class 10)
I recently ordered 5, so I’ll have to see if I have some other brands somewhere. I thought that buying a brand would be a good choice. I could also take the SD card that came with the de0-nano (4GB) and flash your image onto that.

mhaberler, 2016-04-24:
@luminize dont. copy the the SD card which came with the board to an img file, and write that back that file to the SanDisk SD. 

If this does not work as well this confirms the SD card make is the issue.

luminize, 2016-04-24:
> On 24 Apr 2016, at 19:57, Michael Haberler <notifications@github.com> wrote:
> 
> @luminize <https://github.com/luminize> dont. copy the the SD card which came with the board to an img file, and write that back that file to the SanDisk SD.
> 
> If this does not work as well this confirms the SD card make is the issue.
> 
Flashing the stock image onto the SanDisk SD card gives me the nice blinking LED’s.
So another thing that might have gone wrong could be copying the image onto the SD, although I didn’t get errors while doing that.

luminize, 2016-04-25:
Hi,

[the pastebin output of 4 combinations](http://pastebin.com/vEMPddJU)
These are combinations of Michael H his Nano board, Bas' Nano board, Michael H his Machinekit SD image and Bas' stock SD image

absolutely clueless.

I'm reading the following:
```
U-Boot SPL 2016.01-00001-ge56465e (Apr 22 2016 - 01:40:32)
drivers/ddr/altera/sequencer.c: Preparing to start memory calibration
drivers/ddr/altera/sequencer.c: CALIBRATION FAILED
drivers/ddr/altera/sequencer.c: Calibration complete
SDRAM calibration failed.
### ERROR ### Please RESET the board ###
```

while this does not happen on Michael H's board.

How do I reset my board? (i've done a lot of power downs)

cdsteinkuehler, 2016-04-25:
On 4/25/2016 4:45 PM, Bas de Bruijn wrote:
How do I reset my board? (i've done a lot of power downs)

Page 20 of the user manual:

KEY3 = Hard processor system (ARM) warm reset
KEY4 = Hard processor system (ARM) reset

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-04-25:
On 4/25/2016 4:45 PM, Bas de Bruijn wrote:
> |U-Boot SPL 2016.01-00001-ge56465e (Apr 22 2016 - 01:40:32)
> drivers/ddr/altera/sequencer.c: Preparing to start memory calibration
> drivers/ddr/altera/sequencer.c: CALIBRATION FAILED [...]

Looking at your boot logs, I suspect something fundamental is wrong.

I suspect the DDR calibration is actually failing on your board when
using Michael H.'s image.  Since the board seems to work OK with other
images, it is possible that the wrong timing data is getting pulled in
by the u-Boot code.

It may help to define RUNTIME_CAL_REPORT:

https://www.altera.com/support/support-resources/knowledge-base/solutions/rd07172014_261.html

...and verify that the proper file and values are being pulled in from
the hps_isw_handoff directory (which gets built when compiling the
FPGA bitfile).

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-04-26:
ah, I was afraid of that. I pushed the keys, and the message comes up again.
I'll look into your link later, but I think that's going to be beyond my skillset.

what worries me are these lines of the discussion in the link Michael H posted : https://github.com/machinekit/machinekit/issues/915#issuecomment-214005146 

>`### ERROR ### Please RESET the board ###`

>The u-boot maintainers are looking into this.... In the meantime, I bought a second DE0 board and that one works. (I have two boards that don't, failing memory calibration).

mhaberler, 2016-04-26:
I guess I'll take home Bas' 'bad' board and see what I can do; if it's just an u-boot issue should not  be far

mhaberler, 2016-04-26:
@cdsteinkuehler this u-boot is v2016.01 plus config at e56465e, see https://github.com/mhaberler/u-boot/tree/v2016.01-machinekit

I am unclear how those particular knobs would be turned?

cdsteinkuehler, 2016-04-26:
On 4/26/2016 8:00 AM, Michael Haberler wrote:
> @cdsteinkuehler <https://github.com/cdsteinkuehler> this u-boot is v2016.01 plus 
> config at e56465e, see https://github.com/mhaberler/u-boot/tree/v2016.01-machinekit
> 
> I am unclear how those particular knobs would be turned?

As am I.

I have looked through the hps_isw_handoff files and don't see anything
that jumps out as being tied to a particular PCB layout or chip
timings.  I will have to dig through the u-Boot startup code to see
what's going on, but it's probably better to ask the u-Boot folks
about the issue.

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-04-26:
I have Bas' board here now and reproduced his report
I changed the u-boot loader to v2016.03 (which works but has other problems) - same issue.
Currently out of clue..

mhaberler, 2016-04-26:
There is a thread in the denx u-boot mailing list spanning May and April on the issue: "http://lists.denx.de/pipermail/u-boot/2016-April/250539.html"

it seems unresolved as of now

cdsteinkuehler, 2016-04-26:
On 4/26/2016 4:32 PM, Michael Haberler wrote:
> 
> it seems unresolved as of now

Hmm...to me it reads like the calibration issue is resolved, then he
goes on to ask more questions about USB OTG:

http://lists.denx.de/pipermail/u-boot/2016-April/250740.html

...reports success on both the SoCKIT and the DE0-Nano, using the git
branch:

http://git.denx.de/?p=u-boot/u-boot-socfpga.git;a=shortlog;h=refs/heads/ddr

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2016-04-26:
hows that, folks ;): 

`````
U-Boot SPL 2016.05-rc3-00005-ge25b369 (Apr 26 2016 - 23:44:08)
drivers/ddr/altera/sequencer.c: Preparing to start memory calibration
drivers/ddr/altera/sequencer.c: CALIBRATION PASSED
drivers/ddr/altera/sequencer.c: Calibration complete
Trying to boot from MMC1
 
 
U-Boot 2016.05-rc3-00005-ge25b369 (Apr 26 2016 - 23:44:08 +0200)
 
CPU:   Altera SoCFPGA Platform
FPGA:  Altera Cyclone V, SE/A4 or SX/C4, version 0x0
BOOT:  SD/MMC Internal Transceiver (3.0V)
       Watchdog enabled
I2C:   ready
DRAM:  1 GiB
MMC:   dwmmc0@ff704000: 0
*** Warning - bad CRC, using default environment
 
In:    serial
Out:   serial
Err:   serial
Model: Terasic DE0-Nano(Atlas)
Net:   
Error: ethernet@ff702000 address not set.
No ethernet found.
Hit any key to stop autoboot:  0 
...
````
current u-boot master does get past the error!

mhaberler, 2016-04-26:
here is an interim u-boot binary http://static.mah.priv.at/public/socfpga/u-boot-with-spl.sfp
comes up as `U-Boot 2016.05-rc3-00006-gd4c1374-dirty (Apr 27 2016 - 00:12:46 +0200)`

install with `dd if=u-boot-with-spl.sfp of=/dev/ mmcblk0p1 or sdb1 as needed

boots both my and Bas' board, plus runs machinekit fine 

I'll wrap that up tomorrow and create a new image

[81 more replies]

closed 2018-08-01 by ArcEye