machinekit/machinekit #917: documentation to install kernel wrong, missing dependencies

gmoccapy, 2016-04-20:
> Fresh Jessie installed, repos and all stuff updated according to the documentation. The linux-image-4.1.15-rt17mah installed without problems.
But machinekit-rt-preempt  depends on linux-image-rt-686-pae which is not to be installed.

When I wanted to install linux-image-rt-686-pae, then I got, it's depend on linux-image-4.1.0-0.bpo.1-rt-686-pae, but in Jessie backports only ver. 4.4 is available as Charles mentioned earlier.

I didn't go for machinekit-xenomai, although kernel installed without errors, but machinekit-xenomai would require to download 110 new packages (60MB) and I was not sure if this is correct.

The output of  aptitude search rt-686-pae:
p   linux-headers-4.4.0-0.bpo.1-rt- - Header files for Linux 4.4.0-0.bpo.1-rt-68
p   linux-headers-rt-686-pae        - Header files for Linux rt-686-pae configur
i   linux-image-4.4.0-0.bpo.1-rt-68 - Linux 4.4 for modern PCs, PREEMPT_RT     
p   linux-image-4.4.0-0.bpo.1-rt-68 - Debugging symbols for Linux 4.4.0-0.bpo.1-
p   linux-image-rt-686-pae          - Linux for modern PCs (meta-package), PREEM
p   linux-image-rt-686-pae-dbg      - Debugging symbols for Linux rt-686-pae con
v   linux-latest-modules-4.1.0-0.bp -                                          
v   linux-latest-modules-4.4.0-0.bp -       

Best regards:
Janos

<

I have the same problem, for details see:

[https://groups.google.com/forum/#!topic/machinekit/dJFDKSNbCaw](url)

Norbert

mhaberler, 2016-04-20:
Ok, please repair by:

a) discussing what to replace (either no dependency OR a corrected dependency)
b) file a pull request against https://github.com/machinekit/machinekit/blob/master/debian/control.rt-preempt.in

Bujtar, 2016-04-20:
Possible workaround: install linux-image-rt-686-pae package from Jessie backport and try to re-install machinekit-rt-preempt, it worked for me.

Best regards:
Janos

gmoccapy, 2016-04-21:
I think a work around is not the solution. IMHO the package dependencies must be repaired and the the docu must be checked to point users the right way.

I am willing to do the pernickety german testing ;-)

Norbert

mhaberler, 2016-04-21:
Post a PR please.

gmoccapy, 2016-04-24:
Michael,

what kind of PR and how to you refer to?
I have not tested anything yet, as I am waiting for a solution, not a work around.
And I do not know much about packages :-(

Norbert

mhaberler, 2016-04-24:
@gmoccapy - a github Pull Request - this is the way we cooperate here. See http://www.machinekit.io/docs/contributing/

gmoccapy, 2016-04-24:
Hallo Michael,

OK I do know that, but I can only make a pull request having a solution.
I have non! The mentioned work around IMHO is not the solution, or am I wrong?

Sorry, if I ennoy you.

Norbert

mhaberler, 2016-04-24:
Above I pointed to the file which defines the dependency: https://github.com/machinekit/machinekit/blob/master/debian/control.rt-preempt.in

This is a [debian control file](https://www.debian.org/doc/debian-policy/ch-controlfields.html)

The dependency is [in line 8](https://github.com/machinekit/machinekit/blob/master/debian/control.rt-preempt.in#L8)

If you remove line 8, the package dependency is gone. The question is - is this a good idea or do we have a better way. 

It will cause people being able to install machinekit-rt-preempt without a realtime kernel - this needs to be tested and if necessary fixed.

gmoccapy, 2016-10-24:
Hallo,

I tried again to install machinekit, and I am still with the same problem, mentioned on 20th of april.
Nothing has changed! I am not able to solve the dependencies.

Please give some instructions or help.
As far as I did understand, I should git clone the repository, change the debian control file and compile myself.

But that is not the solution. Please tell me how I can help to solve this part. I opened the issue, and I am willing to open a PR, but what should that one contain.

Norbert

gmoccapy, 2016-10-24:
OK, I have it running as rip-environment, I tested gmoccapy, updated it, created a pull request for my changes, all without big problems. I have running 
` Operating System: Debian GNU/Linux 8 (jessie)
            Kernel: Linux 4.1.19-rt22mah
      Architecture: x86-64
`
I am a little afraid to just do a apt-get install machinekit.

Norbert

ArcEye, 2016-10-25:
The problem @mhaberler was referring to, was that by removing the dependency it solves the immediate problem, but leaves users without a kernel that will run in realtime.
For most users that is a complete show stopper.

The dependency could be changed to `linux-image-4.1.19-rt22mah` which is in the machinekit repo, even if it is quite old now.

Does not look like we can rely upon Debian to do anything, they seem to have pulled or changed the 4.4.0-rt kernel, without leaving the previous version accessible.

I think you need to follow https://github.com/machinekit/machinekit/issues/917#issuecomment-214026423, then submit the PR

That will render the machinekit package usable, because it will be able to satisfy the dependencies.

gmoccapy, 2016-10-25:
Thanks for the explanation. As soon as I got time, I will try to change the dependency to match linux-image-4.1.19-rt22mah.

If that works, I will make a PR.

Norbert

ArcEye, 2016-10-25:
To simply amend an existing deb package for dependencies, easiest way is probably:-

`ar xv name-of-package.deb` will extract files from a deb

Change the `control` file in control.tar.gz, to edit dependencies

Then replace the one in the deb with
`ar -r name-of-package.deb control.tar.gz`

gmoccapy, 2016-10-25:
Thanks, so I stopped my work ;-)

I did:
`
apt-get download linux-image-rt-amd64
ar xv linux-image-rt-amd64_4.1+66~bpo8+1_amd64.deb
 `
I modified the control file to be:
`
Package: linux-image-rt-amd64
Source: linux-latest (66~bpo8+1)
Version: 4.1+66~bpo8+1
Architecture: amd64
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>
Installed-Size: 37
Depends: linux-image-4.1.19-rt22mah
Provides: linux-modules-4.1.19-rt22mah
Section: kernel
Priority: optional
Description: Linux for 64-bit PCs (meta-package), PREEMPT_RT
 This package depends on the latest Linux kernel and modules for use on PCs
 with AMD64, Intel 64 or VIA Nano processors.
 .
 This kernel includes the PREEMPT_RT realtime patch set.
`
and paked the deb again with
`
ar -r linux-image-rt-amd64_4.1+66~bpo8+1_amd64.deb control.tar.gz
`
I installed it with
`
sudo dpkg -i ./linux-image-rt-amd64_4.1+66~bpo8+1_amd64.deb
`
then I installed:
`
sudo apt-get install machinekit-rt-preempt
`

and YES!! 
Machinekit runs from the application menue
I did not test with real hardware yet, but I do hope I am able to use it with my MESA 7i76E

So here is the modified deb file:
(added a txt to the name, to be able to upload the file)

[linux-image-rt-amd64_4.1+66~bpo8+1_amd64.deb.txt](https://github.com/machinekit/machinekit/files/551370/linux-image-rt-amd64_4.1.66.bpo8.1_amd64.deb.txt)

So please help with the next step!
If I do understand correct, I must make now a PR to get the changes to the repository. But I only understand making PR to or from github. But this file must go to the deb repository, don't it?

Norbert

Norbert

ArcEye, 2016-10-26:
A bit backwards I'm afraid.
We can't amend a Debian meta-package, because we do not host it.

What needs changing is the dependency in `machinekit-rt-preempt` which causes it to look for
`linux-image-rt-amd64`, which it cannot resolve, so will not continue.

Hopefully #1087 does what is required, perhaps you can test later to be sure

ArcEye, 2016-10-26:
> If I do understand correct, I must make now a PR to get the changes to the repository. But I only understand making PR to or from github. But this file must go to the deb repository, don't it?

The packages are built from the scripts in /debian within the main repo, so everything can be done from there as per the commit above

PR #1087 (2016-10-26): Update control.rt-preempt.in
The debian meta-package listed as dependency, does not resolve to a current rt-preempt kernel any more.
This leaves the package machinekit-rt-preempt uninstallable.

List the latest rt-preempt kernel in our own repo as dependency, enabling installation

Fixes #917 (hopefully)
commit 2c9cf679 (2016-10-26): Update control.rt-preempt.in

The debian meta-package listed as dependency, does not resolve to a current rt-preempt kernel any more.
This leaves the package machinekit-rt-preempt uninstallable.

List the latest rt-preempt kernel in our own repo as dependency, enabling installation

Fixes #917 (hopefully)
closed 2016-10-26 by luminize