machinekit/machinekit #924: Automated build of documentation and website

luminize, 2016-05-01:
Goals (prioritised from high to loa, mark when done!):
- [ ] only work on the content in a minimum way until we get the workflow going, so we can increment and do most of the work on the docs/site thru C4 on machinekit/docs (no hiding [in our cupboard under the stairs](http://i1.wp.com/theverybesttop10.files.wordpress.com/2013/08/the-world_s-top-10-most-amazing-things-to-do-with-under-stairs-spaces-10.jpg?resize=590%2C396) ) **(seriously, this is our nemesis)**
- [ ] defragmenting information, bringing all into 1 place
- [ ] Improving the readability of the documentation
- [ ] lower barrier for improving documentation
- [ ] get better searcheability from search engines
- [ ] when this is all said and done, we can pull the plug on machinekit.github.io (the website repo)

The future place to be for info will be machinekit.io
- for project information (current website)
- for documentation

To get to that situation, all will be generated by Jekyll from machinekit/machinekit-docs repo

step 1: **We aim at the following workflow from the manpages:**
1. a merge on machinekit/machinekit generates asciidoc manpages (static, comp and icomp).
2. asciidoc manpages are brought over to machinekit/machinekit-docs. 
3. machinekit/machinekit-docs merge rebuilds the website, and gets served

(above 3 points see https://github.com/machinekit/machinekit/issues/908)
(We need to find out if every PR to machinekit/machinekit will of that, or only when there are changes in the manpages (static, comp and comp))

step 2: **the rest of the workflow should be like so:**
4. machinekit/machinekit-docs holds content for website as well as the documentation.We need to convert the markdown to asciidoc, and put that in machinekit/machinekit-docs
5. commits via PR's from (local) clone
6. machinekit/machinekit-docs PR triggers a site preview on preview.machinekit.io

step 3: 
7. when OK, merge will trigger final format on machinekit.io

picture depicts the flow from points 4 - 7

[image]

some additions:

- The build is done with Jekyll from asciidoc sources with `jekyll-asciidoc` plugin
- The plugin uses asciidoc**tor**
- Pages need to have front matter (Jekyll spec), this is a block of 2 3 dashes delimiters. Attributes inside the front matter block are not mandatory:
```
---
layout: default
---
```
- because of the front matter being not properly displayed when one builds outside of jekyll asciidoc**tor** has a [skip-front-matter attribute](http://asciidoctor.org/docs/user-manual/#front-matter-added-for-static-site-generators) which can be invoked like so: `asciidoctor my-page.asciidoc -a skip-front-matter`
- linking between pages must be done like so: `link:some-directory/target-adoc-file#some-anchor[foo bar]`

@mhaberler @ArcEye @cdsteinkuehler any additions?

ArcEye, 2016-05-02:
> step 1: We aim at the following workflow from the manpages:
1. a merge on machinekit/machinekit generates asciidoc manpages (static, comp and icomp).

Confused about the 'static' asciidoc manpages.
What are they being generated from?  Thought they *were* static and what was generated was groff manpages and html web pages?

The `--enable-build-documentation=asciidoc` will need to be used in the builders, off by default
Should not reverse that logic or everyone who builds locally, will get unwanted asciidocs built.

> 3. machinekit/machinekit-docs merge rebuilds the website, and gets served
(We need to find out if every PR to machinekit/machinekit will ~~of~~ do that, or only when there are changes in the manpages (static, comp and comp))

I wonder about that too, we may end up with a trigger every merge, but don't know enough about Travis etc
to know if rules can be built in specifying 'only changes to a group of files = trigger'

luminize, 2016-05-02:
> On 02 May 2016, at 11:03, ArcEye <notifications@github.com> wrote:
> 
> step 1: We aim at the following workflow from the manpages:
> 1. a merge on machinekit/machinekit generates asciidoc manpages (static, comp and icomp).
> 
> Confused about the 'static' asciidoc manpages.
> What are they being generated from? Thought they were static and what was generated was groff manpages and html web pages?
> 
yeah, sorry, “static" are not generated, but they need to be copied or whatever to the some location (like where the comp and icomp reside after build). I think gathering all in one directory makes transferring them easier. But that’s just my interpretation.
> The --enable-build-documentation=asciidoc will need to be used in the builders, off by default
> Should not reverse that logic or everyone who builds locally, will get unwanted asciidocs built.
> 
> machinekit/machinekit-docs merge rebuilds the website, and gets served (We need to find out if every PR to machinekit/machinekit will of do that, or only when there are changes in the manpages (static, comp and comp))
> I wonder about that too, we may end up with a trigger every merge, but don't know enough about Travis etc
> to know if rules can be built in specifying 'only changes to a group of files = trigger’
> 
Well, I’m in the dark on this one. Don’t know how we can compare the build result with the previous build result. Would it be a problem to have a merge on every build?

mhaberler, 2016-05-04:
@luminize - your remark: (We need to find out if every PR to machinekit/machinekit will of that, or only when there are changes in the manpages (static, comp and comp))

if built via jenkins (other than travis, which does not retain history), the build works off an existing workspace retained across builds, and is a make run - thus incremental
the playout to the website can be based on the changed files, which for simple fixes would be likely a single file

so I expect the PR- and merge triggered builds to be very fast eventually, certainly less than 30sec

mhaberler, 2016-05-04:
re FAQ: we can work in a FAQ section, for instance using this theme: http://idratherbewriting.com/documentation-theme-jekyll/mydoc_faq_layout/

we need an html/css artist soonish ;)

mhaberler, 2016-05-04:
I'm scouting asciidoctor equivalents for [plugins](http://www.methods.co.nz/asciidoc/plugins.html) which are already used in the docs (not many)

investigating the [asciidoctor-diagram](https://github.com/asciidoctor/asciidoctor-diagram) plugin - these enable great diagrams!

examples:

[blockdiag](http://blockdiag.com/en/)
[seqdiag](http://blockdiag.com/en/seqdiag/)
[actdiag](http://blockdiag.com/en/actdiag/)
[nwdiag](http://blockdiag.com/en/nwdiag/)
[ditaa](http://ditaa.sourceforge.net/)
[mermaid](https://github.com/knsv/mermaid) - does sequence diagrams but nicer than mscgen
[wavedrom](http://wavedrom.com/tutorial.html)
[shaape](https://github.com/christiangoltz/shaape)
[plantUML](http://plantuml.com/)
[graphviz](http://asciidoctor.org/news/2014/02/18/plain-text-diagrams-in-asciidoctor/#controlling-the-output)

ATM I think the only existing plugin which might need attention is mscgen, which could be easily replaced by mermaid

mhaberler, 2016-05-04:
I found an asciidoctor docker container - just what's needed for the docs format: https://hub.docker.com/r/asciidoctor/docker-asciidoctor/

mhaberler, 2016-05-05:
ok, basic automation in place!

- I got the above docker-jekyll-asciidoctor image fixed so it actually works
- jenkins 'jekyll build' job set up: https://jenkins.machinekit.io/job/machinekit-docs-playout/
- example log: https://jenkins.machinekit.io/job/machinekit-docs-playout/39/console
- it is run on the  repo https://github.com/mhaberler/machinekit-docs/tree/work , triggered by a push
- preliminary results are in http://preview.machinekit.io , directly serving the _site directory in the workspace

reformatting run takes 20sec so acceptable IMO

luminize, 2016-05-05:
so far, the setup works.

I'll continue with getting our current information in shape, so all current information is rendered and visible from machinekit.io/docs (see preview.machinekit.io/docs)

At the same time i'll work out/find glitches.

Then after that we can see what we need to restructure, and make more nice.

ArcEye, 2016-05-05:
I have just done a PR to update the comp and instcomp docs

You may be able to use that to test the rebuild / updating?

mhaberler, 2016-05-05:
@ArcEye currently the jenkins job triggers on branch work in my repo

but Bas and you know have write permissions to my repo - so anything you merge into the work branch triggers the rebuild

luminize, 2016-05-06:
@mhaberler @ArcEye 
I only focussed on getting as much content visible.
Please see:
- http://preview.machinekit.io/docs/index-getting-started/
- http://preview.machinekit.io/docs/index-HAL/
- http://preview.machinekit.io/docs/index-integrator/

and the commits I did to get there:
https://github.com/mhaberler/machinekit-docs/pull/4

a lot of links are messed up, includes needed a sed command in the makefile.
Combination of the way Jekyll makes urls and the (relative) references.

ArcEye, 2016-05-07:
That is starting to look quite good.
It follows the current trend to make everything tablet an phone friendly, the page expands easily to fill the screen whilst keeping quite a narrow page width.

The content is another matter, lots of incomplete, inaccurate stuff, but until it is up there and easily amended, won't get changed.

luminize, 2016-05-07:
We need to look into the differences from "stock" .asciidoc, which also can generate pdf, and the links in the generated (by the Makefile) .adoc files.

`docs/index-getting-started.adoc` for example gets rendered with an url like `docs/index-getting-started`/ and the behaviour of `link:` `include::` and `image::`  inside this adoc differ.

`include::../foo/bar.adoc[]` for example allows relative links to include `docs/foo/bar.adoc[]`

But an image in `docs/foo/bar.adoc[]` like `image::baz/boing.png` requires an image file `docs/baz/boing.png`

For PDF to be generated with asciidoctor from .asciidoc we need
`image::baz/boing.png`

And to have it rendered (Jekyll) from adoc we need:
`image::../baz/boing.png`

I've already taken care of that with sed replacements in the Makefile (similar to [this line]( https://github.com/mhaberler/machinekit-docs/commit/9f9ae956e7fcfcc485e594fab5a225fd7640d70d#diff-b67911656ef5d18c4ae36cb6741b7965R21) but then for the `image::` links (haven't committed yet)

So images can only reside on the same or lower directory level of the file.

I think some restructuring of directories wrt the top level document are needed. Or there might be some tweaking from Jekyll I don't know of yet.

ArcEye, 2016-05-07:
Why having taken the trouble to produce .asciidoc, are we converting to .adoc which does not convert properly to .html?

We could produce .html with asciidoctor and jekyll would use that.

I have just created a default jekyll dir, 
`$ jekyll new test1`
and then cloned my machinekit-docs repo into it.

Run this script from the root of the repo
```
#!/bin/sh

for i in $(find ./ -name '*.asciidoc') ; do
    asciidoctor $i ;
done

# do steps seperately in this order, because asciidoctor does not like
# links that don't exist (because of .ext change)

for i in $(find ./ -name '*.asciidoc') ; do
    sed -i -e 's;.asciidoc;.html;g' $i ;
done
```

Then add 2 lines to index.html in Jekyll root dir
```
<h1>Machinekit Documents</h1>

  <a href="machinekit-docs/index.html"> Index </a>
```

`$ jekyll build`
`$ jekyll serve`

and the whole documentation is rendered with all links working and no extra formatting anywhere.

Give it a try, just need to do something like it within a docker or similar.

luminize, 2016-05-07:
> On 07 May 2016, at 17:50, ArcEye wrote:
> Why having taken the trouble to produce .asciidoc, are we converting to .adoc which does not convert properly to .html?
> 

Well, adoc gets converted as well as asciidoc, just a different extension. We insert front matter into the adoc file. For Jekyll sake. As to not screw ourselves with demolishing the sources. Because jekyll/asciidoctor and links is just a tad different. (But might be because missing Jekyll knowledge from my part)

We do the extra step because we have not only the docs, but now also the website from the same repo. One repo to rule all :)

Maybe a faq or tagcloud in the future? Searching? Don't know exactly. But that's what's still possible if we keep Jekyll around. Maybe convert the blog too? No idea, but still possible.
PDF/ePub also still possible for a getting started/Hal/developer manual if we use the unmodified asciidocs, we could automatically make those manuals and provide a download file/link.

ArcEye, 2016-05-08:
I am still not with you, can't see any problem with links when Jekyll renders the converted html

I have just converted the website from .md to .asciidoc using
```
#!/bin/sh

for i in $(find ./ -name '*.md') ; do
    docname=$(basename -s .md "$i") ;
    pandoc -t asciidoc -f markdown $i > $docname.asciidoc ;
done
```
Then convert to .html using the previous script.

The result when built by jekyll displays fine.

The content is rubbish, but that is another matter.

I will fork @mhaberler 's 'work' branch and add a google search engine a bit later and see how that pans out.

luminize, 2016-05-08:
> On 08 May 2016, at 12:02, ArcEye <notifications@github.com> wrote:
> 
> I am still not with you, can't see any problem with links when Jekyll renders the converted html
> 
> I will fork @mhaberler <https://github.com/mhaberler> 's 'work' branch and add a google search engine a bit later and see how that pans out.
> 
yes, please use that branch and try it out. No use comparing apples with pears (literal Dutch translation)

That’s significantly different than the machinekit-docs as it now is, because it combines machinekit.github.io and machinekit-docs into one.

ArcEye, 2016-05-08:
As soon as I try to build the 'work' branch I hit problems

Once I install the required asciidoctor, asciidoctor-diagram and jekyll-asciidoc, the `jekyll build` fails with
`jekyll 2.2.0 | Error:  private method `load' called for Asciidoctor:Module`

So where did you get jekyll from (I am using Jessie package) and where for the other plugins (gems)?

luminize, 2016-05-08:
> On 08 May 2016, at 17:30, ArcEye <notifications@github.com> wrote:
> 
> So where did you get jekyll from (I am using Jessie package) and where for the other plugins (gems)?

`gem install ... ` If I'm correct.
If you already have the gems, then `gem update` will get you the latest version.

ArcEye, 2016-05-08:
makes no difference - uninstalled, reinstalled, updated.
Back to Q
Where did you get Jekyll?

ArcEye, 2016-05-08:
OK, it seems `gem install jekyll` does get version 3.1.3, except it puts it in /usr/local/bin, where it was not found, so I had to wipe the cache in bash before it could find it.

However just get same error with different version number.

luminize, 2016-05-08:
> On 08 May 2016, at 18:05, ArcEye <notifications@github.com> wrote:
> 
> Back to Q
> Where did you get Jekyll?

Either apt-get install or gem install. I didn't do anything special :(

mhaberler, 2016-05-10:
Bas and me did a night shift - goal: great diagrams

result: http://preview.machinekit.io/docs/documenting/diagramtest/
source: https://github.com/mhaberler/machinekit-docs/blob/work/docs/documenting/diagramtest.ad

mhaberler, 2016-05-10:
re jekyll, asciidoctor, asciidoctor-diagram: this is the container serving preview.machinekit.io: https://github.com/mhaberler/docker-jekyll-asciidoctor - could be used locally as well!

maybe some hints & methods can be gleaned - the picking of versions which actually work was fairly tricky - dont got for latest & greatest by default

ArcEye, 2016-05-10:
Finally, I can serve the new site properly, locally.

I installed all the gems in your file, having wiped everything shown as installed using
`gem info --local`
(I do not like having 2 parallel installation mediums, apt-get and gem, immediate scope for conflict)

Everything bar `jekyll-mentions` installed OK, that one required ruby 2.2.2 (Jessie installs 2.1.5), so I just ignored it.

ArcEye, 2016-05-10:
I have pushed a site ID file, which is required by google to be able to have it perform local searches.
https://github.com/mhaberler/machinekit-docs/commit/500bca512ae29120834d6e0de8414ab4310fd812

When the addition propagates to the preview.machinekit.io site and I can test it, will be able to look at adding search bar into the header.

ArcEye, 2016-05-10:
The file is there and verified but this seems to take for ever to get their bots to crawl it.

Will check again tomorrow

luminize, 2016-05-10:
> On 10 May 2016, at 17:52, ArcEye <notifications@github.com> wrote:
> 
> The file is there and verified but this seems to take for ever to get their bots to crawl it.
> 
> Will check again tomorrow
> 
Michael might have put a robots.txt there to disable robots...

ArcEye, 2016-05-10:
> Michael might have put a robots.txt there to disable robots...

It's not present in the clone I have

ArcEye, 2016-05-11:
The search works fine now, it had just never indexed the `preview.machinekit.io` before so had to wait for the indexing request to be dealt with.

Will have the same delay when we switch to the permanent URL.

mhaberler, 2016-05-11:
@ArcEye great! how would I try out this custom search?

ArcEye, 2016-05-11:
https://cse.google.com/cse/publicurl?cx=013132401186084955638:rss8-mkzi3u

I am still experimenting with the best way to fit it into the headers so that it is visible from everywhere in the site

mhaberler, 2016-05-11:
works great!
(doesnt dig deep yet because of the missing links but that will come)

ArcEye, 2016-05-11:
I have just pushed a search bar onto the front page.
Should propagate in a short while.
Then at look one in  the docs sidebar.

So long as my change to _config.yml to enable incremental build does not break things ;)

luminize, 2016-05-11:
Great progress.

One checkbox I added in the first post https://github.com/machinekit/machinekit/issues/924#issue-152372003:
- [ ] We need to look into the link checker

How about getting to machinekit-docs:master a.s.a.p (we need to get out of our cave)?
(see first point)

What I've done so far is pretty "hacky" in the sense that I did a lot of moving directories around. I've just marked 3 of the 6 checkboxes in the first comment. I guess we know what to do, and how to do it.

So I propose I start with making PR's agains machinekit/machinekit-docs:master, working my way thru the past 2 weeks, and putting stuff into place which we know "is good". Keeping mhaberler/machinekit-docs:work as our playground and experimenting location.

Then I can start working on documenting
- how the documenting workflow works. Including diagrams and whatnot.
- get structure on the site in better shape. Meaning that I think we need to split the HAL part in HAL and GUI parts (future)

And if the basics are OK that paves the way to have people start helping out.

Thoughts?

luminize, 2016-05-11:
@ArcEye nice searchbar!

ArcEye, 2016-05-11:
> Thoughts?

Don't know if there is any benefit to changing the main machinekit-docs yet, we can do en-bloc later
(Unless you intend rendering that as well, but then we just present a half finished site to public view)

There are a LOT of asciidocs with the wrong headers in use

If you run `jekyll serve` you get some, but if you run a recursive script to convert all the asciidoc into html and capture the output, such as
```
#!/bin/sh

for i in $(find ./ -name '*.asciidoc') ; do
    asciidoctor $i ;
done
```
you get pages of errors.

Lots of them are using `=` as the first header in a doc which is not the main page, or jumping from
`==` to `====` without an intervening `===`

Trivial but will take a while to correct.  We have used the `==` as the same as ``<H2`>`</H2`>` to get the font header we want, but they are different.

luminize, 2016-05-11:
> There are a LOT of asciidocs with the wrong headers in use

ok, I'll add this as something we must fix before going live

> Lots of them are using = as the first header in a doc which is not the main page, or jumping from
== to ==== without an intervening ===

Yes, that’s true.

asciidoctor supports incremental :leveloffset: +1, so we need to change that where it’s appropriate
- `:leveloffset: 1` becomes `:leveloffset: +1`
- `:leveloffset: 0` becomes `:leveloffset: -1`

like here: https://raw.githubusercontent.com/mhaberler/machinekit-docs/7e28be148ed0eb183c17fbe588b51a885bf7fb97/docs/index-HAL.asciidoc

plus, i think a lot of these errors come from translations like `*_fr.asciidoc` and `*_es.asciidoc` files.
We should remove them (I already removed Master*_es/fr.asciidoc).

Added these points too [above](https://github.com/machinekit/machinekit/issues/924#issue-152372003)

mhaberler, 2016-05-11:
@luminize I do not think making mk/machinekit-docs a 'work-in-progress' repo is a good idea - you saw which confusion just posting the preview.machinekit.io link caused

IMO we should switch when the content is ready, but not before - if anything, half-baked websites are competely repelling and look really unprofessional

mhaberler, 2016-05-11:
re link checker - I will pick one and add it to the formatting run, will look into sending its output to a separate web page but it is easy to use online to get the idea: the  [official w3c checker ](https://validator.w3.org/checklink?uri=preview.machinekit.io&hide_type=all&recursive=on&depth=10&cookie=set&check=Check ) https://wummel.github.io/linkchecker/  

I note the first run of [linkchecker](https://wummel.github.io/linkchecker/ ) reported 438 broken links

luminize, 2016-05-11:
> On 11 May 2016, at 13:11, Michael Haberler <notifications@github.com> wrote:
> 
> you saw which confusion just posting the preview.machinekit.io link caused
> 
Not sure I understand. What confusion?
> IMO we should switch when the content is ready, but not before - if anything, half-baked websites are competely repelling and look really unprofessional
> 
True.
I define content ready as:
- links must be sound
- basic structure must be correct

All other is "extra"

Right now machinekit-docs is stale and doesn't get used for anything. Only viewing on github. Which all agrees is horrible.

IMO We should switch to C4 asap.

Plus the website we see now (machinekit.io) is in machinekit.github.io

But that doesn't mean we cannot commit to machinekit-docs. Right?

Switching machinekit.io is just the 2nd step in this issue. Right?

The longer we stay on our own "invisible" branches, the more we are the only janitors.

ArcEye, 2016-05-11:
> The longer we stay on our own "invisible" branches, the more we are the only janitors.

Your faith in someone coming out of the woodwork to do more than complain or suggest how someone else could do things, is touching :wink: 

But let's hope so.

ArcEye, 2016-05-11:
Added a search bar to all docs window frames

Minimalised it to try to display best in limited screen (mobtel etc)
Don't think you need a button to press to search, everybody knows you just press enter after the search text.

Need to have it rendered remotely to test that easily.

luminize, 2016-05-11:
> On 11 May 2016, at 14:06, ArcEye <notifications@github.com> wrote:
> 
> The longer we stay on our own "invisible" branches, the more we are the only janitors.
> 
> Your faith in someone coming out of the woodwork to do more than complain or suggest how someone else could do things, is touching 
> 
> But let's hope so.
> 

At least then we've done what we can :)

ArcEye, 2016-05-11:
Search bar comes out fine for desktops and for mobile on main page

Just need to add to the mobile only frame to docs without screwing up the display, as happened last time I tried.

the-snowwhite, 2016-05-11:
Those 2 links both lead me to 1 site:

http://www.machinekit.io/ <http://www.machinekit.io/>

From here

Best Wishes
Holotronic

> On 11 May 2016, at 13:24, Bas de Bruijn <notifications@github.com> wrote:
> 
> Plus the website we see now (machinekit.io <http://machinekit.io/>) is in machinekit.github.io <http://machinekit.github.io/>

ArcEye, 2016-05-11:
Mobile search bar for docs added, displays well in portrait mode, not fully optimally in landscape, but perfectly usable.

My mobtel has a 4.5" (115mm) screen approx.
Also tested on my tablets 10" ( 250mm ) screen and displays well in portrait and landscape.

mhaberler, 2016-05-11:
@luminize - re switching:

I would switch both the website AND the docs repo in one go to all-asciidoc - no point IMO to rock the boat twice

re committing to mk-docs now - well defacto my work branch is the current docs devel branch, so we will have to rebase work onto anything committed to mk-docs before switch - problematic

what is the purpose of switching to mk-docs now? redoing what is in mhaberler/mk-docs/work but properly?

I would think as a baseline for a switch we need to:

- decide on the front matter and get that into the asciidoc pages - just one extension in the final repo, too confusing with the make step (I am clueless on the permalink business implied in the markdown front matter)
- make sure the old website still renders ok
- fix all the broken links with automated linkcheck support and that really must happen before a switch
- have a "Edit me" link on every page which gets you right to the mk-docs repo and the underlying page there ; this could likely be done with a stylesheet (?)
- have a coarse cross-read of all major sections for glaring problems
- define an anchor issue for fixing the results (or this will spill all over the tracker)

we can ask for help on tuning the content then, but we cannot realistically ask for help on what are toolchain issues, or need to be mass-fixed by scripts

As for machinekit.github.io - I guess we'll keep it but just fed with the html results of the formatter step, like [described here](http://eshepelyuk.github.io/2014/10/28/automate-github-pages-travisci.html#build-script)

mhaberler, 2016-05-11:
just looking through the issues in https://github.com/machinekit/machinekit.github.io/issues - it seems we can take along https://github.com/machinekit/machinekit.github.io/issues/32 eventually as well if we have migrated docs and website

this would take care of the jump-off-site when hitting 'Blog' which @sliptonic referred to

ArcEye, 2016-05-11:
> this would take care of the jump-off-site when hitting 'Blog' which @sliptonic referred to

I have to question whether it matters, is it that hard to click on the <- Back arrow?
Or open the link in a new tab.

Nothing you can do about going to a different part of github by clicking on the 'GitHub' link, it **is** somewhere else.

Bottom line is probably does it make your and Charles lives harder or easier, it is mostly just your blogs?

mhaberler, 2016-05-11:
@ArcEye Charles and my blog - currently yes; I do think we should open it up for all contributors though

and (while not directly falling under this headline) - I guess me and at least Charles should do periodic updates as to what is happening and on overall direction; @sliptonic was right about that one

probably once per month a short update, phase-shifted among two or three people - should get some traffic

mhaberler, 2016-05-11:
ok - linkchecking: I'll investigate how to add a linkcheck step to the docs run and publish that via a separate html page

luminize, 2016-05-11:
> On 11 May 2016, at 16:58, Michael Haberler <notifications@github.com> wrote:
> 
> have a "Edit me" link on every page which gets you right to the mk-docs repo and the underlying page there ; this could likely be done with a stylesheet (?)

I think this is just a nice to have. Let's have this as "bonus" instead of mandatory

mhaberler, 2016-05-11:
sorry - disagree. Big turnoff for contributions - and lame excuse: "I dont know how to fix this" "dog ate homework" etc etc

just found a solution - might be very easy after all: https://github.com/jekyll/jekyll-help/issues/5

mhaberler, 2016-05-11:
edit button rev -1 (dead link but there): see http://preview.machinekit.io/docs/ 'Edit this page'

linkchecker manual run as html: http://preview.machinekit.io/linkcheck.html

luminize, 2016-05-11:
> On 11 May 2016, at 18:28, Michael Haberler <notifications@github.com> wrote:
> 
> sorry - disagree. Big turnoff for contributions - and lame excuse: "I dont know how to fix this" "dog ate homework" etc etc
> 
Well, I disagree with you disagreeing :) I think it's a fallacy to think that this will make it ultimately easy to contribute. Button, simsalabim and hey presto... Not going to happen.

It's not a question of adding a button, but telling people how to fork and pull too. Update a fork, make a PR. In the end the button only points to the source, and not the committees own personal (and probably) out of date fork.

If you want to 'click and change' then the fork-and-pr model isn't suited IMO and you'll need to switch to a wiki.
Which IMO isn't good because a wiki is set up to not have structure. And structure is the thing we need for beginners.

If we want a "button to point to the source " then we add it. But that isn't the easy contributing function, and hence IMO still a nice to have.
> just found a solution - might be very easy after all: jekyll/jekyll-help#5
> 
@mhaberler @ArcEye would you add these type of goals to the first comment (edit it) in this thread. Then we (me) keep some form of oversight and know when we're done. It tends to keep focus.

cdsteinkuehler, 2016-05-11:
On 5/11/2016 1:21 PM, Bas de Bruijn wrote:
> If you want to 'click and change' then the fork-and-pr model isn't
> suited IMO and you'll need to switch to a wiki. Which IMO isn't good
> because a wiki is set up to not have structure. And structure [...]

What about combining a bit of github magic with a "sandbox" repository?

Github makes it very easy to create a PR directly from the browser:

https://help.github.com/articles/github-flow-in-the-browser/

...the problem is the first step is to create a branch, which for most
casual contributors also means creating a clone of the docs repo.  What
if we have a docs-sandbox repo and craft an "edit me" link that creates
an auto-named branch in the docs-sandbox repo and lands on the edit link
for the current page being viewed:

https://help.github.com/articles/editing-files-in-your-repository/

This likely requires a bit of web magic, but seems like it should be
doable via java-script or something.  The master branch of the
docs-sandbox repo could be auto-synced with the main documentation repo
via hook scripts, and have loose permissions on the sandbox repo, but is
there any fundamental reason this wouldn't work?

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-05-11:
On 5/11/2016 1:47 PM, Charles Steinkuehler wrote:
> 
> ...the problem is the first step is to create a branch, which for most
> casual contributors also means creating a clone of the docs repo.

I should have known, GitHub has already done this for us:

https://help.github.com/articles/editing-files-in-another-user-s-repository/

So just craft an 'edit' link, like so:

https://github.com/machinekit/machinekit-docs/edit/master/machinekit-documentation/getting-help.asciidoc

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-05-11:
> On 5/11/2016 1:47 PM, Charles Steinkuehler wrote:
>
> I should have known, GitHub has already done this for us:
>
> https://help.github.com/articles/editing-files-in-another-user-s-repository/
>
> So just [...]

Getting that to work would take away all my objections. And would be really lowering barriets. One only needs a github account.
Thanks!

mhaberler, 2016-05-11:
yes, that was my plan

[325 more replies]

closed 2016-07-31 by luminize