machinekit/machinekit #928: Status of the machinekit-rt-preempt Debian package on armhf

koppi, 2016-05-02:
On a Beckhoff [CX9020](https://github.com/koppi/mk/blob/master/Machinekit-Beckhoff-CX9020.md), uname -a:
```bash
Linux CX9020 4.1.12-rt13-CX9020-9+ #1 PREEMPT RT Fri Apr 29 01:09:47 CEST 2016 armv7l GNU/Linux
```
I just did a:
```bash
apt install machinekit machinekit-dev machinekit-rt-preempt
```
In ```/usr/share/linuxcnc/Makefile.inc``` I read:
```bash
#                                                          -*-makefile-gmake-*-
# Makefile.inc.  Generated from Makefile.inc.in by configure.
# on Fri Apr 29 11:37:09 UTC 2016
#

# Define libexec dir first, since it contains tools needed to compute
# following definitions
EMC2_LIBEXEC_DIR=${exec_prefix}/libexec/linuxcnc

# Threads systems
#################

# List of thread flavors to build modules
BUILD_THREAD_FLAVORS= posix
```

I want to build https://github.com/sittner/linuxcnc-ethercat with the thread flavour rt-preempt, but rt-preempt is not listed in the file above.

Am I missing anything obvious? Howto enable the rt-preempt thread flavour on armhf?

the-snowwhite, 2016-05-03:
Going a bit out on a thin branch here but nevertheless:

1.   This is machinekit (forked from linuxcnc 2 years ago or so)
Your module clearly states linuxcnc, but no mention of Machinekit compatibility.

2. Machinekit has an arm branch  armhf for 32-bit embedded systems, recently moving towards the rt-preempt kernel flavor.

https://github.com/koppi/mk/blob/master/Machinekit-Beckhoff-CX9020.md

??? I have never heard of Machinekit recommending anything other than Debian for os.

and 4.1.12-rt13-CX9020-9+ is new to the ballpark also afaik.

So your battle seems to be a bit uphill as this seems to be some sort or new machinekit port you are mentioning ,,, unknown at least to me.

koppi, 2016-05-03:
Because LinuxCNC and Machinekit did not yet diverse to much at lease for the HAL layer part

https://github.com/sittner/linuxcnc-ethercat

 is still compatible with ```machinekit-dev```, see:

https://github.com/sittner/linuxcnc-ethercat/blob/master/debian/control

In the docs at https://github.com/koppi/mk/blob/master/Machinekit-Beckhoff-CX9020.md
I'm using Ubuntu just as a build host for cross-compilation, the target is Debian/Jessie armhf.

Yes, it's my ongoing efford to port Machinekit to the Beckhoff CX9020. But I'm stuck at
```/usr/share/linuxcnc/Makefile.inc``` from ```machinekit-dev``` that does not contain the line:
```bash
# List of thread flavors to build modules
BUILD_THREAD_FLAVORS= posix rt-preempt
```
but rather this line:
```bash
# List of thread flavors to build modules
BUILD_THREAD_FLAVORS= posix
```
even when I install the package ```machinekit-rt-preempt```. Any ideas?

the-snowwhite, 2016-05-03:
you should try @mhaberler  he may have some better suggestions

koppi, 2016-05-03:
@the-snowwhite what do you mean by "better suggestions"?

the-snowwhite, 2016-05-03:
than me about the configure rt-preempt for armhf question ... however:

I now see that the hal instcomp / comp compiler is to be used by the build script.

This is one of the components that has been under heavily reconstruction in mk like adding python modules and also some (multi ?) threading changes.  i have no idea if how it works under the new armhf re-preempt kernel system.

the-snowwhite, 2016-05-03:
https://github.com/sittner/linuxcnc-ethercat/blob/master/configure.mk#L1

the-snowwhite, 2016-05-03:
https://groups.google.com/forum/#!searchin/machinekit/comp$20changes/machinekit/hGFtzwi4sw4/IP60179fBwAJ

There are some documention threads about the changes and how to build with the new style.

the-snowwhite, 2016-05-03:
Also:
https://groups.google.com/forum/#!searchin/machinekit/comp$20changes|sort:date/machinekit/r8oRB0AnJt8/GPdczG1d-loJ

Hope this helps
Best wishes
Michael

You need to install machinekit-dev for your kernel flavour

koppi, 2016-05-03:
> You need to install machinekit-dev for your kernel flavour

```bash
# dpkg -l|grep machinekit
ii  machinekit                             0.1.1462206741-1mk.travis.master.gite43c76f6~1jessie armhf        PC based motion controller for real-time Linux
ii  machinekit-dev                         0.1.1462206741-1mk.travis.master.gite43c76f6~1jessie armhf        PC based motion controller for real-time Linux
ii  machinekit-rt-preempt                  0.1.1462206741-1mk.travis.master.gite43c76f6~1jessie armhf        PC based motion controller for real-time Linux
```
@the-snowwhite am I missing a package I'm not aware of?

```bash
root@CX9020:~# comp --print-modinc
/usr/share/linuxcnc/Makefile.modinc
```

```bash
root@CX9020:~# head -n 16 /usr/share/linuxcnc/Makefile.inc
#                                                          -*-makefile-gmake-*-
# Makefile.inc.  Generated from Makefile.inc.in by configure. 
# on Mon May  2 18:22:03 UTC 2016
#

# Define libexec dir first, since it contains tools needed to compute
# following definitions
EMC2_LIBEXEC_DIR=${exec_prefix}/libexec/linuxcnc

# Threads systems
#################

# List of thread flavors to build modules
BUILD_THREAD_FLAVORS= posix

# If $(threads) is not set already, try running 'flavor'; if the
```
```/usr/share/linuxcnc/Makefile.inc``` is from the ```machinekit-dev``` package:

the-snowwhite, 2016-05-03:
Hmmm I'm not sure if you actually are missing anything.. then

What are your build instructions for the ethercat module ?

What happens if you (After making a backup of you current sd os image), run the build
instructions ?

If it fails what error messages do you get in your console ?

the-snowwhite, 2016-05-03:
BTW the docs are being restructured ... atm:
http://machinekit-docs.readthedocs.io/en/latest/src/code/Debugging-RT-components/

the-snowwhite, 2016-05-03:
https://www.google.dk/?gws_rd=cr,ssl&ei=qh8pV_SsB8KpsAHn163ADA#q=machinekit+comp+debugging+rt+components

koppi, 2016-05-03:
When I build linuxcnc-ethercat, only the thread flavour posix gets built, because the thread flavour rt-preempt is missing in the file ```/usr/share/linuxcnc/Makefile.inc```:
```bash
root@CX9020:~# cd linuxcnc-ethercat/
(failed reverse-i-search)`debu': dpkg -i linux-headers-4.1.12.cx9020-rt13+_20160502181701_armhf.deb linux-image-4.1.12.cx9020-rt13+_20160502181701_armhf.^Cb 
root@CX9020:~/linuxcnc-ethercat# vi README.md 
root@CX9020:~/linuxcnc-ethercat# debuild -uc -us
 dpkg-buildpackage -rfakeroot -D -us -uc
dpkg-buildpackage: warning: using a gain-root-command while being root
dpkg-buildpackage: source package linuxcnc-ethercat
dpkg-buildpackage: source version 0.9.2
dpkg-buildpackage: source distribution unstable
dpkg-buildpackage: source changed by Sascha Ittner <sascha.ittner@modusoft.de>
 dpkg-source --before-build linuxcnc-ethercat
dpkg-buildpackage: host architecture armhf
 fakeroot debian/rules clean
dh clean 
   dh_testdir
   dh_auto_clean
make[1]: Entering directory '/root/linuxcnc-ethercat'
make[2]: Entering directory '/root/linuxcnc-ethercat/src'
rm -f *.so *.ko *.o
rm -f *.mod.c .*.cmd
rm -f modules.order Module.symvers
rm -rf .tmp_versions
rm -f lcec_conf
make[2]: Leaving directory '/root/linuxcnc-ethercat/src'
rm -f config.mk config.mk.tmp
make[1]: Leaving directory '/root/linuxcnc-ethercat'
   dh_clean
 dpkg-source -b linuxcnc-ethercat
dpkg-source: warning: source directory 'linuxcnc-ethercat' is not <sourcepackage>-<upstreamversion> 'linuxcnc-ethercat-0.9.2'
dpkg-source: info: using source format `1.0'
dpkg-source: info: building linuxcnc-ethercat in linuxcnc-ethercat_0.9.2.tar.gz
dpkg-source: info: building linuxcnc-ethercat in linuxcnc-ethercat_0.9.2.dsc
 debian/rules build
dh build 
   dh_testdir
   dh_auto_configure
   dh_auto_build
make[1]: Entering directory '/root/linuxcnc-ethercat'
make[2]: Entering directory '/root/linuxcnc-ethercat/src'
make[3]: Entering directory '/root/linuxcnc-ethercat/src'
arm-linux-gnueabihf-gcc -o lcec_conf.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -URTAPI -U__MODULE__ -DULAPI -Os -c lcec_conf.c
arm-linux-gnueabihf-gcc -o lcec_conf lcec_conf.o -Wl,-rpath,/usr/lib -L/usr/lib -llinuxcnchal -lexpat
make[3]: Leaving directory '/root/linuxcnc-ethercat/src'
make[3]: Entering directory '/root/linuxcnc-ethercat/src'
arm-linux-gnueabihf-gcc -o lcec_main.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_main.c
arm-linux-gnueabihf-gcc -o lcec_generic.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_generic.c
arm-linux-gnueabihf-gcc -o lcec_el1xxx.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el1xxx.c
arm-linux-gnueabihf-gcc -o lcec_el1252.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el1252.c
arm-linux-gnueabihf-gcc -o lcec_el2521.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el2521.c
arm-linux-gnueabihf-gcc -o lcec_el2xxx.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el2xxx.c
arm-linux-gnueabihf-gcc -o lcec_el2202.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el2202.c
arm-linux-gnueabihf-gcc -o lcec_el31x2.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el31x2.c
arm-linux-gnueabihf-gcc -o lcec_el40x1.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el40x1.c
arm-linux-gnueabihf-gcc -o lcec_el40x2.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el40x2.c
arm-linux-gnueabihf-gcc -o lcec_el41x2.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el41x2.c
arm-linux-gnueabihf-gcc -o lcec_el5101.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el5101.c
arm-linux-gnueabihf-gcc -o lcec_el5151.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el5151.c
arm-linux-gnueabihf-gcc -o lcec_el5152.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el5152.c
arm-linux-gnueabihf-gcc -o lcec_el7041_1000.o -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__ -I/usr/include/linuxcnc -fPIC -g -funwind-tables -Os -c lcec_el7041_1000.c
[... cut ...]
E: linuxcnc-ethercat: copyright-contains-dh_make-todo-boilerplate
E: linuxcnc-ethercat: extended-description-is-empty
E: linuxcnc-ethercat: section-is-dh_make-template
W: linuxcnc-ethercat: binary-without-manpage usr/bin/lcec_conf
Finished running lintian.
root@CX9020:~/linuxcnc-ethercat# 
```

I'm not willing to hand-edit ```/usr/share/linuxcnc/Makefile.inc``` to add ```rt-prempt``` to
```bash
# List of thread flavors to build modules
BUILD_THREAD_FLAVORS= posix
```
Maybe @mhaberler has an idea then?

koppi, 2016-05-03:
I was a bit annoyed by this and hand-edited ```/usr/share/linuxcnc/Makefile.inc``` to read:
```bash
# List of thread flavors to build modules
BUILD_THREAD_FLAVORS= posix rt-preempt
```
Now the rt-preempt part of the LinuxCNC-EtherCAT HAL module gets built:
```bash
root@CX9020:~# ldd /usr/lib/linuxcnc/rt-preempt/lcec.so 
	linux-vdso.so.1 (0xbeddd000)
	liblinuxcnchal.so.0 => /usr/lib/liblinuxcnchal.so.0 (0xb6ef2000)
	libethercat.so.1 => /usr/lib/libethercat.so.1 (0xb6edc000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6eb7000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb6e8e000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6d9e000)
	liblinuxcnc-pb2++.so.0 => /usr/lib/liblinuxcnc-pb2++.so.0 (0xb6cc8000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb6c10000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6bfd000)
	libprotobuf.so.9 => /usr/lib/arm-linux-gnueabihf/libprotobuf.so.9 (0xb6b21000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6afe000)
	libczmq.so.1 => /usr/lib/arm-linux-gnueabihf/libczmq.so.1 (0xb6ab6000)
	libzmq.so.3 => /usr/lib/arm-linux-gnueabihf/libzmq.so.3 (0xb6a6b000)
	/lib/ld-linux-armhf.so.3 (0x7f56f000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb69f7000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0xb69d5000)
	libsodium.so.13 => /usr/lib/arm-linux-gnueabihf/libsodium.so.13 (0xb6981000)
	libpgm-5.1.so.0 => /usr/lib/libpgm-5.1.so.0 (0xb6931000)
```

I will update document at https://github.com/koppi/mk/blob/master/Machinekit-Beckhoff-CX9020.md
with the steps necessary to get Machinekit running on the Beckhoff CX9020 running.

Except for this single issue, it seems, that everything is working out of the box, so congrats to the Machinekit / ARM / RT-Linux / Travis / Docker / Debian packaging experts, good work so far.

the-snowwhite, 2016-05-03:
https://github.com/the-snowwhite/soc-image-buildscripts/blob/master/MK-Cross-build/mk-rip-build.sh#L89

 configure --with-rt-preempt --with-platform-socfpga

was whart I last used to recomcompile the rip build on @mhaberler  's latest mksocfpga sd-image.

this should do the trick for you:

 configure --with-rt-preempt

the-snowwhite, 2016-05-03:
:-)
you caught the same thought...

the-snowwhite, 2016-05-03:
Great .. Congrats

I think this is the very first new armhf package build install only port, added ...

koppi, 2016-05-03:
@the-snowwhite but please fix this in the Debian packaging mechanism somehow.

– The reason I opened this issue is to point with a big finger to the mechanism that is somehow broken for the different rt flavours with the machinekit-dev package.

the-snowwhite, 2016-05-03:
@mhaberler 
Is the package build master for me.
You should make a PR now you have the fix in hand.
This would be the best way to donate your gratitude back into the porject

the-snowwhite, 2016-05-03:
I don't know where the file you edited is in the source code ?
https://github.com/machinekit/machinekit/search?utf8=%E2%9C%93&q=Makefile.inc

koppi, 2016-05-03:
It gets auto-generated somehow during the configure step I guess.

the-snowwhite, 2016-05-03:
Yep that is also as far as i can see in that direction...

Please leave this issue open ,, and maybe look / search for related issues to link into this thread.

I would expect it to affect any one trying to use the comp component on the new armhf rt-preempt platform... including the latest BBB jessie beta sd-image.

Right now most armhf focus is going into getting the socfpga (Altera / Xilinx ..etc) fully up and running with full image build's .. u-boot, kernel, device-tree support, etc.

So I don't think anyone has had time to try using the comp component here before you did-

koppi, 2016-05-03:
@the-snowwhite can you point me to the current mk-builder scripts? – I can give a try at getting these working for the Beckhoff CX9020 so that one can automate the creation of an sd-image.

A slightly patched u-boot, a slightly patched kernel, device tree file and a bitfile for the CX9020 is required for this platform too, see:

* https://github.com/Beckhoff/CX9020/blob/master/tools/prepare_uboot.sh
* https://github.com/Beckhoff/CX9020/blob/master/tools/prepare_kernel.sh
* https://github.com/Beckhoff/CX9020/blob/master/scripts/50_install_kernel.sh

and

* https://github.com/Beckhoff/CX9020/tree/master/u-boot-patches
* https://github.com/Beckhoff/CX9020/tree/master/kernel-patches

I could work on this, if time permits next weekend or so..

the-snowwhite, 2016-05-04:
Let me link you to the recipie I have used for my current development image Im running now:

I just made a post discribing what you request a few hours ago here in this post:

https://github.com/machinekit/machinekit/issues/903#issuecomment-216618965

the sd-image link (+ some docs) is here:

https://github.com/machinekit/machinekit/issues/915#issuecomment-215223440

-----

The current mksocfpga image builds are put into Robert Nelsonf omap builder system, which is complex and not easy to  customize, or run locally.

---

THe buildscripts are the ones I developed to manifest a whole working sd-image.
You do not now need any of the rootfs gen parts when you use the extracted rootfs.

I you want to use  the u-boot and kernel gen builds uncomment there 2 lines:
(after custommizing the variables  in the top of the script)
https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/SD-Image-Gen/mksocfpga-debian-rel2-sd-image-gen.sh#L711

parameters are then put into the uboot script:
https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/SD-Image-Gen/build_uboot.sh

and the kernel build script:
https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/SD-Image-Gen/build_kernel.sh

from the main script (run the script with cd in a work folder external to the script folder):
(and put any character afterwards like a . )
mksocfpga-debian-rel2-sd-image-gen.sh

    repopath:/soc-image-buildscripts/SD-Image-Gen/mksocfpga-debian-rel2-sd-image-gen.sh .

Remember to checkout the 4.1-ltsi.rt branch first ...

mhaberler, 2016-05-04:
@koppi -  it'd be great to have a packaged ethercat master in the stable!

the flavors list in `/usr/share/linuxcnc/Makefile.inc` seems indeed broken as it is not updated when installing the `machinekit-<flavor>` packages

a possible fix could be a postinstall script for machinekit-dev which fixes Makefile.inc to include the list of flavors installed, and a `machinekit-<flavor>` postinstall/postrm script which runs dpkg-reconfigure on machinekit-dev to do the fixup on flavor install/uninstall

care to look into it and post a PR ?

mhaberler, 2016-05-04:
@koppi -  some code for the postinstall script could be lifted from here: https://github.com/machinekit/machinekit/blob/master/debian/configure#L27-L35

mhaberler, 2016-05-04:
@koppi - as for image build scripts:

currently the de0 image is built with https://github.com/mhaberler/omap-image-builder/tree/machinekit-jessie and [this config](https://github.com/mhaberler/omap-image-builder/blob/machinekit-jessie/configs/machinekit-debian-jessie-de0.conf) 

this works but requires an arm buildhost; it does pull together an image based on packages, other than @the-snowwhite 's build scripts which compile everything from source - this is not supportable downstream so we will not use his scrips for images but just take clues

@kinsamanka has come up with a method doing this on amd64 in a docker container and I think we should migrate to this method as it does not require an arm buildhost and looks easier: https://github.com/kinsamanka/mk_image_builder

I really recommend to look into mk_image_builder - please work with Gemi on that, happy to lurk and take any results into jenkins to have images automatically built on merge to a github repo

what we do need though is unified package lists between the various image building methods so the surprise factor comes down a bit

luminize, 2017-07-26:
closed in favour of https://github.com/machinekit/machinekit/issues/1238

closed 2017-07-26 by luminize