machinekit/machinekit #929: Machinekit How To / How does / How can I tracker missing

the-snowwhite, 2016-05-04:
Problem:

When working on solving an issue a contributer will often face questions like
the ones in this new thead I will use as an example below in message nr2:

The type of questions boil down asking for guidelines of how to do a spesific task.

The issue tracker contains a lot of rerepeted ansews to how do I type of questions,
that could be collected and linked to in a howto type tracker instead.

The issue tracker is geared towards formulating a problem that needs to be solved
and then the issue is closed.

I suggest we invent a likewise tracker where requirements are formulating how to
accomplish a given task / or / explaning providing info about how something works:

The idea is that when we are trying to convey guidance like in the examples below,
we start by making a generic howto, and then link to that in the spesific thread
and the add eventual information needed for the actual context.

Next time we see a need for the same answer It can then be linked in, without having to be written once more.

the-snowwhite, 2016-05-04:
Examplefication:

https://github.com/machinekit/machinekit/issues/928#issuecomment-216748740

	@koppi - as for image build scripts:

	currently the de0 image is built with https://github.com/mhaberler/omap-image-builder/tree/machinekit-jessie and this config

	this works but requires an arm buildhost; it does pull together an image based on packages, other than @the-snowwhite 's build scripts which compile everything from source - this is not supportable downstream so we will not use his scrips for images but just take clues

	@kinsamanka has come up with a method doing this on amd64 in a docker container and I think we should migrate to this method as it does not require an arm buildhost and looks easier: https://github.com/kinsamanka/mk_image_builder

	I really recommend to look into mk_image_builder - please work with Gemi on that, happy to lurk and take any results into jenkins to have images automatically built on merge to a github repo

	what we do need though is unified package lists between the various image building methods so the surprise factor comes down a bit

https://github.com/machinekit/machinekit/issues/928#issuecomment-216695969

can you point me to the current mk-builder scripts?

I this context this is question I have answered a few hours ago in a dirrerent thread,
so I try to link to that answer allready given in a different context with folowwing message here:

https://github.com/machinekit/machinekit/issues/928#issuecomment-216703511

----

	Let me link you to the recipie I have used for my current development image Im running now:

	I just made a post discribing what you request a few hours ago here in this post:

	https://github.com/machinekit/machinekit/issues/903#issuecomment-216618965

	the sd-image link (+ some docs) is here:

	https://github.com/machinekit/machinekit/issues/915#issuecomment-215223440

	-----

	The current mksocfpga image builds are put into Robert Nelsonf omap builder system, which is complex and not easy to  customize, or run locally.

	---

HOWTO -->

	THe buildscripts are the ones I developed to manifest a whole working sd-image.
	You do not now need any of the rootfs gen parts when you use the extracted rootfs.

	I you want to use  the u-boot and kernel gen builds uncomment there 2 lines:
	(after custommizing the variables  in the top of the script)
	https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/SD-Image-Gen/mksocfpga-debian-rel2-sd-image-gen.sh#L711

	parameters are then put into the uboot script:
	https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/SD-Image-Gen/build_uboot.sh

	and the kernel build script:
	https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/SD-Image-Gen/build_kernel.sh

	from the main script (run the script with cd in a work folder external to the script folder):
	(and put any character afterwards like a . )
	mksocfpga-debian-rel2-sd-image-gen.sh

		repopath:/soc-image-buildscripts/SD-Image-Gen/mksocfpga-debian-rel2-sd-image-gen.sh .

	Remember to checkout the 4.1-ltsi.rt branch first ...

---

The revelant info I fail to be able to link in (it took me 20 min to dig it out today) is in this thread here:

https://github.com/machinekit/machinekit/issues/903#issuecomment-215306924

	the only thing specific to the socfpga on this image is kernel, uboot, FPGA blob+dtb's
	so this can be mutated quickly into a Zynq developer image by removing those parts, and adding any Zynq specifics:

	#remove socfpga specifics:
	apt remove -y   linux-headers-socfpga-rt    \
		linux-image-socfpga-rt  \
		socfpga-rbf \
		u-boot-socfpga
	# add any Zynq specifics

the info in @mhaberler s messages from here:

https://github.com/machinekit/machinekit/issues/928#issuecomment-216748740

would also likely be candidates for more generic howtos than having to be givein repetedly in different issues:
There are also how does this work info that could be collected in the hoto / howdoes  tracker

mhaberler, 2016-05-04:
The tracker is the wrong vehicle for this, and just adding noise to an already noisy issue list

what we need is a FAQ vehicle in the manual

luminize, 2016-05-04:
This sort of information on how to set up stuff should be put in 
specific sections in the documentation.
Where in case of need references to other parts of the documentation can 
be made
Applicable sections can be added depending on subject.

see these places for example.
https://github.com/machinekit/machinekit-docs/tree/master/machinekit-documentation
like the subdirs "getting started" and "setting-up"

luminize, 2016-05-04:
Having said the above, this of course applies to information that is relevant to Machinekit, and not a snippets collection from various developers which will rot with time. So if it's not maintained it will be removed.

the-snowwhite, 2016-05-04:
The purpose is to separate the duplicated information in the issue threads out into 1 message containing
(howto like)documentation.

To server as an intermediate step towards moving stuff into the doc repo.

This will make the length of the issue threads much, much shorter and concentrate the problem solving issues into only mentioning unresolved problems.

And giving a new placeholder to link to the solutions, suggestion of solutions, how to do this and that.
Which the naturally get us 1 step closer to have stuff to pull directly into the doc repo that provides info for both developers AND non developers.

Nope:
I disagree with you both and here is why (above). And I'm willing to bet 1 year of wages that if you look through the already created issues and think of sorting out all the messages related to howto do this as suggested above.

And then run a script that does that ,,, a lot of the noise will have been removed from that see of allready generated information.
And much lost information will surface and more clearly stand out as something to migrate into the Machinekit docs.

btw.
I am talking about a placeholder for structuring developer documentation at the very point it is given and written between developers.
Then is can more easiely be pipelined upwards towards the doc repo, with no other effort than a change of habits:

Instead of writing you suggestions for a solution directly in 1 thread where quickly gets lost.
then start by giving it an howxx Title and then link it into the problem issue thread.

Every time you need it.

the-snowwhite, 2016-05-04:
@luminize 

I see the distance of getting info into the docs as to far.

This step can serve as an step to make that distance shorter, while removing a lot of the noise generated because the issue threads contain 2 types of information that can be split out at the source.

the-snowwhite, 2016-05-04:
What I want to clarify is that when you create an issue you start out with having a problem, that needs to get fixed.

When you come into write / add a post to someone else's issue you are coming from from a fokus of how to facilitate information already given and if not provide the steps to create a solution, which the may or may not end up as being added to the documentation later.

A problem formulation is never intended to end up in the documentation, .....unless its unfixable ?

luminize, 2016-05-04:
Michael (B)
sorry to pop the balloon from my point of view, please see below

On 5/4/2016 12:06 PM, Michael Brown wrote:
>
> The purpose is to separate the duplicated information in the issue 
> threads out into 1 message containing
> (howto like)documentation.
>
> To server as an intermediate step towards moving stuff into the doc repo.
>

this will dilute the purpose of documentation and will make developers 
even more lazy in writing documentation with images, flow charts, and 
whatnot. It motivates information getting defragmentated.
Info in an issue thread during development is very quick to become obsolete.
I don't believe that developers will gather and keep these information 
bits into a thread.

> This will make the length of the issue threads much, much shorter and 
> concentrate the problem solving issues into only mentioning unresolved 
> problems.
>

i don't agree. The way to keep issue threads short is to spin out 
separate issues when those can be self contained issues

> And giving a new placeholder to link to the solutions, suggestion of 
> solutions, how to do this and that.
> Which the naturally get us 1 step closer to have stuff to pull 
> directly into the doc repo that provides info for both developers AND 
> non developers.
>

nope, don't agree, see the remark about diluting

> Nope:
> I disagree with you both and here is why (above). And I'm willing to
> bet 1 year of wages that if you look through the already created
> issues and think of sorting out all the messages related [...]

Start writing documentation before you do development will be even 
better than the current way.
Keep development information relevant in a files that can have version 
control

The github issue tracker is IMO an unsuited vehicle for gathering 
information and bringing structure in information. It is in fact an 
unstructured blob of separate issues, and should be treated as an 
unstructured blob.

I will not support any effort in getting structure in the github issue 
tracker. Period

> btw.
> I am talking about a placeholder for structuring developer
> documentation at the very point it is given and written between
> developers.
> Then is can more easiely be pipelined upwards towards [...]

luminize, 2016-05-04:
On 5/4/2016 12:09 PM, Michael Brown wrote:
> @luminize <https://github.com/luminize>
>
> I see the distance of getting info into the docs as to far.
>
> This step can serve as an step to make that distance shorter, while
> removing a lot of the noise [...]
this IMO has nothing to do with making the distance for documenting shorter.
https://github.com/machinekit/machinekit/issues/924 deals with getting 
info into docs and make the flow a tad better.

the-snowwhite, 2016-05-04:
@luminize 

I understand why you from you current view - point maybe with you focus deep into the current mk doc rebuild are totally unable to grasp new the concept / idea Im laying out.
And so totally miss the point ....

I really don't care as to where this new (start by formulating a solution, guide or documentation, issue begin tracker is hosted).

I think the days of fluffy verbose documentation efforts are over and thats why every new Mk documentation effort flies flat to the ground, repetedly.

Fragmented information is however the most obvious way to go, If you give yourself time to look into the idea:

Some people are looking for instructions to copy and paste into their console, others rely on coding examples for learning, others tutorials, or videos / some hate videos. etc.

When I generate information for documentation purposes they tend to come out as snippets of information related to above categories anyway.

So I could just tag that snippet appropriately, and save it somewhere and the save it for later compilation into a rigid  static document as usual old fashion way to do stuff.

Or I could just use a template with links to all the sniplets code examples, book style reading, man style doc pages (which are mostly incomprehensive for me btw.).

Then the reader can determine and select to view which style / type of the documentation on that specific subject / topic works best for them. (And not only have the current 1 static version, where you only can change how pretty it looks in you eyes).

---

The key is being able to link the sniplets in ways that provide the exact information needed for an inquiry in the style that works best for the one seeking that information. 
Which is different for different people.

the-snowwhite, 2016-05-04:
I also have to say that no one seems to have noticed that we actually have way too much information, and this seriously hides the access to finding revelent information, much more that we have recognized so far.

A lot of the pages in the documentation will also be providing the same (staticly) duplicated snippets
used in different subjects, so It would be logicically to only have the same one to pull into the different pages that use it.

Also if that specific instruction needs correction then it can be formulated as a documentation issue problem, filed against the original how...xxx placeholder that lead to that documentation in the first place ...?  and a correction (patch) can then be committed for review.

the-snowwhite, 2016-05-04:
----

BTW are you suggesting that I should have used / suggested the doc repo tracker as the placeholder for the how..xxx --> generate documentation idea ?

As a way to begin documenting a new subject ?

luminize, 2016-05-04:
@the-snowwhite 

>BTW are you suggesting that I should have used / suggested the doc repo tracker as the placeholder for the how..xxx --> generate documentation idea ?

no, I suggest to make commits to the documentation if there is information on how to set up a device, document the steps to take, provide diagrams on the workflow and underlaying processed. Not for a first quick hack of course, for something that's in Machinekit.

>I understand why you from you current view - point maybe with you focus deep into the current mk doc rebuild are totally unable to grasp new the concept / idea Im laying out.
And so totally miss the point ....

Please don't get personal about me not being able to grasp new ideas. I don't like that attitude at all.

I disagree with a lot of things you state above. I think there are some good ideas, but I believe they won't work in real life if they are just a bunch of code snippets without providing explanation on them, providing structure and step by step explaining the concepts.

That does not mean that I want to change your opinion. I just don't agree.

luminize, 2016-05-04:
On 5/4/2016 1:02 PM, Michael Brown wrote:
>
> also have to say that no one seems to have noticed that we actually 
> have way too much information, and this seriously hides the access to 
> finding revelent information, much more that we have recognized so far.
>

the problem is structure, step by step instructions. Not information

> A lot of the pages in the documentation will also be providing the 
> same (staticly) duplicated snippets
> used in different subjects, so It would be logicically to only have 
> the same one to pull into the different pages that use it.
>

and that's what asciidoc does

> Also if that specific instruction needs correction then it can be 
> formulated as a documentation issue problem, filed against the 
> original how...xxx placeholder that lead to that documentation in the 
> first place ...? and a correction (patch) can then be committed for 
> review.
>

That seems to be a very hard way and a lot of steps to change 
documentation, I'd prefer to file an issue against machinekit-docs, or 
better, provide a PR fixing the issue.

mhaberler, 2016-05-04:
@the-snowwhite  - "I see the distance of getting info into the docs as to far."

spot on. And the logical consequence is to reduce that distance by helping refurbish structure, contents, process and appearance of machinekit-docs, not to dilute information into different channels.

Oh, and the side effect of abusing the tracker for FAQ purposes is: dilution of the tracker. What's that going to be - "issues" which are open forever, but arent "issues"? 

Here is an example how we could do FAQ's in jekyll: http://idratherbewriting.com/documentation-theme-jekyll/mydoc_faq_layout/

the-snowwhite, 2016-05-04:
Sorry and thanks for the correction I have no intention of being personal.

What I meant was:

That is not the point I'm trying to make at all.

The first place I looked when I arrived was the doc links you are pointing at.

I am not being ironic but what this link provides right now seems to sum up what I have experienced every time I have tried to add documentation to the formal structure of machinekit in the recomended guidelines:

    actually help improve this documentation without knowing anything of "git". Just read these instructions and you’ll be ready to go in the blink of an eye.

Instead I have generated a considerable amount of documentation regardless of repeatedly being turned of every time i tried to use the Mk doc guidelines in the last year or so.

And instead focused on wondering about what would help me generate better documentation while developing and how to get that linked back into Machinekit docs in the relevant places. 

Remember my main focus the last 7-8 month has been in adding something new to Machinekit so naturally the Idea is based upon how to add new unwritten documentation and exchange way to do this and that, and not so much related to the existing bulk of documentation.

And for me the idea of creating a new issue in the doc repo like the doc I made about how the hm2 driver works, could have very easily been started as a how does the hm2 hdl code work ?  subject and then followed through and made its way into the docs, Instead of just laying about somewhere close.

Also goes for something like how to make a device-tree fragment work.

I could have put directly into the doc repos developer section ? , 
instead of it getting buried in my own repo 8-3 dayes ago, leading to those needing that information yesterday or the day before, found their way to what I had provided correctly a week ago. 24-hours or so to late, as I was caught up in something else at that critical time.
(sorry for the rant).

However I can't see that is how it can be done in the doc repo.
And honestly I am not able to comprehend what Im supposed to do when I go here:

https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/getting-help.asciidoc

I go there ... and nothing happens.

???

https://github.com/machinekit/machinekit-docs/blob/master/documenting/documenting.asciidoc

    {"error":"Not Found"}

the-snowwhite, 2016-05-04:
Thank you both for your thoughts so far.

I have reformulated the first post based upon you feed back trying to paint a more clear picture, of what I should written from the beginning..... :-)

the-snowwhite, 2016-05-04:
I have now also changed the title to be more precise:

Please allow me to try to care out the newest concrete example of what has 
been happening a lot to me, and I do not think I'm the only one having
innues simmalar to this very recent event:

I posted a solution for a working device-tree fragment solution 8 dayes ago:

the-snowwhite commented 8 days ago • edited
https://github.com/machinekit/machinekit/issues/903#issuecomment-214667809

overlay compile load and notes:
https://github.com/the-snowwhite/soc-image-buildscripts/blob/4.1-ltsi-rt/inst-rbf-dtb.sh

THe working dts info (meaning full for both the Altera and Zync related threads) 

could have been placed somewhere much more available (like in the proposed dev doc buffer)
instead of where it got allmost lost.

----

https://github.com/machinekit/machinekit/issues/903#issuecomment-214670642

2 days ago I then extract a readme doc from the script, and try to relocate it to
somewhere more revelant.

However I copy in the information from the wrong file, creating a flaw.

driver with added overlay load, and fpga-mgr (unfinished).
https://github.com/machinekit/machinekit/issues/915#issuecomment-216094490

1 day later "someone" points out the mishap:
@the-snowwhite : note your DTS posted above:
https://github.com/machinekit/machinekit/issues/915#issuecomment-216452896

And Im able to correct it again:

Hi guys
Sorry for my screw up (7-2 days ago) w...
https://github.com/machinekit/machinekit/issues/903#issuecomment-216615690

mhaberler commented a day ago:
puzzle solved - @the-snowwhite had the right dts well hidden: https://github.com/the-snowwhite/soc-image-buildscripts/blob/cdc30f20b9645a051449f5e6f84bf88a740b14d7/dts_overlay.sh#L15-L38

Bottom line:

All the dts related events in the (other) Zync thread from here:
mhaberler commented 3 days ago • edited
#903 Adding overlay based hostmot2 fpga driver required by Zynq SoCs
https://github.com/machinekit/machinekit/issues/903#issuecomment-216259363

Until yesterday, are events of the type I really, really would love to be able to minimize
the potential of being able to happen for anyone involved in the development work and docu effort,
as this puts a damper into being able to be more pruductive within the same timeframe.

This is what I intend to resolve in this issue, and it cannot be solved without some sort of change ...?

luminize, 2017-07-23:
Since this thread seems to be dead, and no work has come from this I'll close it.

closed 2017-07-23 by luminize