machinekit/machinekit #930: Milestone for the mksocfpga missing

the-snowwhite, 2016-05-04:
AFAIK   there has been a clear goal setting for all separate issues created related to this project.

And there has been a spin off of about 10  issue threads of so that can be attached / gathered into 
a properly named milestone that can serve towards providing a higher level placeholder to keep fokus upon unifying these platforms and using shared genetic code and structures as much as possible.

Mksocfpga could a good title for the milestone based collection of issues.

The nano soc image was the first manifestation of a concrete formulatable stepping stone towards the 
overall goal of this sub project.

However the higher purpose, scope and reason is to explore, expand and enhance the possibilities and ways of utilizing dynamic re-configurable programmable logic attached to cpu dies in soc like systems.
(Lets not forget that this is also underway for x86 systems).
And of course for motion control, in the group that has / is providing the infrastructure for manifesting the "new thing".

I will not have any shame in admitting that i see possibilities for more generic (non mk) related uses (also of the sd-images) that I'm trying to keep this project open for.

the-snowwhite, 2016-05-04:
I think this readme from the mksoc repo contains the bulk of the purpose behind what this milestome is about and should have contained from the very beginning of the mksoc repo:

the-snowwhite, 2016-05-04:
# mksoc
Repository for Soc Machinekit ports

(Right now this repo only serves as an container for future development and initial historic purposes)

[current development is here:](https://github.com/the-snowwhite/mksocfpga)  

---

The mksoc name contains the broad inclusive scope of running machinekit on any soc based platform,

with or without fpga.

Initially the link to machinekit is provided by the open sourced hostmot2 hdl cores from Mesa, which is tested and proven to be reliable and also provides compatable physical I/O interfaces that require litte work to connect. 

This whole structure  also targets soc and fpga on separate chips If such opensource projects are to emerge.

This is where the development of the Cyclone V mksocfpga project started, so it seemed natuaral to create

a more limited scope named repo called mksocfpga to move this platform to:

The mksocfpga name scope is limited down to: running Machinekit on Soc's where cpu and fpga are in same chip.

This only includes the Altera and Xilinx socfpga platforms currently.

Starting from the bottom of low cost availability effort has so far been concentrated on Alteras Cyclone V soc family

Manifested thru the lowest cost DE0 Nano Soc board from Terasic with the alias Atlas Soc in the Rocketboards.org eco system.

Terasic maintain a different support and tutorial ecosystem via their web: http://www.terasic.com.tw/

It is hoped to gain traction for adding a Zync based platform simmalar to the DE0 Nano Soc, by a person with such a board and experience with the Xilinx tool's.

! Current and updated souces are at:
mksocfpga repo instead-->
https://github.com/the-snowwhite/mksocfpga

-----------------------

Historic info:

Quartus makefile in mksoc/QuartusProjects/DE0_NANO_SOC_GHRD working

    cd QuartusProjects/DE0_NANO_SOC_GHRD
    make all
    
Scripts:    
    
[Generate sd card image](./scripts/Divided_scripts/readme.md)

~~[Install Quartus generated files to sd card](./Notes/install_Makefile-generated-files-to-sdcard.txt)~~

[Install Quartus generated files to sd card](./scripts/inst-rbf-dtb.sh)

~~[Install and Compile Machinekit](./scripts/mksoc-jessie-mk_rip_build-instal-v2.sh)~~

Just install the Machinekit postix Jessie armhf packages, (minus the kernel), and you are good to go...(once the mk_socfpga uio driver is finished and merged)

the-snowwhite, 2016-05-04:
If there are no protests I wil try adding this Milestone soon, and then try to attach it to all the related issue threads... ?

sirop, 2016-05-06:
@the-snowwhite 
Your last 4 links are corrupt.
I just unpacked the DE0 Nano board.
Is there a step by step guide for beginners?

the-snowwhite, 2016-05-06:
Sorry @sirop 
This is the wrong place.
The Milestone question is about gathering all the issue threads related to the mksocfpga project, making it easier to get an overview, of every related thread.

the-snowwhite, 2016-05-06:
Here you should find links to download an sd-image that will quick start you:

https://github.com/machinekit/machinekit/issues/915

the-snowwhite, 2016-05-06:
And the first link works as it is supposed to:
current development is here:   --> above

luminize, 2017-07-26:
I think this issue should be contained in https://github.com/machinekit/mksocfpga/issues
since there is no activity on this one I'll close this

closed 2017-07-26 by luminize