machinekit/machinekit #931: hm2_soc does not properly probe FPGA configuration

cdsteinkuehler, 2016-05-07:
According to discussions with Michael Brown in machinekit/mksocfpga#9, it appears the hm2_soc drivers are hard-coded as to what is in the FPGA and they do not yet properly probe the IDROM contents to enumerate the appropriate numbers of various functions (stepgen, pwm, etc) and the smart serial buses.

While this simplification helped get things working, the IDROM functionality needs to be restored for proper functionality of any Mesanet or similar daughter cards, particularly if there are multiple FPGA core flavors available for supporting the different daughter-cards (as is the case with virtually all existing Mesa FPGA cards).

mhaberler, 2016-05-08:
not sure I understand:

- there are a [few hardcoded parameters here](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L374-L379) which need work - not sure how these can be retrieved at runtime from the IDROM
- [the same for hm2_soc_ol.c](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c#L357-L362)
- the interpretation of the IDROM itself is completely stock by hm2_register() and friends, like allocation of the number of stepgens/pwmgens etc

it might make sense to mutate the hm2_soc_ol.c driver into an instantiable HAL component to support several "cards" - I can look into that in the evening when I'm back

cdsteinkuehler, 2016-05-08:
On 5/7/2016 10:20 PM, Michael Haberler wrote:
> not sure I understand:
> 
>   * there are a few hardcoded parameters here
>     <https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc.c#L374-L379>
>     which need work - not sure how these can be retrieved at runtime from the IDROM

Hmm...those values are not stored in the IDROM, but need to be pulled
from something that keeps the HW and SW in agreement.  That should
probably be the boardnamelow and boardnamehigh (32-bits each, or eight
ascii characters total).  I'm working on an updated config for the
DE0_Nano_SoC with 1 or 2 DB25 adapter boards that does *NOT* try to
mimic a 5i25.  Once I get an FPGA core compiling, we can work on
trying to get the driver to dynamically recognize the new board.

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-05-08:
This is the missing function that was leftist from the pic driver template.

And reduced to a static 5i25 config.

putting this back in should restore the full functionality in the hm2_socxxxx driver.

https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_pci.c#L460-L663 <https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_pci.c#L460-L663>

Michael Brown

cdsteinkuehler, 2016-05-08:
Note that there are no PCI device IDs to use to identify the board, so
the selection will need to be made using something else.  I suggest
using the boardname high/low elements from the IDROM table (two 32-bit
values, currently coded as 8 ascii bytes):

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/IDROMConst.vhd#L88-L110

I am currently planning on using the following for the DE0-Nano_DB25:

BoardNameLow  = x"41524554";     -- "TERA"
BoardNameHigh = x"4E304544";     -- "DE0N"

...but I am open to other suggestions

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-05-08:
>
> Note that there are no PCI device IDs to use to identify the board, so
> the selection will need to be made using something else. I suggest
> using the boardname high/low elements from the IDROM table (two 32-bit
> values, currently coded as 8 ascii bytes):

Agree.

https://github.com/machinekit/mksocfpga/blob/master/HW/hm2/config/IDROMConst.vhd#L88-L110
>
> I am currently planning on using the following for the DE0-Nano_DB25:
>
> BoardNameLow = x"41524554"; -- "TERA"
> BoardNameHigh = x"4E304544"; -- "DE0N"
>
> ...but I am open to other suggestions

Sounds good from here.

mhaberler, 2016-05-08:
one option is to store these values in the devicetree overlay and have the driver retrieve them

that could happen through reading /proc/device-tree/... or this library: https://github.com/jviki/dtree

mhaberler, 2016-05-08:
hm, looking into a bit deeper - the 'get info from DTB' approach suffers from another chicken-or-egg problem brought from the legacy code:

- the code assumes [basic properties could already be discovered at bus enumeration time](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c#L357-L362)
- if we retrieve those properties from the DTB they are not available at hm2_register() time but only once the programming callback is called with 'config="firmware=zynq_5i25_overlay.dtbo .."' or somesuch
- this is too late for [hm2_register](https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hostmot2.c#L1105-L1137)

not sure how to deal with that, short of reading the dtbo ahead of hm2_register()

dkhughes, 2016-05-08:
Right, that is the problem I ran into when working on the overlay low level driver. I wanted to dig deeper into the hostmot2 software and look where those constant values are used. The large switch case doesn't seem maintainable with all the directions FPGA bitfiles can go in.

@cdsteinkuehler I am not emulating a 5i25 with the example Vivado project in the mksoc repo. It's a custom hostmot2 pinfile and it worked well.

dkhughes, 2016-05-08:
Also, we might want to make the altera side behave like the Xilinx side. Instead of defining a new board constant in IDROM, let the project define its own board name. The other designs don't care about every other board's name. IDROMConst I think should be reserved for constants that are common to all designs (e.g. central to hm2 functionality).

cdsteinkuehler, 2016-05-08:
On 5/8/2016 3:38 PM, dkhughes wrote:
> Also, we might want to make the altera side behave like the Xilinx side. Instead
> of defining a new board constant in IDROM, let the project define its own board
> name. The other designs don't care [...]

I agree, and am already doing this.  Rather than add something like
BoardNameDE0Nano to IDROMConst.vhd, I just put a 32-bit constant in
the board's VHDL file.

The only reason I could see for putting all the board versions into
one file is if we're extracting values that also configure the
software driver.  I do this for some things at my day-job (C header
files are automatically extracted from VHDL code so any updates made
to the hardware automatically turn into appropriate structs and
records for software use, which results in having a single place to
update register documentation when the hardware changes), but we're a
long way from that with the hostmot2 code.

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-05-09:
With the 4.1-ltsi-rt  kernel, that i thin also works for rev_1.

The first step happens by loading the device-tree overlay  .dtbo file.

This then loads the bio driver.

And then copies the bit file into the fpga.

So any (custom)info we put into the (very small) device-tree overlay, is there form the very beginning.

So it is possible to have a .dts file —> dtbp for every board + daughterboard..

Maybe many files, but very small and linked to a specific configuration.

Very easy to make / edit and compile add.

So any info needed up front can be available by probing a custom device tree overlay fragment.

> On 08 May 2016, at 22:21, Michael Haberler <notifications@github.com> wrote:
> 
> the code assumes basic properties could already be discovered at bus enumeration time <https://github.com/machinekit/machinekit/blob/master/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c#L357-L362>
> if we retrieve those properties from the DTB they are not available at hm2_register() time but only once the

dkhughes, 2016-09-07:
With the new fwid code added into the vhdl and the driver layer, I think this issue has been nicely addressed. 

Any outstanding issues we should fix still?

cdsteinkuehler, 2016-09-07:
Yeah, this has been fixed for a while.  The issue was initially not related to the new firmware ID feature, but was due to the fact that the first hm2-soc driver just assumed it was talking to a "stock" 5i25 and didn't even read the existing IDROM.  AFAIK, this has been fixed for a while.

closed 2016-09-07 by cdsteinkuehler