machinekit/machinekit #937: Compile errors with Libwebsockets >1.6

halbritt, 2016-05-13:
Just attempted to compile master against Libwebsockets v1.7 (default on Ubuntu xenial) and received several errors related to functions that were no longer defined.  I presume this is as a result of the fact that functions were renamed from libwebsocket_* to lws_* as of v1.6.

mhaberler, 2016-05-14:
this is why there are specific libwebsockets packages in the machinekit repo

mhaberler, 2016-05-14:
a PR fixing this would be welcome!

halbritt, 2016-05-14:
It is my intention to submit a PR if I can sort out what's required.  Seems pretty obvious, however.

mhaberler, 2016-05-14:
that'd be great - happy to help

libwebsockets has been moving very fast recently, and the packages in debian are bound to stay behind

one option would be to package libwebsockets 2.0 and add that to our repos, but ideally the code should work with what is in the debian repos and with bleeding edge

mhaberler, 2016-05-27:
@halbritt in case you're using webtalk, I'd appreciate if you could give it a test 

we currently do not have a regression test for webalk

mhaberler, 2016-06-12:
fixed AFAICT, closing

closed 2016-06-12 by mhaberler