machinekit/machinekit #954: I'm use raspberry pi. Why cannot install any thing?

meoudon, 2016-05-28:
I'm use raspberry pi. Why cannot install any thing? 2 days ago.
sudo apt-get install machinekit-dev. -> miss
sudo apt-get install machinekit-rt-preempt. -> miss
sudo apt-get install linux-image-rpi2-rt. -> miss

luminize, 2016-05-28:
> On 28 May 2016, at 20:25, meoudon <notifications@github.com> wrote:
> 
> I'm use raspberry pi. Why cannot install any thing? 2 days ago.
> sudo apt-get install machinekit-dev. -> miss
> sudo apt-get install machinekit-rt-preempt. -> miss
> sudo apt-get install linux-image-rpi2-rt. -> miss
> 

Have you correctly configured APT ?

luminize, 2016-05-28:
Do us a favour and have a read thru this for your specific platform: 
http://www.machinekit.io/docs/index-getting-started/#platforms
And then read this:
http://www.machinekit.io/docs/getting-help/ we cannot guess

meoudon, 2016-05-28:
I solved now . 
echo "deb http://deb.machinekit.io/debian raspbian main" > /etc/apt/sources.list.d/machinekit.list
no have in your instruction.

luminize, 2016-05-28:
> I solved now . 
> echo "deb http://deb.machinekit.io/debian raspbian main" > /etc/apt/sources.list.d/machinekit.list
> no have in your instruction.
> 
Good!

Would you close this issue?

closed 2016-05-29 by meoudon