machinekit/machinekit #955: raspberrypi-bootloader package on deb.machinekit.io missing?

koppi, 2016-05-30:
The document at: http://www.machinekit.io/docs/getting-started/install-rt-kernel-RPi2/ states:

"
Updated raspberrypi-bootloader is needed:
```bash
sudo apt-get update
sudo apt-get install -y raspberrypi-bootloader
```
"

But I was not able to find the package ```raspberrypi-bootloader``` on deb.machinekit.io.

Current installed package status:
```bash
# apt-cache show raspberrypi-bootloader

Package: raspberrypi-bootloader
Source: raspberrypi-firmware
Version: 1.20160523-1
Architecture: armhf
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 11979
Depends: raspberrypi-kernel
Homepage: https://github.com/raspberrypi/firmware
Priority: extra
Section: admin
Filename: pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20160523-1_armhf.deb
Size: 3148468
SHA256: 0a60357bf922f15a03a506ef78fe8561c1df639849620b1d5ef61e2f8241f7f0
SHA1: f975012d3ecd6b8147d48f5e3a56e6c366b1e68f
MD5sum: 036587331240e95f23f4e296054368ac
Description: Raspberry Pi bootloader
 This package contains the Raspberry Pi firmware and bootloader.
Description-md5: f3fa59829d377b6fc81f91a4f5aa78da
```

Any ideas where the updated raspberrypi-bootloader package can be fetched from?

mhaberler, 2016-05-30:
@kinsamanka might know more

koppi, 2016-05-30:
Another RPi package related question maybe to @kinsamanka : the following executable files are in x86-86 elf format but they should be an arm executables:
```bash
file /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/*|grep x86
/usr/src/linux-headers-4.4.4-rt9-v7+/scripts/conmakehash:                       ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=6a32ecc7b1a8110f82bb78942608ea555ec2f9d7, not stripped
/usr/src/linux-headers-4.4.4-rt9-v7+/scripts/gcc-x86_32-has-stack-protector.sh: POSIX shell script, ASCII text executable
/usr/src/linux-headers-4.4.4-rt9-v7+/scripts/gcc-x86_64-has-stack-protector.sh: POSIX shell script, ASCII text executable
/usr/src/linux-headers-4.4.4-rt9-v7+/scripts/kallsyms:                          ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=517fea36cfe40942c2cff44433a2d3caef711657, not stripped
/usr/src/linux-headers-4.4.4-rt9-v7+/scripts/pnmtologo:                         ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=b28b214e4e3d7fb3117f45ca9df948a08841c81f, not stripped
/usr/src/linux-headers-4.4.4-rt9-v7+/scripts/recordmcount:                      ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=4f384e148390dede9b0f1d41ee727006a35bac10, not stripped
/usr/src/linux-headers-4.4.4-rt9-v7+/scripts/sortextable:                       ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=88573a87f1af44ea9959844a2810fec8c493fd71, not stripped
/usr/src/linux-headers-4.4.4-rt9-v7+/scripts/unifdef:                           ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=92bac921284282e6d107b85581e4f98e60b4808c, not stripped
```
Were these files created during the cross-compile process of the linux-headers-rpi2-rt package?

kinsamanka, 2016-05-30:
> On 30 May 2016, at 11:35 PM, Jakob Flierl <notifications@github.com> wrote:
> 
> The document at: http://www.machinekit.io/docs/getting-started/install-rt-kernel-RPi2/ states:
> 
> "
> Updated raspberrypi-bootloader is needed:
> 
> sudo apt-get update
> sudo apt-get install -y raspberrypi-bootloader
> 
> "
> 
> But I was not able to find the package raspberrypi-bootloader on deb.machinekit.io.
> 
> Current installed package status:
> 
> # apt-cache show raspberrypi-bootloader
> 
> 
> Package: raspberrypi-bootloader
> Source: raspberrypi-firmware
> Version: 1.20160523-1
> Architecture: armhf
> Maintainer: Serge Schneider 
> <serge@raspberrypi.org>
> 
> Installed-Size: 11979
> Depends: raspberrypi-kernel
> Homepage: https://github.com/raspberrypi/firmware
> Priority: extra
> Section: admin
> Filename: pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20160523-1_armhf.deb
> Size: 3148468
> SHA256: 0a60357bf922f15a03a506ef78fe8561c1df639849620b1d5ef61e2f8241f7f0
> SHA1: f975012d3ecd6b8147d48f5e3a56e6c366b1e68f
> MD5sum: 036587331240e95f23f4e296054368ac
> Description: Raspberry Pi bootloader
>  This package contains the Raspberry Pi firmware and bootloader.
> Description-md5: f3fa59829d377b6fc81f91a4f5aa78da
> 
> Any ideas where the updated raspberrypi-bootloader package can be fetched from?

It’s from the raspbian repo: https://www.raspbian.org/RaspbianRepository

> On 31 May 2016, at 5:20 AM, Jakob Flierl <notifications@github.com> wrote:
> 
> Another RPi package related question maybe to @kinsamanka <https://github.com/kinsamanka> : the following executable files are in x86-86 elf format but they should be an arm executables:
> 
> file /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/*|grep x86
> /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/conmakehash:                       ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=6a32ecc7b1a8110f82bb78942608ea555ec2f9d7, not stripped
> /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/gcc-x86_32-has-stack-protector.sh: POSIX shell script, ASCII text executable
> /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/gcc-x86_64-has-stack-protector.sh: POSIX shell script, ASCII text executable
> /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/kallsyms:                          ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=517fea36cfe40942c2cff44433a2d3caef711657, not stripped
> /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/pnmtologo:                         ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=b28b214e4e3d7fb3117f45ca9df948a08841c81f, not stripped
> /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/recordmcount:                      ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=4f384e148390dede9b0f1d41ee727006a35bac10, not stripped
> /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/sortextable:                       ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=88573a87f1af44ea9959844a2810fec8c493fd71, not stripped
> /usr/src/linux-headers-4.4.4-rt9-v7+/scripts/unifdef:                           ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=92bac921284282e6d107b85581e4f98e60b4808c, not stripped
> Were these files created during the cross-compile process of the linux-headers-rpi2-rt package?
> 
This is already fixed but it lives at http://deb.mah.priv.at :-) I’ll move it later.

koppi, 2016-05-31:
@kinsamanka ok, thank's for the clarification. For the second issue, I just created a pull request, see [https://github.com/kinsamanka/rpi_linux/pull/1](https://github.com/kinsamanka/rpi_linux/pull/1).

closed 2016-05-31 by kinsamanka

koppi, 2016-05-31:
@mhaberler the jenkins job [Project RPi rt-preempt kernel](https://jenkins.machinekit.io/job/RPi%20rt-preempt%20kernel/) did not get triggered when @kinsamanka merged my pull request. And it failed with an error message, when I triggered the job manually from the jenkins console: https://jenkins.machinekit.io/job/RPi%20rt-preempt%20kernel/12/console

mhaberler, 2016-05-31:
yes but we'll have the triggering fixed after the docker issues are resolved and it produces packages, not the other way round.

What is needed is that @kinsamanka adds me as collaborator on the repos the jenkins server should build merges or PR's.

mhaberler, 2016-06-01:
thanks for the help! great to have a reference kernel build which uses stock jenkins means

probably we should update https://github.com/kinsamanka/rpi_linux/blob/rpi-4.4.y_rt/.jenkins/jenkins_start.sh accordingly 

@kinsamanka re collaborator: ideally you do not add/invite my github persona but rather the @machinekit-ci user which I set up as a role account for exactly these things

In theory I get an invitation mail if you do and we can go from there

atm jenkins is still unhappy:
````
github.com:kinsamanka/rpi_linux	There is no credentials with admin access to manage hooks on GitHubRepositoryName[host=github.com,username=kinsamanka,repository=rpi_linux]
````

re bootloader issue: @koppi - can I ask you to exercise the "Edit this page" link ? we're totally proud on the new site and would love to see that used ;)