machinekit/machinekit #959: Conflicting definition of 'float' in HAL documentation

evanmj, 2016-06-07:
In the HAL intro / concepts section, it shows float is 64 bits, yet in the 'Basic HAL' section it is defined as having 32 bits.

http://www.machinekit.io/docs/hal/intro/#hal-concepts-a-id-sec-hal-concepts-a
http://www.machinekit.io/docs/hal/basic_hal/#float

I'd fix it but I'm new to machinekit and I'm not sure which is correct. :)

mhaberler, 2016-06-07:
that is definitely an error - HAL switched from single to double floats a long time ago

go ahead and fix it to 64bits/double, will merge it

evanmj, 2016-06-08:
PR Ready: https://github.com/machinekit/machinekit-docs/pull/163

Hopefully I did everything right.  I'll keep an eye out for any issues as I read the rest of the documentation.

Great job on the recent doc work and the project in general by the way.  I'm very happy to see this project progressing... I'm trying to get up to speed.

sirop, 2016-06-09:
This issue made me pay attention to these lines of https://github.com/sittner/linuxcnc-ethercat:
```
case HAL_FLOAT:
        // check data size
        if (hal_data->bitLength > 32) {
          rtapi_print_msg(RTAPI_MSG_WARN, LCEC_MSG_PFX "unable to export FLOAT pin %s.%s.%s.%s: invalid process data bitlen!\n", LCEC_MODULE_NAME, master->name, slave->name, hal_data->name);
          continue;
        }
....

```
https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_generic.c#L106-L111   

I could submit a PR to linuxcnc-ethercat if there were some kind of `#define MACHINEKIT`
either in `rtapi.h`or `hal.h` .

Where is `#define MACHINEKIT` to be put in?

mhaberler, 2016-06-10:
which source file is that supposed to be? https://github.com/sittner/linuxcnc-ethercat/search?utf8=%E2%9C%93&q=bitlength does not show this.

that said, distinguishing between linuxcnc and machinekit makes no sense since they are (still) the same wrt to HAL_FLOAT size

more context please

ArcEye, 2016-06-10:
>  https://github.com/sittner/linuxcnc-ethercat/search?utf8=%E2%9C%93&q=bitlength does not show this.

It is only showing the top 2 hits, which isn't much use

It is there
https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_generic.c#L106

mhaberler, 2016-06-10:
while I am not familiar with this codebase, it seems to me the per-type bitlength value is set from a configuration: https://github.com/sittner/linuxcnc-ethercat/blob/2d269c58a419f990ae7ba7acdf8946aa9c823c85/src/lcec_main.c

so I think it be more sensible to correct the configuration than patch around in the using code

that said: the test makes no sense to me to start with, neither for linuxcnc nor machinekit

sirop, 2016-06-10:
@mhaberler @ArcEye 

>that said, distinguishing between linuxcnc and machinekit makes no sense since they are (still) the same wrt to HAL_FLOAT size   

Ok, did not know this. Then I let it as it wrt linuxcnc
as linuxcnc circumvents this restriction by using _scale_ and _offset_ for floats.

mhaberler, 2016-06-10:
http://git.linuxcnc.org/gitweb?p=linuxcnc.git;a=blob;f=src/hal/hal.h;h=34a9004aeec55c6c4289d49d5be8913e03b7cc01;hb=f99bdad7625eac603a96e0ee2002d9ee4bc09a07#l300

https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal.h#L492

@sirop 'as linuxcnc circumvents this restriction by using scale and offset for floats.' 

which restriction? what exactly are you referring to? links?

sirop, 2016-06-10:
@mhaberler 
I just get confused about linuxcnc unwilling to work with 64 bits floats
but as you can see for instance in https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_generic.c#L162-L172
one can use `hal_data->floatScale`  and `hal_data->floatOffset` to go beyond a 32 bit float.

Thus the issue I have just raised is not an issue at all.

sirop, 2016-06-10:
@mhaberler 
Sorry, I meant __linuxcnc-ethercat__ and not linuxcnc.

mhaberler, 2016-06-10:
I see a singed/unsigned promoted to float plus a scale/offset operation there plus the inverse in .._generic_write()

I do not see how this has any significance for float size, significant bits etc?

sirop, 2016-06-10:
@mhaberler 

Yes, one can live with that. I just got confused with
https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_generic.c#L106-L111
as `hal_data->bitLength > 32` restricts the float data type to 32 bits, although both MK and EtherCAT
support 64 bit floats.

But with a scale/offset operation it is not an issue then, you are right.

mhaberler, 2016-06-11:
I'd suggest to just comment out that test and see if it breaks things (unlikely)

sirop, 2016-06-11:
Commenting out this test will not break things at that very spot as https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_generic.c#L106-L111 is part of `lcec_conf` - that is a non RT XML parser.  

But it will break things in the RT component of linuxcnc-ethercat as only 32 bit read/write EC_READ_S32/EC_READ_U32  and EC_WRITE_S32/EC_WRITE_U32 are used there. See, for instance, https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_generic.c#L252 

EtherLab offers, of course, 64 bit read/write.

mhaberler, 2016-06-11:
"But it will break things in the RT component of linuxcnc-ethercat as only 32 bit read/write "

I do not understand what "breaks" - the underlying code can do 32bit integer I/O only, fine

hal_floats are scaled/offsetted then casted to/from ints/unsigneds - fine too

where is the problem you are trying to solve?

sirop, 2016-06-11:
My comment https://github.com/machinekit/machinekit/issues/959#issuecomment-225342350 referred solely to the suggestion to  comment out `hal_data->bitLength > 32` test.  

With this test still at its place there is indeed no problem to solve.

mhaberler, 2016-06-12:
original issue resolved, closing

closed 2016-06-12 by mhaberler