machinekit/machinekit #968: client call MT_HALRCOMMAND_DESCRIBE to haltalk breaks haltalk operation

zhivko, 2016-06-13:
I expect I could use ZMQ and MT_HALRCOMMAND_DESCRIBE to retrieve hal status (pin values, and other hal related values), but weirdly it seems that MT_HALRCOMMAND_DESCRIBE breaks haltalk operation.

It breaks it in a way that also MT_PING from client doesn't ẁork anymore ( MT_PING_ACKNOWLEDGE is not received back anymore).

My scenario is as follows:

1. I have ZMQ client with dealer socket.
2. I issue MT_PING request to haltalk and I get MT_PING_ACKNOWLEDGE back to client.
3. In same code I only change line:
builder.setType(ContainerType.MT_PING);
to
builder.setType(ContainerType.MT_HALRCOMMAND_DESCRIBE);
4. Start client again and make MT_HALRCOMMAND_DESCRIBE call to haltalk - from that point on - I don't get any reply back.
5. In same code I change type of container again:
builder.setType(ContainerType.MT_HALRCOMMAND_DESCRIBE);
to
builder.setType(ContainerType.MT_PING);
4. I dont get MT_PING_ACKNOWLEDGE back anymore. I can also see that enumerating services on BBB where haltalk is running doesn't show this 3 services anymore (in start of test they were up and running):
HAL Rcommand service on beaglebone tcp://beaglebone.local.:49155/
HAL Rcomp service on beaglebone tcp://beaglebone.local.:49154/
HAL Group service on beaglebone tcp://beaglebone.local.:49153/

What could be the reason?

my dpkg show this:
```
machinekit@beaglebone:~$ dpkg --list | grep machinekit
ii  machinekit                              0.1.1450479237-1mk.travis.master.git0b8ec05d~1wheezy armhf        PC based motion controller for real-time Linux
ii  machinekit-build-deps                   1.0                                                  all          build-dependencies for machinekit
ii  machinekit-xenomai                      0.1.1450479237-1mk.travis.master.git0b8ec05d~1wheezy armhf        PC based motion controller for real-time Linux
```

mhaberler, 2016-06-13:
run haltalk with the --debug <level> option
observe /var/log/linuxcnc.log

post example code so one can reproduce - the above report is impossible to reproduce

zhivko, 2016-06-14:
Ok will do that - maybe I overcomplicated scenario just to say that PING is working OK.
I was not able to get result from MT_HALRCOMMAND_DESCRIBE anyhow. Will post output from machinekit.

zhivko, 2016-06-14:
machinekit log:
http://pastebin.com/svUppSBK

Since my client is in JAVA I doubt it will help anybody, but actualy there are two calls - first is MT_PING another is MT_HALRCOMMAND_DESCRIBE.

luminize, 2016-06-14:
> On 14 Jun 2016, at 08:49, zhivko <notifications@github.com> wrote:
> 
> machinekit log:
> http://pastebin.com/svUppSBK <http://pastebin.com/svUppSBK>
> Since my client is in JAVA I doubt it will help anybody, but actualy there are two calls - first is MT_PING another is MT_HALRCOMMAND_DESCRIBE.
> 
line 42 in your pastebin is where you see the error.
it says a core is dumped, so you can do a backtrace and see where the error happens.
http://www.machinekit.io/docs/code/Debugging-RT-components/

zhivko, 2016-06-14:
I checked contents of folder it is mentioned - but no dump file there. SO where is it - what is the name of this file?
I guess I can examine it with gdb then?

strahlex, 2016-06-14:
Usually there should be a `core` file in the directory where you started the process.

zhivko, 2016-06-14:
No, cannot find file named core there either.
Tried to find core dump with this:
```
machinekit@beaglebone:/$ pwd
/
machinekit@beaglebone:/$ sudo find /. -name core
/./usr/include/boost/geometry/core
/./usr/include/boost/geometry/multi/core
/./usr/include/boost/spirit/core
/./usr/include/boost/spirit/home/classic/core
/./usr/include/boost/spirit/home/phoenix/core
/./usr/include/boost/phoenix/core
/./usr/include/boost/xpressive/detail/core
/./usr/include/directfb-internal/core
/./usr/include/opencv2/core
/./usr/share/pyshared/numpy/core
/./usr/lib/pymodules/python2.7/numpy/core
/./usr/lib/pymodules/python2.6/numpy/core
/./usr/lib/pyshared/python2.7/numpy/core
/./usr/lib/pyshared/python2.6/numpy/core
/./usr/local/lib/node_modules/bower/lib/core
/./proc/sys/net/core
/./lib/modules/3.8.13-xenomai-r78/kernel/net/core
/./lib/modules/3.8.13-xenomai-r78/kernel/sound/core
/./home/machinekit/git/node-machinetalk-example/node_modules/bower/lib/core
/./dev/core
```

mhaberler, 2016-06-14:
Is this code just too secret to post? I do not get it, and I hate this silly guessing game.

---> POST YOUR CODE <----

zhivko, 2016-06-14:
No no - of course not... Here it is:
https://github.com/zhivko/PipeCutter/blob/master/src/main/java/com/kz/pipeCutter/BBB/BBBHalGroup.java#L34-L35

I will try to follow https://sigquit.wordpress.com/2009/03/13/the-core-pattern/ to get core dump of haltalk.

mhaberler, 2016-06-14:
@luminize @zhivko since it's a haltalk crash the RT comp debugging instructions do not apply - haltalk is a normal user process

zhivko, 2016-06-14:
and here it is:

```
Loaded symbols for /lib/arm-linux-gnueabihf/libnss_files.so.2
0xb69c8f94 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) continue
Continuing.

Program received signal SIGSEGV, Segmentation fault.
0xb6df8bba in halpr_describe_funct(hal_funct*, pb::Function*) ()
   from /usr/lib/liblinuxcnchal.so.0

```

mhaberler, 2016-06-14:
type 'backtrace' and post the output

http://www.gnu.org/software/gdb/documentation/
http://www.unknownroad.com/rtfm/gdbtut/

zhivko, 2016-06-14:
(gdb) backtrace
#0  0xb6df8bba in halpr_describe_funct(hal_funct*, pb::Function*) ()
   from /usr/lib/liblinuxcnchal.so.0
#1  0x00017bda in ?? ()
#2  0x00017bda in ?? ()

mhaberler, 2016-06-14:
well, the packages have no debug symbols, you need to build from source or provide client code which I can reproduce - I cannot reproduce anything java

can't you get it to work with https://github.com/strahlex/pymachinetalk/blob/master/pymachinetalk ?

zhivko, 2016-06-14:
No I havent tried this (yet).
How should I "build from source" to include debug symbols? I tried to find that information in:
http://www.machinekit.io/docs/code/Debugging-RT-components/
but maybe this is not right place?

I started machinekit from BBB image (described by @cdsteinkuehler in some of his blogs).

I wonder which of two approaches (build from source with debug symbols or use of https://github.com/strahlex/pymachinetalk/blob/master/pymachinetalk) will faster yield results...

Probably I could try to use https://github.com/strahlex/pymachinetalk/blob/master/pymachinetalk but what is actually client program here - what file of:
__init__.py
application.py
common.py
dns_sd.py
halremote.py
I need to change to make this scenarion. Actually It should be very simple program like my java code shows.

zhivko, 2016-06-14:
AH OK - I will try to add my scenarion - reading python setup.py --help of this project...

zhivko, 2016-06-14:
Feeling pretty lame here :) - how do I start:
machinekit@beaglebone:~/git/pymachinetalk$ python ./examples/basic/anddemo.py 
Traceback (most recent call last):
  File "./examples/basic/anddemo.py", line 8, in <module>
    from pymachinetalk.dns_sd import ServiceDiscovery
ImportError: No module named pymachinetalk.dns_sd
machinekit@beaglebone:~/git/pymachinetalk$

mhaberler, 2016-06-14:
The process for building from source is described on the website

Remove the machinekit package, the build from source as per website.

For pymachinetalk you need to talk to @strahlex, I do not know if this is current.

zhivko, 2016-06-14:
Here is backtrace with symbols included:
```
Program received signal SIGSEGV, Segmentation fault.
halpr_describe_funct (funct=0xb6753ee8, pbfunct=0xe5d00) at hal/lib/halpb.cc:96
96	    pbfunct->set_runtime(*(funct->runtime));
(gdb) backtrace
#0  halpr_describe_funct (funct=0xb6753ee8, pbfunct=0xe5d00)
    at hal/lib/halpb.cc:96
#1  0x0001005e in describe_funct_cb (funct=0xb6753ee8, arg=0xbecb25e8)
    at machinetalk/haltalk/haltalk_introspect.cc:124
#2  0xb6d8046c in halpr_foreach_funct (name=0x0, 
    callback=0x10019 <describe_funct_cb(hal_funct_t*, void*)>, 
    cb_data=0xbecb25e8) at hal/lib/hal_iter.c:155
#3  0x0001017a in process_describe (self=0xbecb25e8, from=0x36e68, 
    socket=0x32f80) at machinetalk/haltalk/haltalk_introspect.cc:44
#4  0x0000f22c in dispatch_request (socket=0x32f80, from=0x36e68, 
    self=0xbecb25e8) at machinetalk/haltalk/haltalk_command.cc:481
#5  handle_command_input (loop=<optimized out>, poller=<optimized out>, 
    arg=0xbecb25e8) at machinetalk/haltalk/haltalk_command.cc:60
#6  0xb6be7e50 in zloop_start () from /usr/lib/arm-linux-gnueabihf/libczmq.so.1
#7  0x0000b968 in mainloop (self=0xbecb25e8)
    at machinetalk/haltalk/haltalk_main.cc:113
#8  main (argc=3, argv=<optimized out>)
    at machinetalk/haltalk/haltalk_main.cc:467
```

mhaberler, 2016-06-15:
hm, that crash should not happen, but it has no obvious relation the message received

I am still trying to get the pymachinetalk client to run, but  need help from @strahlex for that

- can you finally share which host-side configuration you are running?
- can you run haltalk with --debug 1 (it should print protobuf messages on stderr then)
- can you trace the network exchange with wireshark?

zhivko, 2016-06-15:
Hal and ini files are here:
https://github.com/zhivko/PipeCutter/blob/master/CRAMPS.hal
https://github.com/zhivko/PipeCutter/blob/master/CRAMPS.ini

Original ini and hal is taken from:
https://github.com/machinekit/machinekit/blob/master/configs/ARM/BeagleBone/CRAMPS/CRAMPS.ini
https://github.com/machinekit/machinekit/blob/master/configs/ARM/BeagleBone/CRAMPS/CRAMPS.hal

I got original files initialy by configuration selector Here is patch of ini and hal file:
[patchHal.txt](https://github.com/machinekit/machinekit/files/315915/patchHal.txt)
[patchIni.txt](https://github.com/machinekit/machinekit/files/315914/patchIni.txt)

I run haltalk with following command:
```
loadusr -W haltalk -d 5
```
and on stdout I get:
```
....
[017] 56E58679-B43F81A0
[006] 089402B80301
handle_command_input: req=type: MT_HALRCOMMAND_DESCRIBE
ticket: 1
...
```

mhaberler, 2016-06-15:
it seems to me this is caused by the client socket identity, which looks like a default binary (non-string)

now haltalk should not fail on this and I'll fix it once isolated, but try this:

set the socket identity on the client side to some readable string

mhaberler, 2016-06-15:
if this fixes the issue, can I ask you to try this branch: https://github.com/mhaberler/machinekit/tree/deprecate-send-container-string-dest and revert the client to the default identity as before, and check if it passes?

zhivko, 2016-06-15:
Changing identity to "MyIdentity" chrashes haltalk also.
TEsting your new branch get this:
```
(gdb) backtrace
#0  halpr_describe_funct (funct=0xb675eee8, pbfunct=0xe5d00)
    at hal/lib/halpb.cc:96
#1  0x0000fe26 in describe_funct_cb (funct=0xb675eee8, arg=0xbed505e8)
    at machinetalk/haltalk/haltalk_introspect.cc:129
#2  0xb6d8b46c in halpr_foreach_funct (name=0x0, 
    callback=0xfde1 <describe_funct_cb(hal_funct_t*, void*)>, 
    cb_data=0xbed505e8) at hal/lib/hal_iter.c:155
#3  0x0000ff42 in process_describe (self=0xbed505e8, from=0x3b018, 
    socket=0x32f80) at machinetalk/haltalk/haltalk_introspect.cc:44
#4  0x0000eff4 in dispatch_request (socket=0x32f80, from=0x3b018, 
    self=0xbed505e8) at machinetalk/haltalk/haltalk_command.cc:481
#5  handle_command_input (loop=<optimized out>, poller=<optimized out>, 
    arg=0xbed505e8) at machinetalk/haltalk/haltalk_command.cc:60
#6  0xb6bf2e50 in zloop_start () from /usr/lib/arm-linux-gnueabihf/libczmq.so.1
#7  0x0000b980 in mainloop (self=0xbed505e8)
    at machinetalk/haltalk/haltalk_main.cc:113
#8  main (argc=3, argv=<optimized out>)
    at machinetalk/haltalk/haltalk_main.cc:467
```

zhivko, 2016-06-15:
Here is linuxcnc.log with DEBUG=5

http://pastebin.com/N0Yd7TSS

zhivko, 2016-06-15:
Here is original CRAMPS with linuxcnc debug=5.
Same problem.

[linuxcnc.log.txt](https://github.com/machinekit/machinekit/files/316732/linuxcnc.log.txt)
[CRAMPS.hal.txt](https://github.com/machinekit/machinekit/files/316734/CRAMPS.hal.txt)
[CRAMPS.ini.txt](https://github.com/machinekit/machinekit/files/316736/CRAMPS.ini.txt)

zhivko, 2016-06-15:
You can build standalone java client to simulate this problem with
1. cloning repository at: https://github.com/zhivko/PipeCutter
2. build self contained jar with running this command in cloned repo:
mvn package
3. start MT_HALRCOMMAND_DESCRIBE client with:
java -jar ./package/test_MT_HALRCOMMAND_DESCRIBE-jar-with-dependencies.jar

This client will discover machinekit server (particulary HAL Rcommand service), and send MT_HALRCOMMAND_DESCRIBE to it.

mhaberler, 2016-06-15:
this was a real bug - nothing to do with you, sorry
fixed
please try https://github.com/mhaberler/machinekit/commits/deprecate-send-container-string-dest

zhivko, 2016-06-15:
Thank you for fast resolution.
Bug is fixed I can retrieve hal values back - great. Thanks again.

closed 2016-06-15 by zhivko

strahlex, 2016-06-17:
Here is another test to trigger the problem: https://github.com/strahlex/machinetalk-test