machinekit/machinekit #971: Time Sensitive Networking is coming

koppi, 2016-06-22:
On 2016-06-12 Henrik Austad of Cisco [has published](https://lkml.org/lkml/2016/6/11/213) very early code for implementing a TSN core driver in the Linux kernel.

TSN is short for [Time Sensitive Networking](https://en.wikipedia.org/wiki/Time-Sensitive_Networking). TSN will deprecate all the proprietary deterministic networking vendor standards, notably Profinet IRT, EtherCAT und Powerlink. – [Here](https://standards.ieee.org/events/automotive/08_Teener_TSN.pdf)'s a quick overview of TSN by Michael D. Johas Teener.

And a case study for deterministic OPC UA with Publisher/Subscriber and TSN Ethernet:  https://opcfoundation.org/opc-connect/2016/06/toolkit-case-study-and-pubsub-white-paper

sirop, 2016-06-28:
I do not quite understand the difference between TSN and https://en.wikipedia.org/wiki/Audio_Video_Bridging .

If they have the same goal and application, then there is more than _early code for implementing a TSN core driver_ out there.

For instance:
https://github.com/AVnu/Open-AVB
https://github.com/xcore/sw_avb

koppi, 2016-06-28:
Multimedia streaming over deterministic networks is probably only a subset of applications that is possible with the AVB technology, so the AVB working group renamed their project to TSN to reflect the expanded scope of this technology.

From my understanding, the networking components like sender, receiver, buffers, scheduler can be modeled using the [Network calculus](https://en.wikipedia.org/wiki/Network_calculus), which is helpful for calculating the deterministic (worst case) bounds of the networking system.

koppi, 2016-06-28:
**Update**: here's an example of such a calculator based on the network calculus: http://www.realtimeatwork.com/software/rtaw-pegase/

sirop, 2016-06-30:
I think there is already something in https://github.com/AVnu/Open-AVB if you look at 
https://standards.ieee.org/events/automotive/2015/09_AVnu_Open_Source_Software_for_AVB-TSN.pdf . They say in Sep 2015 that Intel i210 via igb  is already supported, https://github.com/AVnu/Open-AVB/tree/master/lib/igb .

_BTW, as you asked once about Arduino EtherCAT slave: http://www.bausano.net/en/hardware/ethercat-e-arduino/easycat.html_

luminize, 2017-05-18:
@koppi is this an issue? or an anchoring thread for specific development?
Can you rename is or close it so we know the purpose/issue?

closed 2017-05-19 by koppi