machinekit/machinekit #973: haltalk does not stop after hal exit

sirop, 2016-06-28:
How to reproduce:
```
$ halrun
halcmd: loadusr -W haltalk --debug=5
halcmd: exit

$ ps ax | grep hal
 5766 pts/2    Rl     0:10 haltalk --debug=5
 5811 pts/2    S+     0:00 grep hal
```

my machinekit.ini -> https://gist.github.com/sirop/ee69d63974af95f90173fe39150802de

mhaberler, 2016-06-28:
That is correct as the halcmd 'exit' command just terminates halcmd, but not the running realtime stack, and therefore does not unload any components.

that is the same behavior as with loadrt - give it a try:

`````
halcmd: loadrt or2
halcmd: show comp
halcmd: exit
$ halcmd -f -k
halcmd: show comp
halcmd: show comp
Loaded HAL Components:
    ID  Type Flags Inst Name                                      PID   State
    87  User          0 halcmd31460                               31460 ready, u1:0 u2:0
    86  uHAL   i      0 hal_lib31460                              31460 ready, u1:0 u2:0
    77  RT   c        1 or2                                       RT    ready, u1:0 u2:0             <-----
    74  User          0 haltalk                                   31428 ready, u1:0 u2:0
    73  uHAL   i      0 hal_lib31428                              31428 ready, u1:0 u2:0
    66  rHAL   i      0 hal_lib                                   0     ready, u1:0 u2:0
`````
to unload haltalk (like any usercomp) use `halcmd unloadusr haltalk`

sirop, 2016-06-28:
```
slave@debian-slave:~/ecslave_comp$ realtime start
slave@debian-slave:~/ecslave_comp$ realtime status
msgd:0 running
rtapi:0 running
slave@debian-slave:~/ecslave_comp$ halcmd  loadusr -W haltalk --debug=5
<commandline>:0: Component 'haltalk' ready
<commandline>:0: Program 'haltalk' started
slave@debian-slave:~/ecslave_comp$ halcmd unloadusr haltalk
slave@debian-slave:~/ecslave_comp$ halcmd show comp
Loaded HAL Components:
    ID  Type Flags Inst Name                                      PID   State
    78  User          0 halcmd6491                                6491  ready, u1:0 u2:0
    77  uHAL   i      0 hal_lib6491                               6491  ready, u1:0 u2:0
    74  User          0 haltalk                                   6482  ready, u1:0 u2:0
    73  uHAL   i      0 hal_lib6482                               6482  ready, u1:0 u2:0
    66  rHAL   i      0 hal_lib                                   0     ready, u1:0 u2:0

slave@debian-slave:~/ecslave_comp$ ps ax | grep hal
 6482 pts/2    Rl     0:11 haltalk --debug=5
 6495 pts/2    S+     0:00 grep hal
```
What did I miss?

mhaberler, 2016-06-28:
I'm not sure, right now I cannot reproduce, but maybe look at linuxcnc.log:

`````
mah@jessie64:~/machinekit-patch/src$ realtime status 
msgd:0 running
rtapi:0 running
mah@jessie64:~/machinekit-patch/src$ halcmd  loadusr -W haltalk --debug=5
<commandline>:0: Component 'haltalk' ready
<commandline>:0: Program 'haltalk' started
mah@jessie64:~/machinekit-patch/src$ pidof haltalk
32185
mah@jessie64:~/machinekit-patch/src$ halcmd show comp
Loaded HAL Components:
    ID  Type Flags Inst Name                                      PID   State
    76  User          0 halcmd32190                               32190 ready, u1:0 u2:0
    75  uHAL   i      0 hal_lib32190                              32190 ready, u1:0 u2:0
    74  User          0 haltalk                                   32185 ready, u1:0 u2:0
    73  uHAL   i      0 hal_lib32185                              32185 ready, u1:0 u2:0
    66  rHAL   i      0 hal_lib                                   0     ready, u1:0 u2:0

mah@jessie64:~/machinekit-patch/src$ halcmd unloadusr haltalk
mah@jessie64:~/machinekit-patch/src$ halcmd show comp
Loaded HAL Components:
    ID  Type Flags Inst Name                                      PID   State
    80  User          0 halcmd32197                               32197 ready, u1:0 u2:0
    79  uHAL   i      0 hal_lib32197                              32197 ready, u1:0 u2:0
    66  rHAL   i      0 hal_lib                                   0     ready, u1:0 u2:0
`````
the key log entries should be:

`````
Jun 28 12:41:27 jessie64 msgd:0: hal_lib:32194:user hal_exit:258 HAL: removing component 78
Jun 28 12:41:27 jessie64 msgd:0: hal_lib:32185:user haltalk: signal 15 - 'Terminated' received
`````

I have observed the behavior you reported on and off, but the causality is unclear to me
one would have to run haltalk under the debugger and see which signal gets posted if halcmd does an unloadusr

see the line here: https://github.com/machinekit/machinekit/blob/master/src/hal/utils/halcmd_commands.c#L1449

could you try? I'd love to see this resolved

before starting haltalk, export NOSIGHDLR=1 so haltalk, rtapi_app, msgd do not install the default signal handlers which interferes with gdb

sirop, 2016-06-28:
Using gdb I land at
```
Loaded symbols for /lib/x86_64-linux-gnu/libnss_files.so.2
0x00007f1f0d20fd3d in poll () at ../sysdeps/unix/syscall-template.S:81
81      ../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb)
```
where I have to type in `continue` as otherwise I get
```
$ halcmd loadusr -W haltalk --debug=5
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout
```
in the other terminal. From that moment gdb delivers no output ...

After doing `halcmd unloadusr haltalk` I see only these lines in linuxcnc.log which - the lines - might be somehow interesting:
```
Jun 28 04:52:36 debian-slave msgd:0: hal_lib:8411:user hal_exit:258 HAL: removing component 98
Jun 28 04:52:36 debian-slave msgd:0: hal_lib:8411:user hal_exit:258 HAL: removing component 97
Jun 28 04:52:36 debian-slave msgd:0: hal_lib:8411:user ULAPI:0 unknown exit
``` 

Maybe, `strace`could show more?
```
$ strace -v halcmd unloadusr haltalk
 ....
munmap(0x7f42098f4000, 4096)            = 0
kill(8792, SIGTERM)                     = 0
munmap(0x7f420979c000, 524288)          = 0
munmap(0x7f420476c000, 2132944)         = 0
exit_group(0)                           = ?
+++ exited with 0 +++

$ halcmd show comp
Loaded HAL Components:
    ID  Type Flags Inst Name                                      PID   State
    80  User          0 halcmd8807                                8807  ready, u1:0 u2:0
    79  uHAL   i      0 hal_lib8807                               8807  ready, u1:0 u2:0
    74  User          0 haltalk                                   8792  ready, u1:0 u2:0
    73  uHAL   i      0 hal_lib8792                               8792  ready, u1:0 u2:0
    66  rHAL   i      0 hal_lib                                   0     ready, u1:0 u2:0

$ ps ax | grep hal
 8792 pts/2    Rl     0:31 haltalk --debug=5
 8811 pts/2    S+     0:00 grep hal
```
So although `kill(8792, SIGTERM) = 0` succeeded, `ps ax` and `halcmd show comp` still find `haltalk`.

mhaberler, 2016-06-28:
what platform + kernel is this?

could you try this:

- unset NOSIGHDLR
- just start realtime and haltalk as you did above
- in a separate window, run  ` gdb -p <pid of haltalk>`
- `break sigaction_handler` and `continue`
- from a separate shell: `kill -TERM <pid of haltalk>`

does gdb stop in the signal handler?

sirop, 2016-06-28:
It is:
```
~$ uname -a
Linux debian-slave 4.1.19-rt22mah #2 SMP PREEMPT RT Tue Mar 15 09:39:28 CET 2016 x86_64 GNU/Linux
```

If I follow your advice, `haltalk` gets 'killed'.  
gdb output:
```
~$ pidof haltalk
1359
~$ sudo gdb -p 1359
....
Loaded symbols for /lib/x86_64-linux-gnu/libnss_files.so.2
0x00007f4247e4ad3d in poll () at ../sysdeps/unix/syscall-template.S:81
81      ../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) break sigaction_handler
Breakpoint 1 at 0x40afc0: file machinetalk/haltalk/haltalk_main.cc, line 131.
(gdb) c
Continuing.
[Thread 0x7f4246a43700 (LWP 1360) exited]
[Thread 0x7f4246242700 (LWP 1361) exited]
[Inferior 1 (process 1359) exited normally]
(gdb)
```

mhaberler, 2016-06-28:
if just run haltalk and send it a SIGTERM, will it exit? it is a remote chance gdb is interfering
if it exits, there should be a corresponding log entry

sirop, 2016-06-28:
```
$ realtime status
msgd:0 running
rtapi:0 running
~$ halcmd loadusr -W haltalk --debug=5
<commandline>:0: Component 'haltalk' ready
<commandline>:0: Program 'haltalk' started
~$ pidof haltalk
1491
~$ kill -TERM 1491
~$ pidof haltalk
~$ tail -10 /var/log/linuxcnc.log
Jun 28 07:59:45 debian-slave haltalk: zeroconf: registered 'HAL Rcommand service on debian-slave.local pid 1491' _machinekit._tcp 49155 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=f2e25326-3d40-11e6-9b5a-c4e9841daedd" "service=halrcmd" "dsn=tcp://debian-slave.local:49155"
Jun 28 08:00:30 debian-slave haltalk: zeroconf: unregistering 'HAL Group service on debian-slave.local pid 1491'
Jun 28 08:00:30 debian-slave haltalk: zeroconf: unregistering 'HAL Rcomp service on debian-slave.local pid 1491'
Jun 28 08:00:30 debian-slave haltalk: zeroconf: unregistering 'HAL Rcommand service on debian-slave.local pid 1491'
Jun 28 08:00:30 debian-slave msgd:0: hal_lib:1491:user haltalk: signal 15 - 'Terminated' received
Jun 28 08:00:31 debian-slave msgd:0: hal_lib:1491:user haltalk: exiting mainloop (interrupted)
Jun 28 08:00:31 debian-slave msgd:0: hal_lib:1491:user hal_exit:258 HAL: removing component 82
Jun 28 08:00:31 debian-slave msgd:0: hal_lib:1491:user hal_exit:258 HAL: removing component 81
Jun 28 08:00:31 debian-slave msgd:0: hal_lib:1491:user ULAPI:0 unknown exit
```
Yes, the log entry ` hal_lib:1491:user haltalk: signal 15 - 'Terminated' received` is there.

mhaberler, 2016-06-28:
that is rather bizarre

if you send the signal via `halcmd unloadusr haltalk` that does not happen as you already showed above I guess?

I think that halcmd source line warrants more scrutiny, and probably a test with signal=0 if the signal actually can be posted - frankly I do not see what is different from `kill -TERM <pid>`

would you mind strace'ing both halcmd and kill if we can see any difference?

sirop, 2016-06-28:
Meanwhile I think that this has a simple explanation.

I debugged an RT component a day ago setting `export NOSIGHDLR=1` which I unset only now following your https://github.com/machinekit/machinekit/issues/973#issuecomment-229042353 .

Since then I can  reproduce this "bug"  neither in the old terminal ( with NOSIGHDLR unset) nor in a new terminal.

So if this explanation is adequate, shall I close the issue?

mhaberler, 2016-06-28:
oh ok, that all makes sense ;)

closed 2016-06-28 by mhaberler