machinekit/machinekit #974: during `./configure` the `boost::python` shared library is not found

luminize, 2016-06-28:
I run into a ./configure problem where boost::python shared library is not found.

Excercise: dev setup on a BBB with ROS  [as described here](https://github.com/machinekit/machinekit/issues/689#issuecomment-189286263)

steps to reproduce:
* I have taken the latest RCN Jessie Machhinekit image [from here](http://www.machinekit.io/docs/getting-started/machinekit-images/)
* install ROS as described here, the last command executed in this comment was `sudo apt-get install ros-desktop-full-dev libros-filters-plugins-dev` because we don't need more packages
```
mkdir -p ~/ros/src
cd ~/ros/src
git clone https://github.com/luminize/listener2.git
cd ..
catkin_make_isolated --install
echo "source ~/ros/install_isolated/setup.bash" >> ~/.bashrc
source ~/ros/install_isolated/setup.bash
```
* Set up the [APT repo for jessie on Beaglebone](http://www.machinekit.io/docs/getting-started/APT-packages-jessie/)
* follow [instructions](http://www.machinekit.io/docs/developing/machinekit-developing/) where i run `./configure --with-platform-beaglebone`

where I get the message:
```
checking for boost::python shared library... 
configure: error: boost::python is required to build LinuxCNC
```

further info:
```
machinekit@mk-ros-cramps:~/machinekit/src$ uname -a
Linux mk-ros-cramps 3.8.13-xenomai-r79 #1 Wed Jun 15 22:12:09 UTC 2016 armv7l GNU/Linux
```
```
machinekit@mk-ros-cramps:~/machinekit/src$ dpkg -l | grep boost | grep python
ii  libboost-mpi-python-dev                          1.55.0.2                                          armhf        C++ interface to the Message Passing Interface (MPI), Python Bindings (default version)
ii  libboost-mpi-python1.55-dev                      1.55.0+dfsg-3                                     armhf        C++ interface to the Message Passing Interface (MPI), Python Bindings
ii  libboost-mpi-python1.55.0                        1.55.0+dfsg-3                                     armhf        C++ interface to the Message Passing Interface (MPI), Python Bindings
ii  libboost-python-dev                              1.55.0.2                                          armhf        Boost.Python Library development files (default version)
ii  libboost-python1.55-dev                          1.55.0+dfsg-3                                     armhf        Boost.Python Library development files
ii  libboost-python1.55.0                            1.55.0+dfsg-3                                     armhf        Boost.Python Library
ii  python-rosboost-cfg                              1.12.5-3~drp8+20160222                            all          Contains scripts used by the Robot OS rosboost-cfg tool
```

mhaberler, 2016-06-28:
unfortunately configure tests for boost are fickle as boost has no pkg-config descriptor

the way to investigate this is to look at configure.ac and the corresponding log config.log
the error message comes from [here](https://github.com/machinekit/machinekit/blob/master/src/configure.ac#L94) so BOOST_PYTHON_LIBS must be ""

the related config.log I have is:

``````
configure:12474: checking whether the Boost::Python headers are available
configure:12494: g++ -c -g -O2 -std=c++11 -I/usr/include/python2.7  conftest.cpp >&5
configure:12494: $? = 0
configure:12501: result: yes
configure:12517: checking for boost::python shared library
configure:12534: g++ -o conftest -g -O2 -std=c++11 -I/usr/include/python2.7   conftest.cpp  -lXinerama  -lpython2.7 -lboost_python >&5
configure:12534: $? = 0
configure:12545: result: -lboost_python
configure:12568: checking /usr/include/python2.7/Python.h usability
configure:12568: gcc -c -g -O2  conftest.c >&5
configure:12568: $? = 0
configure:12568: result: yes
`````

can you post that section of your config.log?

luminize, 2016-06-28:
```
configure:12462: checking whether the Boost::Python headers are available
configure:12482: g++ -c -g -O2 -std=c++11 -I/usr/include/python2.7  conftest.cpp >&5
configure:12482: $? = 0
configure:12489: result: yes
configure:12505: checking for boost::python shared library
configure:12522: g++ -o conftest -g -O2 -std=c++11 -I/usr/include/python2.7   conftest.cpp  -lXinerama  -lpython2.7 -lboost_python >&5
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
configure:12522: $? = 4
```

Applicable section and everything below until the end of the log see: http://pastebin.com/VBuuMz00

mhaberler, 2016-06-28:
`gcc --version` says what?

it could be you need swap space like here: https://wiki.archlinux.org/index.php/swap#Swap_file_creation

luminize, 2016-06-28:
```
machinekit@mk-ros-cramps:~/machinekit/src$ gcc --version
gcc (Debian 4.9.2-10) 4.9.2
```
this is a totally unmodified image
will add swap space and report back

luminize, 2016-06-28:
not much space left on the device, will extend the 4GB to my 16GB sized SD
```
machinekit@mk-ros-cramps:~$ df /
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/mmcblk0p1   3360336 3360336         0 100% /
```

luminize, 2016-06-28:
adding swap space and extending the partition (3.3GB -> ~16GB) seems to have done the trick.

first I expanded the filesystem
http://elinux.org/Beagleboard:Expanding_File_System_Partition_On_A_microSD

the next piece did not work:
```
# resize2fs /dev/mmcblk0p2
```

led to
```
*Bad magic number in super-block while trying to open /dev/mmcblk0p2. 
Couldn't find valid filesystem superblock.*
After command:
resize2fs /dev/mmcblk0p2
```

https://www.mail-archive.com/beagleboard@googlegroups.com/msg25207.html led me to use `gparted` on a windows host.

then I added the swap space following your link https://wiki.archlinux.org/index.php/swap#Swap_file_creation

BBB is happily building MK now
Thanks @mhaberler

closed 2016-06-28 by luminize