machinekit/machinekit #978: Instcomp broken on Xenomai package install

strahlex, 2016-07-02:
Running the instcomp tool on a Machinekit package install does compile with the wrong kernel flavor. See the following output:
``` bash
machinekit@beaglebone:~/repos/workshop$ sudo instcomp -i mysum2.icomp 
Compiling realtime mysum2.c
Linking mysum2.so
cp mysum2.so /usr/lib/linuxcnc/posix/
cp: cannot create regular file ‘/usr/lib/linuxcnc/posix/’: Not a directory
/usr/share/linuxcnc/Makefile.modinc:114: recipe for target 'install' failed
make: *** [install] Error 1
```

ArcEye, 2016-07-02:
Don't think is anything to do with instcomp per se.
You should find comp will do the same.

I had this problem previously on the machinekit-NG repo.

flavor.c and rtai_compat.c determine the `flavor` and it will default to posix anyway, if the tests for Xenomai fail.
As an added bonus.
If the env var `FLAVOR` gets set to `posix` it will be used regardless and the tests for kernel type, will not be run.

Also Makefile.inc, which is used by comp and instcomp, has the lines
`BUILD_THREAD_FLAVORS=posix`
`HAVE_POSIX_THREADS=yes`
`HAVE_XENOMAI_THREADS=no`

I don't have a Xenomai armf install to play with, but running `flavor` from the command line should return the same incorrect 'posix'.

ArcEye, 2016-07-02:
See also Issue #192 
May be it was never fixed in xenomai?

strahlex, 2016-07-02:
Tried setting the environment variable `FLAVOR=xenomai` did not make a difference. Is the flavor evaluated at runtime?

ArcEye, 2016-07-02:
What does running the binary `flavor` return?

koppi, 2016-07-02:
Is https://github.com/machinekit/machinekit/issues/928#issue-152629454 related to this issue?

ArcEye, 2016-07-02:
> Is #928 (comment) related to this issue?

Looks like it.  I don't use packages or xenomai, so I can't comment on that particular setup.

When I was converting HAL only stack builds to cmake, I changed the Makefile.inc that was used in component builds, because it was defaulting to posix, even though there was a rt-preempt kernel present, so very similar problem.

strahlex, 2016-07-02:
There is binary `flavor` in my setup.

ArcEye, 2016-07-02:
```
mick@INTEL-i7:/usr/src/machinekit-main$ flavor
rt-preempt
mick@INTEL-i7:/usr/src/machinekit-main$ flavor -b
user-dso
mick@INTEL-i7:/usr/src/machinekit-main$ flavor -m
.so
```

strahlex, 2016-07-02:
Ah okay:
``` sh
$ /usr/libexec/linuxcnc/flavor
xenomai
```

zhivko, 2016-12-25:
Runned into this myselve - have you been able to find solution?

machinekoder, 2017-03-29:
It looks like this issue is Jessie related. Can anyone verify this?

ArcEye, 2017-03-30:
See https://github.com/machinekit/machinekit/issues/1060
and https://github.com/machinekit/machinekit/issues/1072

It is build related, the xenomai flavour is not used to save time and keep it under 50 mins

This results in makefile.inc having xenomai set to `no` as detailed above.

@zultron did have a prototype cross-compiler docker image, that produced all packages quite quickly with correct flavours set.
This however would not work with Wheezy and rather fizzled out.

ArcEye, 2017-07-03:
The comp component build issue is hopefully solved by #1230

closed 2017-07-03 by ArcEye