machinekit/machinekit #979: Missing error message when firmware should be set externally

michelrdagenais, 2016-07-04:
When using a card such as the Mesa 5i24, the firmware should be set externally using the mesaflash utility. This is unlike other Mesa cards such as the 5i20 where the firmware is specified in the config argument of the hm2_pci hal statement. Thus, specifying a firmware for the 5i24, while the driver cannot actually program the firmware, should trigger a warning or an error message.

In file src/hal/drivers/mesa-hostmot2/hostmot2.c, we test for the case where we are capable of programming the firmware but none is specified.

    if ((llio->program_fpga != NULL) && (hm2->config.firmware == NULL)) {
        HM2_PRINT_NO_LL("no firmware specified in config modparam!  the board had better have fir
mware configured already, or this won't work\n");
    }

It would be good to test and report when a firmware is specified but we are not capable of programming one ((llio->program_fpga == NULL) && (hm2->config.firmware != NULL).

mhaberler, 2016-07-05:
makes sense 

I'm not near a mesanet card until 13th, will add that

mhaberler, 2016-07-18:
see https://github.com/mhaberler/machinekit/commit/722e6a20a1fec202a4632486b7ed9d233b973906
(part of https://github.com/machinekit/machinekit/pull/1008 which improves support for SoC/FPGA platforms)

(actually I'd be really grateful for a test of this branch on a non-SoC/FPGA platfrom, i.e. classic PC/some Mesanet card - should run without any snags)

closed 2016-08-15 by mhaberler