machinekit/machinekit #986: After upgraded  today. It have problem.

meoudon, 2016-07-10:
I'm use gmoccapy. error:
No module named machinetalk.protobuf.message_pb2

ArcEye, 2016-07-10:
@strahlex updated the protobuf libraries recently, looks like connected.

Please say exactly what you did, on what platform, with which OS and kernel etc. so that the error can be reproduced and fixed.

There are several versions of packages kept in the repo,  but we don't know which you installed, to be able to advise specifically

meoudon, 2016-07-10:
I use Raspbian RT preempt. Pi3.

meoudon, 2016-07-10:
Last week it work perfect. But in morning I'm upgrade it cannot run.

meoudon, 2016-07-10:
Starting Machinekit...
io started
halcmd loadusr io started
task pid=1546
emcTaskInit: using builtin interpreter
Traceback (most recent call last):
  File "/usr/bin/gmoccapy_lcd7", line 33, in <module>
    import hal_glib             # needed to make our own hal pins
  File "/usr/lib/python2.7/dist-packages/hal_glib.py", line 20, in <module>
    from machinetalk.protobuf.message_pb2 import Container
ImportError: No module named machinetalk.protobuf.message_pb2
Shutting down and cleaning up Machinekit...
exiting HAL component buttons
Cleanup done

ArcEye, 2016-07-10:
OK, that should help @strahlex  find the error

Reverting the packages and 'downgrading' to the previous ones can be done many ways.

If you have (or can have) synaptic installed, this is ubuntu oriented but shows how
http://askubuntu.com/questions/292314/how-to-downgrade-packages-on-ubuntu

That will show you what is installed and what you downgrade to.

There are many other ways, but they require a bit more knowledge

Have a look first at http://deb.machinekit.io/debian/pool/main/m/machinekit/
and identify which packages you installed by comparing to the packages in
/var/cache/apt/archives
(However the dir could be empty, don't know what raspbian does in that respect)

Alternately look at /var/log/apt/history.log for the date of install and the packages you installed should be there.
Either way try to ID which package(s) you installed and what the previous good one was and let us know.

The manual method is simply to remove the later package and specifically install the earlier one, but for that you must know the exact version.

ArcEye, 2016-07-10:
Looks like message_pb2 and types_pb2 are being created in /lib/python/machinetalk/protobuf when building a RIP

The package build does not have the /lib/python/dist-packages/machinetalk directory at all
It is missing from debian/machinekit.install

Fixed (hopefully) and PR submitted

ArcEye, 2016-07-10:
When the package `machinekit_0.1.1468151571-1mk.travis.master.git7bcad2ee~1raspbian_armhf.deb` 
is shown as built in 
http://deb.machinekit.io/debian/pool/main/m/machinekit/
you should be able to upgrade machinekit and it will work again

Probably within half an hour or so, it is the last package to build.

Let us know if it doesn't, but the problem that caused your error message is resolved.

closed 2016-07-10 by ArcEye

meoudon, 2016-07-10:
It work! Thankyou very much.

ArcEye, 2016-07-10:
Thanks for confirming.