machinekit/machinekit #995: Remove all troff man pages and move asciidoc pages to mk-docs

ArcEye, 2016-07-14:
This issue is to capture the conversation regards above on PR #980, which will be lost as soon as it is closed.

ArcEye, 2016-07-14:
Original commit notes:
Make generated asciidoc option only.

Asciidoc pages now in /man/man1-3-9
Generated pages are only built if specifically required with
configure switch --enable-build-documentation
(the =type value is deprecated as this is the only document type used)

[1]
Will require some minimal path re-jigging in the site rendering
infrastructure, to pull the new generated asciidoc repo path.
ie. change /man/docs/man? to /man/man?

ArcEye, 2016-07-14:
hat is legacy troff man pages all removed, with an alternative terminal viewing mechanism added.

Ready to merge when the docs side of it has the path amended.

I think that changing https://jenkins.machinekit.io/view/machinekit/job/machinekit-manpages/configure
Post Build Actions - Archive the artifacts - path to /man/man9/*
should just enable the site docs to be updated as required.

@mhaberler if you can confirm that please, not sure if there is a separate destination path elsewhere.

ArcEye, 2016-07-14:
@mhaberler 

hm, still a conflict

are you sure we want a filename rtapi_app_exitrtapi.asciidoc ? (I dont quite see what the trailing rtapi before .asciidoc adds - there's already a leading rtapi_ ? if it is meant as a section, wouldnt that be more like foo.3rtapi.asciidoc?)

as for artefacts and carrying over static manpages: seems you and @luminize want static manpages in mk/mk - so let's clarify what needs to be done:

    what does this now mean for the artefacts carry-over in jenkins - do I need to store ALL of man/* in the artefacts?
    what happens to the existing static manpages in mk/mk-docs/ - note that several pages in https://github.com/machinekit/machinekit-docs/tree/master/docs/man have already been edited so for those somebody needs to decide which is the latest one, and keep that in whatever is the proper place - I assume move back the modified static manpages to mk/mk and erase them in mk/mk-docs?

speaking of artefacts: I think we should keep the generated manpages in a branch of mk/mk and adapt the build process (package build on merge) to store them so they can be easily imported with a git pull

ArcEye, 2016-07-14:
@ArcEye

>    hm, still a conflict

That arises because the files in this PR (principally comp and instcomp) have been amended in other PRs in the interim, I can just rebase this (if that is the correct terminology) onto a current pull of master and they will disappear.

 >   are you sure we want a filename rtapi_app_exitrtapi.asciidoc ? (I dont quite see what the trailing rtapi before .asciidoc adds - there's already a leading rtapi_ ? if it is meant as a section, wouldnt that be more like foo.3rtapi.asciidoc?)

No. That is just an accident of bulk renaming from hopelessly convoluted original file names.
Easy to fix

>    as for artefacts and carrying over static manpages: seems you and @luminize want static manpages in mk/mk - so let's clarify what needs to be done:

I assume that all future references to man pages mean asciidoc manual pages

I think the only sane thing to do is to have all the manual pages in mk/mk and pull en bloc.
Any 'edit this page' should go to mk/mk, not the copy in mk/mk-docs or whatever
So long as the bulk of man9 pages are generated and their contents potentially unknown until build time, I cannot see any other option.

A possible solution from this 'serpent eating its tail' situation is to make all manual documents a separate package.
If that was held in a separate repo, which had generated asciidocs updated from builds if components changed, that might solve the issue.

It might help clarify boundaries at least :unamused:

ArcEye, 2016-07-14:
>    A possible solution from this 'serpent eating its tail' situation is to make all manual documents a separate package.

>    If that was held in a separate repo, which had generated asciidocs updated from builds if components changed, that might solve the issue.

what if that separate package is in fact machinekit-docs containing both static and generated manpages?

Nothing prevents us from pushing generated asciidoc to this repo automatically, it would take care of the "vanishing generated pages", and one would have a single workflow for working on static manpages and other website content

and in case we find a way to generate legacy man pages from asciidoc, the mk-man package build would be tacked on the mk-docs build

ArcEye, 2016-07-14:
>    what if that separate package is in fact machinekit-docs containing both static and generated manpages?

>    Nothing prevents us from pushing generated asciidoc to this repo automatically, it would take care of the "vanishing generated pages", and one would have a single workflow for working on static manpages and other website content

That is fine.
All static manual pages removed to the mk-docs repo,
generation of component manual pages only via an unadvertised configure switch, which we expect only 'buildbots' to use.

It could even be disabled for RIP builds and /man removed completely - but that would probably be inconvenient to those - like me - who only ever use RIPs.
If you write a new component, no way to check the generated manual doc.

If a user wants asciidoc manual pages, they install a package built from mk-docs to a path that mank will search and they can view them.

The only other bone of contention is generated manual docs, these need clearly labelling in some way to show that they cannot be edited directly in any meaningful way.

ArcEye, 2016-07-14:
> If you write a new component, no way to check the generated manual doc.

Stupid me, of course there is, you just use `(inst)comp --view-doc nameof.(i)comp`

So how do we take this forward?

* 1.   Remove all static asciidoc manual pages from mk
* 2.   Move them to mk-docs, ensuring recently edited versions are preserved
* 3.   Remove all troff man page generation code from (inst)comp and Makefiles
* 4.   Have asciidoc generated to /man/man1-3-9, removing man/doc/man1-3-9, 
asciidoc only generated if build specifically configured to do so
* 5.   Sort out a mechanism to push generated asciidoc manual pages to the mk-docs repo and
not include it in the mk packages.
* 6.   Make a man page package build part of mk-docs build?
Have the `mank` script part of that package, so not requiring additional dependencies in the base machinekit package?
* 7.   Add a caveat to generated asciidoc man pages, that they cannot be directly edited and the source code itself must be reviewed in mk

Anything else?

mhaberler, 2016-07-14:
1,2,3 - yes
4: yes; I guess all generated asciidocs go to man9 so man9 is the only dir affected
5: yes, I will look into that; not sure yet if just a plain push, or subtree voodoo needed
6: sure - I use https://github.com/jordansissel/fpm for one-off debs (several examples in jenkins, eg mksocfpga-packaging); can do that - pls point me to mank 
7: I guess a static comment in the front matter section would cover that

ArcEye, 2016-07-14:
OK,
I'll look at it tomorrow. 1 - 4 is done already, just not in that order.

`mank` is here currently : 
https://github.com/ArcEye/machinekit/blob/remove-troff/bin/mank
Will have to think about its location some more as I built its use into (inst)comp to replace man in the `--view-doc` option.

If it stays in /bin in mk, currently you only need install asciidoctor and elinks if you use it and it will prompt for those at the time.

I'll sleep on it

ArcEye, 2016-07-21:
Hurrah - all pesky debian packaging problems sorted and build passes.

Merge when you think the docs infrastructure is ready to do the right things with it.

mhaberler, 2016-07-21:
you win the persistence award ;)

confused by second para: anything I need to do?

ArcEye, 2016-07-22:
I think we need to merge first into machinekit.
Doing so will probably trigger a docs rebuild, which will fail because the path has changed.

Once the packages are built we can change the path in Jenkins to fetch `/man/man9/*` 
```
#move only changed files for minimal rebuild
rsync --archive --verbose  --checksum ${STAGING}/ ${SOURCE}/docs/man/man9/
```

and change the build to just have `--enable-build-documentation` and trigger a rebuild.
*(not essential as it should continue to build, the =asciidoc value is just ignored)*

THEN the components in the docs should be aligned with those produced by a build

It remains then to sort out a package build for manual pages, which by necessity needs to be done from the docs, because that is the only place they all reside.

`mank` is part of the machinekit package,  because it is needed for the `--view-doc` option in (inst)comp,
so that does not need to be in the manual pages package, which would have complicated things.

ArcEye, 2016-07-22:
I have been playing with `fpm` - quite a good tool

Something like
```
cd source/docs
fpm -s dir -t deb -a all --prefix '/usr/share/doc/machinekit' -p $DESTINATION_DIR \
-d machinekit --description "Manual pages in asciidoc format for components, utilities and APIs" \
--license GPL2 --vendor www.machinekit.io --url www.machinekit.io --maintainer arceye@mgware.co.uk \
-n machinekit-manual-pages -v 0.1-$(date +%F) \
man/*
```
will produce a uniquely date based named package from the jenkins workspace

mhaberler, 2016-07-22:
yes, very easy
have a look at the mksocfpga-packaging job on jenkins.machinekit.ii

mhaberler, 2016-07-22:
I also liked the comment in the docs:

'packaging without a two-year hazing process"

ArcEye, 2016-07-22:
> have a look at the mksocfpga-packaging job on jenkins.machinekit.ii

Good example.  
I don't see we need another job, all the required up to date elements are in the ${WORKSPACE}/source dir and nowhere else.

I will try tacking the build onto the end of the script in the Jenkins` website-preview` job.
Then I can test it without affecting anything else, by staging a tickle PR and then deleting it.

Once it is building as required, just need to move it across to `website-production` and ssh the package to the `deb.machinekit.io/debian/pool....`

I'll see how I get on...............

mhaberler, 2016-07-23:
yes, that is a bit involved - due to the split into two jobs I got away with using fpm from the host

this is how it works: https://github.com/mhaberler/docker-jekyll-asciidoctor
the base image is target-independent; the -custom image must be prepared on the build host to match userid/groupid of the jenkins user on _that_ host. Insane, but needed because ssh is super-picky about permissions how to access .ssh key directories - required when pushing the result to machinekit.github.io (production) or machinekit-ci.github.io (preview)

so if you want to add fpm to that docker image, add it to https://github.com/mhaberler/docker-jekyll-asciidoctor/tree/master/jekyll-asciidoctor-base (a merge on this file will have the base image rebuilt on dockerhub)

next step is on mah.priv.at: pull that image, then run https://github.com/mhaberler/docker-jekyll-asciidoctor/blob/master/jekyll-asciidoctor-custom/gen-user-img.sh to generate the per-buildhost image with the userid/groupid aligned

insane? yes. Required? I found no other way of doing this.

mhaberler, 2016-07-23:
thought through it - I guess adding fpm both to the FPGA build and to the jekyll-asciidoctor docker images is the best solution

mhaberler, 2016-07-23:
https://hub.docker.com/r/haberlerm/docker-jekyll-asciidoctor/builds/ building

ArcEye, 2016-07-23:
OK thanks.

I have disabled the build step for now in case any real documentation PRs come along :smile:

Will check again after lunch to see if the build is completed

ArcEye, 2016-07-23:
Hmm,

I can see you have added fpm to https://github.com/mhaberler/docker-jekyll-asciidoctor/blob/master/jekyll-asciidoctor-base/Gemfile and the build is complete.

However still fails with this error

`12:21:16 /tmp/hudson9019017769069241113.sh: line 21: fpm: command not found`

mhaberler, 2016-07-23:
now: mah@mah2:~/mk-src/docker-jekyll-asciidoctor/jekyll-asciidoctor-custom$ docker run -it  mhaberler/jekyll-asciidoctor:jenkins_1030_1032
jenkins@fc1aa719ad9a:/work$ fpm --help
Intro:

  This is fpm version 1.6.2

mhaberler, 2016-07-23:
finished this minute..

ArcEye, 2016-07-23:
Not being found from a shell script.

I removed the # !/bin/bash header, seeing as it appears to be using sh, but made no difference

`14:06:50 /tmp/hudson6181124724734725032.sh: line 21: fpm: command not found`

If you look at https://jenkins.machinekit.io/view/machinekit/job/website-preview/358/console
I changed the script to 
```
ls -l /usr/bin
ls -l /usr/local/bin
echo $PATH
```

The output shows the PATH is set, but `fpm` is not in /usr/bin or usr/local/bin

On my local system it installed to /usr/local/bin

ArcEye, 2016-07-23:
In fact there is only one binary listed with a 2016 file date and that is from january

Somehow, this is not the same filesystem that you updated.

mhaberler, 2016-07-23:
not quite obvious but there:
````
mah@mah2:~/mk-src/QuartusBuildVMs/Jessie-Quartus-15.1.2$ docker run -it mhaberler/jekyll-asciidoctor:jenkins_1030_1032

jenkins@df6d5ea97f91:/work$ which fpm
/usr/local/rvm/rubies/ruby-2.3.0/bin/fpm
```

but that directory is [on the PATH](https://github.com/mhaberler/docker-jekyll-asciidoctor/blob/master/jekyll-asciidoctor-base/Dockerfile#L72)

ArcEye, 2016-07-23:
>  but that directory is on the PATH

Not the one produced by **echo $PATH**
`/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games /usr/local/bin:/usr/bin:/bin:/usr/local
/games:/usr/games`

I will try specifically calling `/usr/local/rvm/rubies/ruby-2.3.0/bin/fpm` and see what happens

ArcEye, 2016-07-23:
Specifically setting the PATH and exporting it works

```
14:52:17 + PATH=/usr/local/rvm/rubies/ruby-2.3.0/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
14:52:17 + export PATH
14:52:17 + echo PATH = /usr/local/rvm/rubies/ruby-2.3.0/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
14:52:17 PATH = /usr/local/rvm/rubies/ruby-2.3.0/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
14:52:17 + fpm --help

14:52:18 Intro:
14:52:18 
14:52:18   This is fpm version 1.6.2
```

Onwards and upwards..............

mhaberler, 2016-07-23:
jeepers, I dont get this ruby stuff
I started this docker-jekyll-asciidoctor image from a fairly convoluted Dockerfile, and it shows

ArcEye, 2016-07-23:
Not to worry, it now works, I just have to finesse it.

[image]

mhaberler, 2016-07-23:
super!

ArcEye, 2016-07-23:
The package is now about right.
It is probably impossible to effectively test the change status of the files, as most of them are freshly generated each run, so just have to create a new package each time.

All that remains is an equivalent to the `/home/debrepo/scripts/manual_upload.sh` script used on 
the mksoc packaging to push the new package to the deb repo.

I expect it calls a script so as not to expose it in plain text in the configuration.

Once the package push can be tested in website-preview, the whole lot can be shifted to website-production and job done.

ArcEye, 2016-07-24:
>  All that remains is an equivalent to the /home/debrepo/scripts/manual_upload.sh script used on
the mksoc packaging to push the new package to the deb repo.

I'll need your help with that one, whatever the ssh logins etc. are, I don't have access.

mhaberler, 2016-07-24:
fair enough, let me have a look

ArcEye, 2016-07-24:
We are both in Jenkins at the same time.

I was just about to add source to the path for the manual_upload.sh and it suddenly appeared!!

ArcEye, 2016-07-24:
Result !!

[image]

mhaberler, 2016-07-24:
super!

I did install the package on a socfpga; see mail
mank is not included?

ArcEye, 2016-07-24:
No, it needs to be part of the main machinekit package, because (inst)comp use it for the --view-doc option

That is all in the pending PR.

The script has populated in website-production from the merge earlier.

Let me know when you have finished and I will move the elements from website-preview into website-production.

Then it will just be a case of merge the PR and adjust the path that mk-docs uses to pull the generated manual pages and it should be complete

ArcEye, 2016-07-24:
Regards dependencies, I just put `machinekit` because it is a platform agnostic package but requires 'mank' which is in that package.

If there is a better way I will use that, but I envisaged it being an extra package install on an existing system, where it would not add any weight save itself.

ArcEye, 2016-07-24:
The final path change in the PR passes
https://travis-ci.org/machinekit/machinekit/builds/146979053

The changes are made in website-production, but not activated.

Let's look at it tomorrow, very little to do now.

ArcEye, 2016-07-24:
See you have merged

The build is failing in that Jenkins job because of the changes in paths and make targets
for the make-manpages stage

PR #1011 should fix it

I have already changed the `Configuration > Post Build Actions` to correct the path

ArcEye, 2016-07-24:
machinekit-manpages works :+1: 

However website-production is failing
```
/bin/bash -e /tmp/hudson1579443330677457229.sh
+ /var/lib/jenkins/workspace/website-production/source/.jenkins/manual_upload.sh --import /var/lib/jenkins/workspace/website-production/package --distribution jessie --user mkjail --pass 86.59.12.252
sftp host argument missing!
Build step 'Conditional step (single)' marked build as failure
```

Being as the script is a cut and paste from website-preview, where it worked, imagine there is a env var which sets host missing, or similar.

ArcEye, 2016-07-24:
Yep, spot the difference.

webiste-preview
```
13:27:01 + /var/lib/jenkins/workspace/website-preview/source/.jenkins/manual_upload.sh --import /var/lib/jenkins/workspace/website-preview/package --distribution jessie --user mkjail --pass ******** 86.59.12.252
```

website-production
```
+ /var/lib/jenkins/workspace/website-production/source/.jenkins/manual_upload.sh --import /var/lib/jenkins/workspace/website-production/package --distribution jessie --user mkjail --pass 86.59.12.252
```
$mkjail is not evaluating to anything

ArcEye, 2016-07-25:
Hurrah, it all now works, @mhaberler spotted that an option which enables global passwords, had not been ticked.

Just needs something to publicise the changes, probably an email, permanent entry in the docs on the man page index and a README in the sources /man directory.

People won't read them, but at least we can point to them when they say the man pages have disappeared :smile:

ArcEye, 2016-07-25:
All finished (famous last words).

closed 2016-07-25 by ArcEye