machinekit/machinekit #996: Python HAL_BIT has wrong type

strahlex, 2016-07-14:
It seems `HAL_BIT` pins are represented as integers in the Python hal module. For example

```python
error = hal.Pin('tool-info.error')
assert(error.type is hal.HAL_BIT)
assert(error.dir is hal.HAL_OUT)
assert(error.linked is False)
assert(error.get() is True)
```

The code fails on the last assert although error is a `HAL_BIT` pin. This is a result of the `HAL_BIT` being represented as int in Python. `assert(error.get() == True)` gives the correct result. However, since this is a boolean data type we would expect the first assert to work as well.

mhaberler, 2016-07-14:
I was shooting for bug compatibility with the legacy hal module.

strahlex, 2016-07-15:
I see, unless `is` is used bool is be compatible when compared with 1 or 0.

machinekoder, 2017-04-07:
On the other hand `error.get() is 1` looks also appropriate for bit comparison.

closed 2017-04-07 by machinekoder