machinekit/machinekit #999: package install: machinetalk/__init__.py missing

mhaberler, 2016-07-15:
see https://groups.google.com/forum/#!topic/machinekit/jdpypzDCsg4

likely caused by https://github.com/machinekit/machinekit/commit/f8598752e000cfb5b989ec41508ffc60cada65d6 (the cp statement in debian/rules.in fails to copy lib/python/machinetalk/__init__.py)

ArcEye, 2016-07-15:
```
mick@INTEL-i7:/downloads$ dpkg -c machinekit_0.1.1468568321-1mk.travis.master.gitba4be43b~1jessie_amd64.deb | grep init
-rw-r--r-- root/root      8786 2016-07-15 09:04 ./usr/share/linuxcnc/examples/sample-configs/sim/axis/vismach/hexapod-sim/minitetra.ini
-rw-r--r-- root/root         0 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/touchy/__init__.py
-rw-r--r-- root/root       911 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/rs274/__init__.py
-rw-r--r-- root/root         0 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/stepconf/__init__.py
-rw-r--r-- root/root        31 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/gladevcp/__init__.py
-rw-r--r-- root/root         0 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/gscreen/__init__.py
-rw-r--r-- root/root         0 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/machinetalk/protobuf/__init__.py
-rw-r--r-- root/root         0 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/gmoccapy/__init__.py
-rw-r--r-- root/root         2 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/fdm/__init__.py
-rw-r--r-- root/root         0 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/machinekit/__init__.py
-rw-r--r-- root/root         2 2016-07-15 09:20 ./usr/lib/python2.7/dist-packages/drivers/__init__.py
```

mhaberler, 2016-07-15:
well yes, there needs to be an empty __init__.py file in /usr/lib/python2.7/dist-packages/machinetalk as well - ./usr/lib/python2.7/dist-packages/machinetalk/protobuf/__init__.py is not enough since it's two levels of nesting

mhaberler, 2016-07-15:
this fix is wrong, pls ignore

mhaberler, 2016-07-15:
this is more like it

ArcEye, 2016-07-15:
It should fail, just as it does locally
```
dh_install --sourcedir=debian/tmp --fail-missing -Xusr/bin/pasm
dh_install: machinekit missing files (usr/lib/python*/*/machinetalk/*), aborting
debian/rules:155: recipe for target 'install' failed
make: *** [install] Error 2
```
I spent hours on various permutations but could not get dh_install to work just using an entry in 
machinekit.install, as is supposed to happen.

Hence the unsatisfactory hard coded mkdir

mhaberler, 2016-07-15:
it occurs to me this is a problem with the install targets; I think debian/machinekit.install is not at fault - the files are just not there since the Submakefile is broken:

if one runs
````
 debian/configure -p 
 debuild -i -us -uc -b
````
there's a step `(cd src; export DESTDIR=`pwd`/../debian/tmp; /usr/bin/make V= install)` 

now replicating post-fail of debuild:
````
mkdir /tmp/fasel
sudo DESTDIR=/tmp/fasel make install
`````
now find /tmp/fasel gives:  http://static.mah.priv.at/public/fa.txt

NB: no trace of machinetalk/*

summary: the install targets for machinetalk/protobuf are broken

@strahlex - can I ask you to take over this bug?

looks like a fallout of the recent protobuf reshuffle, but frankly I am lost in the maze of changes and the commit messages are less than helpful

ArcEye, 2016-07-15:
Shall we 'fix' for now, with a recursive copy in rules.in, to make sure the first __init__py is copied

Then the unsatisfactory hard coded python version can be removed once the install targets are sorted so that machinekit.install works as it should

mhaberler, 2016-07-15:
no, leave as is and mention the touch <path> workaround on the list, should be good for now

strahlex, 2016-07-15:
Not sure why this happens since both `__init__.py` files are created during build

strahlex, 2016-07-15:
How can I check the pacakge build? probably fixed with https://github.com/strahlex/machinekit/tree/proto-install-fix

strahlex, 2016-07-15:
trying
```
debian/configure -p 
debuild -i -us -uc -b
```

mhaberler, 2016-07-15:
yes
if you run this, this likely fails - fine

then try 

````
mkdir /tmp/fasel
sudo DESTDIR=/tmp/fasel make install
````
and inspect /tmp/fasel if all needed files were installed

mhaberler, 2016-07-15:
alternatively, without debuild:

````
mkdir /tmp/install
./configure --help
make OPT=-O0 
sudo DESTDIR=/tmp/install make install
find /tmp/install|grep machinetalk/prot 
<still empty>
`````

mhaberler, 2016-07-15:
the problem is not with PYTARGETS which is ok, I think its in src/Makefile

@ArcEye I think your dir issue can be found in the install-python: target of src/Makefile

mhaberler, 2016-07-15:
fix verified in 0.1.1468610245-1mk.travis.master.git2c0bb2a9~1jessie, closing

closed 2016-07-15 by mhaberler