machinekit/machinekit #1021: "newthread" command is undocumented

jmkasunich, 2016-08-07:
The "newthread" command was added to halcmd over a year ago.  But the in-program help text and the man page were not updated to document the new command.

I have fixed the in-program help (note 1) and will be fixing the man page once I find out where it is stored.

Note 1:  newthread has several options.  Most of them I was able to figure out by referring to the source.  But two, "posix" and "nowait", I cannot figure out.  I have no idea what they mean or why a normal user would ever issue those options.  Michael H (original author of the "newthread" command) will have to explain what he intended those options for.

dkhughes, 2016-08-07:
`nowait` makes a thread able to block - it basically doesn't ever yield. As one example of use, we used it in #687 so we could sync the HAL function processing to an external timing source triggering an interrupt.

`posix` I believe is used to allow extra options that some RT thread providers don't offer, and to warn or fail in situations that would cause a hang, but @mhaberler will have to chime in on specific details.

mhaberler, 2016-08-07:
yes, `nowait` skips the rtapi_wait() at the end of the thread cycle. This makes sense if there is a funct in the chain which does some sort of wait, and the host-timed cyclic behavior is not useful. As @dkhughes noted, this is used in the work for [external HAL thread synchronization ](https://github.com/machinekit/machinekit/issues/687). The only 'config' example of this flag is a test halcmd script [here](https://github.com/machinekit/machinekit/blob/master/configs/hm2-soc-stepper/irqtest.hal#L4) - it is part of the ongoing SoC/FPGA effort ([here](https://github.com/machinekit/machinekit/issues/915) and [here](https://github.com/machinekit/machinekit/issues/903)).  I hope we will be able to use this scheme eventually on a wider range of platforms as it takes out a key source of position feedback noise without resorting to workarounds like the hm2 DPLL.

`posix` starts a thread as a normal Posix thread. In the priority hierarchy it's a very low priority thread without any guarantees. However, functs can be threaded onto it just normally. The use case I had in mind is executing HAL portions which are not time critical but can be assembled with stock HAL 'RT' components, and without actually writing a 'userland' HAL component.

In essence it shifts the meaning of the 'RT' property from the funct to the thread, where it belongs; the legacy distinction where 'RT component' was essentially lipstick on 'kernel module' is marginally helpful nowadays.

`posix` also permits using the Linux API from a component threaded onto a posix thread; that is for instance not possible with a Xenomai RT thread (albeit from an RT-PREEMPT thread). An example would be [here](https://github.com/mhaberler/machinekit/commits/hal-fd-notifiy) (related [issue](https://github.com/machinekit/machinekit/issues/963)) which would typically use nowait and possibly posix in combination.

Both flags are currently not used in any suggested configs, so there is no issue with them remaining undocumented at this time. If any, the `posix` flag could be documented as it has current use case.

jmkasunich, 2016-08-09:
Closed by PRs #1027 (machinekit: fixes halcmd built-in help) and 219 (machinekit-docs: fixes man page)

closed 2016-08-09 by jmkasunich