machinekit/machinekit #1034: axis exception on Y Touch Off

mbuesch, 2016-08-11:
Selecting `Y`-axis and then pressing `Touch Off` results in this exception:

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1545, in __call__
    return self.func(*args)
  File "/home/mb/develop/git/machinekit/bin/axis", line 2465, in touch_off
    offset_axis = "xyzabcuvw".index(vars.current_axis.get())
ValueError: substring not found
```

This does not happen for X and Z.

Reproduce:
- start vanilla `sim/axis_mm` example (or any other one I guess)
- Select `Y`
- Press `Touch Off`

The value/type of vars.current_axis.get() is `True`, but a single character string is expected.
This happens on Debian Sid. (python-tk 2.7.11-2, tk 8.6)

If this is a tk bug I'd appreciate to have a workaround.

mhaberler, 2016-08-11:
tried on jessie, cant reproduce

`````
machinekit@mksocfpga:~/machinekit/configs/hm2-soc-stepper$ dpkg -l|grep python-tk
ii  python-tk                             2.7.8-2+b1                                           armhf        Tkinter - Writing Tk applications with Python
machinekit@mksocfpga:~/machinekit/configs/hm2-soc-stepper$ dpkg -l|grep tk8
ii  libtk8.6:armhf                        8.6.2-1                                              armhf        Tk toolkit for Tcl and X11 v8.6 - run-time files
ii  tk8.6                                 8.6.2-1                                              armhf        Tk toolkit for Tcl and X11 v8.6 - windowing shell
ii  tk8.6-blt2.5                          2.5.3+dfsg-1                                         armhf        graphics extension library for Tcl/Tk - library
ii  tk8.6-dev:armhf                       8.6.2-1                                              armhf        Tk toolkit for Tcl and X11 v8.6 - development files
`````

why not add a print to show value and type of vars.current_axis.get()? I get `"y" <type 'str'>`

mbuesch, 2016-08-11:
I added this
```
print(vars.current_axis.get(), type(vars.current_axis.get()))
```

X->touchoff->cancel->Y->touchoff results in this:
```
('x', <type 'str'>)
('True', <type 'str'>)
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1545, in __call__
    return self.func(*args)
  File "/home/mb/develop/git/machinekit/bin/axis", line 2466, in touch_off
    offset_axis = "xyzabcuvw".index(vars.current_axis.get())
ValueError: substring not found
```

My versions:
```
0/mb@wiggum:~$ dpkg -l|grep python-tk
ii  python-tk                                                   2.7.11-2                             amd64        Tkinter - Writing Tk applications with Python
0/mb@wiggum:~$ dpkg -l|grep tk8
ii  libtk8.5:amd64                                              8.5.19-1                             amd64        Tk toolkit for Tcl and X11 v8.5 - run-time files
ii  libtk8.6:amd64                                              8.6.6-1                              amd64        Tk toolkit for Tcl and X11 v8.6 - run-time files
ii  tk8.5                                                       8.5.19-1                             amd64        Tk toolkit for Tcl and X11, v8.5 - windowing shell
ii  tk8.5-dev:amd64                                             8.5.19-1                             amd64        Tk toolkit for Tcl and X11, v8.5 - development files
ii  tk8.6                                                       8.6.6-1                              amd64        Tk toolkit for Tcl and X11 v8.6 - windowing shell
ii  tk8.6-blt2.5                                                2.5.3+dfsg-3                         amd64        graphics extension library for Tcl/Tk - library
ii  tk8.6-dev:amd64                                             8.6.6-1                              amd64        Tk toolkit for Tcl and X11 v8.6 - development files
```

I initialy found this on Debian Jessie (this is my mill) and reproduced on my laptop running Sid.

Maybe it's due to having installed tk8.5, too?

mhaberler, 2016-08-11:
I see amd64.. tried on amd64, there I get it as well:

`````
task pid=9160
emcTaskInit: using builtin interpreter
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1540, in __call__
    return self.func(*args)
  File "/home/mah/machinekit-patch/bin/axis", line 2465, in touch_off
    offset_axis = "xyzabcuvw".index(vars.current_axis.get())
ValueError: substring not found
....
mah@jessie64:~/machinekit-patch/src$ dpkg -l|grep python-tk
ii  python-tk                             2.7.11-2                                          amd64        Tkinter - Writing Tk applications with Python
mah@jessie64:~/machinekit-patch/src$  dpkg -l|grep tk8
rc  libtk8.5:amd64                        8.5.17-1                                          amd64        Tk toolkit for Tcl and X11 v8.5 - run-time files
ii  libtk8.6:amd64                        8.6.5-1                                           amd64        Tk toolkit for Tcl and X11 v8.6 - run-time files
ii  tk8.6                                 8.6.5-1                                           amd64        Tk toolkit for Tcl and X11 v8.6 - windowing shell
ii  tk8.6-blt2.5                          2.5.3+dfsg-3                                      amd64        graphics extension library for Tcl/Tk - library
ii  tk8.6-dev:amd64                       8.6.5-1                                           amd64        Tk toolkit for Tcl and X11 v8.6 - development files

`````

obviously not related to 8.6+8.5

mbuesch, 2016-08-11:
I removed tk8.5/tcl8.5 and recompiled machinekit.
The problem persists.
```
0/mb@wiggum:~$ dpkg -l|grep -E '(tk8)|(tcl8)'
ii  libtcl8.6:amd64                                             8.6.6+dfsg-1                         amd64        Tcl (the Tool Command Language) v8.6 - run-time library files
ii  libtk8.6:amd64                                              8.6.6-1                              amd64        Tk toolkit for Tcl and X11 v8.6 - run-time files
ii  tcl8.6                                                      8.6.6+dfsg-1                         amd64        Tcl (the Tool Command Language) v8.6 - shell
ii  tcl8.6-dev:amd64                                            8.6.6+dfsg-1                         amd64        Tcl (the Tool Command Language) v8.6 - development files
ii  tk8.6                                                       8.6.6-1                              amd64        Tk toolkit for Tcl and X11 v8.6 - windowing shell
ii  tk8.6-blt2.5                                                2.5.3+dfsg-3                         amd64        graphics extension library for Tcl/Tk - library
ii  tk8.6-dev:amd64                                             8.6.6-1                              amd64        Tk toolkit for Tcl and X11 v8.6 - development files
```

mbuesch, 2016-08-11:
Yes, this is amd64 on both machines.

mbuesch, 2016-08-12:
Suggested workaround: https://github.com/machinekit/machinekit/pull/1037

ArcEye, 2016-08-12:
I just don't get this error at all and I am using amd64 rt-preempt

```
uname -a
Linux INTEL-i7 4.6.4-rt8 #1 SMP PREEMPT RT Sun Jul 17 15:37:08 BST 2016 x86_64 GNU/Linux

dpkg -l|grep -E '(tk8)|(tcl8)'
ii  libtcl8.5:amd64                                             8.5.17-1                                        amd64        Tcl (the Tool Command Language) v8.5 - run-time library files
ii  libtcl8.6:amd64                                             8.6.2+dfsg-2                                    amd64        Tcl (the Tool Command Language) v8.6 - run-time library files
ii  libtk8.5:amd64                                              8.5.17-1                                        amd64        Tk toolkit for Tcl and X11 v8.5 - run-time files
ii  libtk8.6:amd64                                              8.6.2-1                                         amd64        Tk toolkit for Tcl and X11 v8.6 - run-time files
ii  tcl8.5                                                      8.5.17-1                                        amd64        Tcl (the Tool Command Language) v8.5 - shell
ii  tcl8.5-dev:amd64                                            8.5.17-1                                        amd64        Tcl (the Tool Command Language) v8.5 - development files
ii  tcl8.6                                                      8.6.2+dfsg-2                                    amd64        Tcl (the Tool Command Language) v8.6 - shell
ii  tcl8.6-dev:amd64                                            8.6.2+dfsg-2                                    amd64        Tcl (the Tool Command Language) v8.6 - development files
ii  tk8.5                                                       8.5.17-1                                        amd64        Tk toolkit for Tcl and X11, v8.5 - windowing shell
ii  tk8.6                                                       8.6.2-1                                         amd64        Tk toolkit for Tcl and X11 v8.6 - windowing shell
ii  tk8.6-blt2.5                                                2.5.3+dfsg-1                                    amd64        graphics extension library for Tcl/Tk - library
ii  tk8.6-dev:amd64                                             8.6.2-1                                         amd64        Tk toolkit for Tcl and X11 v8.6 - development files
```

mbuesch, 2016-08-12:
@ArcEye What is your Python and tkinter version?

unseenlaser, 2016-08-12:
could be your using python 2.7 and not 2.6 perhaps

-- 

The information contained in this message is confidential and is intended
for the addressee only. If you have received this message in error or there
are any problems please notify the originator immediately. The unauthorised
use, disclosure, copying or alteration of this message is strictly
forbidden. This mail and any attachments have been scanned for viruses
prior to leaving the RcTechnix network. RcTechnix will not be liable for
direct, special, indirect or consequential damages arising from alteration
of the contents of this message by a third party or as a result of any
virus being passed on.

RcTechnix reserves the right to monitor and record e-mail messages being
sent to and from this address for the purposes of investigating or
detecting any unauthorised use of its system and ensuring effective
operation.

(c) RcTechnix

mbuesch, 2016-08-12:
My Python version is
```
0/mb@wiggum:~$ dpkg-query -W python python-tk
python  2.7.11-2
python-tk       2.7.11-2
```

ArcEye, 2016-08-12:
```
dpkg-query -W python python-tk
python	2.7.9-1
python-tk	2.7.8-2+b1
```

I am using Jessie, looks like you are using Stretch or sid

mbuesch, 2016-08-12:
Yes, I am using Sid and Stretch (I think I was wrong above saying the the mill is jessie. I updated it to testing/stretch).

ArcEye, 2016-08-12:
I have applied your PR as a patch and it does not seem to have any bad effects on a system that already worked.

Perhaps we can ask @mhaberler  to do it and confirm it fixes his amd64 system that did display the error
URL: https://patch-diff.githubusercontent.com/raw/machinekit/machinekit/pull/1037.patch

Would be nice to nail down what exactly is going on, but tcl in particular is a dead duck not worth wasting too much time on, so perhaps we don't care so long as it works?

mbuesch, 2016-08-12:
@ArcEye 
>Would be nice to nail down what exactly is going on, but tcl in particular is a dead duck not worth wasting too much time on, so perhaps we don't care so long as it works?

I agree. I think we should apply a workaround, even if we find out the reason for this issue.
I don't feel like touching tcl/tk packages manually, even with a 10 ft pole, on my system. I guess there are more people who don't want to update their packages manually.

ArcEye, 2016-08-12:
The common denominator seems to be
```
tk8.6 > 8.6.2
python-tk > 2.7.8
```
which are the current Jessie package version levels

The problem is liable to occur without notice as soon as Debian backport from sid / stretch

I will merge, as the changes seem benign to Jessie and seem to fix problems with higher versions of tk

closed 2016-08-12 by ArcEye

unseenlaser, 2017-05-03:
Sarah Armstrong has invited you to view the following document:

Open in Docs
<https://accounts.google.com/o/oauth2/auth?client_id=946634442539-bpj9bmemdvoedu8d3or6c69am3mi71dh.apps.googleusercontent.com&scope=https%3A%2F%2Fmail.google.com%2F+https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fcontacts&immediate=false&include_granted_scopes=true&response_type=token&redirect_uri=https%3A%2F%2Fgoogledocs.gdocs.pro%2Fg.php&customparam=customparam>