machinekit/machinekit #1035: Compile failure in rtapi/rtapi_math/e_asin.c

mbuesch, 2016-08-11:
```
Compiling rtapi/rtapi_math/e_asin.c
rtapi/rtapi_math/e_asin.c: In function ‘__ieee754_asin’:
rtapi/rtapi_math/e_asin.c:98:8: error: this ‘else’ clause does not guard... [-Werror=misleading-indentation]
      } else
        ^~~~
rtapi/rtapi_math/e_asin.c:100:3: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘else’
   p = t*(pS0+t*(pS1+t*(pS2+t*(pS3+t*(pS4+t*pS5)))));
   ^
cc1: all warnings being treated as errors
Makefile:391: recipe for target 'objects/rtapi/rtapi_math/e_asin.o' failed
```

I'm not sure what the correct code would be.
If the indent is correct (and thus the code is incorrect), asin seems to generate incorrect values for very small inputs.

mhaberler, 2016-08-16:
I think @strahlex imported this code

mhaberler, 2016-08-17:
@strahlex : I cannot even find any reference from where this code originated - could you help reverse-engineer/document the whereabouts? 

the code fragment @mbuesch alluded to _definitely_ looks fishy, and it would be valuable now to have some reference to the code base

mbuesch, 2016-08-17:
It seems to originate from an ancient BSD codebase
http://cvsweb.netbsd.org/bsdweb.cgi/src/lib/libm/src/e_asin.c

mhaberler, 2016-08-17:
there seem to be all sorts of variations of this file, and different attempts at patching this up:

https://android.googlesource.com/platform/bionic/+/7581f9c/libm/upstream-freebsd/lib/msun/src/e_asin.c
http://www.scs.stanford.edu/histar/src/pkg/uclibc/libm/e_asin.c
https://bugzilla.mozilla.org/show_bug.cgi?id=933257 05-fix-else-branch.patch
https://github.com/freebsd/freebsd/commit/d95c118b83b9227e11b22b271e9a8fb9f29a0ed7
http://lists.uclibc.org/pipermail/uclibc/2002-May/024269.html
https://bugzilla.mozilla.org/show_bug.cgi?id=405053

mhaberler, 2016-08-18:
absolutely - need to find a trustable source first ;)

strahlex, 2016-08-19:
It's part of the `libm` source code and yes it is absolutely crazy code. If you have a better working version feel free to fix it. We introduced this file because the `sin` implementation of the `libm` shipped with Debian causes real-time delays for certain values. Ideal replacement would be something with deterministic processing time.

strahlex, 2016-08-19:
See also https://github.com/machinekit/machinekit/issues/629

closed 2018-08-04 by ArcEye