machinekit/machinekit #1048: Example sim/axis/axis_rostock broken

andreasBihlmaier, 2016-08-24:
Trying to run the sim -> axis -> axis_rostock ini fails on HEAD as shown below:

```
user@host:~$ machinekit
MACHINEKIT - 0.1
Machine configuration directory is '/home/user/machinekit/configs/sim/axis'
Machine configuration file is 'axis_rostock.ini'
Starting Machinekit...
io started
halcmd loadusr io started
task pid=7137
emcTaskInit: using builtin interpreter
waiting for s.axes
Traceback (most recent call last):
  File "/home/user/machinekit/bin/axis", line 3356, in <module>
    activate_axis(0, True)
  File "/home/user/machinekit/bin/axis", line 1303, in activate_axis
    axis.invoke()
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 2808, in invoke
    return self.tk.call(self._w, 'invoke')
_tkinter.TclError: invalid command name "axis_activated"
Shutting down and cleaning up Machinekit...
Cleanup done
Machinekit terminated with an error.  You can find more information in the log:
    /home/user/linuxcnc_debug.txt
and
    /home/user/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
```

Other examples (e.g. axis_mm) work fine.

This is on `Linux host 4.2.0-42-generic #49~14.04.1-Ubuntu SMP Wed Jun 29 20:22:11 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux` with HEAD at b09b7f80478994d020db770aee99bf92c3d55e16.

mhaberler, 2016-08-24:
could be related to https://github.com/machinekit/machinekit/issues/1034

andreasBihlmaier, 2016-08-24:
The tk and python-tk versions on my system are:

```
> dpkg -l | egrep '(tk8|python-tk)'
ii  libtk8.5:amd64                                              8.5.15-2ubuntu3                                        amd64        Tk toolkit for Tcl and X11 v8.5 - run-time files
ii  libtk8.6:amd64                                              8.6.1-3ubuntu2                                         amd64        Tk toolkit for Tcl and X11 v8.6 - run-time files
ii  python-tk                                                   2.7.5-1ubuntu1                                         amd64        Tkinter - Writing Tk applications with Python
ii  python-tk-dbg                                               2.7.5-1ubuntu1                                         amd64        Tkinter - Writing Tk applications with Python (debug extension)
ii  tk8.5                                                       8.5.15-2ubuntu3                                        amd64        Tk toolkit for Tcl and X11, v8.5 - windowing shell
ii  tk8.5-dev:amd64                                             8.5.15-2ubuntu3                                        amd64        Tk toolkit for Tcl and X11, v8.5 - development files
ii  tk8.6                                                       8.6.1-3ubuntu2                                         amd64        Tk toolkit for Tcl and X11 v8.6 - windowing shell
ii  tk8.6-dev:amd64                                             8.6.1-3ubuntu2                                         amd64        Tk toolkit for Tcl and X11 v8.6 - development files
ii  tk8.6-doc                                                   8.6.1-3ubuntu2                                         all          Tk toolkit for Tcl and X11 v8.6 - manual pages
```

ArcEye, 2016-08-24:
I dont know what rostock is, but suspect it goes into world mode.
I can replicate the error.

I have found that @mbuesch only changed the tcl commands for `axes and missed the duplicate set some way down the file for `joints`.

I have amended those and my local copy works, I will submit a PR soon

andreasBihlmaier, 2016-08-24:
I can confirm that your patch fixes the issue for me, too.
Thanks!

With some minor patching, I even got src/hal/user_comps/vismach/rostock.py too work again. However, this is WiP, I'll send a PR further down the road.

ArcEye, 2016-08-24:
OK, thanks for the confirmation, I'll close this.

closed 2016-08-24 by ArcEye