machinekit/machinekit #1059: Interest for Machinekit meeting in the states?

machinekoder, 2016-09-26:
It is almost a year since the last Machinekit meeting at the Tormach facility in Wisconsin. Therefore, I would like to gauge some interest for another Machinekit meeting in the states this Fall or Spring 2017.

Is anyone else interested? Are there any location or time preferences?

prcdslnc13, 2016-09-28:
Would love to see it happen. I missed last year and would like to make it this year.

pmcstone, 2016-10-04:
Tormach make it happen :+1:

machinekoder, 2016-11-24:
Does not look like there is a lot of discussion going on here.

Early February would be the optimal time.

machinekoder, 2016-11-28:
I created a doodle poll: 
http://doodle.com/poll/32kkc5pnkyfuxpun

machinekoder, 2017-01-17:
I updated the poll for April, since February and March turned out to be too busy.

luminize, 2017-07-26:
This meeting has happened.

closed 2017-07-26 by luminize