machinekit/machinekit #1060: comp --install fails under xenomai

cdsteinkuehler, 2016-09-28:
On the Beaglebone, when compiling a *.comp file out-of-tree, the install fails.  The comp command is attempting to copy the component to the /usr/lib/linuxcnc/posix directory, which does not exist.

If I manually copy the generated *.so file to /usr/lib/linuxcnc/xenomai, I can load the component as usual and everything seems to be OK.

I'm not sure what changes (if any) need to be made when compiling for different RT flavors (ie: if comp is really making a posix *.so file and I'm just lucky that it works under xenomai), but there might be an issue with this as well.  I assume the *.so files are not interchangeable, or we wouldn't be compiling all of them into different directories based on RT flavor.

Can someone review the comp file and make sure it's doing the right thing based on the current running RT environment?

zhivko, 2016-09-28:
I am doing same workflow and since I use BBB with xenomai that wf worked
for me. I thought it can be better to lnk it but apart from that....
copyingis workingjust fine.
BEside of this and maybe unrelated - its there any advantage using rtpreemt
on BBB? Is it in plan to distiribute machinekit for BBB with rtpreemt?

cdsteinkuehler, 2016-09-28:
On 9/28/2016 8:57 AM, zhivko wrote:
> BEside of this and maybe unrelated - its there any advantage using rtpreemt
> on BBB? Is it in plan to distiribute machinekit for BBB with rtpreemt?

The Xenomai kernel still performs better than the RT_PREEMPT kernels,
but the RT kernels continue to get better.

You can already run Machinekit on the -rt kernels, just install the
kernel of your choice (there are several available in the standard BBB
packaging stream) and "apt-get install machinekit-rt-preempt".

The various flavors of Machinekit (xenomai, rt-preempt, posix) can all
be installed at the same time.  Which gets used will depend on the
flavor of kernel you are running.

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-09-28:
Reviewing things a bit, it appears HAL code compiled in-tree gets built with various different options for the different RT flavors, while the comp command seems to ignore this and just does one thing.

So the question is can the single *.so generated by comp be safely used across all RT flavors?

ArcEye, 2016-09-28:
> Can someone review the comp file and make sure it's doing the right thing based on the current running RT environment?

Will do.

Did you enable posix when you built MK?

I have had problems in the past with the `flavor` binary defaulting to posix, because posix was built, even though the running kernel was rt-preempt.

cdsteinkuehler, 2016-09-28:
This is with the packaged version for the BeagleBone, so I believe all flavors were built (posix, rt-preempt, and xenomai).  Currently I only have the machinekit-xenomai flavor installed, along with machinekit-dev to build a comp file.

cdsteinkuehler, 2016-09-28:
The flavor binary seems to work as expected:

''''
$ /usr/libexec/linuxcnc/flavor
xenomai
''''

ArcEye, 2016-09-28:
A bit of a route map.

The Makefile created by comp, just #includes Makefile.modinc

This defines install as 
```
install:
	cp $(patsubst %.o,%.ko,$(obj-m)) $(DESTDIR)$(RTLIBDIR)/
```

It gets these string values from an earlier conditional block (the paths are mine)
```
ifeq ($(RUN_IN_PLACE),yes)
include /usr/src/machinekit/src/Makefile.inc
else
include /usr/src/machinekit/share/linuxcnc/Makefile.inc
endif
```
Therefore comp does not make any decisions, it follows the RTLIBDIR set in Makefile.inc

Makefile.inc has labyrinthine code to determine the THREAD type starting at line 10.
which tries to use the `flavor`  binary, which can be fallible.

Can you forward me the Makefile.modinc and Makefile.inc from your system?

cdsteinkuehler, 2016-09-28:
Here you go...with a *.txt extension to keep github happy.
The files came from:
/usr/share/linuxcnc/Makefile.inc
/usr/share/linuxcnc/Makefile.modinc

...starting with the modinc file as reported by comp:

````
$ comp --print-modinc
/usr/share/linuxcnc/Makefile.modinc
````
[Makefile.modinc.txt](https://github.com/machinekit/machinekit/files/498554/Makefile.modinc.txt)
[Makefile.inc.txt](https://github.com/machinekit/machinekit/files/498555/Makefile.inc.txt)

cdsteinkuehler, 2016-09-28:
Oh...if you're messing with the comp command, you might fix the usage documentation.  The short command line switches don't match the long version.  Otherwise, I'll fix it sometime.

ArcEye, 2016-09-28:
Extract from Makefile.inc which I suspect is relevant
```
HAVE_POSIX_THREADS=yes
HAVE_RT_PREEMPT_THREADS=no
HAVE_XENOMAI_THREADS=no
HAVE_XENOMAI_KERNEL_THREADS=no
HAVE_RTAI_KERNEL_THREADS=no
```
This in turn goes right back via Makefile.inc.in to configure.ac
```
# Xenomai userland settings
if test "$with_xenomai" = yes; then
    AC_DEFINE([HAVE_XENOMAI_THREADS],[],[Build Xenomai userland threads])
fi
AC_SUBST([HAVE_XENOMAI_THREADS],[$with_xenomai])
AC_SUBST([XENOMAI_THREADS_FLAVOR_ID],_cfg_flavor_id(xenomai))
AC_SUBST([XENOMAI_THREADS_BUILD_SYS],_cfg_flavor_build_sys(xenomai))
AC_SUBST([XENOMAI_THREADS_SOURCE],[xenomai])
AC_SUBST([XENOMAI_THREADS_RTFLAGS])
AC_SUBST([XENOMAI_THREADS_LDFLAGS])
AC_SUBST([XENOMAI_THREADS_RTS])
AC_SUBST([XENOMAI_THREADS_RTDIR],[])
AC_SUBST([XENOMAI_THREADS_EXTRA_KMODS],[])
AC_SUBST([PRECISE_MULTILIB_BUG],[$PRECISE_MULTILIB_BUG])

# Xenomai kernel settings
if test "$with_xenomai_kernel" = yes; then
    AC_DEFINE([HAVE_XENOMAI_KERNEL_THREADS],[],[Build Xenomai kernel threads])
fi
AC_SUBST([HAVE_XENOMAI_KERNEL_THREADS],[$with_xenomai_kernel])
AC_SUBST([XENOMAI_KERNEL_THREADS_FLAVOR_ID],_cfg_flavor_id(xenomai-kernel))
AC_SUBST([XENOMAI_KERNEL_THREADS_BUILD_SYS],
	_cfg_flavor_build_sys(xenomai-kernel))
AC_SUBST([XENOMAI_KERNEL_THREADS_SOURCE],[xenomai-kernel])
# force hard-float for kernel modules
AC_SUBST([XENOMAI_KERNEL_THREADS_RTFLAGS])
AC_SUBST([XENOMAI_KERNEL_THREADS_LDFLAGS])
AC_SUBST([XENOMAI_KERNEL_THREADS_RTS])
AC_SUBST([XENOMAI_KERNEL_THREADS_KERNEL_DIRS],[$xenomai_kernels])
AC_SUBST([XENOMAI_KERNEL_THREADS_EXTRA_SYMBOLS])
# in-kernel math options
AC_SUBST([XENOMAI_KERNEL_THREADS_KERNEL_MATH_CFLAGS])
AC_SUBST([XENOMAI_KERNEL_THREADS_RTDIR],[])
AC_SUBST([XENOMAI_KERNEL_THREADS_EXTRA_KMODS],
	"_cfg_flavor_extra_kmods(xenomai-kernel)")
```
so it has to do with the actual build which originated it.

ArcEye, 2016-09-28:
> Oh...if you're messing with the comp command, you might fix the usage documentation. The short command line switches don't match the long version. Otherwise, I'll fix it sometime.

It is probably my error, I changed the switches for comp and instcomp, when we removed man page production and doubtless missed something - will do.

cdsteinkuehler, 2016-09-28:
This is on a fresh Machinekit Jessie build:
````
machinekit@beaglebone:~/machinekit/Tecnica/pru_custom$ which comp
/usr/bin/comp
machinekit@beaglebone:~/machinekit/Tecnica/pru_custom$ dpkg -S /usr/bin/comp
machinekit-dev: /usr/bin/comp
machinekit@beaglebone:~/machinekit/Tecnica/pru_custom$ dpkg -l machinekit-dev
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                               Version                Architecture           Description
+++-==================================-======================-======================-=========================================================================
ii  machinekit-dev                     0.1.1474450922-1mk.tra armhf                  PC based motion controller for real-time Linux
````

ArcEye, 2016-09-28:
I have just downloaded the .deb and confirmed that it is set for posix only.

This will take a bit of thought and experimentation, I don't have access to the Travis docker builds to see the configuration details there.

It is possible that the build is kernel agnostic, defaulting to posix, but this is having an unforeseen effect on 'out of tree' builds with other kernels.

ArcEye, 2016-09-28:
This appears to be confirmed by https://travis-ci.org/machinekit/machinekit/jobs/159413628
(a Travis log from a recent build for Jessie armhf)
```
./debian/configure -p
debian/control:  copied base template
debian/rules:  copied base template
debian/control:  added POSIX threads package
debian/rules:  enabled posix threads
.......
cd src && env  \
    ./configure --prefix=/usr \
    --build=arm-linux-gnueabihf \
    --host=arm-linux-gnueabihf \
    --with-posix \
    --without-rt-preempt \
    --without-xenomai \
    --without-xenomai-kernel \
    --without-rtai-kernel \
```

ArcEye, 2016-09-28:
I have built a RIP clone of MK, on a partition running a rt-preempt kernel.
I used the configure switches as previous, specifically selecting posix and excluding other kernels.

If I build a new component, it is installed to `.../rtlib/posix` not `.../rtlib/rt-preempt`

The output of flavor is `posix` despite the running rt-preempt kernel.

If I build for rt-preempt, a new component is installed to `.../rtlib/rt-preempt` and flavor returns `rt-preempt`

This is as close as I can get to reproducing and showing the situation at present.

ArcEye, 2016-09-29:
If these lines in debian/rules.in are removed
https://github.com/machinekit/machinekit/blob/master/debian/rules.in#L215
```
#	# delete files that should be in machinekit-dev package
	rm -f debian/machinekit/usr/bin/comp
	rm -f debian/machinekit/usr/share/linuxcnc/Makefile.modinc
	rm -f debian/machinekit/usr/share/linuxcnc/Makefile.inc
```
each Machinekit package for a particular flavour will contain comp, plus a Makefile.inc and Makefile.modinc which relate to that flavour.

This should eliminate at source the potential for these problems.

Having comp still requires headers and some extra libs, so the larger question is, do we actually need a 
machinekit-dev package?

We are just following an old practice from Linuxcnc, which originates from the far more restricted HDD space that prevailed and very few users having any need for its contents.

Our project users are arguable far more technical and likely to need the tools, plus storage technology has moved on to an extent that even a palm sized ARM board, commonly is used with a 16GB SD card, more disk space than generally assumed to be available on a desktop in the days of Ubuntu 6 or 8 based distros.

ArcEye, 2016-09-29:
The reason why this problem only occurs with xenomai armhf installs, appears to be because armhf machinekit-dev package alone is built with only posix threads enabled (both mainstream and raspbian)

Both amd64 and i386 are built with
```
HAVE_POSIX_THREADS=yes
HAVE_RT_PREEMPT_THREADS=yes
HAVE_XENOMAI_THREADS=yes
```

This is either a mistake, or more likely a requirement of the cross compilation / the platform it is built on.

Perhaps another reason to do away with the machinekit-dev package altogether by incorporating it into the main machinekit-flavour package.

@mhaberler @machinekoder @luminize 
do you have any views / insights on this?

ArcEye, 2016-09-29:
> @cdsteinkuehler 
Oh...if you're messing with the comp command, you might fix the usage documentation. The short command line switches don't match the long version. Otherwise, I'll fix it sometime.

You are going to have to be specific, struggling to find the problem
https://github.com/machinekit/machinekit/blob/master/src/hal/utils/comp.g#L1008

The short switches in the Usage print match those in the getopts, or is it something else you are referring to?

cdsteinkuehler, 2016-09-29:
On 9/29/2016 4:42 AM, ArcEye wrote:
> @cdsteinkuehler <https://github.com/cdsteinkuehler>
>     Oh...if you're messing with the comp command, you might fix the usage
>     documentation. The short command line switches don't match the long [...]

Usage:
%(name)s [--compile -c|--preprocess -p|--document -d|--view-doc -v]
compfile...

This shows no short option for --compile, -c or --preprocess, -p or
--document, -d or --view-doc.

The short switches are offset by one.

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-09-29:
On 9/29/2016 10:19 AM, ArcEye wrote:
> Perhaps another reason to do away with the machinekit-dev package
> altogether by incorporating it into the main machinekit-flavour package.
>
> @mhaberler <https://github.com/mhaberler> @machinekoder [...]
@ArcEye, I'm afraid i'm of little use here, sorry

ArcEye, 2016-09-29:
> This shows no short option for --compile, -c or --preprocess, -p or --document, -d or --view-doc. The short switches are offset by one.

We are obviously at cross purposes somewhere, I can't see a problem with the short switches

```
mick@INTEL-i7:/usr/src/machinekit$ comp -h
comp: Build, compile, and install Machinekit HAL components

Usage:
           comp [--compile -c|--preprocess -p|--document -d|--view-doc -v] compfile...
    [sudo] comp [--install -i|--install-doc -j] compfile..
           comp --compile --userspace cfile...
    [sudo] comp --install --userspace cfile...
    [sudo] comp --install --userspace pyfile...
           comp --print-modinc

mick@INTEL-i7:/usr/src/machinekit$ comp -c depth.comp
Compiling realtime depth.c
Linking depth.so

mick@INTEL-i7:/usr/src/machinekit$ comp -p depth.comp 

mick@INTEL-i7:/usr/src/machinekit$ comp -d depth.comp 

mick@INTEL-i7:/usr/src/machinekit$ comp -i depth.comp 
Compiling realtime depth.c
Linking depth.so
cp depth.so /usr/src/machinekit/rtlib/rt-preempt/

mick@INTEL-i7:/usr/src/machinekit$ comp -j depth.comp 
INSTALLDOC /usr/src/machinekit/man/man9/depth.asciidoc
```

cdsteinkuehler, 2016-09-29:
On 9/29/2016 5:56 AM, ArcEye wrote:
>     This shows no short option for --compile, -c or --preprocess, -p or
>     --document, -d or --view-doc. The short switches are offset by one.
> 
> We are obviously at cross purposes somewhere, I can't see a problem with the 
> short switches

I think it is how I am reading things.

I see "--compile -c|--preprocess ...", and I read that as "use either
-c or --preprocess", when the -c short switch is really supposed to be
joined with the previous --compile and not the trailing |--process.

I guess things could be clearer, but when interpreted correctly there
are no mistakes.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-09-29:
OK, I am with you now, the bar is supposed to indicate only one switch can be used at a time, by dividing the switches.
I'll have a think about making it clearer.

dkhughes, 2016-09-29:
Just a thought, but the usage line could look like:

```
Usage:
          comp [option] compfile ...
Options:
            -c, --compile Compile a comp...
            ...
```

ArcEye, 2016-09-29:
The usage print came from linuxcnc, which does not give the short options at all 
( as well as some other undocumented switches. )
I added the short options largely because no-one could guess that `-j` was the same as `--install-doc`.
I didn't anticipate the bars being misconstrued as 'OR', just a divider between switch pairs  :smile: 

Hopefully parenthesis around the short option and extra spaces will serve to make the eye see them as alternatives to the long version
```
comp: Build, compile, and install Machinekit HAL components

Usage:
           comp [ --compile (-c) | --preprocess (-p) | --document (-d) | --view-doc (-v) ] compfile...
    [sudo] comp [ --install (-i) | --install-doc (-j) ] compfile..
           comp --compile --userspace cfile...
    [sudo] comp --install --userspace cfile...
    [sudo] comp --install --userspace pyfile...
           comp --print-modinc
```
I will do a PR when I have sorted another matter that seems to have arisen regards comp, which previously worked fine.

cdsteinkuehler, 2016-09-29:
On 9/29/2016 8:32 AM, ArcEye wrote:
> 
> Hopefully parenthesis around the short option and extra spaces will serve to 
> make the eye see them as alternatives to the long version
> 
> |comp: Build, compile, and install Machinekit HAL components Usage: comp [ 
> --compile (-c) | --preprocess (-p) | --document (-d) | --view-doc (-v) ] 

That is *WAY* more readable to me.  Both the parenthesis and the space
around the | help to make the meaning clearer.

-- 
Charles Steinkuehler
charles@steinkuehler.net

darenschwenke, 2016-09-29:
Closing #873 as this is the same issue, and this one is active.

ArcEye, 2016-09-30:
I have come up with a way to test my hypothesis https://github.com/machinekit/machinekit/issues/1060#issuecomment-250401337

I have amended the debian/ setup, so that the files included into machinekit-dev.deb currently, get pulled into the machinekit-{flavour}.deb instead

This should result in a Makefile.inc that has `HAVE_XENOMAI_THREADS=yes` set for a Xenomai build and all the required files for building components are part of the base machinekit-{flavour}.deb package

An easy way to test this, without getting involved in setting up a new qemu within docker to build armhf  etc, is to use the existing build setup on a PR which makes this change.
( I will leave the creation of a machinekit-dev.deb package, because there will be a deployment script that will otherwise fail if it is missing, however it will just have a changelog in it. )

Once the packages are built, another PR removing the debian/ changes can be merged and then we have a set of packages of a known serial No and gitref to try out, whilst anyone updating will get the newer stock packages.

If they work and the xenomai package solves the `comp --install` problem, then I need to get access to the docker scripts etc. to remove the references to machinekit-dev deployment prior to submitting a final PR.

@cdsteinkuehler what do you think?
The PR's are at
https://github.com/ArcEye/machinekit/tree/machinekit-dev-test
https://github.com/ArcEye/machinekit/tree/machinekit-dev-test-reversal

When you have time to test the resultant package on a xenomai armhf system, let me know.

cdsteinkuehler, 2016-09-30:
On 9/30/2016 6:56 AM, ArcEye wrote:
> 
> @cdsteinkuehler <https://github.com/cdsteinkuehler> what do you think?

Sounds like a good plan!

> When you have time to test the resultant package on a xenomai armhf system, let 
> me know.

I can test anytime through the weekend, then I'll be out of town and
it will be difficult to test.

It looks like you need to make a PR (but don't merge) to the main
machinekit repo to kick off a build?

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-09-30:
> It looks like you need to make a PR (but don't merge) to the main machinekit repo to kick off a build?

Yes, I will do a PR now, it will take 30 mins to get checked and then another 1 hr approx after merge, before the packages hit the repo.

Once we have confirmed the packages, I can PR the reversal.

The only reason I don't just revert, is I am unsure what happens to the packages from a reverted merge.
Don't want to find they are deleted!

cdsteinkuehler, 2016-09-30:
On 9/30/2016 8:39 AM, ArcEye wrote:
> 
> The only reason I don't just revert, is I am unsure what happens to the packages 
> from a reverted merge.
> Don't want to find they are deleted!

I'm also unsure what happens if several packages try to install the
same file.  That may be why the files became part of the -dev package
in the first place.  It's also unclear to me why the -dev package is
built with different settings than the main packages (which are built
will all RT flavors defined).

...I guess we'll see what happens.  :)

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-09-30:
I think the problem is how the packages are built.  The x86 builds generate all flavors at once, but the arm builds are one at a time (I believe due to limits on Travis run-time).  See the matrix in the .travis-yml file: https://github.com/machinekit/machinekit/blob/master/.travis.yml#L32-L47

I'm not sure what to do about this...

ArcEye, 2016-09-30:
> I'm also unsure what happens if several packages try to install the same file. That may be why the files became part of the -dev package in the first place.

Should it become an issue, I can simply add a preinst script that deletes the common files, should they exist. 

> It's also unclear to me why the -dev package is built with different settings than the main packages (which are built will all RT flavors defined).

Only the Xenomai armhf package is built that way and I suspect that has to do with the limitations of building inside a VM in a docker, just using posix does not add further layers of difficulty to the build.

The `build-deb` tests have already succeeded, so there should be no surprises hopefully
https://travis-ci.org/machinekit/machinekit/builds/164019437
approx 6 minutes to go

I am out for an hour and will check back on the main build then

cdsteinkuehler, 2016-09-30:
On 9/30/2016 9:07 AM, ArcEye wrote:
> It's also unclear to me why the -dev package is built with different
>     settings than the main packages (which are built will all RT flavors defined).
>
> Only the Xenomai armhf package is built that [...]

So digging into the Travis-CI config, the ARM packages are built one
flavor at a time.  So if the makefiles are installed for one flavor,
the other flavors will not work.  IIRC, the issue is a full build of
all flavors for ARM (as is done on x86) exceeds the Travis-CI maximum
single job time of 50 minutes.

I think the real fix is to either get the full ARM build under the 50
minute limit (which probably requires switching to builds using a
cross-compiler...a non-trivial task) or adding a build target that can
*JUST* build the machinekit-dev package with all RT flavors defined
(hopefully easy?).

Thoughts?

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-09-30:
> The build-deb tests have already succeeded, so there should be no surprises hopefully......

Famous last words, build failed on machinekit-posix, which is peculiar.

Not only have I already built this locally, but the files it says are missing are the same ones it managed to find for rt-preempt and xenomai.

I will look again with fresh eyes tomorrow

ArcEye, 2016-09-30:
> I think the real fix is to either get the full ARM build under the 50 minute limit (which probably requires switching to builds using a cross-compiler...a non-trivial task) or adding a build target [...]

The 50 minute issue rings a bell now you mention it.

I need to get access to the Travis config scripts, I have it for Jenkins but not Travis as far as I am aware.
Simply changing the configure switches to include the other flavours may be the answer, if it builds OK.

cdsteinkuehler, 2016-09-30:
You should have full access to the Travis config, which is in the Machinekit repo:
https://github.com/machinekit/machinekit/tree/master/.travis

I've been reviewing cross building, and it may not be much harder than the qemu emulation we're currently running under...and if it works it would be *MUCH* faster.  I'll play around with this some as time allows.

ArcEye, 2016-09-30:
>  You should have full access to the Travis config, which is in the Machinekit repo:
https://github.com/machinekit/machinekit/tree/master/.travis

Last time I looked, there was no clue as to how $FLAV was set only to posix for machinekit-dev when building for armhf, that should only happen for PRs.
I'll look again tomorrow

>  I've been reviewing cross building, and it may not be much harder than the qemu emulation we're currently running under...and if it works it would be MUCH faster. I'll play around with this some as time allows.

If you could get that going, could be a big improvement

zultron, 2016-09-30:
I'm actually working on Travis CI builds right now for another reason, and some of these details are fresh in my head.  Here are several ways of reducing the `armhf` build time to fit within the Travis CI limits.

### ccache

Travis CI now allows ccache persistence on free accounts.  The trouble is ccache won't help in some cases (e.g. if no compiles hit the cache, the build could still go over time), so it's not a good general solution, even though it would help.

### cross-building

IMO, @cdsteinkuehler has the best approach:  revisit cross-building.

Debian has an elegant way to install packages needed for cross-building using Multiarch.  Last time I tried was a couple of years ago, and some of Machinekit's dependencies didn't have proper Multiarch support.  I think I managed to forcefully cram all the packages together once in Wheezy, but it was ugly.  Even if this has been fixed for Jessie, it won't help if Wheezy is still a build target.

The traditional cross-build setup, if not using the above Debian setup, needs these elements (I never quite got it working):
- Cross-compile tools installed; crib from the [`mk-builder`][2] scripts to install emdebian
- The host-arch libs installed in a directory somewhere; this might be done trivially with `multistrap`, also as seen in `mk-builder`
- The correct incantation for `./configure`

### monkey patching

If cross-building doesn't work, it might be worth trying to monkey-patch the chroot.  The `mk-builder` scripts do that for the compiler (see the scripts in `bin/`).  Even more simply, common executables like `bash` could be replaced with a script that sets `LD_LIBRARY_PATH` and calls the build-arch executable in e.g. the `/host-rootfs/usr/bin/` directory.  This could theoretically bring build times down pretty close to the native builds.

### rasing Travis CI limit

I just read a case where a project requested the build time limit be raised.  The Travis CI folks raised it to 90 minutes and said they'd raise it to 120 minutes if needed.

### Problem:  Docker builder images

One other problem:  Most any of these solutions will require modifications to the [`mk-builder` images][2], which can no longer be built on Docker Hub.  (The `ptrace()` call, needed by `proot`, is now prohibited in the container build environment, and [setting `PROOT_NO_SECCOMP=1`][1] doesn't help.)  I'm working on those scripts and the instructions right now to find the most optimal workaround, and I can let you know when they're done later today.

[1]: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1202161
[2]: https://github.com/machinekit/mk-builder/

zultron, 2016-10-01:
I've got a cross-building environment that's easy to set up and builds 99% of Machinekit.  When I figure out how to build the other 1%, I'll have a docker image and some simple instructions to share.

ArcEye, 2016-10-01:
>  I've got a cross-building environment that's easy to set up and builds 99% of Machinekit. When I figure out how to build the other 1%, I'll have a docker image and some simple instructions to share.

Simple will be good :laughing: 

My problem with docker and like systems, is that all that effort goes to producing something I don't use, ie. packages, so I have never progressed very far with it.

ArcEye, 2016-10-01:
Re https://github.com/machinekit/machinekit/issues/1060#issuecomment-250767945
It now builds and is ready for merge

I guessed that the PR build problem was that the armhf-posix deb build, does not use USER_PCI and I had commented out the test for it.
Didn't matter in a full local build, since it was defined.

Reinstated a test and the .install files conditionally install the include/linuxcnc/user_pci headers

ArcEye, 2016-10-01:
The packages are built, the serial and git ref is
...........0.1.1475326532-1mk.travis.master.git96a90a9f~1...........
I have copies in case of problems.

It should work OK, unless a user installs more than one kernel flavour, in which case the last to be installed would be reflected in the Makefile.inc flavour selections.

If this combined packaging method is used, that problem could be negated by putting the 
Makefile.(mod)inc files into a kernel flavour titled sub directory from /usr/share/linuxcnc and changing (inst)comp to fetch kernel flavour and act accordingly.

I will prepare a fresh reversal PR

zultron, 2016-10-02:
@ArcEye, if you have this licked, I'll lay off the cross-build question.

As for packaging, there's still going to be a problem with a number of libraries that aren't yet `Multi-Arch` and some tools like python, cython and tcl/tk.  The libraries are trivially fixed, and some I looked at already are in Stretch.  The tools aren't trivially fixed, but I see some experimental support for python that could get into Stretch (I wonder if that'll ever come for tcl/tk).

For RIP die-hards, it's actually pretty easy to get a full cross-build environment working.  I'd done this when I boasted about the 99%, myself forgetting about packages, even though that's the point of this issue:  to fix Travis CI.

Apologies for the diversion!

ArcEye, 2016-10-02:
> @ArcEye, if you have this licked, I'll lay off the cross-build question.

I merely have a work-around which hopefully will cure this one issue.
A proper package cross-build environment for armhf would still be the optimum solution if it can be done.

cdsteinkuehler, 2016-10-02:
On 10/1/2016 10:55 PM, John Morris wrote:
> @ArcEye <https://github.com/ArcEye>, if you have this licked, I'll lay off the
> cross-build question.
>
> As for packaging, there's still going to be a problem with a number of libraries
> that aren't [...]

Being able to run a full build on Travis-CI under the 50 minute time
limit is still very important, IMHO.  If the tools are a problem, what
about a mixed approach, where the C files are built using a cross
compiler, but everything else uses "native" ARM tools via qemu as
we're doing now.

IIRC, there is a full ARM rootfs on the Docker build instance, meaning
the armhf libraries and everything are already there.  Can we maybe do
something as simple as install an ARM cross-compiler (the compiler is
an x86 binary that generates ARM code) and defining CC=?

-- 
Charles Steinkuehler
charles@steinkuehler.net

zultron, 2016-10-03:
On 10/02/2016 09:15 AM, cdsteinkuehler wrote:
> Being able to run a full build on Travis-CI under the 50 minute time
> limit is still very important, IMHO. If the tools are a problem, what
> about a mixed approach, where the C files are built using a [...]

The [Docker images][1] that Travis CI runs already have the 
cross-compilers; see the `bin/` directory in that repo.  This approach 
can be extended to use even more build-arch binaries during the build 
process.  If I went that way, I'd try to figure out which executables 
were using the most CPU time and address those first.

I'm actually trying out one more thing before giving up on package 
cross-builds.  I'll report any positive results in the next day or so, 
and otherwise assume I've given up.  ;)

I suppose it's time for a new issue to collect this stuff?

[1]: https://github.com/machinekit/mk-builder

cdsteinkuehler, 2016-10-04:
On 10/3/2016 5:32 PM, John Morris wrote:
>
> I'm actually trying out one more thing before giving up on package
> cross-builds. I'll report any positive results in the next day or so,
> and otherwise assume I've given up. ;)

<fingers crossed!>

> I suppose it's time for a new issue to collect this stuff?
>
> [1]: https://github.com/machinekit/mk-builder

That makes sense, particularly if you get positive results and it seems 
likely we can get cross builds working.  If you post anything over 
there, include a breadcrumb linking back to this issue.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2017-07-03:
The build issue is hopefully solved by #1230 

Testing on a replacement cross-build system continue, using @zultron 's mk-cross-builder system

closed 2017-07-03 by ArcEye