machinekit/machinekit #1067: comp is broken with userspace components

cdsteinkuehler, 2016-10-03:
I have had a couple of problems with using the comp command to compile a userspace component.

The first problem is with include files.  Any include file in the local directory is not found by the comp command when actually doing the compile.  This can be worked around by first running comp -p --userspace <compfile> to generate a *.c file, then building the *.c file with the command line spit out by the failed "comp --compile --userspace <compfile>" command.

The second problem is the resulting file crashes when accessing HAL pins for a singleton component.  This appears to be a result of how things are setup by default for an array of components.  Specifically, in the initialization code export() routine, __comp_first_inst and __comp_last_inst are defined.  But the pin access #define is setup to use the __comp_inst static, not __comp_first_inst or __comp_last_inst.  When the user_mainloop function is executed, nothing has set the __comp_inst static, so any references to HAL I/O pins results in a null pointer dereference.

If you copy __comp_first_inst to __comp_inst as the first part of the user_mainloop function, the component runs as expected:

````
machinekit@beaglebone:~/badcomp$ ls -l
total 8
-rw-r--r-- 1 machinekit machinekit 267 Oct  3 00:16 badcomp.comp
-rw-r--r-- 1 machinekit machinekit  21 Oct  2 23:55 badcomp.h
machinekit@beaglebone:~/badcomp$ comp --compile --userspace badcomp.comp
arm-linux-gnueabihf-gcc -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__   -I/usr/include/linuxcnc -fPIC -g -funwind-tables -URTAPI -U__MODULE__ -DULAPI -Os  -o badcomp /tmp/tmpKU0Dog/badcomp.c -Wl,-rpath,/usr/lib -L/usr/lib -llinuxcnchal
badcomp.comp:9:21: fatal error: badcomp.h: No such file or directory
compilation terminated.
Makefile:2: recipe for target 'badcomp' failed
make: *** [badcomp] Error 1
machinekit@beaglebone:~/badcomp$ comp --preprocess --userspace badcomp.comp
machinekit@beaglebone:~/badcomp$ arm-linux-gnueabihf-gcc -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__   -I/usr/include/linuxcnc -fPIC -g -funwind-tables -URTAPI -U__MODULE__ -DULAPI -Os  -o badcomp badcomp.c -Wl,-rpath,/usr/lib -L/usr/lib -llinuxcnchal
machinekit@beaglebone:~/badcomp$ ./badcomp
Segmentation fault
machinekit@beaglebone:~/badcomp$ realtime stop
machinekit@beaglebone:~/badcomp$ realtime start
machinekit@beaglebone:~/badcomp$ sed -i '/mainloop/s/$/ __comp_inst = __comp_first_inst;/' badcomp.comp
machinekit@beaglebone:~/badcomp$ comp --preprocess --userspace badcomp.comp
machinekit@beaglebone:~/badcomp$ arm-linux-gnueabihf-gcc -I. -pthread -DTHREAD_FLAVOR_ID=0 -DRTAPI -D_GNU_SOURCE -D_FORTIFY_SOURCE=0 -D__MODULE__   -I/usr/include/linuxcnc -fPIC -g -funwind-tables -URTAPI -U__MODULE__ -DULAPI -Os  -o badcomp badcomp.c -Wl,-rpath,/usr/lib -L/usr/lib -llinuxcnchal
machinekit@beaglebone:~/badcomp$ ./badcomp
^C
machinekit@beaglebone:~/badcomp$

````

[badcomp.h.txt](https://github.com/machinekit/machinekit/files/505334/badcomp.h.txt)
[badcomp.comp.txt](https://github.com/machinekit/machinekit/files/505333/badcomp.comp.txt)

ArcEye, 2016-10-03:
Nothing to do with any changes I made :laughing: 

There are 2 issues.

The local include failure is strange, because the gcc stanza contains `-I.`
I can only surmise that because the compilation is directed to /tmp, it is looking there and not finding the header. 
I will experiment with that a bit later, it finds the header if moved to the RIP includes.

Regards the __comp_inst, everything works as intended if you use the FOR_ALL_INSTS() macro, irrespective of how many instances there are

eg.
```
component badcomp "Fails to compile with comp";
pin out float out_;
pin in  float in_;
option userspace;
option singleton;
license "public domain";
;;

#include "badcomp.h"

static void user_mainloop(void).
{
    while(true).
	{
	FOR_ALL_INSTS()
    	    {
    	    out_ = in_ + increment;
    	    sleep(1);
            }
	}
}
```
Results in

```
mick@INTEL-i7:/prog/EMC/Components/badcomp$ DEBUG=5 realtime restart
mick@INTEL-i7:/prog/EMC/Components/badcomp$ halcmd loadusr -W ./badcomp
<commandline>:0: Component 'badcomp' ready
<commandline>:0: Program './badcomp' started
mick@INTEL-i7:/prog/EMC/Components/badcomp$ halcmd show pin badcomp
Component Pins:
  Comp   Inst Type  Dir         Value  Name                                             Epsilon         Flags
    74        float IN              0  badcomp.in                               	0.000010	0
    74        float OUT             2  badcomp.out                              	0.000010	0

mick@INTEL-i7:/prog/EMC/Components/badcomp$ realtime stop
<commandline>:0: Realtime threads stopped
```
I have always used this method for userspace components, singleton or otherwise, as if you don't you have to address the __comp_state directly, because the convenience macros don't work.

cdsteinkuehler, 2016-10-03:
I realize if I used the FOR_ALL_INSTS() macro things would work, but my actual userspace component is much more complex than the simple example shown.  Rather than just assign an output, my user_mainloop function does a bunch of setup (allocates memory, opens a network socket, sets up signal handlers, and does other stuff I never want to do more than once).  I think simply initializing the __comp_inst value in the export() function (where __comp_first_inst and __comp_last_inst are assigned) would fix the problem for the singleton case and not break anything if there are multiple instances.

If you can help figure out the problem with the include issue, that would be great!  Oddly, includes seem to work OK with realtime comp files, it's just userspace files that are broken.

cdsteinkuehler, 2016-10-03:
It looks like the problem with the includes is related to some extra flags missing from the userspace build.  I think these need to be added to the build_usr() function: https://github.com/machinekit/machinekit/blob/master/src/hal/utils/comp.g#L676-L677

I'll try to put together a PR

ArcEye, 2016-10-03:
>  Rather than just assign an output, my user_mainloop function does a bunch of setup (allocates memory, opens a network socket, sets up signal handlers, and does other stuff I never want to do more than once).

I think I have always done this before the while() loop in user_mainloop(), or used the EXTRA_SETUP() macro to initialise things before user_mainloop() is called.

Your PR works whether FOR_ALL_INSTS() is used or not, so fixes it nicely

closed 2016-10-03 by ArcEye