machinekit/machinekit #1072: Reduce Machinekit `armhf` architecture build times

zultron, 2016-10-06:
Current approaches to building ARM executables on Intel architecture rely heavily on emulation and take many times longer than a native build.  This is especially visible in the Machinekit automated builds on x86 architecture Travis CI build hosts.  In order to finish within the Travis 50 minute build limit, the armhf package builds were split into three (one for each RT thread flavor).  By comparison, native x86 builds take just 12 minutes.

In [another issue][1] I outlined some approaches to speeding up ARM builds.  Among those, the best option would be cross-building, if it could be made to work.  The advantages:

- Shortest build times:  same as native builds
- Future-proof:  Multi-Arch awareness is a goal of the Debian distribution, and cross-building will have better and better support over time
- No special OS requirements:  e.g. `ptrace()`, needed by `proot` in the current emulation method, is now disallowed in the Docker Hub image build environment, making it difficult to build the ARM-arch chroot image
- Machinekit is ready for cross-building:  the build system has already had a great deal of work to be cross-compile-ready

The disadvantage is Debian Multi-Arch support is not yet complete.  In Jessie, most core packages have been converted, but not all.  Although this situation will improve over time, currently, Machinekit depends on a number of packages without Multi-Arch support, and it is difficult to install all the needed packages in the correct architectures at once.  Working around this will be the main challenge in setting up the cross-build environment.

My initial experiments show that with some force, Machinekit can be cross-compiled against Jessie packages.  Travis package builds take 10 minutes on Travis CI (4 minutes with `ccache` enabled).  In RIP build tests, the build result could be transferred to an ARM architecture host and all regression tests pass.

This issue is to track this particular effort.  I will follow up soon with details for reproducing these initial results.

[1]: https://github.com/machinekit/machinekit/issues/1060#issuecomment-250810451

cdsteinkuehler, 2016-10-07:
On 10/6/2016 5:23 PM, John Morris wrote:
> My initial experiments show that with some force, Machinekit can be
> cross-compiled against Jessie packages. Travis package builds take 10
> minutes on Travis CI (4 minutes with |ccache| enabled). In [...]

Wow!  Great progress John!

I'm off-line for the next 4 days, but look forward to your setup 
details, and I should have some time to help get things working when I 
return next week.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-10-07:
That is a huge improvement.

Still look forward to following the 'simple' instructions to replicate :laughing:

zultron, 2016-10-07:
Thanks, @cdsteinkuehler.  @ArcEye, here are the 'simple' instructions; the complexity is hidden in the Dockerfile.  The instructions depend on changes in #1074.

The Dockerfile that builds the cross-build environment is in my [`mk-builder-2` repo][1].  That image is on the Docker hub, so in theory you should be able to `cd` to your Machinekit source tree and spin up the Docker container.  The current (MK source) directory must be bind-mounted, and I find it's useful to also bind-mount `$HOME`.  The `mk-builder.sh` script from that repo will do that, or try something like the following:

    docker run --rm -it	-u `id -u`:`id -g` \
        -v $PWD:$PWD -v $HOME:$HOME -e HOME=$HOME -w $PWD \
        zultron/mk-builder-2

From the shell prompt inside the Docker container, you should be able to cross-build either Machinekit packages or an RIP config:

    # RIP
    ./autogen.sh && ./configure --host=arm-linux-gnueabihf
    make -j4 && sudo make setuid

    # Packages
    debian/configure -prxt 8.6
    dpkg-buildpackage -uc -us -a armhf -B -d

The container is *not* suited to run the regression tests.  To transfer the cross-built Machinekit sources to an ARM host for running tests, be sure to unpack them in the exact same directory location, since the RIP build hard-codes a lot of absolute paths.

[1]: https://github.com/zultron/mk-builder-2

ArcEye, 2016-10-08:
> ... here are the 'simple' instructions; the complexity is hidden in the Dockerfile. The instructions depend on changes in #1074.

Thanks, I will give it a spin a bit later

ArcEye, 2016-10-08:
I have learnt a lot about docker in a couple of hours!

I am building on amd-64 - kernel / lsb results from inside container are:
```
2f66dbe10be7:libexec$ uname -r
3.16.0-4-amd64
2f66dbe10be7:libexec$ grep "VERSION=" /etc/os-release |awk -F= {' print $2'}|sed s/\"//g
8 (jessie)
```

I thought I was doing things wrong and tried this with a bind-mounted local copy of Machinekit and with a modified docker image, with Machinekit preloaded into it, but results are the same.

If you build a RIP with `./configure --host=arm-linux-gnueabihf` it works and the build completes.

I checked what had actually been produced, and they look like Arm binaries and libs - success!!
```
2f66dbe10be7:libexec$ file flavor
flavor: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=137cb33de1e200ca86e1076f2c732170aced5e18, not stripped
2f66dbe10be7:rtlib$ file ulapi-posix.so
ulapi-posix.so: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=97c655c74d84ddb49f137c14bff14fc47d23ec73, not stripped
```

However running
```
debian/configure -prxt 8.6
dpkg-buildpackage -uc -us -a armhf -B -d
```
triggers an error from configure, both on local files and on the MK repo clone inside the container
```
checking for boostlib >= 1.34... yes
checking whether the Boost::Serialization library is available... yes
configure: error: Could not link against boost_python !
debian/rules:73: recipe for target 'build-stamp' failed
make: *** [build-stamp] Error 1
dpkg-buildpackage: error: debian/rules build-arch gave error exit status 2
```

I will continue trying some hacks to get it built, currently even discarding the detected
`--host=$(DEB_HOST_MULTIARCH) `
and replacing with
`--host=arm-linux-gnueabihf`
in debian/rules.in, does not fix it.

I am probably doing something wrong, but cannot find it :cry:

ArcEye, 2016-10-08:
I have tried all sorts of changes to rules.in and got nowhere

Got configure to get past the libboost_python issue and finish a few times, but then debian/rules errored, being unable to resolve `--with-posix`
(or `--without-posix` if you remove it from the configuration).

This was all from a container where I cloned a copy of my repo of MK and then made changes and pulled them to the container, because it seems impossible to run an editor within the container.

I found if I used a bind-mounted local repo directory, the changes I made to that directory were never reflected in the file system seen from the container. 

My head is scrambled now - giving up.

zultron, 2016-10-08:
Hmm, sorry @ArcEye, I'll go through my instructions again and make sure they work.  I'm touching too many things at once.  Thanks for trying.

(Docker might be useful to you even if you don't do packages.  I've been using it on my laptop for a year now so I can run Trusty but still develop on Lucid.  :D )

I might've found a more general way of doing this that would work for Wheezy too. I'll update the issue soon.

ArcEye, 2016-10-09:
>  Hmm, sorry @ArcEye, I'll go through my instructions again and make sure they work. I'm touching too many things at once. Thanks for trying.

Don't apologise, it is a great step forward to be able to build x86 ARM binaries, on a amd64 Intel platform.  
They build at the same speed as my normal builds, all done in just over a minute with 8 cores engaged.

The debpackage problem must be some kind of environment hiccup.
When I got it past one problem, it was then not finding czmq libs etc, all of which were perfectly discoverable to previous RIP builds, which also used configure.

I will have another go later and create a new user account to do it, just to ensure none of my environment is interfering.

ArcEye, 2016-10-09:
> I will have another go later and create a new user account to do it, just to ensure none of my environment is interfering.

Same problem as before, builds a RIP fine, but configure falls over at libboost_python when building a package.

Regards users, there are some peculiarities.

I set up the Dockerfile with my user UID and a GID of one of my groups.
However docker will not run for an ordinary user, so you have to use sudo.
This means that having launched with `sudo sh ./mk-builder.sh`, when you do `cd ~/`, irrespective of where you launch it from, you will get the `/root` dir not your users home dir.

This also affects the environment variables.

Doing `printenv` from inside my container results in
```
ZDEBUG=1
HOSTNAME=mk-builder
DEBCONF_NONINTERACTIVE_SEEN=true
SHELL=/bin/bash
TERM=dumb
ALTERAOCLSDKROOT=/root/altera/15.1/hld
KERNEL=x86_64
QT_INSTALL_PLUGINS=/opt/Qt/5.4/gcc_64/plugins
QSYS_ROOTDIR=/root/altera/15.1/quartus/sopc_builder/bin
QTDIR=/opt/Qt/5.4/gcc_64
INCLUDE_PATH=/usr/include:/usr/local/include:/usr/local/include/machinekit:/opt/Qt/5.4/gcc_64/include:
LC_ALL=C.UTF-8
USER=mick
LD_LIBRARY_PATH=/usr/lib:/usr/local/lib:/usr/local/lib/machinekit:/opt/Qt/5.4/gcc_64/lib:/opt/Qt/Tools/QtCreator/lib:/opt/Qt/5.4/gcc_64/qml/Machinekit/Application:
USE_ZMQ=1
QTCREATEDIR=/opt/Qt/Tools/QtCreator
ENV=/root/.bashrc
USERNAME=root
QTVERSION=5.4
PATH=/sbin:/usr/sbin:/bin:/usr/bin:/usr/X11R6/bin:/usr/local/bin:/opt/Qt/5.4/gcc_64/bin:/opt/Qt/Tools/QtCreator/bin
GID=1050
PWD=/
LANG=C.UTF-8
SHLVL=1
HOME=/root
LANGUAGE=C.UTF-8
UID=1000
DEBIAN_FRONTEND=noninteractive
DISPLAY=:0.0
```
which is a mixture of the env setting for root on my local system ( which includes Qt5 and the Altera / Quartus FPGA building system ) and the env vars set in the docker.
Note $USERNAME=root,  $USER=mick

This is not the cause of the problem, I swapped my root dir for a bare one from another partition which contained no env vars in .bashrc and the result was the same.

zultron, 2016-10-09:
On 10/09/2016 09:36 AM, ArcEye wrote:
>     I will have another go later and create a new user account to do it,
>     just to ensure none of my environment is interfering.
>
> Same problem as before, builds a RIP fine, but configure falls over at
> libboost_python when building a package.

I have fixes related to Boost::Python in my tree.  I'll push a branch 
today or tomorrow.

> Regards users, there are some peculiarities.
>
> I set up the Dockerfile with my user UID and a GID of one of my groups.
> However docker will not run for an ordinary user, so you have to use sudo.
> This [...]

Add yourself to the `docker` group and you should be able to run Docker 
as yourself.  Let me know if that fixes it.

ArcEye, 2016-10-09:
> Add yourself to the `docker` group and you should be able to run Docker as yourself. Let me know if that fixes it.

Ah, didn't realise there was a `docker` group, that makes more sense.
I'll try it once you push your updates

dkhughes, 2016-10-10:
@ArcEye Per users:

I'm not a fan of giving my normal user sudo access without requiring a password (or a reminder to me that I'm running something as root) and adding myself to the docker group essentially does just that when you share folders from the image to your local machine. So, in docker scripts I usually create a normal user with a matching uid:gid and then launch the docker image as that user. Wrapping it up into a couple of bash scripts makes docker a little easier for me to come back to after being away for a bit. User creation see:

[https://github.com/dkhughes/ubker-docker/blob/master/Dockerfile.in#L61](https://github.com/dkhughes/ubker-docker/blob/master/Dockerfile.in#L61)

And I launch it via this generated script which sets up the environment automatically:

[https://github.com/dkhughes/ubker-docker/blob/master/build_docker_image.sh#L35](https://github.com/dkhughes/ubker-docker/blob/master/build_docker_image.sh#L35)

We are using this template to crossbuild the uboot/kernels packages for the zynq work now.

Crossbuilding arm packages sounds great - all the trouble I hit with qemu made me go back to a native arm target, which is just another machine to maintain.

ArcEye, 2016-10-10:
@dkhughes 
Looks like a good system, I will 'borrow it with pride' (filtch it :smiley: )

The docker system works well, I must persevere with it.

I have had experience with qemu a long time ago, building programs and packages for the Sharp Zaurus.
Compiling was bad enough but simulated graphical running of programs to test them, inside a qemu virtual screen buffer was even worse as I recall. Took ages to get it right.

zultron, 2016-10-12:
@dkhughes, thanks for pointing that out.  I should be careful not to gloss over security risks in my recommendations!  Other than that, I'm doing something similar but different in my scripts, which make different assumptions while replicating parts of the external user environment within the Docker container.  Once my work matures a little (and I stop rebasing), I hope you'll review.

zultron, 2016-10-12:
I tried a (sorta) completely new approach to cross-building Debian packages in Docker.  See the [`mk-builder-3` repo][1] for details about how to use it.  It depends on changes in [this branch][4].

If this approach can be made to work, it may be the cleanest, fastest and most easily maintained way to cross-build Machinekit, whether on Travis CI or not.  It borrows @kinsamanka's `multistrap` to unpack (but not configure) build dependencies in `/sysroot/armhf`.  It tweaks a few symlinks and build tools, and installs the Emdebian cross-compile toolchain.  Then, the `README` gives the right incantations to run either RIP or package cross-builds against the headers, libraries and configs in the foreign architecture system root.

This is an improvement on the last version (of which hopefully @ArcEye was the only victim), which used a traditional `Multi-Arch:` configuration.  This method isn't subject to the build dependency resolution problems coming from the incomplete roll-out of `Multi-Arch:` in Jessie, since the packages are installed in a separate tree, not alongside the native architecture packages.  For the same reason, it may be immune to `Multi-Arch:` dependency problems building Raspbian and other `dpkg`-based OSs where packages may diverge from mainstream (even if just by a time lag).

I was able to do an RIP build, transfer the built source tree to a native ARM host, and successfully complete all regression tests (`posix` threads).  I was also able to successfully [build packages on Travis CI][2] (12 minutes).  That's all the testing I've had the time for so far, but it looks pretty promising.

[1]: https://github.com/zultron/mk-builder-3
[2]: https://travis-ci.org/Dovetail-Automata/emc2/builds/166951211
[3]: https://travis-ci.org/Dovetail-Automata/emc2/builds/166953587
[4]: https://github.com/zultron/machinekit/tree/mk-1072-xbuild-prep

ArcEye, 2016-10-12:
> I tried a (sorta) completely new approach to cross-building Debian packages in Docker. See the mk-builder-3 repo for details about how to use it. It depends on changes in this branch.

Well done @zultron 
I was able to build the packages using your repos. ( see [1] )

My work flow, for the benefit of anyone else trying it.

Create a new branch in your local repo and merge @zultron 's branch into it
```
# go to your machinekit dir
cd ~/src/machinekit
git checkout master
# update it
git pull main +master (main is what machinekit main repo is aliased to in my remote settings)
# create new branch from master
git checkout -b crossbuild
# add zultron to remotes and merge
git remote add zultron http://github.com/zultron/machinekit
git remote update
git merge zultron/mk-1072-xbuild-prep
```
clone the new mk-builder-3 repo, in the same parent dir as machinekit (for me it was ~/src)
change settings and build it
```
cd ~/src
git clone https://github.com/zultron/mk-builder-3.git
cd mk-builder-3

# open Dockerfile and change user / UID / GID see [1]
# open mk-builder.sh and change to
    # IMAGE=mk-builder
    # NAME=mk-builder
# else you will download @zultrons image not use your own

docker build -t mk-builder .
```
Then return to your machinekit repo 
```
cd ../machinekit
sh ../mk-builder-3/mk-builder.sh
```
and you have started the docker
```
mk-builder:machinekit $ debian/configure -prxt 8.6

# Using the equivalent of make -j8 reduced build to 3 mins
mk-builder:machinekit $ DEB_BUILD_OPTIONS=parallel=$(nproc) SYSROOT=$ARM_ROOT dpkg-buildpackage -uc -us -a armhf -B -d
```
The packages will be built into ../

[1]. 
There are still issues with users and permissions.
Setting up my user as a member of `docker` group, solved the issues with docker but created a peculiar issue when dh_builddeb came to write the packages
```
dh_builddeb
dpkg-deb: building package `machinekit-dev' in `../machinekit-dev_0.1.1-1_armhf.deb'.
dpkg-deb: error: unable to create `../machinekit-dev_0.1.1-1_armhf.deb': Permission denied
dh_builddeb: dpkg-deb --build debian/machinekit-dev .. returned exit code 2
dpkg-deb: building package `machinekit' in `../machinekit_0.1.1-1_armhf.deb'.
dpkg-deb: error: unable to create `../machinekit_0.1.1-1_armhf.deb': Permission denied
dh_builddeb: dpkg-deb --build debian/machinekit .. returned exit code 2
dpkg-deb: building package `machinekit-posix' in `../machinekit-posix_0.1.1-1_armhf.deb'.
dpkg-deb: error: unable to create `../machinekit-posix_0.1.1-1_armhf.deb': Permission denied
dh_builddeb: dpkg-deb --build debian/machinekit-posix .. returned exit code 2
dpkg-deb: building package `machinekit-rt-preempt' in `../machinekit-rt-preempt_0.1.1-1_armhf.deb'.
dpkg-deb: error: unable to create `../machinekit-rt-preempt_0.1.1-1_armhf.deb': Permission denied
dh_builddeb: dpkg-deb --build debian/machinekit-rt-preempt .. returned exit code 2
dpkg-deb: building package `machinekit-xenomai' in `../machinekit-xenomai_0.1.1-1_armhf.deb'.
dpkg-deb: error: unable to create `../machinekit-xenomai_0.1.1-1_armhf.deb': Permission denied
dh_builddeb: dpkg-deb --build debian/machinekit-xenomai .. returned exit code 2
debian/rules:215: recipe for target 'binary-arch' failed
make: *** [binary-arch] Error 1
dpkg-buildpackage: error: fakeroot debian/rules binary-arch gave error exit status 2
mick@mk-builder:/media/sde/sde7/home/mick/src/machinekit$
```

I found that the `../` it was trying to build to was not the parent dir (to which my user had full access)
, but an equivalent dir inside the docker, which for some reason it didn't

I built and ran the whole thing as root and confirmed that the packages were in fact written inside the docker
```
mk-builder:machinekit $ cd ../
mk-builder:src$ ls -l
total 6256
   4 drwxr-xr-x  3 root root    4096 Oct 12 10:38 ./
   4 drwxr-xr-x 12 root root    4096 Oct 12 10:27 ../
   4 drwxr-xr-x 24 root root    4096 Oct 12 10:37 machinekit/
 328 -rw-r--r--  1 root root  334192 Oct 12 10:38 machinekit-dev_0.1.1-1_armhf.deb
 432 -rw-r--r--  1 root root  440104 Oct 12 10:38 machinekit-posix_0.1.1-1_armhf.deb
 432 -rw-r--r--  1 root root  440080 Oct 12 10:38 machinekit-rt-preempt_0.1.1-1_armhf.deb
 432 -rw-r--r--  1 root root  442074 Oct 12 10:38 machinekit-xenomai_0.1.1-1_armhf.deb
   4 -rw-r--r--  1 root root    2654 Oct 12 10:38 machinekit_0.1.1-1_armhf.changes
4616 -rw-r--r--  1 root root 4724082 Oct 12 10:38 machinekit_0.1.1-1_armhf.deb
```
Then I just copied the debs to the machinekit dir, which is on the local machine
`mk-builder:src$ cp machinekit-*.deb machinekit `

I didn't try @dkhughes approach (there were just too many other things to get my head around :wink: )  that may have solved the permissions problem for the .deb writes.

If anyone recalls what the original issue was #1060 
The machinekit-dev package produced should cure this because `Makefile.inc` contains
```
HAVE_POSIX_THREADS=yes
HAVE_RT_PREEMPT_THREADS=yes
HAVE_XENOMAI_THREADS=yes
HAVE_XENOMAI_KERNEL_THREADS=no
HAVE_RTAI_KERNEL_THREADS=no
```
Great job.

ArcEye, 2016-10-12:
Fat finger syndrome :smile:

zultron, 2016-10-13:
@ArcEye I think it'll solve your permissions issue if you start the Docker container in the parent directory of the MK git repo rather than in the MK git repo directory.

The directory you start the container in is bind-mounted inside the container.  The parent directory is part of the container file system, and only writeable by root.  I didn't think about this, since my git repo is under my home directory, which is also bind-mounted inside the container.  I'll fix my instructions.

zultron, 2016-10-13:
I've fleshed this out a little more, and now can build for `amd64`, `i386` and `armhf`.  The packages are still untested, but I'll give them a try today.

Here are the [Travis CI results][1].  Without `ccache`, the times are around 13 minutes (less for `armhf`, not sure why).

[1]:  https://travis-ci.org/Dovetail-Automata/emc2

zultron, 2016-10-13:
I forgot, you'll have to pull a new Docker image if you want to test this.

Also, big thanks, @ArcEye, for testing.  It's very encouraging to have someone else repeat my results.

ArcEye, 2016-10-13:
> @ArcEye I think it'll solve your permissions issue if you start the Docker container in the parent directory of the MK git repo rather than in the MK git repo directory.

OK that makes sense.  I'll give it another spin later tomorrow.

> Also, big thanks, @ArcEye, for testing. It's very encouraging to have someone else repeat my results.

No problem, happy to try it and it brought me up to speed a lot better with docker.
This is a big improvement for armhf builds, which have always been the Achilles heel of the Travis CI system, look forward to testing the rest.

zultron, 2016-10-14:
PR #1077 needs to be fixed (for Wheezy again), but it contains all updates to the build system to cross-build Jessie packages or RIP for armhf and i386.

I also have a Travis CI configuration that demonstrates [1] Jessie package builds for the three architectures, plus working RIP build + regression tests for amd64.

I still haven't made time to actually test the build results by running `linuxcnc` on an ARM host.  That'll be an important indicator of whether this approach to cross-building could replace emulation to build packages in Travis CI.

[1]: https://travis-ci.org/Dovetail-Automata/emc2

ArcEye, 2016-10-14:
>     @ArcEye I think it'll solve your permissions issue if you start the Docker container in the parent directory of the MK git repo rather than in the MK git repo directory.
>> OK that makes sense. I'll give it another spin later tomorrow.

Yep that works fine, I'll check the status of your PRs and proceed as req tomorrow

zultron, 2016-10-17:
At this point the workflow is finished, and [automated cross-builds and regression tests][0] for Jessie amd64/i386/armhf and Raspbian (no Wheezy) produce packages, now pushed to [packagecloud][1].

After adding Raspbian, the Docker image was about 2.5GB, so I split it up into four, which allowed arch-specific things to be hidden inside the containers.  Then the super-complicated command lines could be simplified; see [this MK branch][2].  Now the scripts are not only useful for Travis, but also for building by hand.  Check out these examples:
```
# Print "Usage:  scripts/build_docker [ -i DOCKER-IMAGE ] [ -t DOCKER-TAG ]
#          [ -c ( deb [ -n ] | test | [ shell ] [ COMMAND ARG ... ] ) ]"
scripts/build_docker -?
# build `armhf` binary packages (-n: no source)
scripts/build_docker -t armhf -c deb -n
# build `amd64` (default) source and binary packages
scripts/build_docker -c deb
# build `amd64` RIP and run unit tests (only on `amd64`)
scripts/build_docker -c test
# run a shell command (default) in the Raspbian image (`CPPFLAGS`/`LDFLAGS` already set)
scripts/build_docker -t armhf ./configure --host=arm-linux-gnueabihf
# run an interactive shell in the i386 image
scripts/build_docker -t i386
```

After some thorough testing and review, if we want this cross-build method to become the new Machinekit CI system, we'll have to decide what to do about Wheezy packages.  At this point, I don't know if building a Docker image for Wheezy is possible or not.

[0]: https://travis-ci.org/Dovetail-Automata/emc2
[1]: https://packagecloud.io/dovetail-automata/machinekit
[2]: https://github.com/zultron/machinekit/tree/mk-travis-xbuild

cdsteinkuehler, 2016-10-17:
On 10/17/2016 10:29 AM, John Morris wrote:
> 
> After some thorough testing and review, if we want this cross-build method to 
> become the new Machinekit CI system, we'll have to decide what to do about 
> Wheezy packages. At this point, I don't know if building a Docker image for 
> Wheezy is possible or not.

My personal opinion is that Wheezy has been "oldstable" for 6+ months
already and if it's a significant burden to get cross builds going on
Wheezy they can be deprecated like RTAI.

Normally, I'd like to see the older versions have a bit longer life,
but Wheezy has *SO* many unmet dependencies for everything from 0mq to
cython to (I can't even remember them all).

But we're just talking about ARM...there would still be Wheezy
versions for x86, right?  And Wheezy could be built RIP or as packages
by someone if they wanted.  Actually, throwing an ARM or private build
machine (that doesn't have the build time-limit) might be a good way
to deal with the problem.

...but if ARM Wheezy packages get booted off the Travis-CI / Docker
platform in the name of cross-build progress and/or disappear, I don't
think it's a serious problem.

-- 
Charles Steinkuehler
charles@steinkuehler.net

zultron, 2016-10-17:
On 10/17/2016 10:54 AM, cdsteinkuehler wrote:
>
> My personal opinion is that Wheezy has been "oldstable" for 6+ months
> already and if it's a significant burden to get cross builds going on
> Wheezy they can be deprecated like RTAI.

I was hoping someone besides me would suggest this.  :)

> Normally, I'd like to see the older versions have a bit longer life,
> but Wheezy has *SO* many unmet dependencies for everything from 0mq to
> cython to (I can't even remember them all).

That's right, there's quite a stack of custom packages we carry to 
support Wheezy.

Also, RCN is no longer building new updates for his ARM Wheezy images, 
so esp. ARM Wheezy is less valuable a target

> But we're just talking about ARM...there would still be Wheezy
> versions for x86, right? And Wheezy could be built RIP or as packages
> by someone if they wanted. Actually, throwing an ARM or private build
> machine (that doesn't have the build time-limit) might be a good way
> to deal with the problem.

If we threw only Wheezy ARM out of Travis CI, that would achieve the 
goal stated in the issue title.

My Docker images won't build Wheezy at all, but if that turns out to be 
too hard to fix, then we can figure out how to keep the old Travis 
system running just for Wheezy amd64 and i386.  Or we could build 
special images that don't need --sysroot.  Or maybe ARM Wheezy can be 
made to work with --sysroot, despite all my fuss.

> ...but if ARM Wheezy packages get booted off the Travis-CI / Docker
> platform in the name of cross-build progress and/or disappear, I don't
> think it's a serious problem.

Awesome, that resolves one of my two main worries.

My other big worry is that after testing, we find that using --sysroot 
isn't viable for some unanticipated reason (though the ARM build passing 
regression tests is very encouraging).  I may not get to test until next 
week, but if anyone wants to load up the i386 or armhf packages here to 
see whether `linuxcnc` can successfully run a sim config, I'd love to 
hear the results.

https://packagecloud.io/dovetail-automata/machinekit

cdsteinkuehler, 2016-10-17:
On 10/17/2016 12:24 PM, John Morris wrote:
> If we threw only Wheezy ARM out of Travis CI, that would achieve the
> goal stated in the issue title.
>
> My Docker images won't build Wheezy at all, but if that turns out to be
> too hard to fix, then we [...]

I think the existing Wheezy builds would be fine as a fall-back.  I
seem to be the only one who's run into this issue with comp caused by
not building all three flavors at once.

>  > ...but if ARM Wheezy packages get booted off the Travis-CI / Docker
>  > platform in the name of cross-build progress and/or disappear, I don't
>  > think it's a serious problem.
> 
> Awesome, that resolves one of my two main worries.

...but I still wouldn't be too concerned if Wheezy builds just died
for a while (or forever).

> My other big worry is that after testing, we find that using --sysroot
> isn't viable for some unanticipated reason (though the ARM build passing
> regression tests is very encouraging). I may not get [...]

My currently active & running BBB needs to stay as-is since I'm
currently in the middle of a job getting some custom HAL components
written.  I'll try to test on a different system soon, but you should
send a request for help out to the mailing list.  The more folks who
test things the better!

-- 
Charles Steinkuehler
charles@steinkuehler.net

cdsteinkuehler, 2016-10-17:
No good.  It looks like the packages didn't build correctly (hard-coded /sysroot/ paths in the linuxcnc script):

````
machinekit@beaglebone:~$ machinekit
MACHINEKIT - 0.1
/usr/bin/linuxcnc: line 281: /sysroot/armhf/usr/bin/wish8.6: No such file or directory
/usr/bin/linuxcnc: line 190: /sysroot/armhf/usr/bin/wish8.6: No such file or directory
machinekit@beaglebone:~$ grep -n sysroot `which linuxcnc`
190:        echo "catch {send -async popimage destroy .}; destroy ." | /sysroot/armhf/usr/bin/wish8.6
216:            /sysroot/armhf/usr/bin/wish8.6 $LINUXCNC_TCL_DIR/show_errors.tcl $DEBUG_FILE $PRINT_FILE
248:PICKCONFIG="/sysroot/armhf/usr/bin/wish8.6 $LINUXCNC_TCL_DIR/bin/pickconfig.tcl"
249:LINUXCNC_EMCSH=/sysroot/armhf/usr/bin/wish8.6
628:    input=$(/sysroot/armhf/usr/bin/wish8.6 <<EOF
machinekit@beaglebone:~$
````

zultron, 2016-10-18:
A couple of minor fixes to the Docker images, and `grep` doesn't find those executables (tcl-related, plus `xeno-config`) hard-coded with `/sysroot` anymore.

It is finding `/sysroot` in the binaries though, both executables and shared libs.  The sample I looked at appeared to be header file names, and since the regression tests run, I'm guessing it's debug symbols.  I'm still thinking about the implications of this.

So, new packages are in the repo.  I don't expect you to keep playing whack-a-mole with me, but thanks @cdsteinkuehler for the testing.  I should probably get through a complete initial test before bothering others for help.

ArcEye, 2016-10-18:
Jessie is currently on v8.6 and has been out for 11/2 years.
I personally have not used Wheezy for over a year, except on the one machine I still have running a rtai kernel in the workshop, that uses the parport.

Will not be long before we have to think about supporting Stretch, especially if we are ever going to
break away from tcl / tk / wish and get Qt5 properly integrated (Jessie still defaults to Qt4)

Afraid I don't have anything to test on ARM wise, so you may have to go wider afield for confirmation tests.

cdsteinkuehler, 2016-10-18:
On 10/17/2016 11:07 PM, John Morris wrote:
> 
> So, new packages are in the repo. I don't expect you to keep playing 
> whack-a-mole with me, but thanks @cdsteinkuehler 
> <https://github.com/cdsteinkuehler> for the testing. I should probably get 
> through a complete initial test before bothering others for help.

I dug out another BeagleBone, so now it's easy for me to test w/o
potentially killing my other work.

...but I can't figure out how to install the latest packages.  On your
packagecloud, the armhf files show up as version 0.1.1-1, missing the
long git hash.  But after an apt-get update, I don't see either the
0.1.1-1 packages or the latest git hash (975ab7a- per the x86 package
files) available for armhf.

Did something go wrong with the package push, or how do I install the
packages?  I didn't have any problems installing the previous version.

-- 
Charles Steinkuehler
charles@steinkuehler.net

zultron, 2016-10-18:
Aargh, I screwed up some logic, and the packages were build wrong.  Worse, I've hit the 250MB "storage" limit on the free packagecloud account (it turns out "storage" really means "storage at beginning of billing period + cumulative push bandwidth through end of billing period").  It looks like we're going to have to start building from the command line, and I'll finish up the flow in a few weeks (what can I say, I'm cheap).

Thanks again for the tests!

cdsteinkuehler, 2016-10-18:
No worries.

If you get packages from the command line, feel free to email them or
otherwise just put them somewhere and I can install them manually to
test.  Otherwise, waiting for a bit isn't a problem...this isn't a
super time-critical issue.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2018-08-01:
This has been addressed by @zultron s new cross-builder which has been in operation for over a year

closed 2018-08-01 by ArcEye