machinekit/machinekit #1076: Cannot delete a ring with halcmd

cdsteinkuehler, 2016-10-14:
I'm not sure I completely understand the syntax, but if I'm not confused it doesn't look like you can delete a ring with halcmd once it's created:

````
machinekit@beaglebone:~$ halcmd newring test 1024
machinekit@beaglebone:~$ halcmd show ring
Rings:
Name           Size       Type   Rdr Wrt Ref Flags
test           1024       record 0/0 0/0 0   recmax:1008

machinekit@beaglebone:~$ halcmd delring test
delring NIY: ring='test'
machinekit@beaglebone:~$ halcmd show ring
Rings:
Name           Size       Type   Rdr Wrt Ref Flags
test           1024       record 0/0 0/0 0   recmax:1008

machinekit@beaglebone:~$ halcmd newring test 1024
<commandline>:0: newring: failed to create new ring test: File exists
machinekit@beaglebone:~$

````

cdsteinkuehler, 2016-10-14:
It looks like "NIY" means Not Implemented Yet.  I guess this is a feature request and not a bug...here's the code:
````
int do_delring_cmd(char *ring)
{
    halcmd_output("delring NIY: ring='%s'\n", ring);
    // return halpr_group_delete(group);
    return 0;
}
````

mhaberler, 2016-10-14:
A ringbuffer is a toplevel HAL object (=independent of any other) and there is no dependency tracking of which comps reference a ringbuffer - deleting a ringbuffer which happens to be referenced by a comp will likely lead to a crash

so it is best to consider a ringbuffer "eternal" (for HAL session lifetime)

the delring command is implemented in the [last version I worked on 13 months ago](https://github.com/mhaberler/machinekit/tree/disentangle/) but I do not recommend it except after all comps are unloaded

luminize, 2017-05-18:
@cdsteinkuehler do you want to keep this issue open?