machinekit/machinekit #1079: Merge features from multicore branch

cdsteinkuehler, 2016-10-15:
I got an email from Michael Haberler detailing many of the changes to the Machinekit HAL layer vs. it's origins in LinuxCNC (see attached) and it mentioned both the mainline Machinekit version of HAL and the not-yet-merged multicore work Michael did some time ago.

This issue is intended to identify the multicore changes that have not yet been merged and to hopefully help make that happen.
[rundown - features.txt](https://github.com/machinekit/machinekit/files/531217/rundown.-.features.txt)

ArcEye, 2016-10-15:
Some while ago, I got the multicore features enabled work ported across to a stand alone, cmake built HAL layer here.
https://github.com/ArcEye/machinekit-NG/tree/cmake-halck-inrepo

This however lacks the git history, I copied the files across in replacement of existing ones, the aim being to test it rather than an end user suitable branch.

I quite often use the 'ck-integration' branch, as we called it, but it now lags behind the main machinekit repo in a lot of areas, and would take quite a bit of work to integrate AND preserve history.
There would be far too many clashes for anything but a bit by bit manual merge.

cdsteinkuehler, 2016-10-15:
On 10/15/2016 10:34 AM, ArcEye wrote:
> 
> I quite often use the 'ck-integration' branch, as we called it, but it now lags 
> behind the main machinekit repo in a lot of areas, and would take quite a bit of 
> work to integrate AND preserve history.
> There would be far too many clashes for anything but a bit by bit manual merge.

Yes, 600+ commits between the two branches on each side.  But it's
just going to get worse the longer any merge gets delayed...

Is the problem greatly simplified if we don't worry about the git history?

Are there any particular pieces or features that might make sense to
merge individually, or is it sort of all-or-nothing?

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-10-15:
> Are there any particular pieces or features that might make sense to merge individually, or is it sort of all-or-nothing?

It really is a complete package, because of the inter-dependence.

All the components are created using pin-pointers instead of pins for instance with a different version of instcomp. Halcmd and lots of the hal code, is changed to work with the new components, in a way that is not backwardly compatible. 

I would need to get my head back around it, we finished the work about 18 months ago.

AFAIR there remained the issue of detection of version2 over version1 components, I don't think that was finally resolved but I cannot recall the details at the moment.

ArcEye, 2016-10-15:
I would be very happy to get the ck-integration branch into mainstream Machinekit, Michael and I spent a lot of time on it, don't think I am reluctant in that respect, it is just that I am aware of the disruption it may cause to the main repo and the work it will take to get it there.

One possible interim solution whilst keeping to C4, is just to have a parallel repo forked from the current master, in the same way as the Machinekit-NG exists, and do the work there, where it does not effect users.

cdsteinkuehler, 2016-10-15:
On 10/15/2016 11:26 AM, ArcEye wrote:
> One possible interim solution whilst keeping to C4, is just to have a parallel 
> repo forked from the current master, in the same way as the Machinekit-NG 
> exists, and do the work there, where it does not effect users.

I'm happy to have a semi-official branch somewhere for doing the
integration work.  It does look like this is a fairly major chunk of code.

My concern, and the reason I opened this issue, is that with Michael
stepping back, and the large number of commits already separating the
multicore work from the official Machinekit, it's going to just get
harder as time goes on to include the multicore work.  And that would
be a significant loss...a lot of the multicore work makes fundamental
improvements to the HAL and real-time layers.

This was mostly an issue to make sure multicore isn't forgotten about,
and to determine what might make sense as a path to get the changes
integrated...things like should it happen before or after a separation
of the CNC and HAL/RTAPI layers.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-10-15:
OK, I am all for that.

I think it will be simplest to merge the ck-integration files into an up to date branch, rather than try updating the existing ck-integration with the subsequent commits from the main repo, a huge number of which will clash.

The painstaking bit will be looking at all the changes to the current files since ck-integration was checked out and ascertaining which can stay in and which are not compatible.

I will look at it tomorrow, seems I am best placed to attempt this, if I can just remember what we did and why :stuck_out_tongue:

ArcEye, 2016-10-16:
Had a look at my Machinekit-NG/cmake-halck-inrepo branch, which is probably the easiest way to get a snapshot of the actual changes required

**1,259 changed files with 18,341 additions and 687 deletions**

Quite a lot of changes are from new components / v2 versions of existing components and runtest modifications to cater for those, so it is not *quite* as mammoth a task as it appears, but still decidedly non trivial.

I also want to reference the original commits in the ck-integration branch to maintain at least most of the history, so it will take a while.

ArcEye, 2016-10-16:
Now I have got my head back into it, I remembered this
https://github.com/mhaberler/machinekit/commits/disentangle-merged-into-master

@mhaberler started work on a merge into master in June.
This branch builds and would be the best base to update.

Shortens the number of commits missing considerably

ArcEye, 2016-10-18:
The biggest single problem with merging multicore, is not the HAL stuff, but the protobuf / machinetalk etc. 

This was changed massively since the original work and even since @mhaberler started his merge, some of it has not only changed but has been moved.
Trying to merge 6 pages of subsequent commits, one of the closest of which is a squash of 20 other commits(!) proved very problematic

I am back to trying to merge into the current master and fixing the conflicts and build errors as I go along.

What I have not ascertained yet, is if the multicore protobuf headers extend the existing headers, as they appear to for ring buffers say, or whether there are fundamental clashes in other places which require deeper changes to the existing setup.

Could be WIP for a while yet.

ArcEye, 2016-10-18:
After a major session today, I now have a working multicore branch
https://github.com/ArcEye/machinekit/tree/multicore2

Just cloned and built it from the repo and it works
```
mick@INTEL-i7:/media/sde/sde7/home/mick/src/machinekit-multicore$ halcmd newinst debounce db-non-atomic pincount=2
mick@INTEL-i7:/media/sde/sde7/home/mick/src/machinekit-multicore$ halcmd newinst debouncev2 db-atomic pincount=2
mick@INTEL-i7:/media/sde/sde7/home/mick/src/machinekit-multicore$ halcmd show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon Flags  linked to:
   206    240 bit   IN          FALSE  db-atomic.0.in                		----
   206    240 bit   OUT         FALSE  db-atomic.0.out               		----
   206    240 s32   I/O             0  db-atomic.0.state             		----
   206    240 bit   IN          FALSE  db-atomic.1.in                		----
   206    240 bit   OUT         FALSE  db-atomic.1.out               		----
   206    240 s32   I/O             0  db-atomic.1.state             		----
   206    240 s32   I/O             5  db-atomic.delay               		----
   206    240 s32   OUT             0  db-atomic.funct.time          		----
   206    240 s32   I/O             0  db-atomic.funct.tmax          		----
   206    240 bit   OUT         FALSE  db-atomic.funct.tmax-inc      		----
   170    224 bit   IN          FALSE  db-non-atomic.0.in            		--l-
   170    224 bit   OUT         FALSE  db-non-atomic.0.out           		--l-
   170    224 s32   I/O             0  db-non-atomic.0.state         		--l-
   170    224 bit   IN          FALSE  db-non-atomic.1.in            		--l-
   170    224 bit   OUT         FALSE  db-non-atomic.1.out           		--l-
   170    224 s32   I/O             0  db-non-atomic.1.state         		--l-
   170    224 s32   I/O             5  db-non-atomic.delay           		--l-
   170    224 s32   OUT             0  db-non-atomic.funct.time      		----
   170    224 s32   I/O             0  db-non-atomic.funct.tmax      		----
   170    224 bit   OUT         FALSE  db-non-atomic.funct.tmax-inc  		----
```

I think it requires libck.so.0.4.6 which I have locally, need to check that tomorrow.
Originally we used it for atomic ops etc. but I can't remember if it was replaced with internal functions
Can guarantee the master of the repo has changed out of all recognition :stuck_out_tongue_closed_eyes: 
https://github.com/concurrencykit/ck/commits/master

It has undoubtedly broken some of the protobuf stuff , because that had evolved with no knowledge of the uses multicore put it to and changed field names and numbering in some cases.
Need to speak with @machinekoder re this

There were some hacks required to get it to build, largely regards header locations, which need resolving.

But it runs and is based on the current master. :smile:

cdsteinkuehler, 2016-10-18:
On 10/18/2016 11:50 AM, ArcEye wrote:
> 
> But it runs and is based on the current master. 😄

Great news!

Let us know if there is anything us mere mortals can do to help!

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-10-19:
> Let us know if there is anything us mere mortals can do to help!

I don't think there are any 'mere mortals' on these pages. 

The main clash that I need to resolve next is regards ring buffers.

When @mhaberler collaborated with @luminize on the ROS and CanOpen stuff, he rewrote and improved the ring buffers he wrote for the Multicore project.
It was these that made their way into the master, with subsequential changes to protobuf and cython too probably.

Once Multicore uses the ring buffers already in the repo, it should be a lot simpler to finish the conversion and get somewhere near a point where it can all be merged in.

cdsteinkuehler, 2016-10-19:
On 10/19/2016 3:13 AM, ArcEye wrote:
> 
> Once Multicore uses the ring buffers already in the repo, it should be a lot 
> simpler to finish the conversion and get somewhere near a point where it can all 
> be merged in.

This I can maybe help with...I've been using the ring buffers a bunch
in my recent custom HAL components so am familiar with this code in
master.  What needs to be fixed up?

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-10-19:
Thanks,  looks like you will be the ideal person to test the end result.

I think it is relatively minor, the errors were largely regards number and type of arguments in functions like hal_ring_detach(....)

I need to do a detailed diff, but seem to recall it looked like the new code maintained a count of the number of rings within an instance, whereas the old ones didn't, the new ones didn't use the mutex by default and the old ones did etc.

Plus annoying changes like
`typedef __s32 ring_size_t;` had started out as `typedef __s32 ringsize_t;`

I'll be having a look this afternoon and will give a shout if I get stuck.

ArcEye, 2016-10-19:
It will take a while.

The main difference, besides argument types and numbers, is that the `hal_plug_t` that used to be associated with each rings read/write is subsumed into the new `hal_ring_t`.
On top of this, the Multicore code can use atomic data types and hal_accessors.

In pbring.c in the master for example
```
record_shift(&p->to_rt_rb);
    *(p->received) += 1;
```
the internal receiving ringbuffer of the pbring instance is accessed directly and the received count
pin is incremented directly, by de-referenced pointer arithmetic. 

In the multicore version this becomes
```
record_shift(plug_rb(p->to_rt_rb));
    set_u32_pin(p->received, get_u32_pin(p->received) + 1);
```
where the plug is accessed for the same ringbuffer and the received count pin_ptr is incremented
using atomic accessors.
(I actually wrote increment and decrement accessors for all data types as macros in instcomp, so I could add these to the code whilst I am about it, to save the longhand above) 

Having examples using both old and new should make it easier to patch up as a combination of both.

No quick fix, I need to immerse myself in it for a while and do plenty of testing once done.

ArcEye, 2016-10-23:
@cdsteinkuehler 
If you have any C examples of components using rings, I would be interested to see them.
@mhaberler changed things considerably when he put rings and ringbuffers into the main repo.

I have removed hal_plugs from multicore2 and have pbring.c working using atomic accessors etc. but it requires the explicit creation of rings before attaching them and other stuff which differs from the current master.

I probably need to 'backport' rings from master to multicore, so that everything works as it does now,
if I can find some more examples of how that is being done.

cdsteinkuehler, 2016-10-24:
I previously sent some links to code, but apparently it didn't make it to github...here it is again:

On 10/23/2016 11:14 AM, ArcEye wrote:
> > @cdsteinkuehler <https://github.com/cdsteinkuehler>
> > If you have any C examples of components using rings, I would be interested to 
> > see them.
> > @mhaberler <https://github.com/mhaberler> changed things considerably when he 
> > put rings and ringbuffers into the main repo.

Here's a reader on the real-time side, with the writer process in
python.  Basically the process is:

Someone creates the ring in HAL (reader or writer or halfile or ...)

* Writer:
** Attaches to ring
** Checks for enough space to fit message
** Writes message

https://github.com/cdsteinkuehler/ros_hal_tests/blob/master/src/listener/scripts/listener.py#L72-L81

* Reader:
** Attaches to ring
** Checks for data
** Processes data if available
** Marks data as consumed

https://github.com/cdsteinkuehler/ros_hal_tests/blob/master/src/listener/scripts/listener.py#L72-L81

...but there are various ways to break this down on either side into
slightly more fine-grained processes.  Ask if you have questions, I've
got other usage examples but they were written for someone else and
are not open source.

cdsteinkuehler, 2016-10-24:
Do any of the multicore changes seriously affect the HAL API and header files?  I'm going to start looking at splitting the CNC stuff off from HAL/RTAPI, and am wondering if this split should happen before or after merging the multicore work.  Which do you think would work better?

ArcEye, 2016-10-25:
> Here's a reader on the real-time side

Thanks, I'll look at it in a bit.

I am pretty close now, everything builds and loads. The only glitch is a segfault in halcmd when threads are started with rings attached.
This happens with the rings - plugs - ringbuffers from the original multicore and the modified code I wrote to remove plugs.
Need to do a carefull diff of all the halcmd files, suspect it will be accessing a buffer address or array that has changed size, ie the usual segfault suspects.

> Do any of the multicore changes seriously affect the HAL API and header files? I'm going to start looking at splitting the CNC stuff off from HAL/RTAPI, and am wondering if this split should happen before or after merging the multicore work. Which do you think would work better?

They make wholesale changes, with a lot of new apis and variant functions.

A good place to look at what is in the split is my repo I posted a link to above.  There is a branch for standard MK and one for Multicore.
That will show what was split off for the HAL package.

The bit I am unclear about is how protobuf etc fits between and whether it can standalone or needs to be in both HAL and CNC for them to communicate

ArcEye, 2016-10-25:
The usage I had in mind was very raw C component stuff, but now I have got past that.
I have narrowed down the segfault to a protobuf `buf_read()` function at present.

This is what happens
```
mick@INTEL-i7:/usr/src/testbed2$ DEBUG=5 realtime restart
mick@INTEL-i7:/usr/src/testbed2$ halcmd newring to-rt1 16384
mick@INTEL-i7:/usr/src/testbed2$ halcmd newring from-rt1 16384
mick@INTEL-i7:/usr/src/testbed2$ halcmd loadrt pbmsgs
<commandline>:0: Realtime module 'pbmsgs' loaded
mick@INTEL-i7:/usr/src/testbed2$ halcmd newinst pbring first rring=to-rt1 wring=from-rt1
<commandline>:0: Realtime module 'pbring' loaded
mick@INTEL-i7:/usr/src/testbed2$ halcmd show ring
Rings:
ID    Name           Size       Type   Rdr Wrt Ref Flags 
83    from-rt1       16384      record 0/0 103/0 1   shmseg recmax:16368 
                                             <== from-rt1.first.write id=103 owner=96
78    to-rt1         16384      record 102/0 0/0 1   shmseg recmax:16368 
                                             ==> to-rt1.first.read id=102 owner=96

mick@INTEL-i7:/usr/src/testbed2$ halcmd newthread servo 1000000 fp
mick@INTEL-i7:/usr/src/testbed2$ halcmd addf first.update servo
<commandline>:0: Function 'first.update' added to thread 'servo', rmb=0 wmb=0
mick@INTEL-i7:/usr/src/testbed2$ halcmd start
<commandline>:0: Realtime threads started
mick@INTEL-i7:/usr/src/testbed2$ halcmd show ring
halcmd: cant connect to rtapi_app: -1 (uri= uuid=a42c8c6b-4025-4f83-ba28-dad21114744a): rtapi_rpc(): reply timeout

halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
halcmd: the msgd:0 logger demon is not running - please investigate /var/log/linuxcnc.log
```

Some more painstaking diff'ing to come I expect, to work out what has changed in the machinetalk side that one of the buffers is NULL or out of range, so that similar to
```
while (count--)
    *(buff)++ = *(source)++;
```
is hitting the buffers (pun intended)

cdsteinkuehler, 2016-10-25:
On 10/23/2016 11:14 AM, ArcEye wrote:
> @cdsteinkuehler <https://github.com/cdsteinkuehler>
> If you have any C examples of components using rings, I would be interested to 
> see them.
> @mhaberler <https://github.com/mhaberler> changed things considerably when he 
> put rings and ringbuffers into the main repo.

Here's a reader on the real-time side, with the writer process in
python.  Basically the process is:

Someone creates the ring in HAL (reader or writer or halfile or ...)

* Writer:
** Attaches to ring
** Checks for enough space to fit message
** Writes message

https://github.com/cdsteinkuehler/ros_hal_tests/blob/master/src/listener/scripts/listener.py#L72-L81

* Reader:
** Attaches to ring
** Checks for data
** Processes data if available
** Marks data as consumed

https://github.com/cdsteinkuehler/ros_hal_tests/blob/master/src/listener/scripts/listener.py#L72-L81

...but there are various ways to break this down on either side into
slightly more fine-grained processes.  Ask if you have questions, I've
got other usage examples but they were written for someone else and
are not open source.

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-10-25:
@ArcEye
In the interpolator (not in master) is a ring being attached to and read from.
https://github.com/luminize/machinekit/blob/hm2-no-params-plus-interpolator/src/hal/interpolator/interpolate.c

There's also a delayline component i wrote with Michaels help. It's in master. It creates, writes and reads from a ring. https://github.com/machinekit/machinekit/blob/master/src/hal/components/delayline.c

I don't know if this is any help since it isn't from the multicore branch

ArcEye, 2016-10-26:
The problem now is not the usage, but most likely protobuf internals.
I would expect any of your examples that use the same functions to fail also.

I have been using the pbring example in machinetalk/msgcomponents.
This works in the old ck-integration branch, but segfaults in the new merged multicore2 branch.

The problem is in pb_decode.c, in the buf_read() function section
```
while (count--)
    *(buff)++ = *(source)++;
```

I have debugged the core file and the exception is being thrown when count reaches `count=0`
Presumably, as anticipated it is the usual segfault error from trying to write beyond the bounds of a buffer.

The problem however is not the code itself, as it works in the other branch.

I am surmising that the protobuf code has changed subtly, maybe a message length is one byte smaller than expected, that is all it would take.

Unfortunately all the protobuf stuff has been moved around recently (for no real reason I can see) which makes it even harder to compare like with like and makes old files instantly incompatible with new versions because all the header paths have changed.
Add to this the fact that lots of it is generated and does not exist until build time.

Diff'ing has thus far not thrown up any obvious discrepancies between versions in this area.

I will rewrite the function to give more info about exactly when the error is occurring and try to prevent it, then see what happens to the rest of the program flow.
Will need to trace though to find where values and size assumptions are coming from and hopefully nail it down.

ArcEye, 2016-10-26:
Arggghhh  -  I have been chasing my tail  - the pbring example in multicore has never worked. :scream: 
The multicore version segfaults at the same place in @mhaberlers' remerge branch.

When I tested in what I thought was the `remerge` branch, either the repo branch must have been set to `master` or I didn't start the thread.
The version in master does work, albeit in a different way without hal_plugs etc.

I am going to walk away from this for a while and try to wipe my braincells clear of the last couple of days and then use your examples to test whether rings work OK.
 
Or rewrite pbring.c, which I don't like because it codes in assumptions about naming which only appear in pbring-demo.hal, making it only workable in a strict sequence with other factors present
( ie. specifically named pre-created rings) 

The basic merge does work, machinekit launches and runs, v1 and v2 pins are created and work.

This particular cul-de-sac now resolved, hopefully I will progress better.

ArcEye, 2016-10-31:
OK, back to ring examples that do work

@luminize I used your delayline component and hopefully you will agree this is working
```
mick@INTEL-i7:/usr/src/testbed2$ DEBUG=5 realtime restart
mick@INTEL-i7:/usr/src/testbed2$ halcmd loadrt delayline names=foo max_delay=2500 samples=ffb
<commandline>:0: Realtime module 'delayline' loaded
mick@INTEL-i7:/usr/src/testbed2$ halcmd newthread servo 1000000 fp
mick@INTEL-i7:/usr/src/testbed2$ halcmd addf foo.sample servo
<commandline>:0: Function 'foo.sample' added to thread 'servo', rmb=0 wmb=0
mick@INTEL-i7:/usr/src/testbed2$ halcmd addf foo.output servo
<commandline>:0: Function 'foo.output' added to thread 'servo', rmb=0 wmb=0
mick@INTEL-i7:/usr/src/testbed2$ halcmd show ring
Rings:
ID    Name           Size       Type   Rdr Wrt Ref Flags 
80    foo.samples    120000     record 0/0 0/0 1   shmseg recmax:119984  scratchpad:320 

mick@INTEL-i7:/usr/src/testbed2$ halcmd show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon Flags  linked to:
    78        bit   IN          FALSE  foo.abort                     		--l-
    78        u32   IN     0x00000000  foo.delay                     		--l-
    78        bit   IN          FALSE  foo.enable                    		--l-
    78        float IN              0  foo.in0-flt                   	0.000010	--l-
    78        float IN              0  foo.in1-flt                   	0.000010	--l-
    78        bit   IN          FALSE  foo.in2-bit                   		--l-
    78        float OUT             0  foo.out0-flt                  	0.000010	--l-
    78        float OUT             0  foo.out1-flt                  	0.000010	--l-
    78        bit   OUT         FALSE  foo.out2-bit                  		--l-
    78        s32   OUT             0  foo.output.time               		----
    78        s32   I/O             0  foo.output.tmax               		----
    78        bit   OUT         FALSE  foo.output.tmax-inc           		----
    78        u32   OUT    0x00000000  foo.read-fail                 		--l-
    78        s32   OUT             0  foo.sample.time               		----
    78        s32   I/O             0  foo.sample.tmax               		----
    78        bit   OUT         FALSE  foo.sample.tmax-inc           		----
    78        u32   OUT    0x00000000  foo.too-old                   		--l-
    78        u32   OUT    0x00000000  foo.write-fail                		--l-
    66        s32   OUT       1004431  servo.curr-period             		----
    66        s32   OUT             0  servo.time                    		----
    66        s32   I/O             0  servo.tmax                    		----

mick@INTEL-i7:/usr/src/testbed2$ halcmd start
<commandline>:0: Realtime threads started
mick@INTEL-i7:/usr/src/testbed2$ halcmd setp foo.in0-flt 99.99
<commandline>:0: setting parameter 'foo.in0-flt' to '99.99'
<commandline>:0: Pin 'foo.in0-flt' set to 99.99
mick@INTEL-i7:/usr/src/testbed2$ halcmd setp foo.in1-flt 333.333
<commandline>:0: setting parameter 'foo.in1-flt' to '333.333'
<commandline>:0: Pin 'foo.in1-flt' set to 333.333
mick@INTEL-i7:/usr/src/testbed2$ halcmd show pin foo
Component Pins:
  Comp   Inst Type  Dir         Value  Name                             Epsilon Flags  linked to:
    78        bit   IN          FALSE  foo.abort                     		--l-
    78        u32   IN     0x00000000  foo.delay                     		--l-
    78        bit   IN          FALSE  foo.enable                    		--l-
    78        float IN          99.99  foo.in0-flt                   	0.000010	--l-
    78        float IN        333.333  foo.in1-flt                   	0.000010	--l-
    78        bit   IN          FALSE  foo.in2-bit                   		--l-
    78        float OUT         99.99  foo.out0-flt                  	0.000010	--l-
    78        float OUT       333.333  foo.out1-flt                  	0.000010	--l-
    78        bit   OUT         FALSE  foo.out2-bit                  		--l-
    78        s32   OUT           564  foo.output.time               		----
    78        s32   I/O         10749  foo.output.tmax               		----
    78        bit   OUT         FALSE  foo.output.tmax-inc           		----
    78        u32   OUT    0x00000000  foo.read-fail                 		--l-
    78        s32   OUT           974  foo.sample.time               		----
    78        s32   I/O         11057  foo.sample.tmax               		----
    78        bit   OUT         FALSE  foo.sample.tmax-inc           		----
    78        u32   OUT    0x00000000  foo.too-old                   		--l-
    78        u32   OUT    0x00000000  foo.write-fail                		--l-

mick@INTEL-i7:/usr/src/testbed2$ halrun -U
<commandline>:0: Realtime threads stopped
<commandline>:0: Realtime module 'delayline' unloaded
<commandline>:0: Realtime threads stopped
mick@INTEL-i7:/usr/src/testbed2$ halrun -R
msgd:0 stopped
```
certainly appears that values at `in` are mirrored in `out` 
(albeit I did not set the period long enough to test the delay aspect)

Next, I will probably write a v2 version of delayline, using pin_ptrs and atomic accessors, which should pretty much prove the functionality.

luminize, 2016-10-31:
@ArcEye

> @luminize I used your delayline component and hopefully you will agree this is working

Looks good to me so far, but to know for sure:

- If you load a siggen signal to one of the pins
- set the foo.delay to 10 (periods)
- set foo.enable to 1

you should see the (delayed) signal in the output pin from halscope for example (or show different values when you do a "halcmd show pin")

There's also a halfile to do that: https://github.com/machinekit/machinekit/blob/master/src/hal/components/delayline.hal

if you load that one all should be set up and you just need to load halscope

> Next, I will probably write a v2 version of delayline, using pin_ptrs and atomic accessors, which should pretty much prove the functionality.

That would also be a good way to see what needs changing (if any) for the multicore stuff (of which I'm pretty blank).

ArcEye, 2016-10-31:
OK, I'll try that this afternoon, hedge cutting beckons. :smile:

ArcEye, 2016-10-31:
delayline works fine, once I got halscope built, so some progress.

Found halscope was not building, because I had not patched the branch with my fix for the HAVE_GTK bug https://github.com/machinekit/machinekit/commit/1872d9b97681b089bc35a088ca578d25163f610f

Consequently then found, once classicladder began building, that classicladder needs a lot of patching to work with the new `hal_pin_t` setup, so commented out the build for now.

Two steps forward / one back

ArcEye, 2016-11-01:
Classicladder fixed, so simply that I am suspicious, but it builds.

Decided not to build a v2 variant of delayline at this time.
It uses 'clever' command line data type specifications for the samples, which result in a pin initialisation later after data type is determined, as:
```
if (((retval = hal_pin_newf(type, HAL_IN, (void **) &hd->pins_in[i], comp_id,
				    "%s.in%d-%s", hd->name, i, str_type)) < 0)
```
The `halx_......()` pin creation routines actually return a pin_ptr, rather than a result value, 
so there is no equivalent to `hal_pin_newf(type......)`.
Because you require to know the pointer type, you have it specified in the assignment and call the relevant halx_... variant eg.
```
ip->delay = halx_pin_u32_newf(HAL_IN, owner_id, "%s.delay",  ip->name);
if (u32_pin_null(ip->delay))
    return _halerrno;
```
The null pointer checking function, has a variant defined for each type too.

It can be done with rewriting, probably would be best as an instcomp, but some other time. :smile:

ArcEye, 2016-11-02:
I have now converted and tested all the ring specific comps and code in machinetalk/msgcomponents

The pbring code has been moved out of the build, until someone with knowledge of and interest in using protobuf with rings (I have neither), has time to work out why pbdecode.c is hitting a NULL pointer when iterating through to the last record.
It didn't work in the merge that @mhaberler started either and the original one definitely won't work.

I can only speculate that a buffer length has changed or something like that, but cannot see anything obvious in the headers etc.

**It is now probably ready for testing by people who are actually coding using hal functions, ringbuffers etc.**
https://github.com/ArcEye/machinekit/tree/multicore2

The v2 versions of all the components are there too, currently just named `{componentname}v2`
The .icomps for those components show how they differ in pin types and the use of accessor functions instead of dereferenced pointers.

We were going to devise an automatic preference loading strategy, to load v2 components in preference to v1 if they existed, using kernel module tags.

At this time at least I am not sure that is necessary, everything will be backwardly compatible, with just a few hal file tweaks [1] to the majority of users as it is for now.

[1]
The primary differences are things like, `component_name` with underscores are NOT converted into dashes, (this is the case now actually) so the name that is loaded is the name you use.

If you `loadrt charge_pump` you get a component called charge_pump.0 by default, because it is now instantiable with numbered instances, unless you name it.
Its pins are `charge_pump.0.out` for instance.

If you are directly creating / attaching / detaching rings in code, as opposed to using HAL calls, you need to look at hal_ring.h and ring specific comps and code in machinetalk/msgcomponents for examples of the new code.

I am assuming that the cython generated python bindings make the right calls, but that may not be so.
Hopefully someone who uses them will test and advise.

luminize, 2016-11-02:
> From: ArcEye
> 
> I have now converted and tested all the ring specific comps and code in machinetalk/msgcomponents

Sounds like a major step!

> The pbring code has been moved out of the build, until someone with knowledge of and interest in using protobuf with rings (I have neither), has time to work out why pbdecode.c is hitting a NULL pointer when iterating through to the last record.
> It didn't work in the merge that https://github.com/mhaberler started either and the original one definitely won't work.
> I can only speculate that a buffer length has changed or something like that, but cannot see anything obvious in the headers etc.

Ok, I might eventually run into this when I take over the interpolator component which is not in master yet (planning to do)

> It is now probably ready for testing by people who are actually coding using hal functions, ringbuffers etc. https://github.com/ArcEye/machinekit/tree/multicore2

I'll try it out.

> If you are directly creating / attaching / detaching rings in code, as opposed to using HAL calls, you need to look at hal_ring.h and ring specific comps and code in machinetalk/msgcomponents for [...]
Hopefully someone who uses them will test and advise.

I use the cython bindings a lot, so I'll see what happens when I run your branch

ArcEye, 2016-11-02:
Great, thanks.

ArcEye, 2016-11-03:
There is now an axis_mm based sim which just uses v2 components
https://github.com/ArcEye/machinekit/tree/multicore2/configs/sim/v2-icomp-demo-sim

Does not really add anything, just quick proof that everything runs as anticipated using v2 versions.

ArcEye, 2016-11-03:
I have rebased manually, to put all the commits in multicore2 on top of the current master.
This is in https://github.com/ArcEye/machinekit/commits/reverse-engineer

Hopefully will make it easier to merge at some point, as it has an unbroken start - stop commit range

ArcEye, 2016-11-04:
I am having a peculiar problem with Submakefiles not copying over headers to /include

The standard rule at the top of most is similar to
```
../include/%.h: ./rtapi/%.h
	cp  $^ $@

../include/%.hh: ./rtapi/%.hh
	cp  $^ $@

../include/%.h: ./rtapi/shmdrv/%.h
	mkdir -p ../include/shmdrv
	cp  $^ $@
```

This is not being evaluated and acted upon.  However if I insert
```
cp_rtai_headers:
	cp -f ./rtapi/*.h* ../include/
	mkdir -p ../include/shmdrv
	cp -f ./rtapi/shmdrv/*.h* ../include/shmdrv/
....
.PHONY:  modclean-rtapi cp-rtai-headers
....
modules:  ../libexec/rtapi_app_$(threads) cp-rtai-headers
```
ie make it a rule, add it to where others are executed - it works

Given the Submakefiles were unchanged from master (in that respect at least) any ideas what is going on?

cdsteinkuehler, 2016-12-07:
I'm returning to this after being distracted by some other projects for a while.

What is the current status?  If things are ready for wider testing, I'd like to get together some details (howto install packages or build from source) and send out a request for testers to the list and the blog.

If things aren't ready for testing, what can I do to help?

...as for the makefile question, the problem is likely in what is creating a dependency on ../include/*, not with the rules, although I think the last rule ought to be:

````
../include/shmdrv/%.h: ./rtapi/shmdrv/%.h
    mkdir -p ../include/shmdrv
    cp  $^ $@
````

ArcEye, 2016-12-07:
The includes problem is sidestepped for now, by explicit copying across in Makefiles.
What has caused it is baffling, nothing has changed, it was only when I started fixing up the runtests that I realised several headers were not present in /include.

The branch you want is `reverse-engineer`

I reordered so that all the commits are contiguous, on top of the then updated master, so as to make a later merge easier.

I need to look at subsequent commits and make sure they don't break anything.
In particular all the moving about and renaming of protobuf and machinetalk stuff from @machinekoder

ArcEye, 2016-12-07:
> What is the current status? If things are ready for wider testing, I'd like to get together some details (howto install packages or build from source) and send out a request for testers to the list and the blog.

As it is merged into the current master, it just needs to be built as normal.

Usage is another matter.  Specifying the `{componentname}v2.icomp` in a configuration, will use the new component.
As above, there is already a sample config which does this.
This has not been widely tested in real world configs.

Looking at the C file produced for an .icomp file which uses pin_ptrs, will show the accessor methods
used to assign and read data.

Probably need a 'killer' demo showing that the same data can be accessed from 2 different cores and updated / read correctly.  I can't remember what @mhaberler did in that respect and whether it is in the examples in the repo.

I likewise have been doing other stuff and need to get my head back around it and clarify what else needs doing.

ArcEye, 2016-12-07:
> I need to look at subsequent commits and make sure they don't break anything.
In particular all the moving about and renaming of protobuf and machinetalk stuff from @machinekoder

As I suspected everything merges OK until the last lot of commits which shift things around wholesale
https://github.com/machinekit/machinekit/commit/2362ac826e47ca0c1259cab99b549da7079f093c

I'll have to look at those another day.

cdsteinkuehler, 2016-12-08:
On 12/7/2016 11:12 AM, ArcEye wrote:
> 
> As I suspected everything merges OK until the last lot of commits which shift things around wholesale
> https://github.com/machinekit/machinekit/commit/2362ac826e47ca0c1259cab99b549da7079f093c
> 
> I'll have to look at those another day.

A great example of why this code needs to get merged soon-ish.  It will
just continue to get harder to merge as time goes on.  :-/

I'm fairly busy for the next week or so, but am taking off two weeks for
the holidays and can probably devote a couple solid days work to getting
things cleaned up or whatever else needs to happen (testing, packaging,
etc) if the protobuf cleanup happens before then.

Should we set a target date of new year's to have something folks on the
list could start testing?

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-12-08:
> A great example of why this code needs to get merged soon-ish. It will just continue to get harder to merge as time goes on. :-/

It is just the last 3 commits to master, I haven't looked at the clashes yet, it could just be connected to the renaming.
But yes it needs to be done before too long.

I will be able to look at it in a couple of days and try to get a snag list drawn up
Besides the ../include issue, I had to hack the hm2 runtest to skip it, you will probably know better than I what the problem was there.

An unresolved matter was component loading preference, using module tags to establish if a v2 version was available and loading that.
It may be that viewed from a greater distance, we might decide that you would not ever want a v1 component, if a v2 version was available and just replace them.

There is already an example in /src/hal/iring-example which can be built upon to show atomic access from different threads.

As usual the main problem will be documenting the API changes, changes to writing components using accessors etc. 
Albeit this could be aided by the fact that only the python bindings have anything approaching existing documentation:stuck_out_tongue:

ArcEye, 2016-12-12:
All the commits up to and including  https://github.com/machinekit/machinekit/commit/1df3d83011d20fb8e94ad52d76fab4a8728e0182 are combined into `reverse-engineer` branch and the repo builds OK.

The commits https://github.com/machinekit/machinekit/commit/295ae9e00ab0cf0207bae60aaaf16ff02f3e87b0 / https://github.com/machinekit/machinekit/commit/cbbbbe7aed35eee554accc8c7d073ef6554464dc / https://github.com/machinekit/machinekit/commit/37b2ed64f37d66011cdfa2a38f15b993de1107cb are horribly convolute squashes and merges.

The individual component commits fail with fatal errors, eg.
https://github.com/machinekit/machinekit/commit/295ae9e00ab0cf0207bae60aaaf16ff02f3e87b0 makes changes to /python/setup.py
This fails because that file has never existed.

I have tried merging the overall commit https://github.com/machinekit/machinekit/commit/d0d25d2781fc53580cf3fc348c2998e88113d46e, but even after fixing clashes nothing relating to machinetalk etc will build.

Looks like I may have to implement the changes manually, but that will have to wait.
I have little knowledge of or interest in the python machinetalk stuff.

cdsteinkuehler, 2016-12-12:
On 12/12/2016 10:28 AM, ArcEye wrote:
> All the commits up to and including 1df3d83 
> <https://github.com/machinekit/machinekit/commit/1df3d83011d20fb8e94ad52d76fab4a8728e0182> 
> are combined into |reverse-engineer| branch and the repo builds OK.
> 
> The commits 295ae9e 
> <https://github.com/machinekit/machinekit/commit/295ae9e00ab0cf0207bae60aaaf16ff02f3e87b0> 
> / cbbbbe7 
> <https://github.com/machinekit/machinekit/commit/cbbbbe7aed35eee554accc8c7d073ef6554464dc> 
> / 37b2ed6 
> <https://github.com/machinekit/machinekit/commit/37b2ed64f37d66011cdfa2a38f15b993de1107cb> 
> are horribly convolute squashes and merges.

Thanks for all your work on this!!!

I'm going to try to take a few days (probably next week) and see if I
can get a "mashup" of the multi-core and protobuf namespace changes
into a working state.

Assuming that happens, how can we get some more people testing this to
make sure everything is OK before merging into the main repo?  Do we
need to make packages available or is there an easy way to test the PR
build artifacts from the Travis CI?

I _really_ don't want to loose the multi-core stuff, and I also
*REALLY* don't want to have to keep spending developer cycles on
getting it to track master.  :-)  Let's see if we can't get things
merged and into github.com/machinekit/machinekit by the new year!

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-12-13:
>I'm going to try to take a few days (probably next week) and see if I can get a "mashup" of the multi-core and protobuf namespace changes into a working state.

I'll let you know if I get there before that.
I think this all arises because the code already existed in his own repo.
Instead of a transparent copy across of affected files and commit, he did a 'clever' squash of the commit range and merged into a single commit.
A lot of the files have only very trivial alterations, so the 26 total is not really as bad as it sounds.

The errors in build seemed to be largely coming from the use of old API calls, which is another good reason to do things manually, have a good look at the code  and ensure only what needs doing is changed.

>Assuming that happens, how can we get some more people testing this to make sure everything is OK before merging into the main repo? Do we need to make packages available or is there an easy way to test the PR build artifacts from the Travis CI?

The sort of people we will want to test initially, should be capable of just building a RIP.
I would anticipate that sims and normal configs should just run with v2 components, it will be things like ROS using rings which might hilight differences / problems.

luminize, 2016-12-13:
> On 12 Dec 2016, at 22:58, cdsteinkuehler <notifications@github.com> wrote:
> 
> how can we get some more people testing this to
> make sure everything is OK before merging into the main repo?

I'm willing to test this, but for that 2 things need to be done (to start testing)
- i need to bring the interpolator component @mhaberler wrote into that branch somehow and that probably needs work (guess), so for any changes to be done i likely need some help/examples. Would it be ok if i made a pr against machinekit/master?
- the ringbuffers and protobuf needs to work, since protobuf messages are put into the ringbuffer that's used by the interpolator component

ArcEye, 2016-12-13:
> - i need to bring the interpolator component @mhaberler wrote into that branch somehow and that probably needs work (guess), so for any changes to be done i likely need some help/examples. Would it be ok if i made a pr against machinekit/master?

If it is a stand alone component should not be any problem doing a PR for it.

If you post a link to the actual code, I will look once the current problem is sorted.

> - the ringbuffers and protobuf needs to work, since protobuf messages are put into the ringbuffer that's used by the interpolator component

The ringbuffers work, as per https://github.com/machinekit/machinekit/issues/1079#issuecomment-257256586 above.

Hopefully we will get the protobuf / machinetalk back working soon.  
However any user python programs written using the old `pb.` interface will be broken by commit https://github.com/machinekit/machinekit/commit/d0d25d2781fc53580cf3fc348c2998e88113d46e I imagine and require changing.

ArcEye, 2016-12-13:
Just finished manually inserting the changes to machinetalk / protobuf and added last merged commit on the top.

Now my `reverse-engineer` is up to date with machinekit/master.
The commits do not match but the contents of them do.

The resultant repo builds and the v2 test sim runs.
( https://github.com/ArcEye/machinekit/tree/reverse-engineer/configs/sim/v2-icomp-demo-sim )

zhivko, 2016-12-13:
> engineer/configs/sim/v2-icomp-demo-sim )

I am willing to test this as soon there will be testable branch available...

ArcEye, 2016-12-13:
> I am willing to test this as soon there will be testable branch available...

Thanks

The branch can be fetched with
`git clone -b reverse-engineer --single-branch http://github.com/arceye/machinekit.git`

Just build it and run whatever config etc. you normally run, without any changes to begin with.
Don't really want any esoteric set-ups that cannot easily be replicated to start with.

It should be backwardly compatible for any 'normal' usage.

Then later we can try the new components etc.

ArcEye, 2016-12-13:
As a final test, I have now merged `reverse-engineer` into a clone of `master`.

After fix ups of a few clashes, mainly changes to new API, this branch builds and runs
https://github.com/ArcEye/machinekit/tree/test-merge

ArcEye, 2016-12-13:
> @luminize I think the interpolator component is probably somewhere on your private repo server, which is currently off-line. Let me know when you have it visible and give me a link.

Finally found it in your github repo
https://github.com/luminize/machinekit/blob/hm2-no-params-plus-interpolator/src/hal/interpolator/interpolate.c

This should work 'as is'.  That would be the first test really, backward compatibility.

Then we would need to convert the component to use pin_ptrs and the atomic accessor functions to create a v2 version of it.

Should not be too difficult (famous last words).
I will add the v1 component to my repo and look at converting it in a while.

luminize, 2016-12-13:
@ArcEye, I will make a clean branch from current master. That would make life easier, cause a few changes (pins instead of params) have already been fixed. The branch you mention is the correct one, but polluted :)

luminize, 2016-12-13:
I'm rebasing a cleaned-up branch unto latest master. So far I need to fix some locations of .proto files.

zhivko, 2016-12-14:
I have builded machinekit from:
```
git clone -b reverse-engineer --single-branch http://github.com/arceye/machinekit.git
```

I build protobuf java machinkekit classes with cloning repository
```
git clone https://github.com/machinekoder/machinetalk-protobuf.git
```
checkout java branch with:
```
git checkout java
```
Start build of java remote protobuff classes for machinekit with
```
make clean all
```

Now there seems to be something broken because at the end of this procedure build script issues:
```
rm -rf build
```

full output of building java protobuff classes is here:
```
machinekit@beaglebone:~/git/ArcEye/machinetalk-protobuf$ make clean all
protoc create dependencies for src/machinetalk/protobuf/canon.proto
protoc create dependencies for src/machinetalk/protobuf/status.proto
protoc create dependencies for src/machinetalk/protobuf/object.proto
protoc create dependencies for src/machinetalk/protobuf/test.proto
protoc create dependencies for src/machinetalk/protobuf/preview.proto
protoc create dependencies for src/machinetalk/protobuf/rtapicommand.proto
protoc create dependencies for src/machinetalk/protobuf/task.proto
protoc create dependencies for src/machinetalk/protobuf/emcclass.proto
protoc create dependencies for src/machinetalk/protobuf/motcmds.proto
protoc create dependencies for src/machinetalk/protobuf/nanopb.proto
protoc create dependencies for src/machinetalk/protobuf/value.proto
protoc create dependencies for src/machinetalk/protobuf/log.proto
protoc create dependencies for src/machinetalk/protobuf/rtapi_message.proto
protoc create dependencies for src/machinetalk/protobuf/config.proto
protoc create dependencies for src/machinetalk/protobuf/types.proto
protoc create dependencies for src/machinetalk/protobuf/message.proto
protoc create build/cpp/machinetalk/protobuf/message.pb.cc from src/machinetalk/protobuf/message.proto
protoc create build/cpp/machinetalk/protobuf/types.pb.cc from src/machinetalk/protobuf/types.proto
protoc create build/cpp/machinetalk/protobuf/config.pb.cc from src/machinetalk/protobuf/config.proto
protoc create build/cpp/machinetalk/protobuf/rtapi_message.pb.cc from src/machinetalk/protobuf/rtapi_message.proto
protoc create build/cpp/machinetalk/protobuf/log.pb.cc from src/machinetalk/protobuf/log.proto
protoc create build/cpp/machinetalk/protobuf/value.pb.cc from src/machinetalk/protobuf/value.proto
protoc create build/cpp/machinetalk/protobuf/nanopb.pb.cc from src/machinetalk/protobuf/nanopb.proto
protoc create build/cpp/machinetalk/protobuf/motcmds.pb.cc from src/machinetalk/protobuf/motcmds.proto
protoc create build/cpp/machinetalk/protobuf/emcclass.pb.cc from src/machinetalk/protobuf/emcclass.proto
protoc create build/cpp/machinetalk/protobuf/task.pb.cc from src/machinetalk/protobuf/task.proto
protoc create build/cpp/machinetalk/protobuf/rtapicommand.pb.cc from src/machinetalk/protobuf/rtapicommand.proto
protoc create build/cpp/machinetalk/protobuf/preview.pb.cc from src/machinetalk/protobuf/preview.proto
protoc create build/cpp/machinetalk/protobuf/test.pb.cc from src/machinetalk/protobuf/test.proto
protoc create build/cpp/machinetalk/protobuf/object.pb.cc from src/machinetalk/protobuf/object.proto
protoc create build/cpp/machinetalk/protobuf/status.pb.cc from src/machinetalk/protobuf/status.proto
protoc create build/cpp/machinetalk/protobuf/canon.pb.cc from src/machinetalk/protobuf/canon.proto
protoc create build/java/machinetalk/protobuf/message.java from src/machinetalk/protobuf/message.proto
protoc create build/java/machinetalk/protobuf/types.java from src/machinetalk/protobuf/types.proto
protoc create build/java/machinetalk/protobuf/config.java from src/machinetalk/protobuf/config.proto
protoc create build/java/machinetalk/protobuf/rtapi_message.java from src/machinetalk/protobuf/rtapi_message.proto
protoc create build/java/machinetalk/protobuf/log.java from src/machinetalk/protobuf/log.proto
protoc create build/java/machinetalk/protobuf/value.java from src/machinetalk/protobuf/value.proto
[libprotobuf WARNING google/protobuf/compiler/java/java_file.cc:197] machinetalk/protobuf/value.proto: The default outer class name                                                                                                          , "Value", conflicts with a type declared in the proto file and an alternative outer class name is used: "ValueOuterClass". To avoi                                                                                                          d this warning, please use the java_outer_classname option to specify a different outer class name for the .proto file.
protoc create build/java/machinetalk/protobuf/nanopb.java from src/machinetalk/protobuf/nanopb.proto
protoc create build/java/machinetalk/protobuf/motcmds.java from src/machinetalk/protobuf/motcmds.proto
protoc create build/java/machinetalk/protobuf/emcclass.java from src/machinetalk/protobuf/emcclass.proto
protoc create build/java/machinetalk/protobuf/task.java from src/machinetalk/protobuf/task.proto
protoc create build/java/machinetalk/protobuf/rtapicommand.java from src/machinetalk/protobuf/rtapicommand.proto
[... 21 lines cut ...]
protoc create build/python/machinetalk/protobuf/canon_pb2.py from src/machinetalk/protobuf/canon.proto
cp "python/setup.py" "build/python/setup.py"
cp "python/machinetalk/__init__.py" "build/python/machinetalk/__init__.py"
cp "python/machinetalk/protobuf/__init__.py" "build/python/machinetalk/protobuf/__init__.py"
[... cut ...] a different outer class name for the .proto file.
protoc create build/java/machinetalk/protobuf/test.java from src/machinetalk/protobuf/test.proto
protoc create build/java/machinetalk/protobuf/object.java from src/machinetalk/protobuf/object.proto
protoc create build/java/machinetalk/protobuf/status.java from src/machinetalk/protobuf/status.proto
protoc create build/java/machinetalk/protobuf/canon.java from src/machinetalk/protobuf/canon.proto
rm -rf build
make: Nothing to be done for 'all'.
```

zhivko, 2016-12-14:
Ah ... :) nonsense - it seems build script should not be passed multiple goals like ```all``` and ```clean```.
it should be issed like 
```
make clean
make
```

ArcEye, 2016-12-14:
The testing needs to be for a basic RIP build only, as I said above.
There is no point trying something exotic that no one else uses.

You will have to refer to @machinekoder for any queries regards java protobuff, 
the repo contains the same `machinetalk::` prefixed code as the `master` does.

ArcEye, 2016-12-14:
> I'm rebasing a cleaned-up branch unto latest master. So far I need to fix some locations of .proto files.

OK, you will have to let me know what files are necessary and how the component is built.  
There is no Submakefile and it does not appear in the Makefile.
Are you compiling outside the main build?

zhivko, 2016-12-14:
> "testing needs to be for a basic RIP build only"
What kind of testing? As the build itself is concerned build of this branch on my side was succesfull.
But to ultimately confirm something is not broken shouldn't we test also use of changed protobuff stuff (communication and usage - and this is what I tried to do) as one of the test scenarios?

ArcEye, 2016-12-14:
The protobuf stuff is not changed in this branch, it is the same as was committed to master.

You need to test first with a new pull of the master, if it works with that and not with this branch, that is a problem we can look at.
It looks like the java protobuff namespace is not the same as the one in master.  That needs to be sorted first.

> What kind of testing?

Going into your existing configs and replacing the components used, with v2 versions (where they exist) and testing.

Converting your own components to use pin_ptrs instead of pins and using the accessors to read and write to them (lots of examples in the /hal/i_components dir)

luminize, 2016-12-14:
I've added the files for the interpolator and jplan in this branch https://github.com/luminize/machinekit/tree/interpolator-cleanup-rebase.
- I took @mhaberler his original branch ( https://github.com/mhaberler/machinekit/tree/interpolator ) minus the 3 top mux component commits (haven't tried nor tested that yet)
- applied some fixes on top of that
- rebased onto master from recent commit 0db30dcff887f9056389093d65e5137b097633f1
- fixed some paths wrt proto file locations, Makefile merging stuff, and include paths in the interpolator and jplan source

 I seem to run into a build issue like this message:
```
Compiling realtime hal/jplanner/jplan.c
Linking ../rtlib/posix/ufdemo.so
Linking ../rtlib/posix/lutn-demo.so
Compiling realtime hal/interpolator/interpolate.c
hal/jplanner/jplan.c: In function ‘rtapi_app_main’:
hal/jplanner/jplan.c:257:4: warning: passing argument 4 of ‘hal_xinit’ from incompatible pointer type
    instantiate_jplan,
    ^
In file included from hal/jplanner/jplan.c:4:0:
./hal/lib/hal.h:252:5: note: expected ‘hal_constructor_t’ but argument is of type ‘int (*)(const char *, const int,  const char **)’
 int hal_xinit(const int type,
     ^
Compiling realtime emc/tp/tpmain.c
hal/interpolator/interpolate.c: In function ‘rtapi_app_main’:
hal/interpolator/interpolate.c:323:4: warning: passing argument 4 of ‘hal_xinit’ from incompatible pointer type
    instantiate_interpolate,
    ^
In file included from hal/interpolator/interpolate.c:4:0:
./hal/lib/hal.h:252:5: note: expected ‘hal_constructor_t’ but argument is of type ‘int (*)(const char *, const int,  const char **)’
 int hal_xinit(const int type,
     ^
```

What would you want me to do to get this into your multicore branch?

[159 more replies]

closed 2017-02-07 by ArcEye