machinekit/machinekit #1080: PRU debugger cannot find files

cdsteinkuehler, 2016-10-15:
When using the PRU debugger, it cannot find the source files specified in the <prucode>.dbg file, because it is looking for them with a 'dot' in front of them instead of for the actual file name.  I have taken a quick look at this, but my Python skills are not up to the challenge of figuring out what's going wrong.

Note when the user component is launched, it cannot find the file '.pruexample.p', but it should be looking for 'pruexample.p' (without the leading dot).  If symlinks are created so the 'dot' filename points to the real filename, everything works as expected.

Seems like a simple fix for someone who knows Python well.

````
machinekit@beaglebone:~$ cd
machinekit@beaglebone:~$ rsync -Pva /usr/share/linuxcnc/examples/sample-configs/ARM/PRU-Debugger .
sending incremental file list
PRU-Debugger/

sent 147 bytes  received 20 bytes  334.00 bytes/sec
total size is 46,287  speedup is 277.17
machinekit@beaglebone:~$ cd PRU-Debugger/
machinekit@beaglebone:~/PRU-Debugger$ pasm -b -L -d -V2 pruexample.p

PRU Assembler Version 0.86
Copyright (C) 2005-2013 by Texas Instruments Inc.

Pass 2 : 0 Error(s), 0 Warning(s)

Writing Code Image of 72 word(s)

machinekit@beaglebone:~/PRU-Debugger$ halcmd loadrt hal_pru_generic prucode=/home/machinekit/PRU-Debugger/pruexample.bin disabled=1
machinekit@beaglebone:~/PRU-Debugger$ halcmd loadrt hal_prudebug
machinekit@beaglebone:~/PRU-Debugger$ halcmd loadusr -Wn ui gladevcp -u ui.py -c ui  -U filename=\"pruexample.dbg\" prudebug.ui
Waiting for component 'ui' to become ready.......gladevcp: trouble looking for handlers in 'ui': [Errno 2] No such file or directory: '.pruexample.p'
Traceback (most recent call last):
  File "/usr/bin/gladevcp", line 164, in load_handlers
    objlist = h(halcomp,builder,useropts, compname)
  File "./ui.py", line 218, in get_handlers
    return [HandlerClass(halcomp,builder,useropts,compname)]
  File "./ui.py", line 210, in __init__
    self.load_file(filename)
  File "./ui.py", line 154, in load_file
    self.dbtxt = DebugText(filename)
  File "./ui.py", line 94, in __init__
    self.decode_filenames()
  File "./ui.py", line 43, in decode_filenames
    self.files[i] = open(fname, "r").read().splitlines()
IOError: [Errno 2] No such file or directory: '.pruexample.p'
/usr/bin/gladevcp:297: RuntimeWarning: missing handler 'populate_popup'
  builder.connect_signals(handlers)
/usr/bin/gladevcp:297: RuntimeWarning: missing handler 'pc_changed'
  builder.connect_signals(handlers)
/usr/bin/gladevcp:297: RuntimeWarning: missing handler 'file_set'
  builder.connect_signals(handlers)
.
machinekit@beaglebone:~/PRU-Debugger$ ls -la
total 96
drwxr-xr-x  2 machinekit machinekit  4096 Oct 15 17:12 .
drwxr-xr-x 26 machinekit machinekit  4096 Oct 15 17:10 ..
-rw-r--r--  1 machinekit machinekit   148 Oct 15 15:20 Makefile
-rw-r--r--  1 machinekit machinekit 30956 Oct 15 15:20 prudebug.ui
-rw-r--r--  1 machinekit machinekit   288 Oct 15 17:12 pruexample.bin
-rw-r--r--  1 machinekit machinekit  2068 Oct 15 17:12 pruexample.dbg
-rw-r--r--  1 machinekit machinekit  6697 Oct 15 15:20 pruexample.p
-rw-r--r--  1 machinekit machinekit 13454 Oct 15 17:12 pruexample.txt
-rw-r--r--  1 machinekit machinekit  1075 Oct 15 15:20 README
-rw-r--r--  1 machinekit machinekit  7411 Oct 15 15:20 ui.py
-rw-r--r--  1 machinekit machinekit  7601 Oct 15 17:12 ui.pyc
machinekit@beaglebone:~/PRU-Debugger$ ln -s pruexample.p .pruexample.p
machinekit@beaglebone:~/PRU-Debugger$ halcmd loadusr -Wn ui gladevcp -u ui.py -c ui  -U filename=\"pruexample.dbg\" prudebug.ui
Waiting for component 'ui' to become ready........
machinekit@beaglebone:~/PRU-Debugger$ ls -la
total 96
drwxr-xr-x  2 machinekit machinekit  4096 Oct 15 17:14 .
drwxr-xr-x 26 machinekit machinekit  4096 Oct 15 17:10 ..
-rw-r--r--  1 machinekit machinekit   148 Oct 15 15:20 Makefile
-rw-r--r--  1 machinekit machinekit 30956 Oct 15 15:20 prudebug.ui
-rw-r--r--  1 machinekit machinekit   288 Oct 15 17:12 pruexample.bin
-rw-r--r--  1 machinekit machinekit  2068 Oct 15 17:12 pruexample.dbg
lrwxrwxrwx  1 machinekit machinekit    12 Oct 15 17:14 .pruexample.p -> pruexample.p
-rw-r--r--  1 machinekit machinekit  6697 Oct 15 15:20 pruexample.p
-rw-r--r--  1 machinekit machinekit 13454 Oct 15 17:12 pruexample.txt
-rw-r--r--  1 machinekit machinekit  1075 Oct 15 15:20 README
-rw-r--r--  1 machinekit machinekit  7411 Oct 15 15:20 ui.py
-rw-r--r--  1 machinekit machinekit  7601 Oct 15 17:12 ui.pyc
machinekit@beaglebone:~/PRU-Debugger$
````

cdsteinkuehler, 2016-10-16:
The extra 'dot' is present in the files generated by the BBB default installed pasm (/usr/bin/pasm) but is not present in the pasm included with machinekit (available if you do a RIP build).  So I've got to figure out what's up with the difference, and if a bug crept into the official BBB pasm.

cdsteinkuehler, 2016-10-17:
Since this is not a problem with Machinekit, I am closing the issue here.  Follow along in beagleboard/am335x_pru_package#53 if interested in the ultimate outcome.

closed 2016-10-17 by cdsteinkuehler

zultron, 2016-10-17:
I got sucked in by this one and submitted a potential fix; see the [other issue][1].

[1]: https://github.com/beagleboard/am335x_pru_package/issues/53