machinekit/machinekit #1083: halscope and halmeter  not built after 0b36e4c ?

sirop, 2016-10-20:
Hello.

The RIP build on xenomai amd64 does not produce halscope and halmeter.

Reverting  to https://github.com/machinekit/machinekit/commit/5d17a92e914a2dabe6560dfdc843a03d6edfdb09 yields both halmeter and halscope.

Make log for  https://github.com/machinekit/machinekit/commit/d531ae9b55bfd7ccb7c9c3eaba69bc5b34999b03 : 
https://gist.github.com/sirop/018a41cf3aa30fa44162cafe83db9386
Make log for https://github.com/machinekit/machinekit/commit/5d17a92e914a2dabe6560dfdc843a03d6edfdb09 :
https://gist.github.com/sirop/91c39c9f73f89add63779e4ee069beaa

ArcEye, 2016-10-21:
That is rather bizarre, I can see that halmeter and halscope are not being linked, but struggling to see what removing the attempt right at the end of the Makefile to stat linuxcnc_module_loader (which now does not get built for non kthreads) could do to cause it.

I cannot do anything today, so could I ask you to manually restore the Makefile as it was, into  d531ae9 and rebuild.
That at least should eliminate some other untracked corruption or change in that '\<SHA\>'

sirop, 2016-10-21:
My "suspect" is https://github.com/machinekit/machinekit/commit/0b36e4ceedca3216dfaea1b7545406afa133c236
that's why I reverted to the one commit before https://github.com/machinekit/machinekit/commit/5d17a92e914a2dabe6560dfdc843a03d6edfdb09 .

ArcEye, 2016-10-21:
OK, I wasn't at a proper computer before, just assumed you had gone back one commit.

I just tested with a build of the latest (d531ae9) on rt-preempt and posix and it does not build halmeter or halscope.

It builds halrmt, which is directly above it in the Submakefile (src/hal/utils) and the buildspecs for halrmt and halmeter are pretty much the same.

I can only assume it is connected to the `#ifdef HAVE_GTK` block that the meter and scope build sits in, both requiring miscgtk.o

@zultron -  something to do with `HAVE_GTK` not being set as expected, due to changed tests for tk / tcl ?

ArcEye, 2016-10-22:
Autoconf is arcane at the best of times.

Found by trial and error that `HAVE_GTK` was set properly when all the args inside the `PKG_CHECK_MODULES` test were square bracketed ( [ ] ).

Now halmeter, halscope, classicladder and all the other gtk dependent binaries are built

PR  #1084 refers

ArcEye, 2016-10-22:
PR #1084 has passed

@sirop
Once it has been merged, could you please test on the xenomai machine that showed the problem and if OK, close the issue.
Thanks

sirop, 2016-10-22:
ok, i will do that.

closed 2016-10-22 by luminize

sirop, 2016-10-22:
Confirmed.
```
master@debian-master:~/machinekit/src$ git log | head -15
commit c0fff168f63968943532339f1fd813d018054bc4
Merge: d531ae9 1872d9b
Author: Bas de Bruijn <bdebruijn@luminize.nl>
Date:   Sat Oct 22 16:14:40 2016 +0200

    Merge pull request #1084 from ArcEye/configure-ac-gtk

    Amend configure.ac HAVE_GTK test to use square bracketed args.

commit 1872d9b97681b089bc35a088ca578d25163f610f
Author: Mick <arceye@mgware.co.uk>
Date:   Sat Oct 22 13:30:57 2016 +0100

    Amend configure.ac HAVE_GTK test to use square bracketed args.

master@debian-master:~/machinekit/src$ ls ../bin
5axisgui       encdec        halcmd                halsampler        halui           linmove      mank        npbdecode     rostock        teach-in       webtalk
adxl345        genserkins    hal_gpio_mcp23017     halscope          hbmgui          lintini      max5gui     pncconf       rs274          tooledit       xhc-hb04
axis           gladevcp      hal_gy86              halshow           hexagui         linuxcnclcd  mb2hal      position      rtprintf       touchy         xlinuxcnc
axis-remote    gmoccapy      hal_input             hal_storage       image-to-gcode  linuxcncrsh  mdi         profile_axis  scaragui       tracking-test  yapps
classicladder  gremlin       hal_manualtoolchange  halstreamer       instcomp        linuxcncsvr  milltask    puma560gui    schedrmt       unionread
comp           gremlin_view  halmeter              haltalk           io              linuxcnctop  mklauncher  pumagui       shuttlexpress  vfdb_vfd
configserver   gs2_vfd       hal_pwm_pca9685       hal_temp_ads7828  iov2            machinekit   mkwrapper   pyngcgui      sizes          vfs11_vfd
debuglevel     gscreen       halrmt                hal_temp_bbb      keystick        maho600gui   ngcgui      pyvcp         stepconf       videoserver
```

ArcEye, 2016-10-22:
Thanks @sirop

luminize, 2016-10-25:
I just merged the PR.

PR #1084 (2016-10-22): Amend configure.ac HAVE_GTK test to use square bracketed args.
Changes in 0b36e4c did not work as anticipated and HAVE_GTK
was not defined, resulting in halmeter, halscope, classicladder
and all gtk dependent binaries not being built

Fixes #1083
commit 1872d9b9 (2016-10-22): Amend configure.ac HAVE_GTK test to use square bracketed args.

Changes in 0b36e4c did not work as anticipated and HAVE_GTK
was not defined, resulting in halmeter, halscope, classicladder
and all gtk dependent binaries not being built

Fixes #1083