machinekit/machinekit #1088: traj pins missing in halmeter

zhivko, 2016-10-26:
I miss: 
MAX_ANGULAR_VELOCITY
and
DEFAULT_ANGULAR_VELOCITY

[image]

I am using ARM.BeagleBone.CRAMPS configuration for ini and hal files.

Maybe somebody knows the reason?

ArcEye, 2016-10-27:
AFAIK, nothing has changed (there hasn't been an edit to meter.c since 2012)

Halmeter displays pins, params and signals
MAX_ANGULAR_VELOCITY and DEFAULT_ANGULAR_VELOCITY
are just ini file values

You will not find them with the Hal Configuration window or by doing this with machinekit running in the background either.
```
mick@INTEL-i7:/usr/src/machinekit$ halcmd show pin ini.traj
Component Pins:
  Comp   Inst Type  Dir         Value  Name                                             Epsilon         Flags
   799        float IN          1e+99  ini.traj_default_acceleration            	0.000010	0
   799        float IN             10  ini.traj_default_velocity                	0.000010	0
   799        float IN          1e+99  ini.traj_max_acceleration                	0.000010	0
   799        float IN          1e+99  ini.traj_max_velocity                    	0.000010	0
```

If you need the values, you need to read them as axis does for instance
```
inifile.find("TRAJ","MAX_ANGULAR_VELOCITY")
```

closed 2016-10-27 by ArcEye

zhivko, 2016-10-27:
UPs... sorry - maybe I stated wrong question - My question was more like "Why" they are not showed - since ini.traj_max_velocity is showed...
ini.traj_max_velocity
why
ini.traj_max_angular_velocity

I do not only to read them but want to change them in run-time rather than in offline when MK is not started. If I could change traj_max_velocity through remote components, I would also like to change traj_max_angular_velocity.
I guess there is some analogy here.

ArcEye, 2016-10-27:
It is a long time since I looked at something similar, but I don't think you can change the values on the fly, because they are cached and those cached values are read until MK stops and reloads them on restart.

There was a thread on it in the Linuxcnc forum, from someone who wanted to restrict rapids at corners with a water cutting machine, don't think it was possible to do what he wanted that way.

Why they are not pins, I have no idea, you would have to ask whoever at Linuxcnc decided to leave them as values, or delve into the code and change it.

zhivko, 2016-10-27:
SO I guess this is in:

./git/machinekit/src/emc/ini/inihal.cc:    INIT_PIN(traj_max_acceleration);
./git/machinekit/src/emc/ini/inihal.cc:    if
(CHANGED(traj_max_acceleration)) {
./git/machinekit/src/emc/ini/inihal.cc:        if (debug)
SHOW_CHANGE(traj_max_acceleration)
./git/machinekit/src/emc/ini/inihal.cc:
 UPDATE(traj_max_acceleration);
./git/machinekit/src/emc/ini/inihal.cc:        if (0 !=
emcTrajSetMaxAcceleration(NEW(traj_max_acceleration))) {

 ?

zhivko, 2016-10-27:
I am trying to find
Max_angular_velocity in documentation of linuxcnc in
http://linuxcnc.org/docs/2.5/html/config/ini_config.html#sub:TRAJ-section
And I cannot find it there...
Is this MK specific?

There is only max_velocity and max_acceleration there and it refers to any
axis (also angular).
In mk there are max linear and max angular velocities and I wonder why.