machinekit/machinekit #1090: CRAMPS ini configuration inconsistent with linuxcnc docs

zhivko, 2016-10-28:
CRAMPS ini configuration:
https://github.com/machinekit/machinekit/blob/master/configs/ARM/BeagleBone/CRAMPS/CRAMPS.ini
has 2 additional items:
MAX_ANGULAR_VELOCITY
and
DEFAULT_ANGULAR_VELOCITY
that are additional regarding to linuxcnc:
[linuxcnc traj section](http://linuxcnc.org/docs/2.5/html/config/ini_config.html#sub:TRAJ-section)

this 2 pins are not exposed as linuxcnc pins after MK is started as:
\machinekit\src\emc\ini\inihal.cc
doesn't expose them.

Should the correct be to modify inihal.cc to expose those pins?

The conflict is that linuxcnc documentation says:
MAX_VELOCITY = 5.0 - The maximum velocity for **any** axis or coordinated move, in machine units per second. The value shown equals 300 units per minute.
This means that MAX_VELOCITY is also applicable to rotational axis so MAX_ANGULAR_VELOCITY is not necessary.
Same is true for DEFAULT_ANGULAR_VELOCITY - it seems to be not needed DEFAULT_VELOCITY should be sufficient.
On other hand CRAMPS.ini is missing MAX_[ANGULAR]_ACCELERATION.

Perhaps @cdsteinkuehler can elaborate a bit, why this was needed and what is then true when limiting angular axis.

cdsteinkuehler, 2016-10-28:
On 10/28/2016 4:09 AM, zhivko wrote:
> 
> Perhaps @cdsteinkuehler <https://github.com/cdsteinkuehler> can elaborate a bit, 
> why this was needed and what is then true when limiting angular axis.

The CRAMPS configuration was copied and modified from the
configuration for my personal 3D printer, which started out well
before Machinekit split from LCNC, and even before running on ARM
boards (my first setup used an x86 parallel port with a bit-banged I2C
ADC for temperature sensing).  I also haven't used a rotary axis and
wouldn't have noticed any missing angular config settings.

So feel free to make appropriate changes and send a PR!

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-10-28:
I think you need to check whether it is necessary ( by removing and running and checking if there are any differences ).
Although Axis contains code to read the ini file for these values, the only other place they are referred to is in gscreen and mkwrapper.py  (ie in python code )
DEFAULT_XXXX_VELOCITY refers to jog speed default used in the Axis GUI, 
only MAX_XXXX_VELOCITY has any potential for altering rapid speeds

Then one comes to the situation alluded to in the docs, where any coordinated move is governed by MAX_ACCELERATION (with an 20% overhead in MAX_STEPGEN_ACCELERATION to try to allow the stepgen to move all axes in a coordinated manner whilst maintaining the specified feed rate).
Very often rotary axes are perceived as slow, because of the coordination to finish moves when all axes reach the commanded location. 

The other issue that I mentioned previously, is that AFAIR, the values are read in and cached when a program is started, so having access to a pin will not enable you to change values on the fly, because the cached values are used throughout the run.

zhivko, 2016-10-30:
I already prepared changes in ini to remove _angular_ settings. As you say and i checked in src/emc/*.cc it seems values are not used anywhere to influence motion.
Now I need to fix pru_generic encoder issue #1093,  to confirm machine is moving same without *_angle_* settings in [traj] section.

zhivko, 2016-10-30:
OK As I found out I removed all _angle_ settings from [traj] section of ini file and linuxcnc runs OK without those values.
So for me issue closed...

closed 2016-10-30 by zhivko