machinekit/machinekit #1093: Unable to set parameter on thc.icomp

zhivko, 2016-10-29:
I tried to set  scale_offset parameter on thc icomponent within hal like this:
```
newinst thc thc.0 scale_offset=119000.0
```
but I get:
```
THC.hal:4: Realtime module 'thc' loaded
THC.hal:4: rc=-1
Unknown parameter `scale_offset=119000.0'
passing args for 'thc.0' failed: 'scale_offset=119000.0
Shutting down and cleaning up Machinekit...
```
In thc.icomp I find:
```
// Parameters
pin io float vel_scale "The scale to convert the Velocity signal to Volts";
pin io float scale_offset "The offset of the velocity input at 0 volts";
```
so scale_offset is parameter - what is wrong with my newinst call in my hal file?

If I try with moving parameter out of newinst to setp in my hal file like:
```
setp thc.0.scale_offset 119000.0
```
I get:
```
THC.hal:46: setting parameter 'thc.0.scale_offset' to '119000.0'
THC.hal:46: parameter or pin 'thc.0.scale_offset' not found
Shutting down and cleaning up Machinekit...
```

cdsteinkuehler, 2016-10-29:
On 10/29/2016 7:48 AM, zhivko wrote:
> In thc.icomp I find:
>
> |// Parameters pin io float vel_scale "The scale to convert the Velocity signal
> to Volts"; pin io float scale_offset "The offset of the velocity input at 0
> volts"; |
>
> so [...]

comp and icomp typically turn underscores into dashes when creating
HAL names.  Try:

setp thc.0.scale-offset 119000.0

...if that doesn't work, just manually load the thc component with
halcmd and examine the pins and parameters with "halcmd show".

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-10-29:
@zhivko wrote:
> newinst thc thc.0 scale_offset=119000.0

You can only use `count=N names=....., .....` and `pincount=N` as args to a newinst call, plus other args after a --- separator, which must be resolved in your component code

> setp thc.0.scale-offset 119000.0

As @cdsteinkuehler  says, it is a parameter, so set after loading with setp

> ...if that doesn't work, just manually load the thc component with halcmd and examine the pins and parameters with "halcmd show".

When I looked at this, to do just that, I found something a bit alarming.
Loading `thc` or `thcud` locked up halcmd, but no obvious error in linuxcnc.log

I then wrote a test script for all the i_components, and found that thc(ud) alone did this,
remove them and all the others load and unload.

```
# !/bin/bash
############  icomp_tester  ################

oldpwd=$PWD
cd /usr/src/machinekit/src/hal/i_components # (change this to your own path)
DEBUG=5 realtime restart
for i in *.icomp
    {
    module=$(basename -s .icomp "$i")
    echo "Name is $module" ;
    halcmd loadrt $module ;
    halcmd show pin $module ;
    halcmd unloadrt $module ;
    };
halrun -U
halrun -R
cd $oldpwd
```

Run with ./icomp_tester | tee test.result
If you have the same problem it will only get as far as `sum2` and lock on `thc`

@zhivko can you run this script on your copy of machinekit, so I can be sure I have not got something wrong locally
At first looks I cannot see anything wrong with the code, especially as it is just being loaded and not even added to a thread.

zhivko, 2016-10-29:
I tried the script and when it comes to thc I get this:
```
Name is thc
<commandline>:0: Realtime module 'thc' loaded
Component Pins:
  Comp   Inst Type  Dir         Value  Name                                             Epsilon         Flags
  1662   1663 bit   IN          FALSE  thc.0.arc-ok                                                     0
  1662   1663 float I/O             0  thc.0.correction-vel                             0.000010        0
  1662   1663 float IN              0  thc.0.current-vel                                0.000010        0
  1662   1663 bit   IN          FALSE  thc.0.enable                                                     0
  1662   1663 float IN              0  thc.0.encoder-vel                                0.000010        0
  1662   1663 float IN              0  thc.0.requested-vel                              0.000010        0
  1662   1663 float I/O             0  thc.0.scale-offset                               0.000010        0
  1662   1663 s32   OUT             0  thc.0.time                                                       0
  1662   1663 bit   IN          FALSE  thc.0.torch-on                                                   0
  1662   1663 float I/O             0  thc.0.vel-scale                                  0.000010        0
  1662   1663 bit   OUT         FALSE  thc.0.vel-status                                                 0
  1662   1663 float IN              0  thc.0.vel-tol                                    0.000010        0
  1662   1663 float I/O             0  thc.0.velocity-tol                               0.000010        0
  1662   1663 float I/O             0  thc.0.voltage-tol                                0.000010        0
  1662   1663 float OUT             0  thc.0.volts                                      0.000010        0
  1662   1663 float IN              0  thc.0.volts-requested                            0.000010        0
  1662   1663 float OUT             0  thc.0.z-fb-out                                   0.000010        0
  1662   1663 float IN              0  thc.0.z-pos-in                                   0.000010        0
  1662   1663 float OUT             0  thc.0.z-pos-out                                  0.000010        0

<commandline>:0: Realtime module 'thc' unloaded
```
The for loop doesn't lock anywhere on my side. I attached test.result.
[test.txt](https://github.com/machinekit/machinekit/files/560078/test.txt)

Also I try setting parameters - now with
```
newinst thc thc.0
addf thc.0 servo-thread

##############################
# HAL PLASMA SETUP
##############################
setp thc.0.scale_offset 119000.0
setp thc.0.vel_scale 0.00037866834
setp thc.0.velocity_tol 3.0
setp thc.0.voltage_tol 2.0
setp thc.0.correction_vel 0.0001
```
but with same error:
```
THC.hal:11: Realtime module 'thc' loaded
THC.hal:12: Function 'thc.0' added to thread 'servo-thread'
THC.hal:17: setting parameter 'thc.0.scale_offset' to '119000.0'
THC.hal:17: parameter or pin 'thc.0.scale_offset' not found
```

I have similar problem with hpg encoder (maybe not related to this issue topic, but in hal I have:
```
THC.hal:18: setting parameter 'hpg.encoder.00.out.00.counter-mode' to '2'
THC.hal:18: parameter or pin 'hpg.encoder.00.out.00.counter-mode' not found
Shutting down and cleaning up Machinekit...
```

cdsteinkuehler, 2016-10-30:
On 10/29/2016 1:21 PM, zhivko wrote:
> 
> but with same error:
> 
> |THC.hal:11: Realtime module 'thc' loaded THC.hal:12: Function 'thc.0' added to 
> thread 'servo-thread' THC.hal:17: setting parameter 'thc.0.scale_offset' to 
> '119000.0' THC.hal:17: parameter or pin 'thc.0.scale_offset' not found |

Re-read my previous post.  You are using an underscore (scale_offset)
but the pin name uses a dash (scale-offset).  That's why you are
getting this error.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2016-10-30:
> I tried the script and when it comes to thc I get this:
Name is thc
<commandline>:0: Realtime module 'thc' loaded
Component Pins:
  Comp   Inst Type  Dir         Value  Name   

I'm glad about that, just need to find out what is screwy with my copy now :frowning_face:

zhivko, 2016-10-30:
Yes you are correct, underscoes in param names are changed to dashes for
usage in hal.
Knowing that I'm able to set parameter values now with setup.

Maybe just a reminder for _ to - replacement in documentation - it should
be mentioned in docs with warning ! sign.

Should I open new issue for setting pins on encoder within
xenomai/pru_generic.bin?

ArcEye, 2016-10-30:
> Maybe just a reminder for _ to - replacement in documentation - it should be mentioned in docs with warning ! sign.

I think what has happened since the removal of man pages and direct .asciidoc generation, is that one of the conversion functions in instcomp is not being used.

At some point the pin entries are being generated with for instance
`print >> f, "%s.N.%s" % (comp_name, name)`
whereas it should be
`print >> f, "%s.N.%s" % (comp_name, to_hal(name))`

I will look at it before too long.

ArcEye, 2016-10-30:
>  I'm glad about that, just need to find out what is screwy with my copy now :frowning_face:

I am not going mad, there IS a problem with the `thc` and `thcud` components.

see #1094

zhivko, 2016-10-30:
OK find out with:
```
halcmd show pin hp.*
```

should be named:
```
setp hpg.encoder.00.chan.00.counter-mode 	2
setp hpg.encoder.00.chan.00.B-pin			16
setp hpg.encoder.00.chan.00.B-pin			27
setp hpg.encoder.00.chan.00.index-pin		27
setp hpg.encoder.00.chan.00.scale			1
setp hpg.encoder.00.chan.00.vel-timeout      3.0
```

So it seems not correct with documentation at http://www.machinekit.io/docs/man/man9/hal_pru_generic/ (or I do not get it right):
It says:
"
Encoders have names like "hpg.encoder.<Instance>.out.<Channel>"
"

As for the encoder pin numbering scheme - in documentation it says numbering is special so maybe @cdsteinkuehler can help me with what should I used for 
CRAMPS_P503_2 MISO
or
P929
pin ?

cdsteinkuehler, 2016-10-30:
On 10/30/2016 11:02 AM, zhivko wrote:
> 
> As for the encoder pin numbering scheme - in documentation it says numbering is 
> special so maybe @cdsteinkuehler <https://github.com/cdsteinkuehler> can help me 
> with what should I used for
> CRAMPS_P503_2 MISO
> or
> P929
> pin ?

For the encoder, you use the PRU direct input pin number.  P929 is
signal pr1_pru0_pru_r31_1, so you would use "1".

This should probably be migrated to allow using normal 8xx/9xx numbers
like everything else...pull requests welcome, otherwise I'll probably
get to it someday.

-- 
Charles Steinkuehler
charles@steinkuehler.net

zhivko, 2016-10-30:
Yes that would be really great that other naming scheme could be used... Do
you have formula or this ODS you posted long time ago that maps pins
according different schemes?

cdsteinkuehler, 2016-10-30:
On 10/30/2016 11:44 AM, zhivko wrote:
> Yes that would be really great that other naming scheme could be used... Do
> you have formula or this ODS you posted long time ago that maps pins
> according different schemes?

There is no formula, just the PRU direct in register bit number, which
is available from a variety of places.  The canonical reference is the
AM335x data-sheet, but I find the pinmux spreadsheet easier to use:

https://github.com/cdsteinkuehler/beaglebone-black-pinmux/blob/hal_pru_generic/pinmux.ods

-- 
Charles Steinkuehler
charles@steinkuehler.net

zhivko, 2016-11-05:
Character "_" in pin name was changed to "-". Having this in mind I was abl e to set pin values with

closed 2016-11-05 by zhivko