machinekit/machinekit #1094: option singleton causes halcmd lock up with icomps

ArcEye, 2016-10-30:
Ref  #1093

I am not going mad, there IS a problem with the `thc` and `thcud` components.

If you do a fresh clone and build then
`DEBUG=5 realtime restart`
`halcmd loadrt thc`

I get halcmd locking with high single CPU activity until I kill the process

`/var/log/linuxcnc.log` just shows that thc was loaded from thc.so, no error at all

I was rather a loss as to what could be causing it either locally or globally

By process of elimination, discovered that commenting out
`option singleton yes`
cures it

This only occurs in thc and thcud and is an anachronism from the .comp file, since singleton does not really have a useful meaning in relation to instantiated components.  
(If you just want one, don't create more than one)

There is a kernel module marker put in components that use that option and halcmd_commands.c
has code that checks for existing instances if it is found.
That is where I need to look next.

zhivko, 2016-10-30:
Its funny how it doesnt happen on bbb.
So commenting out singleton fixes it?

closed 2016-10-30 by cdsteinkuehler

ArcEye, 2016-10-30:
Yes, albeit I have now deprecated the whole `option singleton` so it definitely won't happen again.

PR #1095 (2016-10-30): Deprecate `option singleton` in instantiated components
The option was originally supported for backward compatibility,
however it really contradicts the basic concept of instantiation.

If a component cannot ever have more than one instance, it should
be written as a standard .comp

Arose initially from Issue #1094 where checking for other instances
of a component marked as a singleton, without proper mutex lock
was causing problems
On reflection the whole concept within instcomps was removed.

Whilst editing instcomp, corrected an error that had crept in during
changes to document generation [... cut]
commit 9b84d0d1 (2016-10-30): Deprecate `option singleton` in instantiated components

The option was originally supported for backward compatibility,
however it really contradicts the basic concept of instantiation.

If a component cannot ever have more than one instance, it should
be written as a standard .comp

Arose initially from Issue #1094 where checking for other instances
of a component marked as a singleton, without proper mutex lock
was causing problems
On reflection the whole concept within instcomps was removed.

Whilst editing instcomp, corrected an error that had crept in during
changes to document generatio [... cut]