machinekit/machinekit #1097: legacy component 'thc' loaded, but not instantiable

zhivko, 2016-11-05:
I pulled last MK sources, build MK and I get following error when using

```
newinst thc thc.0
addf thc.0 servo-thread 
```
it outputs:
```
THC.hal:16: legacy component 'thc' loaded, but not instantiable
Shutting down and cleaning up Machinekit...
handle_command_input: req=type: MT_PING
```

Maybe wrong usage from my side? Or something new that came in with last @ArcEye changes in thc.icomp?

zhivko, 2016-11-05:
OK it seems it was schanged to normal component.loadrt thc

I was able to load it with:
```
loadrt thc
```

but it refues to load with:
```
loadrt thc names=thc0
```
I get:
```
THC.hal:17: insmod failed, returned -1:
Unknown parameter `names=thc0'
```

ArcEye, 2016-11-06:
thc and thcud, were conceived of and written as singleton comps.

In the mass conversions to instantiated components, they were converted, but extra code had to be added t o halcmd to prevent more than one being loaded.
This is contrary to the basic premise of instcomps, so when there were other problems with 'singleton' I reverted the thc and thcud components back to standard comps, which can be singletons.

singleton comp components do not take count= names= args, because that is meaningless, why would you number a component when you can only have one of them?
It should produce a component called 'thc' or 'thcud' respectively.

It happens that previously you could rename by using newinst, but not now.

zhivko, 2016-11-06:
OK.

ArcEye, 2016-11-06:
```
mick@INTEL-i7:/usr/src/machinekit$ DEBUG=5 realtime restart
mick@INTEL-i7:/usr/src/machinekit$ halcmd loadrt thc
<commandline>:0: Realtime module 'thc' loaded
mick@INTEL-i7:/usr/src/machinekit$ halcmd show pin thc
Component Pins:
  Comp   Inst Type  Dir         Value  Name                                             Epsilon         Flags
    73        bit   IN          FALSE  thc.arc-ok                               			0
    73        float IN              0  thc.current-vel                          	0.000010	0
    73        bit   IN          FALSE  thc.enable                               			0
    73        float IN              0  thc.encoder-vel                          	0.000010	0
    73        float OUT             0  thc.offset-value                         	0.000010	0
    73        float IN              0  thc.requested-vel                        	0.000010	0
    73        s32   OUT             0  thc.time                                 			0
    73        bit   IN          FALSE  thc.torch-on                             			0
    73        bit   OUT         FALSE  thc.vel-status                           			0
    73        float IN              0  thc.vel-tol                              	0.000010	0
    73        float OUT             0  thc.volts                                	0.000010	0
    73        float IN              0  thc.volts-requested                      	0.000010	0
    73        float OUT             0  thc.z-fb-out                             	0.000010	0
    73        float IN              0  thc.z-pos-in                             	0.000010	0
    73        float OUT             0  thc.z-pos-out                            	0.000010	0
```

zhivko, 2016-11-06:
Yes you are correct I was able to work with loart thc... it reates thc named component.

zhivko, 2016-11-06:
@arce regarding: "why would you number a component when you can only have one of them?"
I have now a case that in would be good to rename instance of component so it is not named like its original name.
I copied thc.comp to mythc.comp, now hal automatically names component as mythc, but all my hal configuration is done as thc.[pin name] - so now I need to go through all net names and change thc.[pin name] to mythc.[pin name]
So yes - it gives integrator much more flexibility if one could just name a component with ```names=...``` syntax that kind of change would be welcomed.

ArcEye, 2016-11-06:
For normal use, the convention for .comp singleton components is as is presently.
The removal of args which could be used to try to circumvent the singleton tag, is deliberate.

No-one is likely to be using 2 plasma torches at the same time, in the same configuration and since the component takes no account of that possibility, it is tagged as `singleton`.

If you have a special developmental case where you wish to differentiate between different configs, by giving different names to the component, the method you chose is probably best.
It is not going to be a situation you share with (m)any others.

luminize, 2016-11-06:
> On 06 Nov 2016, at 14:22, zhivko <notifications@github.com> wrote:
> 
> @arce <https://github.com/arce> regarding: "why would you number a component when you can only have one of them?"
> I have now a case that in would be good to rename instance of component so it is not named like its original name.
> I copied thc.comp to mythc.comp,
> 
is this only about the name “mythc”?
> now hal automatically names component as mythc, but all my hal configuration is done as thc.[pin name] - so now I need to go through all net names and change thc.[pin name] to mythc.[pin name]
> 
Simplest would be to use your editor function and bulk-replace thc to mythc if you’re insistent on the mythc name.

in other words is that all worth the trouble for just a name?

zhivko, 2016-11-06:
I agree that this is minor issue - but I thought if linuxcnc has this functionality already for naming components (with names= word behind loadrt) why we are now deliberately throwing this functionality away?

@luminize yes this is exactly what i did, but with multiple hal files it can be few trial & error with restarting MK in between

luminize, 2016-11-06:
> On 06 Nov 2016, at 15:15, zhivko <notifications@github.com> wrote:
> 
> I agree that this is minor issue - but I thought if linuxcnc has this functionality already for naming components (with names= word behind loadrt) why we are now deliberately throwing this functionality away?
> 
These components are _exactly_ as in LCNC. This is as @arceye mentioned because of the singleton option http://linuxcnc.org/docs/html/hal/comp.html#_instance_creation therefore names and count parameters cannot be used. Also in LCNC.
so it’s not about being an instantiatiable component, but more about it being a singleton.

zhivko, 2016-11-06:
> hal/comp.html#_instance_creation therefore names and count parameters

Ah ok! Thanks for clarification.

closed 2016-11-07 by zhivko