machinekit/machinekit #1106: stepconf error: No such file or directory: '/usr/share/machinekit/examples/sample-configs/common/tool.tbl'

dwrobel, 2016-11-16:
stepconf seems to use completely different paths than exists on file system. Basically it refers to hard-coded: /usr/share/machinekit/ directory instead of /usr/share/linuxcnc/

Steps to reporduce:
- build with --prefix=/usr (standard in any linux distribution)
- if exists, remove ~/machinekit directory
- run stepconf (with all default values) and click "Next" until you reach the following:
```Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/stepconf/pages.py", line 82, in on_button_fwd_clicked
    self['%s_finish'%name]()
  File "/usr/lib/python2.7/site-packages/stepconf/pages.py", line 713, in finished_finish
    self.a.buid_config()
  File "/usr/bin/stepconf", line 783, in buid_config
    self.copy(base, "tool.tbl")
  File "/usr/bin/stepconf", line 774, in copy
    shutil.copy(os.path.join(distdir, filename), dest)
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: '/usr/share/machinekit/examples/sample-configs/common/tool.tbl'
```
The following commit https://github.com/machinekit/machinekit/commit/d31956892a5c4a167e7076537d1dd001856afbb7#diff-f9fc25de596bb6877b5af1f9110e3cbfR100 seems to break the stepconf. BTW, it looks that the generated lib/python/machinekit/config.py file contains self.sampleconfsdir field which might be used to determine the location of the sample directory.

Workaround: for a given --prefix (e.g. /usr) a symlink from share/linuxcnc to share/machinekit mitigates the issue (e.g. ln -s /usr/share/linuxcnc /usr/share/machinekit).

ArcEye, 2016-11-16:
Not something widely used any more if it has taken this long to surface :smile: 
The rebranding by @unseenlaser was 2015.

> if exists, remove ~/machinekit directory

The `linuxcnc` script creates the dir or a symlink if ~/linuxcnc pre-exists.
Where `~/machinekit` is used in stepconf, it should be a valid path, deleting it will just introduce errors which would otherwise not occur.
Perhaps we need to change the docs to suggest running a sim to test before doing anything else, this will guarantee it does exist.

There were quite a few instances of `linuxcnc` in use which I did not touch in my original re-branding.
This is one of the reasons why, paths hard coded into applications.

Given that the machinekit package creates a `/usr/share/linuxcnc` directory, not `/usr/share/machinekit`, `/etc/linuxcnc` not `/etc/machinekit`, these are not valid name re-branding issues, but changing to a non-existent path, which differs from the one that pickconfig etc also use.

I will look at fixing by reversing the 're-branding' where it leads to an incorrect hard coded path.

Then wholesale re-branding at some future point, needs to include Makefiles, /debian install paths, libraries etc etc as well as any remaining textual occurrences of `linuxncnc`.

ArcEye, 2016-11-16:
@dwrobel when https://github.com/machinekit/machinekit/pull/1107 is merged, can you test please.

I don't do `prefix=/usr` build/installs and a RIP will not throw the same errors.

I have made some docs amendments at https://github.com/machinekit/machinekit-docs/pull/246

PR #1107 (2016-11-16): Partially revert some re-branding from d319568
Had the effect of introducing non-existent paths which prevented stepconf
from functioning properly

Fixes #1106 (hopefully)
commit 74411c0e (2016-11-16): Partially revert some re-branding from d319568

Had the effect of introducing non-existent paths which prevented stepconf
from functioning properly

Fixes #1106
closed 2016-11-16 by luminize