machinekit/machinekit #1110: upgrading breaks working machine, conjecture: `struct __attribute__ ((__packed__))`

luminize, 2016-11-19:
I'm having trouble upgrading my 1+ yr old setup.

I'm using a BeBoPr++ with a Pepper board (5 DRV8825 deeper drivers which are configured by an AVR chip over SPI), My hardware setup is unchanged and I use an identical configuration. I cannot get the Pepper board to work with the latest code. It will enable, I can hear the whine of the stepper motors being powered, but no motion is forthcoming. 

During compiling from source I noticed the following message:
```
Compiling realtime hal/components/pepper.c
hal/components/pepper.c:540:1: note: offset of packed bit-field 'z' has changed in GCC 4.4
 };
 ^
```

So this led me to look into the component to these lines:
https://github.com/machinekit/machinekit/blob/master/src/hal/components/pepper.c#L534-L540

and further on here is another packed struct:
https://github.com/machinekit/machinekit/blob/master/src/hal/components/pepper.c#L562-L572

during power on (disable e-stop) the configuration string is sent to the pepper board, and I managed to catch the debug messages from /var/log/linuxcnc.log for both (working and non-working) setups.

non-working:
```
Nov 19 15:48:19 beaglebone msgd:0: hal_lib:4772:rt PEPPER: configuration sent 29 bytes, crc16 is $3C7C
```

working:
```
Nov 19 16:01:32 bbb-mk-02 msgd:0: hal_lib:1526:rt PEPPER: configuration sent 29 bytes, crc16 is $31F8
```

I [searched the net](https://www.google.nl/search?client=safari&rls=en&q=note:+offset+of+packed+bit-field+has+changed+in+GCC+4.4&ie=UTF-8&oe=UTF-8&gfe_rd=cr&ei=SXowWN7vCcKK8QfDlLXAAw) and found info, but I'm unable to think up a way of getting this to work, or determine if this is really the case of my broken hardware.

Anybody got ideas or tips how to go ahead with this?

luminize, 2016-11-19:
update, I made a wrong remark above, both messages _are_ the same.
the non-working setup also shows 
```
Nov 19 16:50:35 beaglebone msgd:0: hal_lib:8516:rt PEPPER: configuration sent 29 bytes, crc16 is $31F8
```

luminize, 2016-11-19:
guess I'm on the wrong path, and will change the title accordingly

cdsteinkuehler, 2016-11-19:
I don't think the bitfields are your problem, but you might try changing the uint8_t values for the ms_struct to uint16_t (it looks  like the intent is to put 5 3-bit values into a 16-bit word).

Otherwise, I'd recommend poking around in HAL to see if everything else is OK.  From what I can tell, the Pepper component basically sets up micro-step and current values.  Actual movement comes from step/dir lines driven by the PRU.  IIRC Bas L. was using some sort of dynamic sharing to get extra step/dir signals.  It is possible some recent update has broken that code.

cdsteinkuehler, 2016-11-19:
...but it doesn't look like anything has changed with that code within the last two years.  There was a fix to the PWM logic a year ago, and some updates to the encoder handling 7 months ago, but nothing that should have affected the pepper/decamux.

cdsteinkuehler, 2016-11-19:
There are some anomalies with how the bitfields are defined in the pepper component, so it's possible something changed between the Wheezy gcc 4.7 likely used on your 1 year old system and the current packages being built under Jessie (gcc 4.9).

Can you try building the pepper component out-of-tree (using the comp command from machinekit-dev) and see if that version works?

luminize, 2016-11-19:
What I tried was [saving the configuration settings in the pepper EEPROM in the old situation] (https://github.com/machinekit/machinekit/blob/master/src/hal/components/pepper.c#L49) and then I modified the pepper component so it would not send a new configuration string in the new situation.
Which didn't work.
I'll plug in some pollulu's and unplug the pepper to determine if it's the pepper or the BB.

I build the component (and MK) from source in both setups.

- Setup 1 (working) is loaded from emmc loaded
- Setup 2 (not-working) is loaded from SD on same BB as Setup 1.

Setup 1 (working, BeagleBoard.org BeagleBone Debian Image 2014-09-03)
```
machinekit@bbb-mk-02:~$ uname -a
Linux bbb-mk-02 3.8-1-xenomai.beaglebone-omap #1 Debian 3.8.13-9 armv7l GNU/Linux
machinekit@bbb-mk-02:~$ cat /etc/debian_version
7.8
machinekit@bbb-mk-02:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.6.3-14' --with-bugurl=file:///usr/share/doc/gcc-4.6/README.Bugs --enable-languages=c,c++,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.6 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.6 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.6.3 (Debian 4.6.3-14)
```

Setup 2 (non-working, Machinekit Debian Image 2016-10-30)
```
machinekit@beaglebone:~$ uname -a
Linux beaglebone 3.8.13-xenomai-r81 #1 Fri Oct 14 16:24:43 UTC 2016 armv7l GNU/Linux
machinekit@beaglebone:~$ cat /etc/debian_version 
8.6
machinekit@beaglebone:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Debian 4.9.2-10) 
```

luminize, 2016-11-19:
I changed to stepstick drivers. and I've got a similar same situation.

behaviour:
- clear e-stop
- power on
- manual mode
- home joint0

there is no movement, and looking at the pins gives me the following info
- axis.n.amp-enable-out FALSE
- hpg.stepgen.nn.enable FALSE
- hpg.stepgen.00.counts gives me a number (negative because of negative scale)

```
machinekit@beaglebone:~$ halcmd show pin axis | grep amp-enable
   100        bit   OUT         FALSE  axis.0.amp-enable-out                    			0 ==> emcmot.00.enable
   100        bit   OUT         FALSE  axis.1.amp-enable-out                    			0 ==> emcmot.01.enable
   100        bit   OUT         FALSE  axis.2.amp-enable-out                    			0 ==> emcmot.02.enable
   100        bit   OUT         FALSE  axis.3.amp-enable-out                    			0 ==> emcmot.03.enable
machinekit@beaglebone:~$ halcmd show pin hpg.stepgen | grep enable
   409        bit   IN          FALSE  hpg.stepgen.00.enable                    			0 <== emcmot.00.enable
   409        bit   IN          FALSE  hpg.stepgen.01.enable                    			0 <== emcmot.01.enable
   409        bit   IN          FALSE  hpg.stepgen.02.enable                    			0 <== emcmot.02.enable
   409        bit   IN          FALSE  hpg.stepgen.03.enable                    			0 <== emcmot.03.enable
machinekit@beaglebone:~$ halcmd show pin hpg.stepgen | grep counts
   409        s32   OUT        -63511  hpg.stepgen.00.counts                    			0
   409        s32   OUT             0  hpg.stepgen.01.counts                    			0
   409        s32   OUT             0  hpg.stepgen.02.counts                    			0
   409        s32   OUT             0  hpg.stepgen.03.counts                    			0
```

luminize, 2016-11-19:
I'll have to double check pinouts for the bebopr++ and differences wit a pepper setup tomorrow, bah, contemplating using the CRAMPS board, but that's a waste of a board (and also wiring) that just should work. Off for today... Pizza...

cdsteinkuehler, 2016-11-19:
On 11/19/2016 12:09 PM, Bas de Bruijn wrote:
> 
> I build the component (and MK) from source in both setups.
> 
>   * Setup 1 (working) is loaded from emmc loaded
>   * Setup 2 (not-working) is loaded from SD on same BB as Setup 1.
> 
> Setup 1 (working, BeagleBoard.org BeagleBone Debian Image 2014-09-03)

That's a _really_ old image.

I suspect you are running into problems with the device tree overlay
and/or the kernel (which now has device-tree-overlays compiled in).

Verify the contents of your slots file between the two versions:
/sys/devices/bone_capemgr.*/slots

...and look in the output of dmesg for any suspicious messages
regarding cape loading.  There were a bunch of different BeBoPr device
tree overlays floating around at various points.  I think the most
likely problem is that you're getting the wrong overlay loaded somehow
on the new install.

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-11-19:
I'll do that. I disable cape-bebopr-brdg in /boot/uEnv.txt and install the cape-bebopr-brdg-R3 and cape-bebopr-pp-R3 from [here with dtc.sh script](https://github.com/machinekit/machinekit/tree/master/configs/ARM/BeagleBone/BeBoPr%2B%2B)(as in the old situation)

Do you have a working very recent master btw?

I would expect that if this is not a board problem (bebopr) then others with capes would notice any breakage. But then again, nothing configuration specific changed.

cdsteinkuehler, 2016-11-20:
On 11/19/2016 3:20 PM, Bas de Bruijn wrote:
> 
> Do you have a working very recent master btw?

Yes, I have been working with recent (Oct. 30) Jessie pre-built images
and source builds, and have not had any unexpected problems.

> I would expect that if this is not a board problem (bebopr) then others with 
> capes would notice any breakage. But then again, nothing configuration specific 
> changed.

As I said, my best guess is that this is a problem with device tree
overlays.  But diagnostics from your slots file and dmesg should help
figure out if I'm guessing correctly.  :)

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-11-20:
I think you're right.

I have looked at dmesg and saw there are errors with pinmux, I made sure I have disabled the HDMI, HDMIN and EMMC, as well as bone-bebopr-brgd in /boot/uEnv.txt

after loading MK
```
machinekit@beaglebone:~$ cat /sys/devices/bone_capemgr.*/slots
 0: 54:P---- BeBoPr (3D-printer/CNC) Cape,R2,www.bebopr.info,cape-bebopr-brdg
 1: 55:PF---
 2: 56:PF---
 3: 57:PF---
 4: ff:P-O-- Bone-LT-eMMC-2G,00A0,Texas Instrument,BB-BONE-EMMC-2G
 5: ff:P-O-- Bone-Black-HDMI,00A0,Texas Instrument,BB-BONELT-HDMI
 6: ff:P-O-- Bone-Black-HDMIN,00A0,Texas Instrument,BB-BONELT-HDMIN
 7: ff:P-O-L Override Board Name,R3,Override Manuf,cape-bebopr-pp
```

I've saves all output of `dmesg`, `dmesg | grep bone` and `dmesg | grep pinmux` to [this pastebin](http://pastebin.com/bh98ZJFw)
here's a short version where I can see errors or failures
```
[    0.394252] arm-pmu: probe of arm-pmu failed with error -28
[    0.394618] omap2_mbox_probe: platform not supported
...
[    0.556146] io scheduler cfq registered (default)
[    0.557072] tps65217-bl tps65217-bl: no platform data provided
[    0.557108] tps65217-bl: probe of tps65217-bl failed with error -22
[    0.557537] Serial: 8250/16550 driver, 4 ports, IRQ sharing enabled
[    0.558698] omap_uart 44e09000.serial: did not get pins for uart0 error: -19
...
[    0.730118] device-mapper: ioctl: 4.23.1-ioctl (2012-12-18) initialised: dm-devel@redhat.com
[    0.730337] of_get_named_gpio_flags exited with status 6
[    0.730347] of_get_named_gpio_flags: can't parse gpios property
[    0.730354] of_get_named_gpio_flags: can't parse gpios property
[    0.730374] omap_hsmmc mmc.5: of_parse_phandle_with_args of 'reset' failed
[    0.737586] omap_hsmmc mmc.5: Failed to get rstctl; not using any
[    0.737902] edma-dma-engine edma-dma-engine.0: allocated channel for 0:25
[    0.737935] edma-dma-engine edma-dma-engine.0: allocated channel for 0:24
[    0.738098] mmc.5 supply vmmc_aux not found, using dummy regulator
[    0.738376] omap_hsmmc mmc.5: pins are not configured from the driver
[    0.765498] pinctrl-single 44e10800.pinmux: pin 44e10854 already requested by 44e10800.pinmux; cannot claim for gpio-leds.8
[    0.777159] pinctrl-single 44e10800.pinmux: pin-21 (gpio-leds.8) status -22
[    0.784445] pinctrl-single 44e10800.pinmux: could not request pin 21 on device pinctrl-single
[    0.793363] leds-gpio gpio-leds.8: pins are not configured from the driver
[    0.793384] of_get_named_gpio_flags exited with status 53
[    0.793393] of_get_named_gpio_flags exited with status 54
[    0.793403] of_get_named_gpio_flags exited with status 55
[    0.793411] of_get_named_gpio_flags exited with status 56
[    0.793423] of_get_named_gpio_flags exited with status 53
[    0.793580] of_get_named_gpio_flags exited with status 54
[    0.793696] of_get_named_gpio_flags exited with status 55
[    0.793785] of_get_named_gpio_flags exited with status 56
...
machinekit starting, e-stop off, machine on, homing joint0 not moving
...
[  119.895207] gpio-of-helper bebopr_io_enables.11: could not find pctldev for node /ocp/interrupt-controller@48200000, deferring probe
[  119.895238] platform bebopr_io_enables.11: Driver gpio-of-helper requests probe deferral
[  119.896135] ehrpwm 48302200.ehrpwm: unable to select pin group
[  119.896798] ehrpwm 48304200.ehrpwm: unable to select pin group
[  119.898004] bone-pinmux-helper bebopr_pwm_J2_pinmux.12: could not find pctldev for node /ocp/i2c@44e0b000/tps@24/regulators/regulator@1, deferring probe
[  119.898022] bone-pinmux-helper bebopr_pwm_J2_pinmux.12: Failed to get pinctrl
[  119.905537] platform bebopr_pwm_J2_pinmux.12: Driver bone-pinmux-helper requests probe deferral
[  119.905872] bone-pinmux-helper bebopr_pwm_J3_pinmux.13: could not find pctldev for node /ocp/gpio@4804c000, deferring probe
[  119.905889] bone-pinmux-helper bebopr_pwm_J3_pinmux.13: Failed to get pinctrl
[  119.913374] platform bebopr_pwm_J3_pinmux.13: Driver bone-pinmux-helper requests probe deferral
[  119.913661] pinctrl-single 44e10800.pinmux: pin 44e10988 already requested by 44e0b000.i2c; cannot claim for bebopr_pwm_J4_pinmux.14
[  119.926127] pinctrl-single 44e10800.pinmux: pin-98 (bebopr_pwm_J4_pinmux.14) status -22
[  119.934496] pinctrl-single 44e10800.pinmux: could not request pin 98 on device pinctrl-single
[  119.943424] bone-pinmux-helper bebopr_pwm_J4_pinmux.14: Failed to select default state
[  119.951726] bone-pinmux-helper: probe of bebopr_pwm_J4_pinmux.14 failed with error -22
[  119.953373] gpio-of-helper bebopr_steppers.18: could not find pctldev for node /ocp/edma@49000000, deferring probe
[... 57 lines cut ...]
[  120.380627] pwm_test bebopr_pwm_J4.17: Unable to select pin group
[  120.380933] gpio-of-helper bebopr_io_enables.11: could not find pctldev for node /ocp/interrupt-controller@48200000, deferring probe
[  120.380959] platform bebopr_io_enables.11: Driver gpio-of-helper requests probe deferral
[  120.381214] bone-pinmux-helper bebopr_pwm_J2_pinmux.12: could not find pctldev for node /ocp/i2c@44e0b000/tps@24/regulators/regulator@1, deferring probe
[  120.381228] bone-pinmux-helper bebopr_pwm_J2_pinmux.12: Failed to get pinctrl
[  120.388784] platform bebopr_pwm_J2_pinmux.12: Driver bone-pinmux-helper requests probe deferral
[  120.388963] bone-pinmux-helper bebopr_pwm_J3_pinmux.13: could not find pctldev for node /ocp/gpio@4804c000, deferring probe
[  120.388979] bone-pinmux-helper bebopr_pwm_J3_pinmux.13: Failed to get pinctrl
[... cut ...]
[  120.537861] bone-pinmux-helper bebopr_pwm_J3_pinmux.13: could not find pctldev for node /ocp/gpio@4804c000, deferring probe
[  120.537876] bone-pinmux-helper bebopr_pwm_J3_pinmux.13: Failed to get pinctrl
[  120.545356] platform bebopr_pwm_J3_pinmux.13: Driver bone-pinmux-helper requests probe deferral
[  120.545482] gpio-of-helper bebopr_steppers.18: could not find pctldev for node /ocp/edma@49000000, deferring probe
[  120.545499] platform bebopr_steppers.18: Driver gpio-of-helper requests probe deferral
```

So I guess you're right, something is messing up. And the cape/overlay does't work

cdsteinkuehler, 2016-11-20:
Is this slots file from the working or broken version?

My guess is the slots files will be different and you will need to
adjust something (perhaps black-listing an overlay) to get your board
working with Jessie.

Your device tree file may also need to be recompiled.  I've seen
strange issues like the dmesg output you posted when I had an overlay
compiled with the wrong version of the device-tree compiler.  Note
that I think the dtc provided on the images is too new (it's for the
4.x kernels).

Try with the dtc from RCN's tools repo:

git clone https://github.com/RobertCNelson/tools.git
cd tools
pkgs/dtc.sh

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2016-11-20:
> On 20 Nov 2016, at 12:03, cdsteinkuehler <notifications@github.com> wrote:
> 
> Is this slots file from the working or broken version?

the broken version

> My guess is the slots files will be different and you will need to
> adjust something (perhaps black-listing an overlay) to get your board
> working with Jessie.
>
> Your device tree file may also need to [...]

will do

luminize, 2016-12-15:
recompiling the device tree didn't work. Instead I have used the EMMC wheezy setup, and got it working.

closed 2016-12-15 by luminize