machinekit/machinekit #1124: Problems with GCC 4.9

cdsteinkuehler, 2017-02-08:
While related to the recent merge of the multicore features into the main branch and Issue #1123 , I believe this is not really a merge bug.  The Wheezy armhf Travis builds are failing early when linking libmachinetalk-pb2++.so with the error " undefined reference to `__cxa_throw_bad_array_new_length'"

The issue appears to be related to the use of gcc-4.9, which John Morris apparently setup for the armhf proot build environments.  From the proot-helper rootfs on a locally running machinekit/mk-builder docker image:

````
root@0666472dd3d7:/usr/src/machinekit/src# gcc --version
arm-linux-gnueabihf-gcc-4.9 ( 4.9.2-10) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

````

Trying to google this error turns up lots of not very helpful messages, but it appears this function has been removed from libstdc++ in gcc-4.9, yet calls to the function are still being generated by the compiler.  [This commit](https://git.merproject.org/jpetrell/gecko-dev/commit/31ad3d202afc92ef7755705a0635af132cfb37be?view=parallel) to add compatibility for gcc 4.9 has some details.  Perhaps we will need to do something similar if/when Debian moves to gcc 4.9 or later?

When building on a standard Wheezy armhf system (and not the machinekit/mk-builder docker image) using gcc 4.6 or 4.7 this isn't a problem.

cdsteinkuehler, 2017-02-08:
Digging into this more, it appears to be an incompatibility between gcc versions specific to the Wheezy armhf build images.  The script which has replaced "gcc-4.6" on this build image is:
````
#!/bin/sh
exec /host-rootfs/usr/bin/arm-linux-gnueabihf-gcc-4.9 \
        -I/usr/include \
        -Wl,-rpath-link=/usr/lib/gcc/arm-linux-gnueabihf/4.7 \
        -Wl,-rpath-link=/usr/lib/arm-linux-gnueabihf \
        -Wl,-rpath-link=/lib/arm-linux-gnueabihf \
        -Wl,-rpath-link=/usr/lib \
        -Wl,-rpath-link=/lib \
        -L/usr/lib/gcc/arm-linux-gnueabihf/4.7 \
        -L/usr/lib/arm-linux-gnueabihf \
        -L/lib/arm-linux-gnueabihf \
        -L/usr/lib \
        -L/lib \
        "$@"
````

I am nearly certain the problem is caused by using arm-linux-gnueabihf-gcc-4.9 as the compiler and linking against arm-linux-gnueabihf/4.7 for the library.

...still looking at possible ways to fix this.

ArcEye, 2017-02-09:
I assume the Wheezy builds on Travis are not actually building on a Wheezy platform?

I could not find gcc-4.9 available anywhere for Wheezy, just 4.7

Let me know if I can do anything

cdsteinkuehler, 2017-02-09:
On 2/9/2017 3:11 AM, ArcEye wrote:
> I assume the Wheezy builds on Travis are not actually building on a Wheezy platform?

They build in a Wheezy chroot from the mk-builder Docker image.

> I could not find gcc-4.9 available anywhere for Wheezy, just 4.7

The gcc 4.9 armhf cross compiler is installed in the Docker image from
the emdebian repo:

https://github.com/machinekit/mk-builder/blob/wheezy-armhf/mkdocker-base/Dockerfile#L25-L32

> Let me know if I can do anything

If I can't get it fixed I'll probably have to see if John can help
out, he's the one who setup the cross-compile in the first place.

For now I'm going to switch to native builds to see if I can get past
the issue with atomics (which seems likely, since the x86 Wheezy
builds work OK).  If I get to that point, we can dig into why the
Wheezy armhf Travis builds don't work.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2017-02-09:
> The gcc 4.9 armhf cross compiler is installed in the Docker image from the emdebian repo

They seem to have ditched everything below 4.9

http://linux-sunxi.org/Toolchain states that embedian is deprecated, being superseded by Debian.
What they don't mention is that Debian support starts at 4.9 for Jessie too, so same situation.

Apparently  "The embedian toolchain collection for wheezy was never quite right" according to the debian-lists.
I'm sure I had it working fine at one stage.

So there is always linaro, right?
Wrong, it has dropped gcc-4.7 too (http://releases.linaro.org/components/toolchain/gcc-linaro/)

Finding it very hard to discover if the ABI version of 4.7 is compatible with 4.9 (v1.2), thus whether linking with 4.9 would fare better.
They must be fairly compatible for it to have ever worked, but 50% of the listed atomic operations under Concurrency seem to require gcc 4.8 and above.
So building with a gcc that supports the standard more fully and then trying to link with gcc and attendent libs that only partially supports it, is likely creating extra problems.

Overall am leaning towards our initial thoughts, just drop Wheezy.

cdsteinkuehler, 2017-02-09:
On 2/9/2017 9:16 AM, ArcEye wrote:
> 
> Overall am leaning towards our initial thoughts, just drop Wheezy.

Interesting data point:

Building multicore on Wheezy with 4.7 from backports works fine.  The
problems I was having previously related to not fully swapping out
*ALL* of the gcc related symlinks in /usr/bin from 4.6 to 4.7.

So...we should be able to build with the native gcc 4.7, but that will
be slow in a Travis/Docker container.  Better is probably switching
_everything_ to the gcc 4.9 cross compiler and not use the libraries
from 4.7, but I'm not sure if that has any run-time implications.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2017-02-09:
> Building multicore on Wheezy with 4.7 from backports works fine. The problems I was having previously related to not fully swapping out *ALL* of the gcc related symlinks in /usr/bin from 4.6 to 4.7.

Ah, I used to create my own symlinks and specifically edit them to point to the version I wanted.
Otherwise a mess.

> Better is probably switching _everything_ to the gcc 4.9 cross compiler and not use the libraries from 4.7, but I'm not sure if that has any run-time implications.

That is what concerns me.
If the libstdc++ etc has changed, as it must have, any dynamically linked functions that are only found within that environment, or whose implementation has changed, will fail when run on a 4.7 based env.

Probably only one way to find out :smile:

As an aside, version hell continues in Stretch.
I thought I would give it a spin to see what happened.
The version of libczmq which installs, has deprecated lots of functions that machinetalk relies upon, probably meaning we need to compile our own earlier version lib or undertake a big rewrite.

Hey Ho.

ArcEye, 2017-02-09:
> As an aside, version hell continues in Stretch.
I thought I would give it a spin to see what happened.
The version of libczmq which installs, has deprecated lots of functions that machinetalk relies upon, probably meaning we need to compile our own earlier version lib or undertake a big rewrite.

I have built local versions of the czmq-2.2.0 and zeromq-4.0.5.
With those, some hacks of cython, alterations to rtapi_maths functions and a ton of warnings from
gcc-6.3.0 ........ it builds and runs on Stretch :clinking_glasses: 

I'll give up for the day before something goes wrong.

cdsteinkuehler, 2017-02-09:
I have dug through the gcc source and it looks like the __cxa_throw_bad_array_new_length function was added to the ABI about four years ago to the 4.7 branch via this commit: https://github.com/gcc-mirror/gcc/commit/bcb3170cd7700c03e5017c0a2b4e845b5cb1d763

...but the Debian 4.7 library is not quite new enough to have picked up this specific change.

So the solution available solutions would appear to be:

1. Find or build a working gcc 4.7 cross compiler
2. Add the __cxa_throw_bad_array_new_length to our source when doing wheezy builds.  The code is simple (it just needs to crash and/or throw an exception), the only hard bit would be working out some #ifdef logic for when to include the function
3. Switch to using the armhf native gcc 4.7 present in the Wheezy build rootfs via qemu

(1) appears to be hard, (3) will be very slow, so I'll look at what it will take to do something for (2).

cdsteinkuehler, 2017-02-09:
I have an ugly hack working that gets me past the problem with the missing API function.  It requires changing the Docker build hosts rather than the source, which seems appropriate since the issue is with the mis-matched g++/libstdc++ versions on the build machine vs. any sort of problem with the code (which builds fine on a Wheezy system with it's native gcc-4.7 from back-ports).

I will try to clean up the code and send a PR to the mk-builder repo.

ArcEye, 2017-02-10:
Well done.
I am in and out a lot today, but will keep an eye out for the PR.

cdsteinkuehler, 2017-02-10:
On 2/10/2017 12:46 AM, ArcEye wrote:
> Well done.
> I am in and out a lot today, but will keep an eye out for the PR.

I have a PR waiting for the mk-builder repo.  I'm not sure how the C4
workflow is applied to this repo, since it requires a force-push to
all the branches.  I figured someone should at least review the hack I
applied for sanity (and accept the merge), then we can update and
force-push all the branches.

I'm not sure about making the resulting Docker images, however.  Is
that automatic or does it need to be done by hand?

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2017-02-10:
On 10 Feb 2017, at 12:42, cdsteinkuehler  wrote:

On 2/10/2017 12:46 AM, ArcEye wrote:
> Well done.
> I am in and out a lot today, but will keep an eye out for the PR.

I have a PR waiting for the mk-builder repo. I'm not sure how the C4
workflow is applied to this repo, since it requires a force-push to
all the branches. I figured someone should at least review the hack I
applied for sanity (and accept the merge), then we can update and
force-push all the branches.

I'm not sure about making the resulting Docker images, however. Is
that automatic or does it need to be done by hand?

 

IIRC when a push is made to machinekit/mk-builder repo a new docker image is built. But I have no credentials to actually look into that part

See https://hub.docker.com/r/machinekit/mk-builder/

ArcEye, 2017-02-10:
Because the machinekit checks do not actually test a build on armhf, it needs a PR to be pushed before we test properly

I am about to merge the mk-builder PR and will then do a `do nothing` whitespace PR to machinekit, which when merged will force a build on armhf and test the hack

ArcEye, 2017-02-10:
> IIRC when a push is made to machinekit/mk-builder repo a new docker image is built. But I have no credentials to actually look into that part

The package side of it is downloaded every time it is run, so it should just use the new control files and build.   Soon find out :smile:

ArcEye, 2017-02-10:
Unfortunately, at present armhf builds are failing with the same error
```
../lib/libmachinetalk-pb2++.so: undefined reference to `__cxa_throw_bad_array_new_length'
```

cdsteinkuehler, 2017-02-10:
On 2/10/2017 11:34 AM, ArcEye wrote:
> Unfortunately, at present armhf builds are failing with the same error
> 
> |../lib/libmachinetalk-pb2++.so: undefined reference to 
> `__cxa_throw_bad_array_new_length' |

When did you test?

The travis build for the wheezy-armhf  image just finished ~27 minutes
ago:

https://travis-ci.org/machinekit/mk-builder/branches

...and the wheezy-64 build failed, but that was due to a package
download timeout, so nothing wrong with the images (just a hiccup with
the internet).  I'm trying to see if I can kick off a build w/o
another commit.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2017-02-11:
> When did you test?

I was going by the build being run for some reason for my PR
https://travis-ci.org/machinekit/machinekit/builds/200415670
with error at
https://travis-ci.org/machinekit/machinekit/jobs/200415679

It did actually build the other packages even though the PR has not been merged.eg. 
http://deb.machinekit.io/debian/pool/main/m/machinekit/machinekit-dev_0.1.1486746094-1mk.travis.ArcEyepatch1.git523044c8~1wheezy_i386.deb

I was careful not to select 'commit directly' but it still effectively did just that.
Must be something to do with my having push authorisations etc. to the repo.

I shall have to make sure I do any web edit PRs to my own repo, which hopefully will prevent that happening again

ArcEye, 2017-02-11:
> The travis build for the wheezy-armhf image just finished ~27 minutes ago:

AFAICT the last time a wheezy-armhf package was produced was 4th Feb 2017

ArcEye, 2017-02-12:
New error, the build for wheezy-armhf is succeeding but the link stage fails

https://travis-ci.org/machinekit/machinekit/builds/200748522

```
/usr/lib/wheezy_hacks.o: file not recognized: File format not recognized

collect2: error: ld returned 1 exit status
```

cdsteinkuehler, 2017-02-13:
On 2/12/2017 1:13 AM, ArcEye wrote:
> New error, the build for wheezy-armhf is succeeding but the link stage fails
> 
> https://travis-ci.org/machinekit/machinekit/builds/200748522
> 
> |/usr/lib/wheezy_hacks.o: file not recognized: File format not recognized 
> collect2: error: ld returned 1 exit status |

Yes, I've been trying to patch things up over the weekend.  I think I
finally have a working fix, but won't know for a few hours once
everything gets rebuilt (again).

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2017-02-13:
Excellent - I am watching the checks progress and will merge once they pass.

cdsteinkuehler, 2017-02-13:
On 2/13/2017 6:12 AM, ArcEye wrote:
> Excellent - I am watching the checks progress and will merge once they pass.

Looks like the wheezy-armhf builds are _finally_ passing!

https://travis-ci.org/machinekit/machinekit

Hooray!!!  :)

-- 
Charles Steinkuehler
charles@steinkuehler.net

luminize, 2017-02-13:
> On 13 Feb 2017, at 15:10, cdsteinkuehler <notifications@github.com> wrote:
> 
> On 2/13/2017 6:12 AM, ArcEye wrote:
> > Excellent - I am watching the checks progress and will merge once they pass.
> 
> Looks like the wheezy-armhf builds are _finally_ passing!
> 
> https://travis-ci.org/machinekit/machinekit <https://travis-ci.org/machinekit/machinekit>
> 
> Hooray!!! :)

Congrats guys!

machinekoder, 2017-03-13:
Is the issue fixed?

ArcEye, 2017-03-13:
The issue in so far as it affected package builds using embedded toolchain, yes

machinekoder, 2017-03-29:
Okay. @cdsteinkuehler Can we close this issue?

cdsteinkuehler, 2017-03-29:
The issue is fixed, closing.

closed 2017-03-29 by cdsteinkuehler