machinekit/machinekit #1127: lincurve icomp goes nuts

propcoder, 2017-02-10:
lincurve and lincurvev2 icomponents return enormous values, totally out of scale, sometimes negative when only positive input numbers exist. (master branch, latest version)

The difference from LinuxCNC lincurve.comp is in line 60.
Shouldn't it be:
sf(out_, gf(y_val(i)) + f * (gf(y_val(i+1)) - gf(y_val(i)) )) ;
instead of:
sf(out_, ( gf(y_val(i)) + f) * (gf(y_val(i+1)) - gf(y_val(i)) )) ;
?

ArcEye, 2017-02-11:
I think @machinekoder fixed it a long while back, but the code has got reverted in the recent merge.

I will look at it in a while.

Thanks for the report.

luminize, 2017-02-11:
I think the issues are #818 #784

ArcEye, 2017-02-11:
You are right and it was me that fixed it, @machinekoder reported it.
All blurs into one after a while. :smile:

I will sort it soon.

ArcEye, 2017-02-11:
PR #1128 hopefully fixes back to how it was.
Will take about 90 minutes from when it is merged for all the packages to appear, armhf ones still not building right now.

closed 2017-02-11 by ArcEye

ArcEye, 2017-02-12:
0.1.1486851828-1mk.travis.master.gitcbd2dfd3
is the relevant serial for the build with lincurve corrected.

Unfortunately wheezy-armhf builds not working yet.

PR #1128 (2017-02-11): Fix lincurve and lincurvev2 bracketing auto-completion typo error
Previously fixed in #783 but reverted by old code in multicore merge

Fixes #1127
commit f14417b6 (2017-02-11): Fix lincurve and lincurvev2 bracketing auto-completion typo error

Previously fixed in #783 but reverted by old code in multicore merge

Fixes #1127