machinekit/machinekit #1129: Removing Webtalk from build?

machinekoder, 2017-02-12:
Webtalk adds a dependency to libwebsocket which is complicated to resolve on many systems since a special version of the library is required and Websockets are a moving target right now. However, Webtalk is not used in any project I know of.

I suggest removing Webtalk from the Machinekit main repository or deactivate it in the default build configuration. Even better would be to replace it with a Node.js based solution like this one: https://github.com/machinekoder/Webvcp

ArcEye, 2017-02-12:
I am all for simplifying builds and removing dependencies.

I would suggest that a configure option of  `--with-option-webtalk` would be the easiest way to remove the dependency but retain the code to use it if required.

If webtalk is not in use, replacing it with another solution does not seem a profitable use of time.

ArcEye, 2017-02-12:
It can be done quite simply as at
https://github.com/ArcEye/machinekit/commit/5d8a7de21bce4e1459b1c2f5984722ca55d9a446

Tested and `./configure --enable-webtalk` will build, otherwise is ignored.

(ignore the other headers the commit removes, I based it on the wrong branch initially)

machinekoder, 2017-02-12:
Excellent! This makes the build a lot easier, especially for new users.

ArcEye, 2017-02-12:
OK based it upon the proper branch now
https://github.com/ArcEye/machinekit/commit/c357fe1e79a453caaf1a68482d295119c113de75

I can do a PR if you want.
Depends how sure you are that no-one uses it?

ArcEye, 2017-02-18:
Email sent out to the list as a poll.

If no responses or only from those using RIPs, this could be dropped from Travis builds and remain as an option for local builds.

ArcEye, 2017-02-24:
Only response has been one person who says he doesn't use it but would like it left in just in case he ever changes his mind (to paraphrase) :stuck_out_tongue_closed_eyes: 

PR #1149 refers

PR #1149 (2017-02-24): Changes to make webtalk build and libwebsockets dependency optional
Fixes #1129
closed 2017-02-24 by ArcEye